kaloz/mwlwifi #1: Wifi drops on 2.4Ghz signal only

dpdurst, 2015-01-08:
I have been noticing wifi drops on the 2.4ghz network only on the 1900. Monitoring from the overview screen and running ping against specific devices confirms this.

The 5ghz seems fine, the SNR ratio seems to be steady, the noise fluctuates between 91% and 89% and the signal never goes away on the devices I have connected

On the 2.4ghz the SNR fluctuates really bad, it starts solid at 95% then drops specifically to 76% then drops to 46% where at that time the devices loose contact for about 12ms of time. This can be noted when running ping -t command on those devices they are normally 1-3ms response times till the noise ratio drops to 46% then goes to 12ms time and starts back up. 

Screenshots of OpenWRT area and ping times included

The devices are as follows:

Router WRT1900AC running CC with new Wifi 

Running in 2.4ghz mode:

Foscam 8918w IP Cameras -3 devices
HP 1102 Wireless Laserjet - 1 device
HP 4630 Wireless Inkjet - 1 device
Nest Thermostat - 1 device
Nest Protect Fire and CO - 4 devices
Android Tables - 4 devices

Running in 5ghz mode with no problems:

Android Tables - 2 devices
Ipad tablet - 1 device

[image]
[image]
[image]
[image]
[image]
[image]

dpdurst, 2015-01-10:
I tested a few things out on the wifi with the drops. I have no encryption wide open network and I have 2 devices connected and the signal is -31 but the noise is -95, -75, -48 and continues to toggle. At -48 the drop out occurs on both a tablet and printer. Meanwhile the 5G signal signal is -65 and noise never moves off -90 with no drops.

jkoelndorfer, 2015-07-20:
Can anyone acknowledge whether or not this is a bug or a configuration issue? I'm experiencing the same problem, also on a first gen WRT1900AC. Currently running Chaos Calmer RC2.

I haven't been able to test 5GHz as I don't have a 5GHz compatible device handy, but I definitely notice the 2.4GHz signal is flaky sometimes. It seems to almost completely cut out for a little while (maybe 100ms?) - really noticeable when playing games over wifi. I can usually see the bars drop off in the wifi indicator in my taskbar.

I can do some more empirical testing if that would be helpful.

closed 2015-11-27 by yuhhaurlin