kaloz/mwlwifi #3: Wifi seems to freeze

creiners, 2015-01-08:
I have noticed this the most on my iphone 5s and my Macbook Pro early 2013.  I am checking with the rest of the family when they get home today to see if they have the same issues.   It seems to maintain the actual WiFi connection but stops transmitting data.  A simple disco and reconnect of the client device seems to restore data for a while 15 - 20 minutes max.

It seems to happen on both the 2.4 and 5 Mhz radios.  I generally have about 25 devices online during peak, maybe 6-7 wireless devices max online though.

I am running gen1 WRT1900AC.  Please let me know if you need any further info or logs.  I am more than willing to also do a screen share if some one wishes to look at the router directly.

okrasit, 2015-01-13:
See issue #4.

yuhhaurlin, 2015-03-30:
Still got problem on 10.2.8.5.p0?

yuhhaurlin, 2015-11-12:
Does anyone still encounter this problem with 10.3.0.14?

closed 2015-11-24 by yuhhaurlin