kaloz/mwlwifi #4: wifi disconnects

tusc, 2015-01-08:
I'm having issues with wireless disconnects on both bands. Initially I thought it was a problem with the 5ghz band so I've moved most of the devices to 2.4 but still see the disconnects, mostly with an Ipad 2 and 3 device. I've played around with Legacy and N mode on the 2.4 bandas well as channel width but the problem persists.

I have to go into setitngs, wifi, turn off/on on the Ipad to fix the problem. Both devices use Broadcom WIFI chipsets:  BCM4329 for Ipad 2 and BCM4330for Ipad 3.

yuhhaurlin, 2015-01-09:
deauthenticated due to inactivity (timer DEAUTH/REMOVE) => The problem happened because there is no traffic within the inactivity time period. Should be fixed in next release.

okrasit, 2015-01-13:
If using owrt set hostapd disassoc_low_ack=0 with:
wireless.@wifi-iface[0].disassoc_low_ack=0

creiners, 2015-01-13:
I have done this and my apple products have stayed very stable for the past 4 hours.

thagabe, 2015-01-13:
I agree, this has solved my problem

gibsonj, 2015-01-27:
I am also experiencing a similar issue, but only on one of my Apple devices. I have tried the fix described above, but it didn't help.

The device that is experiencing the problem is a Late 2010 MacBook Air. It is able to connect to the 2.4 and 5 GHz networks, but then after ~30 seconds it disconnects. It is able to connect to one of my other routers without an issue.

I have attached the system logs from the Mac.

2015-01-26 8:42:17.737 PM UserEventAgent[17]: Captive: [CNInfoNetworkActive:1709] en0: SSID '...text removed...' making interface primary (protected network)
2015-01-26 8:42:17.744 PM UserEventAgent[17]: Captive: CNPluginHandler en0: Evaluating
2015-01-26 8:42:17.747 PM UserEventAgent[17]: Captive: en0: Probing '...text removed...'
2015-01-26 8:42:17.758 PM networkd[144]: +[NETLedBelly stopFastFail] Clearing ledbelly failure cache
2015-01-26 8:42:17.765 PM networkd[144]: +[NETLedBelly stopFastFail] Clearing ledbelly failure cache
2015-01-26 8:42:17.772 PM configd[26]: network changed: v4(en0!:...text removed...) DNS+ Proxy+ SMB
2015-01-26 8:42:17.000 PM kernel[0]: en0: BSSID changed to ...text removed...
2015-01-26 8:42:17.813 PM sharingd[248]: 20:42:17.812 : SDStatusMonitor::kStatusWirelessPowerChanged
2015-01-26 8:42:18.017 PM UserEventAgent[17]: Captive: CNPluginHandler en0: Authenticated
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  Restoring session ...text removed...
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  get the reconnect token
2015-01-26 8:42:18.172 PM KernelEventAgent[74]: tid 54485244 received event(s) VQ_NOTRESP (1)
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  Restoring session ...text removed...
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  get the reconnect token
2015-01-26 8:42:18.172 PM KernelEventAgent[74]: tid 54485244 received event(s) VQ_NOTRESP (1)
2015-01-26 8:42:20.469 PM com.apple.xpc.launchd[1]: (com.apple.imfoundation.IMRemoteURLConnectionAgent) The _DirtyJetsamMemoryLimit key is not available on this platform.
2015-01-26 8:42:42.000 PM kernel[0]: MacAuthEvent en0   Auth result for: ...text removed...  MAC AUTH succeeded
2015-01-26 8:42:42.000 PM kernel[0]: wlEvent: en0 en0 Link UP virtIf = 0
2015-01-26 8:42:42.000 PM kernel[0]: wl0: Roamed or switched channel, reason #8, bssid ...text removed...
2015-01-26 8:42:42.000 PM kernel[0]: en0: BSSID changed to ...text removed...
2015-01-26 8:42:42.000 PM kernel[0]: en0::IO80211Interface::postMessage bssid changed

yuhhaurlin, 2015-03-30:
Still got problem on 10.2.8.5.p0?

raven-au, 2015-03-30:
We will need to wait for others to comment on this but I thought I would comment on the disconnects I had seen in the previous release as it may be related.

With the previous release I was unable to watch streamed video from my Plex server for more than 1 to 2 hours before getting a network error which would then persist quite frequently.

With the current release I'm not getting these errors.

However I've not had the my device as the primary router for a longer period than a day, basically due to the disconnects. I'm now testing extended usage time.

closed 2015-05-26 by yuhhaurlin

Rovinatore, 2016-05-10:
Please help, in which file i have to put the disassoc_low_ack=0?

kainolophobia, 2018-02-07:
@Rovinatore Using Openwrt's UCI:
`uci set wireless.@wifi-iface[0].disassoc_low_ack=0`

dmascord, 2018-07-16:
Hi guys,

There is a patch for other chipsets which supposedly resolves this "disconnect" problem.

https://github.com/openwrt/mt76/issues/169

Perhaps it is related.

Cheers,

Damien