kaloz/mwlwifi #5: WiFi Freeze Error

thagabe, 2015-01-08:
On MacBook 2008 unibody using Broadcom's BCM94322USA successfully connects to CC build of OpenWRT running 3.18.1 kernel and mwlwifi. However the connection last for a few minutes (or seconds) before it timeout and disconnects. My macbook shows it still receiving an ip, subnetmask, router, external ip; the router show the same ip assignment to the macbook. Running the macbook under Ubuntu 14.10 kernel 3.16 it presents no wifi problem (proprietary module loaded bcmwl-kernel-source). My intuition is either the mwlwifi (under a build for mvebu WRT1900AC) is having authentication errors or the mac driver is not being kept up to date. I have tested again using my build (http://tinyurl.com/CC-lifehacksback-Known-Good) and Kaloz's build along with the beta of OS X Yosemite 10.10.2 which claims to fix wifi issues but had not fixed my particular issue. Currently running McWRT AA v1.0.8 using wifi driver wlan-7 ported to the WRT1900AC. Thank You

yuhhaurlin, 2015-01-09:
If establishing a ping session to AP, do you still see this kind of problem? It looks like there are two problems now: 1. No traffic within inactivity time checking period. 2. Current driver has IOT problem with some clients. Establishing a ping session to AP can clarity which one is the root cause.

thagabe, 2015-01-10:
Ping doesn't establish a connection

okrasit, 2015-01-13:
See issue #4, it's probably related.

thagabe, 2015-01-13:
Issue still persists

okrasit, 2015-01-14:
Please verify, from running config, that you have actually disabled disassoc_low_ack on all hostapd instances.

thagabe, 2015-01-14:
nano /etc/config/wireless

config wifi-iface
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Network_Name'
        option encryption 'psk2+ccmp'
        option key 'password'
        option disassoc_low_ack '0'

-----------------------------------------------------------\

config wifi-iface
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Network'
        option encryption 'psk2+ccmp'
        option key 'password'
        option disassoc_low_ack '0'

creiners, 2015-01-15:
This resolved my 2.4 but not my 5.0 wifi ...

gufus, 2015-01-18:
I tryed this fix, still  get the same error on 2.4ghz
https://forum.openwrt.org/viewtopic.php?pid=261726#p261726

thagabe, 2015-01-18:
2.4 is now connecting fine, 5GHz the hack doesnt work

gibsonj, 2015-01-27:
I am experiencing an issue exclusively on one of my Apple devices. I have tried the fix described in the <a href="https://github.com/kaloz/mwlwifi/issues/4">other posts</a>, but it did not help in this case.

The device that is experiencing the problem is a Late 2010 MacBook Air (OS X 10.10.1). It is able to connect to the 2.4 and 5 GHz networks, and access the Internet, but after 25 seconds it consistently disconnects. It is able to connect to one of my other routers without an issue.

Model Linksys WRT1900AC (Mamba)
Firmware Version OpenWrt Chaos Calmer r43974 / LuCI Master (git-14.359.33351-5e6c33e)
Kernel Version 3.18.2

I have attached the system logs from the Mac.

2015-01-26 8:42:17.737 PM UserEventAgent[17]: Captive: [CNInfoNetworkActive:1709] en0: SSID '...text removed...' making interface primary (protected network)
2015-01-26 8:42:17.744 PM UserEventAgent[17]: Captive: CNPluginHandler en0: Evaluating
2015-01-26 8:42:17.747 PM UserEventAgent[17]: Captive: en0: Probing '...text removed...'
2015-01-26 8:42:17.758 PM networkd[144]: +[NETLedBelly stopFastFail] Clearing ledbelly failure cache
2015-01-26 8:42:17.765 PM networkd[144]: +[NETLedBelly stopFastFail] Clearing ledbelly failure cache
2015-01-26 8:42:17.772 PM configd[26]: network changed: v4(en0!:...text removed...) DNS+ Proxy+ SMB
2015-01-26 8:42:17.000 PM kernel[0]: en0: BSSID changed to ...text removed...
2015-01-26 8:42:17.813 PM sharingd[248]: 20:42:17.812 : SDStatusMonitor::kStatusWirelessPowerChanged
2015-01-26 8:42:18.017 PM UserEventAgent[17]: Captive: CNPluginHandler en0: Authenticated
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  Restoring session ...text removed...
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  get the reconnect token
2015-01-26 8:42:18.172 PM KernelEventAgent[74]: tid 54485244 received event(s) VQ_NOTRESP (1)
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  Restoring session ...text removed...
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  get the reconnect token
2015-01-26 8:42:18.172 PM KernelEventAgent[74]: tid 54485244 received event(s) VQ_NOTRESP (1)
2015-01-26 8:42:20.469 PM com.apple.xpc.launchd[1]: (com.apple.imfoundation.IMRemoteURLConnectionAgent) The _DirtyJetsamMemoryLimit key is not available on this platform.
2015-01-26 8:42:42.000 PM kernel[0]: MacAuthEvent en0   Auth result for: ...text removed...  MAC AUTH succeeded
2015-01-26 8:42:42.000 PM kernel[0]: wlEvent: en0 en0 Link UP virtIf = 0
2015-01-26 8:42:42.000 PM kernel[0]: wl0: Roamed or switched channel, reason #8, bssid ...text removed...
2015-01-26 8:42:42.000 PM kernel[0]: en0: BSSID changed to ...text removed...
2015-01-26 8:42:42.000 PM kernel[0]: en0::IO80211Interface::postMessage bssid changed

closed 2015-01-27 by thagabe