kaloz/mwlwifi #6: Nexus 5 low/inconsistent throughput on 5GHz (and possibly 2.4GHz)

bsoft16384, 2015-01-08:
AP:
- 5GHz 802.11ac
- Channel 36 (also tried other channels); all low block channels are clear
- 80MHz (also tried narrower channels)
- US regulatory domain
- 50mW power (also occurs with lower values)

Device:
- Nexus 5
- Android 5.0.1
- BCM4339 1x1 802.11ac

OpenWRT built from trunk gets ~20Mbps of TCP throughput.
Stock WRT1900AC firmware (1.1.8.164461) with the same configuration gets ~110Mbps of TCP throughput.
Link rate shows 433Mbps on the Nexus 5 on both devices.

My Windows machine (BCM4352 2x2 802.11ac) gets >200Mbps on both OpenWRT and stock firmware.

alirz1, 2015-01-14:
I have the exact same issue with a Nexus 5. Also tested with an Ipad Air2, connected to the router also and get similar low thoroughput of about 20-30 mbps. Whereas stock linksys FW gives over 240mbps

The client links to the router at 433mbps,Have tried diffident channels etc.
Currently running: OpenWrt Chaos Calmer r43853

yuhhaurlin, 2015-01-28:
1. Get OpenWrt from main trunk.
2. Apply configuration from  https://downloads.openwrt.org/people/kaloz/mvebu-config.
3. Come out OpenWrt image (kernel 3.18.2 + mwlwifi 10.2.6.1-p4).
3. Test with LG Nexus 5 - Android 5.0.1, we can't find the problem described by you.

yuhhaurlin, 2015-01-28:
CHAOS CALMER (Bleeding Edge, r43965)

closed 2015-05-26 by yuhhaurlin