kaloz/mwlwifi #7: 5.0ghz Disconnect issues

LonnieReeder, 2015-01-09:
Using stock Linksys firmware or CC kernel version of Openwrt for the WRT1900 I can get the 5.0ghz to start, but I cannot get my Samsung Galaxy SIII cell phone to stay connected to it, once I am about 20-25 feet from the router. Once the phone disconnects from the 5.0Ghz range and connects to the 2.4ghz range the 5.0ghz range will turn off on the router and will not turn back on unless return to close enough range and force the phone to reconnect to that SSID There no other routers on my block that have 5.0Ghz signals and I live in a rural area.

DavidMcWRT, 2015-07-27:
@LonnieReeder 

5GHz radio waves have a much shorter range than 2.4GHz.  I lose 5GHz just outside my house, whereas I'll still be able to connect to 2.4GHz.  It could just be that, unless it's something else you're reporting?

LonnieReeder, 2015-07-27:
This was dealing with disconnect / lock issues that are still being reported. When I was running an earlier version of open wrt the router would lock, just after 5.0ghz would disconnect. The only way I could get the router to respond was to power cycle it. It was so consistent, every 20-24 hours, that I have returned back to the stock firmware, though I keep following the fourm. Currently I have no issues with the stock a day with issues I still see with the lock disconnect I have not ventured back to open source. 

This ticket can be closed as there are active tickets being worked that are dealing with the lock/disconnect issue. 

Thanks, 

Lonnie Reeder 

https://github.com/kaloz/mwlwifi/issues/7#issuecomment-125362325

closed 2015-07-27 by LonnieReeder