kaloz/mwlwifi #9: Use request_firmware()

mmilburn, 2015-01-13:
You can probably lift a bunch of code from mwl8k to do this.  
Some documentation here:
https://github.com/torvalds/linux/tree/master/Documentation/firmware_class

yuhhaurlin, 2015-01-13:
Thanks. I will check it.

raven-au, 2015-01-19:
Forgive me if I'm misunderstanding what your recommending here @mmilburn. If that's he case please explain further.

I think what your recommending is to change to depending on a user space script to load the firmware into the wireless device.

I really don't think that is a good idea and might not be doable anyway.

It's easy enough to see that the mwlwifi driver already uses request_firmware() to get the firmware and then loads it into the card itself and finally release the firmware once completed.

Personally, I prefer the self contained firmware handling rather than a user space (I don't trust user space) script for loading the firmware but also there's some specific register interactions done by the driver during the upload. They may be required by the wireless device so user space script upload might not be possible.

So I think it's fine the way it is and that's why I'm not recommending the change.
Anyway, just my thoughts on it.

Ian

mmilburn, 2015-01-20:
@raven-au You're not misunderstanding.  I'm incorrect.  I took a quick look at the firmware loading code and thought "Oh no, this is just like mrvl_wlan_v7drv". I also saw the firmware name being passed as a module parameter, then failed to do the requisite grep for request_firmware().  This ticket is not useful.

closed 2015-01-20 by mmilburn