kaloz/mwlwifi #11: Speed issues with AC wifi on WRT1900AC running OpenWrt Chaos Calmer r43853

alirz1, 2015-01-14:
My AC wifi speeds are really low with the router running openwrt. Tested with a Nexus 5 and an Apple Ipad Air2. Both devices connect to the router at 433mbps but get  low thoroughput of about 20-30 mbps(megabits NOT megabytes). Whereas with stock linksys FW i can transfer files at over 240mbps

I have have tried different channels, tramit power etc.
Currently running: OpenWrt Chaos Calmer r43853

yuhhaurlin, 2015-01-28:
1. Get OpenWrt from main trunk.
2. Apply configuration from  https://downloads.openwrt.org/people/kaloz/mvebu-config.
3. Come out OpenWrt image (kernel 3.18.2 + mwlwifi 10.2.6.1-p4).
3. Test with LG Nexus 5 - Android 5.0.1, we can't find the problem described by you.

yuhhaurlin, 2015-01-28:
CHAOS CALMER (Bleeding Edge, r43965)

closed 2015-05-26 by yuhhaurlin