kaloz/mwlwifi #12: Status information display

raven-au, 2015-01-19:
The detail of the status information display is dear to my heart and it isn't available yet.
I'm trying to work out what needs to be done to add it.

I've looked around quite a bit (at other mac80211 drivers too) and I wonder if adding rate information to the ieee80211_rx_status structure for received packets would help.

In any case Marvel plans to do this based on the comment:
       /* TODO: rate & antenna
        */
in mwl_rx.c:mwl_rx_prepare_status().

The problem is I'm having difficulty working out the encoding of the rate field of the mwl_rx_desc structure (that comes from the wireless device) for received packets.

Is there any way we can get this answered by Marvel please?

yuhhaurlin, 2015-03-30:
Should be included in next release (10.2.8.5.p1).

raven-au, 2015-03-30:
That's great news.
This is functionality that is really important to me too, thanks for working to include it.

yuhhaurlin, 2015-04-22:
Report receive frame rate information. 

commit 07df0cf2f71dec9c8b51d4707492540773b9944d

closed 2015-05-26 by yuhhaurlin