kaloz/mwlwifi #13: Marvel please add wireless station support

raven-au, 2015-01-22:
I hadn't noticed the driver doesn't support station mode.
I found out after tracking a WARN() splat in the log when trying to setup the device as a routed bridge.

Client mode is quite important to me but I believe it will be a primary use case for the coming WRT1200AC, assuming the wireless on the device (whatever it may be) is to be supported by this driver.

dangowrt, 2015-01-24:
So is Ad-Hoc/IBSS (and 802.11s?) as well as having multiple AP/STA/... interfaces sitting on top of the same wifi phy.
I suggest that at least IBSS/Ad-Hoc+AP(+AP(+...)) (common setup in wireless community mesh networks) as well as WDS-STA+AP (repeater done the right way[tm]), both setups working on a single wifi phy, should be included in the imaginary list of common use cases.

raven-au, 2015-01-26:
I thought I had posted the kernel WARN() log output.

I originally thought the problem was caused by an error exit from  mwl_mac80211_set_key() but that function isn't called before the WARN() so this is something else unexpected.

One thing is for sure the interface ends up not functional regardless of whether dhcp is used or a fixed ip address is assigned.

The information below shows a system configured as a pseudo-bridge but the WARN() can be duplicated from a default install by going to  the wireless configuration selecting "scan" and joining a wireless BSS SSID.

Anyway here is the log and system information:
Please press Enter to activate this console.
[   14.135525] NET: Registered protocol family 10
[   14.145802] tun: Universal TUN/TAP device driver, 1.6
[   14.150882] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[   14.161193] gre: GRE over IPv4 demultiplexor driver
[   14.166985] ip_gre: GRE over IPv4 tunneling driver
[   14.175788] ip6_tables: (C) 2000-2006 Netfilter Core Team
[   14.186635] Loading modules backported from Linux version master-2014-11-04-0-gf3660a2
[   14.194651] Backport generated by backports.git backports-20141023-2-g4ff890b
[   14.206864] cfg80211: Calling CRDA to update world regulatory domain
[   14.217377] cfg80211: World regulatory domain updated:
[   14.222642] cfg80211:  DFS Master region: unset
[   14.227015] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   14.236818] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   14.244864] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   14.252907] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   14.260935] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   14.269014] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   14.278574] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   14.286737] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   14.294811] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   14.305403] mwl_probe(): <<Marvell 802.11ac Wireless Network Driver version 10.2.6.1.p4>>
[   14.313702] pci 0000:00:02.0: enabling device (0140 -> 0142)
[   14.319440] mwlwifi 0000:02:00.0: enabling device (0140 -> 0142)
[   14.325660] mwl_alloc_pci_resource(): priv->iobase0 = d1000000
[   14.331607] mwl_alloc_pci_resource(): priv->iobase1 = d1200000
[   14.337486] mwl_alloc_pci_resource(): priv->pcmd_buf = d082a000  priv->pphys_cmd_buf = f8c0000
[   14.349199] mwl_fwdl_download_firmware(): fw download start 88
[   14.467702] mwl_fwdl_download_firmware(): FwSize = 113244 downloaded Size = 113244 curr_iteration 65509
[   14.986777] mwl_fwdl_download_firmware(): 0;
[   15.272028] mwl_fwdl_download_firmware(): 0;
[   15.334318] mwl_fwdl_download_firmware(): complete
[   15.339255] mwl_load_tx_pwr_tbl(): open tx power table </lib/firmware/Mamba_FCC_v1.2_5G4TX.ini>: OK
[   15.356446] mwl_fwcmd_get_hw_specs(): pcmd = d082a000
[   15.371628] mwl_fwcmd_set_hw_specs(): mwl_fwcmd_set_hw_specs ...
[   15.377668] mwl_fwcmd_set_hw_specs():   -->pPhysTxRing[0] = f8d0000
[   15.384010] mwl_fwcmd_set_hw_specs():   -->pPhysTxRing[1] = f8d4000
[   15.390306] mwl_fwcmd_set_hw_specs():   -->pPhysTxRing[2] = f8d8000
[   15.396646] mwl_fwcmd_set_hw_specs():   -->pPhysTxRing[3] = f8dc000
[   15.402990] mwl_fwcmd_set_hw_specs():   -->pPhysRxRing    = f8c4000
[   15.409289] mwl_fwcmd_set_hw_specs():   -->numtxq 4 wcbperq 256 totalrxwcb 256
[   15.453096] pci 0000:00:03.0: enabling device (0140 -> 0142)
[   15.458829] mwlwifi 0000:03:00.0: enabling device (0140 -> 0142)
[   15.465025] mwl_alloc_pci_resource(): priv->iobase0 = d1400000
[   15.470951] mwl_alloc_pci_resource(): priv->iobase1 = d1600000
[   15.476878] mwl_alloc_pci_resource(): priv->pcmd_buf = d0832000  priv->pphys_cmd_buf = f8c8000
[   15.487913] mwl_fwdl_download_firmware(): fw download start 88
[   15.605306] mwl_fwdl_download_firmware(): FwSize = 113244 downloaded Size = 113244 curr_iteration 65498
[   16.083764] mwl_fwdl_download_firmware(): 0;
[   16.343144] mwl_fwdl_download_firmware(): 0;
[   16.482497] mwl_fwdl_download_firmware(): complete
[   16.487397] mwl_load_tx_pwr_tbl(): open tx power table </lib/firmware/Mamba_FCC_v1.2_5G4TX.ini>: OK
[   16.504843] mwl_fwcmd_get_hw_specs(): pcmd = d0832000
[   16.520043] mwl_fwcmd_set_hw_specs(): mwl_fwcmd_set_hw_specs ...
[   16.526142] mwl_fwcmd_set_hw_specs():   -->pPhysTxRing[0] = f8e0000
[   16.532489] mwl_fwcmd_set_hw_specs():   -->pPhysTxRing[1] = f8e4000
[   16.538786] mwl_fwcmd_set_hw_specs():   -->pPhysTxRing[2] = f8e8000
[   16.545138] mwl_fwcmd_set_hw_specs():   -->pPhysTxRing[3] = f8ec000
[   16.551465] mwl_fwcmd_set_hw_specs():   -->pPhysRxRing    = f8cc000
[   16.557795] mwl_fwcmd_set_hw_specs():   -->numtxq 4 wcbperq 256 totalrxwcb 256
[   16.660817] i2c /dev entries driver
[... 312 lines cut ...]

eth0      Link encap:Ethernet  HWaddr 94:10:3E:05:D0:4E  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:709 errors:0 dropped:0 overruns:0 frame:0
          TX packets:591 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:532 
          RX bytes:65971 (64.4 KiB)  TX bytes:267827 (261.5 KiB)
          Interrupt:28 

lo        Link encap:Local Loopback  
[... cut ...]
          collisions:0 txqueuelen:0 
          RX bytes:137088 (133.8 KiB)  TX bytes:137088 (133.8 KiB)

wlan1     Link encap:Ethernet  HWaddr 00:25:9C:13:40:92  
          inet6 addr: fe80::225:9cff:fe13:4092/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2 errors:0 dropped:0 overruns:0 frame:0
          TX packets:89 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:282 (282.0 B)  TX bytes:30126 (29.4 KiB)

dangowrt, 2015-01-26:
ok, that is caused by misconfiguration. you cannot at a station to a bridge.
read the docs:
http://wiki.openwrt.org/doc/howto/clientmode

raven-au, 2015-01-26:
OK, I'll check http://wiki.openwrt.org/doc/recipes/relayclient again, which is what I was after.

I just noticed the WARN() back trace is devoid of useful information but I'll only fix that if it really looks like I have a valid configuration.

Thanks for the feedback.

raven-au, 2015-01-26:
Not setting the lan interface as type bridge does nothing to prevent the WARN() and the wlan1 still never gets an IP address via DHCP which I believe it should regardless of other configuration mistakes.

 The WARN() trace is:
[   32.228093] WARNING: CPU: 0 PID: 1543 at /work/openwrt/openwrt.git/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/compat-wireless-2014-11-04/net/mac80211/tx.c:2767 ieee80211_csa_is_complete+0x70c/0x7f0 [mac80211]()
[   32.249269] Modules linked in: pppoe ppp_async iptable_nat pptp pppox ppp_mppe ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE armada_thermal xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY ts_kmp ts_fsm ts_bm thermal_sys slhc nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_irc nf_nat_ftp nf_nat nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_irc nf_conntrack_ftp iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress tmp421 gpio_fan hwmon i2c_mv64xxx i2c_dev mwlwifi mac80211 cfg80211 compat ledtrig_usbdev ip6t_REJECT nf_reject_ipv6 ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat regmap_spi ipv6 sha1_generic ecb arc4 crypto_blkcipher usb_storage leds_tlc59116 regmap_i2c regmap_core i2c_core xhci_pci xhci_hcd sata_mv libata ehci_orion ehci_platform ehci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crypto_hash
[   32.368728] CPU: 0 PID: 1543 Comm: wpa_supplicant Not tainted 3.18.2 #6
[   32.375428] Backtrace:
[   32.377936] [<c001acbc>] (dump_backtrace) from [<c001b024>] (show_stack+0x18/0x1c)
[   32.385586]  r6:bf2861b4 r5:00000000 r4:c0410b18 r3:00000000
[   32.391358] [<c001b00c>] (show_stack) from [<c017f154>] (dump_stack+0x8c/0x9c)
[   32.398679] [<c017f0c8>] (dump_stack) from [<c0027008>] (warn_slowpath_common+0x78/0x94)
[   32.406854]  r5:00000009 r4:00000000
[   32.410504] [<c0026f90>] (warn_slowpath_common) from [<c00270c8>] (warn_slowpath_null+0x24/0x2c)
[   32.419372]  r8:cf308b40 r7:ce739a38 r6:cf351c60 r5:cf351c60 r4:00000000
[   32.426317] [<c00270a4>] (warn_slowpath_null) from [<bf2861b4>] (ieee80211_csa_is_complete+0x70c/0x7f0 [mac80211])
[   32.436894] [<bf285e64>] (ieee80211_csa_is_complete [mac80211]) from [<bf2862e0>] (ieee80211_beacon_get_tim+0x30/0x160 [mac80211])
[   32.448784]  r10:cf351480 r9:00000005 r8:cf30a7d0 r7:cf351c68 r6:cf351c60 r5:00000000
[   32.456798]  r4:00000000
[   32.459440] [<bf2862b0>] (ieee80211_beacon_get_tim [mac80211]) from [<bf2d2a54>] (mwl_mac80211_bss_info_changed+0x98/0x110 [mwlwifi])
[   32.471542]  r5:cf308b40 r4:000000e0
[   32.475243] [<bf2d29bc>] (mwl_mac80211_bss_info_changed [mwlwifi]) from [<bf2666a4>] (ieee80211_bss_info_change_notify+0xd8/0x124 [mac80211])
[   32.488039]  r8:cf30a7d0 r7:00000008 r6:00000000 r5:00000007 r4:00000060 r3:000000e0
[   32.496054] [<bf2665cc>] (ieee80211_bss_info_change_notify [mac80211]) from [<bf2925ec>] (ieee80211_cqm_rssi_notify+0x5f0/0x7f0 [mac80211])
[   32.508685]  r4:00000060
[   32.511352] [<bf292014>] (ieee80211_cqm_rssi_notify [mac80211]) from [<bf296ff0>] (ieee80211_mgd_auth+0x1dc/0x290 [mac80211])
[   32.522753]  r10:cf30ac50 r9:00000000 r8:00000000 r7:cf308b40 r6:cf351480 r5:cf886f40
[   32.530702]  r4:ce739b98
[   32.533429] [<bf296e14>] (ieee80211_mgd_auth [mac80211]) from [<bf27ae74>] (ieee80211_auth+0x18/0x1c [mac80211])
[   32.543712]  r8:00000000 r7:00000000 r6:cf308000 r5:cf351000 r4:cf308160
[   32.550622] [<bf27ae5c>] (ieee80211_auth [mac80211]) from [<bf246d70>] (cfg80211_mlme_auth+0x100/0x134 [cfg80211])
[   32.561194] [<bf246c70>] (cfg80211_mlme_auth [cfg80211]) from [<bf2435cc>] (nl80211_authenticate+0x254/0x2b4 [cfg80211])
[   32.572162]  r9:ce63c220 r8:00000000 r7:ce63c234 r6:00000000 r5:cf351000 r4:00000000
[   32.580068] [<bf243378>] (nl80211_authenticate [cfg80211]) from [<c028e460>] (genl_rcv_msg+0x2dc/0x37c)
[   32.589551]  r10:cf36e000 r9:ce63c210 r8:00000000 r7:ce694f40 r6:ce63c200 r5:bf24f484
[   32.597553]  r4:bf257bac
[   32.600125] [<c028e184>] (genl_rcv_msg) from [<c028d8a8>] (netlink_rcv_skb+0x58/0xb4)
[   32.608037]  r10:00000000 r9:00000000 r8:00000000 r7:00000044 r6:c028e184 r5:ce694f40
[   32.616049]  r4:ce63c200
[   32.618620] [<c028d850>] (netlink_rcv_skb) from [<c028e178>] (genl_rcv+0x30/0x3c)
[   32.626184]  r6:ce694f40 r5:ce694f40 r4:c040e05c r3:00000001
[   32.632016] [<c028e148>] (genl_rcv) from [<c028d204>] (netlink_unicast+0x144/0x230)
[   32.639781]  r5:ce6b6400 r4:cf9eb400
[   32.643488] [<c028d0c0>] (netlink_unicast) from [<c028d6f8>] (netlink_sendmsg+0x360/0x3bc)
[   32.651834]  r8:00000000 r7:ce6b6400 r6:ce739dd8 r5:ce739f5c r4:00000000
[   32.658668] [<c028d398>] (netlink_sendmsg) from [<c024f7a0>] (sock_sendmsg+0x90/0xa4)
[   32.666581]  r10:00000000 r9:00000000 r8:cf563040 r7:00000000 r6:ce739e60 r5:00000044
[   32.674596]  r4:c028d398
[   32.677171] [<c024f710>] (sock_sendmsg) from [<c025106c>] (___sys_sendmsg.part.36+0x22c/0x240)
[   32.685867]  r4:ce739f5c
[   32.688456] [<c0250e40>] (___sys_sendmsg.part.36) from [<c0252124>] (__sys_sendmsg+0x54/0x7c)
[   32.697066]  r10:00000000 r9:ce738000 r8:c0009024 r7:00000128 r6:00000000 r5:bea7d054
[   32.705079]  r4:cf563040
[   32.707649] [<c02520d0>] (__sys_sendmsg) from [<c025215c>] (SyS_sendmsg+0x10/0x14)
[   32.715300]  r6:01da4a50 r5:bea7d054 r4:01da4a50
[   32.720009] [<c025214c>] (SyS_sendmsg) from [<c0008e80>] (ret_fast_syscall+0x0/0x30)
[   32.727847] ---[ end trace 6bc900ad2cd8d915 ]---

raven-au, 2015-01-26:
Umm .. if I edit the comment with the WARN() trace I can see the expected function call information but it never shows in the comment itself ..... bit odd.

yuhhaurlin, 2015-03-30:
Station mode is added on 10.2.8.5.p0.

raven-au, 2015-03-30:
Thanks very much.

I've tested client mode by setting up my device as a pseudo bridge using relayd (as described in http://wiki.openwrt.org/doc/recipes/relayclient) and that worked as expected.

However, I haven't done more than check that it functioned, so I haven't used it for an extended period.

closed 2015-03-30 by raven-au