kaloz/mwlwifi #14: wireless disconnects after 25 seconds - Roamed or switched channel, reason #8

gibsonj, 2015-01-27:
I experiencing an issue exclusively on one of my Apple devices. I have tried the fix described in the other posts, but it does not help.

I have created this as a separate issue, because the symptoms appear to be slightly different than some of the other wireless disconnects that are being reported.

The device that is experiencing the problem is a Late 2010 MacBook Air. It is able to connect to the 2.4 and 5 GHz networks, and access the Internet, but after 25 seconds it consistently disconnects. It is able to connect to one of my other routers without an issue.

I have attached the system logs from the Mac.

2015-01-26 8:42:17.737 PM UserEventAgent[17]: Captive: [CNInfoNetworkActive:1709] en0: SSID '...text removed...' making interface primary (protected network)
2015-01-26 8:42:17.744 PM UserEventAgent[17]: Captive: CNPluginHandler en0: Evaluating
2015-01-26 8:42:17.747 PM UserEventAgent[17]: Captive: en0: Probing '...text removed...'
2015-01-26 8:42:17.758 PM networkd[144]: +[NETLedBelly stopFastFail] Clearing ledbelly failure cache
2015-01-26 8:42:17.765 PM networkd[144]: +[NETLedBelly stopFastFail] Clearing ledbelly failure cache
2015-01-26 8:42:17.772 PM configd[26]: network changed: v4(en0!:...text removed...) DNS+ Proxy+ SMB
2015-01-26 8:42:17.000 PM kernel[0]: en0: BSSID changed to ...text removed...
2015-01-26 8:42:17.813 PM sharingd[248]: 20:42:17.812 : SDStatusMonitor::kStatusWirelessPowerChanged
2015-01-26 8:42:18.017 PM UserEventAgent[17]: Captive: CNPluginHandler en0: Authenticated
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  Restoring session ...text removed...
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  get the reconnect token
2015-01-26 8:42:18.172 PM KernelEventAgent[74]: tid 54485244 received event(s) VQ_NOTRESP (1)
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  Restoring session ...text removed...
2015-01-26 8:42:18.000 PM kernel[0]: AFP_VFS afpfs_DoReconnect:  get the reconnect token
2015-01-26 8:42:18.172 PM KernelEventAgent[74]: tid 54485244 received event(s) VQ_NOTRESP (1)
2015-01-26 8:42:20.469 PM com.apple.xpc.launchd[1]: (com.apple.imfoundation.IMRemoteURLConnectionAgent) The _DirtyJetsamMemoryLimit key is not available on this platform.
2015-01-26 8:42:42.000 PM kernel[0]: MacAuthEvent en0   Auth result for: ...text removed...  MAC AUTH succeeded
2015-01-26 8:42:42.000 PM kernel[0]: wlEvent: en0 en0 Link UP virtIf = 0
2015-01-26 8:42:42.000 PM kernel[0]: wl0: Roamed or switched channel, reason #8, bssid ...text removed...
2015-01-26 8:42:42.000 PM kernel[0]: en0: BSSID changed to ...text removed...
2015-01-26 8:42:42.000 PM kernel[0]: en0::IO80211Interface::postMessage bssid changed

gibsonj, 2015-01-27:
I have closed this issue, as it is very similar to the issue noted <a href="https://github.com/kaloz/mwlwifi/issues/5">here</a>.

gibsonj, 2015-01-27:
I have re-opened as this is an ongoing issue that has not been resolved.

yuhhaurlin, 2015-03-30:
Can you check if this problem is still exited on 10.2.8.5.p0? Thanks.

pdfruth, 2015-03-30:
The 10.2.8.5.p0 updates have resolved this issue in my environment.

closed 2015-05-26 by yuhhaurlin