kaloz/mwlwifi #15: performance of wifi connection to usb storage drive attached to router is bad

LogicoZone, 2015-03-28:
Use case: 

USB 3 External HD formatted as ext4 and connect to router's USB 3 port. 

Wired connection to FTP, NFS, SAMBA share of the attached USB HD behaves normal. 

Wireless connection to FTP, NFS, SAMBA share of the same USB HD is very slow and pretty much stall on OS X machine. 

Wireless to other NFS/FTP/SAMBA share on the network (but not physically attached to router) works normal. 

The problem only exists on wireless access to usb storage attached to the router.

LogicoZone, 2015-03-29:
wrtpat provides a possible patch or fix to address this issue. Maybe this information will help Marvell to fix this issue. 

Post link:
https://forum.openwrt.org/viewtopic.php?pid=270354#p270354

yuhhaurlin, 2015-03-30:
Thanks. I will add this fix in next release. However, the fix should be added after delay free queue is full and it should also include QoS data.

https://forum.openwrt.org/viewtopic.php?pid=270354#p270354

pdfruth, 2015-03-31:
Hi yuhhaurlin,  I noticed the changes to mwl_tx.c and mwl_main.c in today's commit.
After checking out the latest commit, rebuilding, and testing, it appears the performance problem is still present.

yuhhaurlin, 2015-03-31:
Thanks. I will check it.

LogicoZone, 2015-03-31:
Thanks. It seems the latest checkin improved performance for myself.

yuhhaurlin, 2015-03-31:
Glad to know that. Thanks for your help.

pdfruth, 2015-04-01:
Latest commit, with changes to mwl_tx.c and mwl_rx.c, seems to have resolved the performance problem for me as well.
Thank you very much for your timely support and attention yuhhaurlin.

yuhhaurlin, 2015-04-01:
Thanks for your help. The fix would be included in 10.2.8.5.p1. Once if this version is ready, I will change version number.

closed 2015-04-22 by yuhhaurlin