kaloz/mwlwifi #17: IPv4-Address is shown as "?"

gufus, 2015-04-27:
Linux 3.18.11 #1 SMP Thu Apr 23 22:38:47 UTC 2015 armv7l GNU/Linux
openwrt r45566
trunk

I set up 2 wifi devices in Client mode both are connected and everything is working

WRT54GL dd-wrt r21676
 20:AA:4B:3E:5E:39  192.168.1.10
 
DIR-615-E  dd-wrt r14896
 BE:A3:86:4B:61:EE  192.168.1.11

 Luci > Network > Wifi

Note that the network to which we are connected is shown under "Associated Stations", and that the "IPv4-Address" is shown as "?".

see http://www.gypsy-designs.com/098.jpg

gufus, 2015-04-27:
http://www.gypsy-designs.com/446.jpg

yuhhaurlin, 2015-04-28:
LuCI uses the arptable to look up a suitable IP for the associated mac.

gufus, 2015-04-28:
Okay, got it.

Thanks.

closed 2015-04-28 by gufus