kaloz/mwlwifi #19: Wireless connection lost requiring hard reboot

thagabe, 2015-05-06:
Router: Linksys WRT1900AC
Version: OpenWrt Chaos Calmer r45610 / LuCI Master (git-15.125.30735-c38b4cd)
Kernel: Linux Kernel: 3.18.11

- Wireless clients stutter (Chromecast when watching a video 2.4GHz)
- Wireless connection signals Limited connection (Windows 10 build 10072 with proper wireless drivers + Windows 7 both on the same machine [Both intel AC wireless chip] + Mac OS X Yosemite 10.10.3 [Unibody 2008 using broadcom N] all in 5GHz)
- Wireless (Only/mainly on 5GHz; haven't seen this on 2.4GHz but [stevef9432204](https://forum.openwrt.org/viewtopic.php?pid=275308#p275308) has reported 2.4GHz) locking up and requiring a hard reboot [manually power cycling it]
- I have noticed this on kernel 4.0.1 (less frequent) which was the build I used before switching to 3.18.11.

yuhhaurlin, 2015-05-07:
1. Wireless clients stutter: can you give more detail information? Every channel of 2.4GHz has same phenomenon? Does the environment of 2.4GHz is clean?
2. Does it mean you can't connect to internet? If yes and you can ping to WRT1900AC, it is not problem of wireless.
3. Only wireless can't work? System is all right? If yes, can you check if related hostapd is running?

yuhhaurlin, 2015-05-08:
About wireless locking, can you specify the way to reproduce it? And if system is gone, can you hook console to see if there is any messages coming out when the problem happened?

closed 2015-05-22 by thagabe