kaloz/mwlwifi #21: Router randomly halts after connect/disconnect to wireless AP

Vanav, 2015-05-20:
Halt means router doesn't respond to ping wired or wireless. The only way to recover is power reset. Most often triggered after wireless clients connect or disconnect to router. I think this may be related to #20.

Syslog backtrace:
```
Mon May 18 23:19:42 2015 kern.err kernel: [ 5833.037794] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=350693 c=350692 q=1682)
Mon May 18 23:19:42 2015 kern.info kernel: [ 5833.047192] Task dump for CPU 0:
Mon May 18 23:19:42 2015 kern.info kernel: [ 5833.050437] kworker/u4:0    R running      0     6      2 0x00000002
Mon May 18 23:19:42 2015 kern.info kernel: [ 5833.056883] Workqueue: phy1 ieee80211_ba_session_work [mac80211]
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.062924] Backtrace: 
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.065403] [<c001ae24>] (dump_backtrace) from [<c001b18c>] (show_stack+0x18/0x1c)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.073002]  r6:c044d000 r5:c044d0c0 r4:cf830400 r3:00000006
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.078748] [<c001b174>] (show_stack) from [<c0044dd8>] (sched_show_task+0xc8/0xfc)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.086431] [<c0044d10>] (sched_show_task) from [<c0046fcc>] (dump_cpu_task+0x40/0x44)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.094376]  r5:c044d0c0 r4:00000000
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.098010] [<c0046f8c>] (dump_cpu_task) from [<c005f550>] (rcu_dump_cpu_stacks+0x70/0xb4)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.106299]  r4:00000000 r3:00000001
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.109929] [<c005f4e0>] (rcu_dump_cpu_stacks) from [<c00620e8>] (rcu_check_callbacks+0x24c/0x668)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.118921]  r9:c044d000 r8:c044d000 r7:0f999000 r6:00000692 r5:c0440214 r4:cfdd9214
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.126757] [<c0061e9c>] (rcu_check_callbacks) from [<c0064190>] (update_process_times+0x48/0x68)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.135661]  r10:c046fc80 r9:c046fc80 r8:c046fc00 r7:00000000 r6:00000000 r5:cf830400
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.143586]  r4:cf84a000
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.146146] [<c0064148>] (update_process_times) from [<c007224c>] (tick_sched_timer+0x210/0x254)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.154962]  r7:cf84b640 r6:cfdd9570 r5:0000054e r4:181ae998
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.160707] [<c007203c>] (tick_sched_timer) from [<c0064efc>] (__run_hrtimer.isra.35+0xa8/0x144)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.169525]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.177437]  r4:cfdd9570
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.180007] [<c0064e54>] (__run_hrtimer.isra.35) from [<c00651ac>] (hrtimer_interrupt+0x138/0x2ac)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.188998]  r7:00000001 r6:cfdd9368 r5:0000054e r4:181ae088
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.194731] [<c0065074>] (hrtimer_interrupt) from [<c0266c1c>] (armada_370_xp_timer_interrupt+0x4c/0x54)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.204245]  r10:c0487f98 r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c04516e8
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.212169]  r4:cfddcc40
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.214728] [<c0266bd0>] (armada_370_xp_timer_interrupt) from [<c005bbe8>] (handle_percpu_devid_irq+0x74/0x8c)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.224764]  r4:cf803c40 r3:c0266bd0
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.228400] [<c005bb74>] (handle_percpu_devid_irq) from [<c0057fd0>] (generic_handle_irq+0x34/0x44)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.237471]  r8:cf805000 r7:00000001 r6:cf84bd20 r5:00000000 r4:00000010 r3:c005bb74
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.245314] [<c0057f9c>] (generic_handle_irq) from [<c00582d8>] (__handle_domain_irq+0x90/0xa4)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.254045]  r4:c04408f8 r3:0000005f
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.257664] [<c0058248>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.266916]  r8:cf84b640 r7:00000001 r6:c0487fac r5:20000153 r4:c0011e14 r3:cf84b640
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.274761] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.283145] Exception stack(0xcf84b640 to 0xcf84b688)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.288228] b640: ce4a2838 00000000 00008ed6 00008ed5 ce648100 00000060 ccc62620 00000000
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.296435] b660: 00000000 000859b6 00000001 cf84b694 cf84b698 cf84b688 bf303658 c0011e14
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.304644] b680: 20000153 ffffffff
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.308154]  r10:00000001 r9:000859b6 r8:00000000 r7:cf84b674 r6:ffffffff r5:20000153
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.316067]  r4:c0011e14
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.318658] [<c0011de0>] (_raw_spin_lock) from [<bf303658>] (mwl_tx_xmit+0x458/0x8bc [mwlwifi])
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.327391] [<bf303200>] (mwl_tx_xmit [mwlwifi]) from [<bf2fe574>] (mwl_mac80211_tx+0x60/0x64 [mwlwifi])
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.336903]  r10:00000000 r9:00000002 r8:00000000 r7:ce405ca8 r6:ce4a0b40 r5:ce648100
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.344826]  r4:cf84b770
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.347417] [<bf2fe514>] (mwl_mac80211_tx [mwlwifi]) from [<bf2af424>] (ieee80211_nullfunc_get+0x15d0/0x168c [mac80211])
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.358364] [<bf2af084>] (ieee80211_nullfunc_get [mac80211]) from [<bf2b007c>] (ieee80211_get_buffered_bc+0x210/0x82c [mac80211])
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.370056]  r10:c0446350 r9:00000000 r8:ce4a0b40 r7:000005dc r6:ce4054c2 r5:00000000
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.377978]  r4:ce648100
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.380569] [<bf2affb0>] (ieee80211_get_buffered_bc [mac80211]) from [<bf2b1550>] (ieee80211_xmit+0xf0/0xf8 [mac80211])
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.391389]  r9:ce648100 r8:cc80f36e r7:ccc62000 r6:ce4a0b40 r5:ce4054c0 r4:ce648100
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.399265] [<bf2b1460>] (ieee80211_xmit [mac80211]) from [<bf2b18e0>] (__ieee80211_subif_start_xmit+0x88/0x90 [mac80211])
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.410346]  r8:cc9fb868 r7:cce83e00 r6:cc9fb800 r5:ce4054c0 r4:ce405000
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.417163] [<bf2b1858>] (__ieee80211_subif_start_xmit [mac80211]) from [<bf2b18fc>] (ieee80211_subif_start_xmit+0x14/0x1c [mac80211])
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.429288]  r5:ce405000 r4:ce648100
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.432927] [<bf2b18e8>] (ieee80211_subif_start_xmit [mac80211]) from [<c028ba5c>] (dev_hard_start_xmit+0x270/0x31c)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.443496] [<c028b7ec>] (dev_hard_start_xmit) from [<c02a598c>] (sch_direct_xmit+0xb4/0x218)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.452055]  r10:ce405000 r9:ce648100 r8:cc9fb868 r7:cce83e00 r6:cc9fb800 r5:cc9fb800
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.459979]  r4:00000000
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.462538] [<c02a58d8>] (sch_direct_xmit) from [<c028bd44>] (__dev_queue_xmit+0x23c/0x524)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.470919]  r10:00000000 r9:cc9fb868 r8:00000000 r7:cce83e00 r6:ce405000 r5:cc9fb800
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.478840]  r4:ce648100
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.481402] [<c028bb08>] (__dev_queue_xmit) from [<c028c040>] (dev_queue_xmit+0x14/0x18)
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.489523]  r10:c0446350 r9:cf3be4c0 r8:cf3be000 r7:cc80f382 r6:ce648100 r5:ce405000
Mon May 18 23:19:42 2015 kern.warn kernel: [ 5833.497434]  r4:ce648100
[... 85 lines cut ...]
Mon May 18 23:22:42 2015 kern.info kernel: [ 6013.082847] kworker/u4:0    R running      0     6      2 0x00000002
[... cut ...]
Mon May 18 23:22:42 2015 kern.warn kernel: [ 6013.201929]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
Mon May 18 23:22:42 2015 kern.warn kernel: [ 6013.209840]  r4:cfdd9570
Mon May 18 23:22:42 2015 kern.warn kernel: [ 6013.212411] [<c0064e54>] (__run_hrtimer.isra.35) from [<c00651ac>] (hrtimer_interrupt+0x138/0x2ac)
Mon May 18 23:22:42 2015 kern.warn kernel: [ 6013.221401]  r7:00000001 r6:cfdd9368 r5:00000578 r4:02baac82
...
```
Cross-post: https://forum.openwrt.org/viewtopic.php?pid=276687#p276687

yuhhaurlin, 2015-05-20:
Thanks. We will check it.

tusc, 2015-05-20:
I reported this last month in the openwrt forum: https://forum.openwrt.org/viewtopic.php?pid=268305#p268305

This seemed to occur periodically while on kernel 3.8.x. Ever since moving to 4.0 I haven't experienced this.

[24496.233412] INFO: rcu_sched self-detected stall on CPU { 0}  (t=60009 jiffies g
=83102 c=83101 q=61450)
[24496.242810] Task dump for CPU 0:
[24496.246055] kworker/u4:2    R running      0   359      2 0x00000002
[24496.252508] Workqueue: phy1 ieee80211_iface_work [mac80211]
[24496.258116] Backtrace: 
[24496.260594] [<c001ad94>] (dump_backtrace) from [<c001b0fc>] (show_stack+0x18/0x
1c)
[24496.268192]  r6:c0424540 r5:c0424600 r4:cfafb800 r3:00000006
[24496.273932] [<c001b0e4>] (show_stack) from [<c0044bf8>] (sched_show_task+0xc8/0
xfc)
[24496.281617] [<c0044b30>] (sched_show_task) from [<c0046dec>] (dump_cpu_task+0x4
0/0x44)
[24496.289554]  r5:c0424600 r4:00000000
[24496.293187] [<c0046dac>] (dump_cpu_task) from [<c005f36c>] (rcu_dump_cpu_stacks
+0x70/0xb4)
[24496.301483]  r4:00000000 r3:00000001
[24496.305125] [<c005f2fc>] (rcu_dump_cpu_stacks) from [<c0061f04>] (rcu_check_cal
lbacks+0x24c/0x668)
[24496.314116]  r9:c0424540 r8:c0424540 r7:0f9d7000 r6:0000f00a r5:c0402214 r4:cfd
d9214
[24496.321966] [<c0061cb8>] (rcu_check_callbacks) from [<c0063fac>] (update_proces
s_times+0x48/0x68)
[24496.330871]  r10:c0444a10 r9:c0444a10 r8:c044e1f0 r7:00000000 r6:00000000 r5:cf
afb800
[24496.338808]  r4:cf270000
[24496.341370] [<c0063f64>] (update_process_times) from [<c0072068>] (tick_sched_t
imer+0x210/0x254)
[24496.350186]  r7:cf2715d0 r6:cfdd9570 r5:00001647 r4:6308c1d4
[24496.355943] [<c0071e58>] (tick_sched_timer) from [<c0064d18>] (__run_hrtimer.is
ra.35+0xa8/0x144)
[24496.364761]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cf
dd93a0
[24496.372687]  r4:cfdd9570
[24496.375265] [<c0064c70>] (__run_hrtimer.isra.35) from [<c0064fc8>] (hrtimer_int
errupt+0x138/0x2ac)
[24496.384257]  r7:00000001 r6:cfdd9368 r5:00001647 r4:6308b7fc
[24496.390010] [<c0064e90>] (hrtimer_interrupt) from [<c02461a0>] (armada_370_xp_t
imer_interrupt+0x4c/0x54)
[24496.399525]  r10:c042c43c r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c0
410630
[24496.407461]  r4:cfddcc40
[24496.410023] [<c0246154>] (armada_370_xp_timer_interrupt) from [<c005ba04>] (han
dle_percpu_devid_irq+0x74/0x8c)
[24496.420059]  r4:cf803c40 r3:c0246154
[24496.423696] [<c005b990>] (handle_percpu_devid_irq) from [<c0057dec>] (generic_h
andle_irq+0x34/0x44)
[24496.432776]  r8:cf805000 r7:00000001 r6:cf271da0 r5:00000000 r4:00000010 r3:c00
5b990
[24496.440633] [<c0057db8>] (generic_handle_irq) from [<c00580f4>] (__handle_domai
n_irq+0x90/0xa4)
[24496.449363]  r4:c04028f8 r3:0000005f
[24496.452986] [<c0058064>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp
_handle_irq+0x64/0xe0)
[24496.462238]  r8:cf2715d0 r7:00000001 r6:c044a7d8 r5:20000153 r4:c0011d94 r3:cf2
715d0
[24496.470094] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_sv
c+0x40/0x54)
[24496.478479] Exception stack(0xcf2715d0 to 0xcf271618)
[24496.483561] 15c0:                                     ce4f23e4 00000000 0000d2f
0 0000d2ef
[24496.491779] 15e0: ce697280 00000061 00000000 00000000 cdc6f48c 00000001 0000072
6 cf271624
[24496.499995] 1600: cf271628 cf271618 bf3b4a68 c0011d94 20000153 ffffffff
[24496.506638]  r10:00000726 r9:00000001 r8:cdc6f48c r7:cf271604 r6:ffffffff r5:20
000153
[... 146 lines cut ...]
[24497.115946]  r4:cdc6f000
[24497.118552] [<bf34d0f0>] (ieee80211_send_bar [mac80211]) from [<bf34df28>] (iee
e80211_process_addba_resp+0xbc/0x11c [mac80211])
[24497.130067]  r7:00000000 r6:cdc6f21c r5:cdc6f000 r4:cd9c1280
[24497.135864] [<bf34de6c>] (ieee80211_process_addba_resp [mac80211]) from [<bf352
78c>] (ieee80211_iface_work+0x1d0/0x398 [mac80211])
[24497.147644]  r10:ce4f0efc r9:ce4f0b40 r8:cea3e8cc r7:cea3e480 r6:cea3e8bc r5:cd
cd6088
[24497.155570]  r4:cc87df40 r3:00000021
[24497.159230] [<bf3525bc>] (ieee80211_iface_work [mac80211]) from [<c00396f8>] (p
rocess_one_work+0x1f4/0x33c)
[24497.169006]  r10:00000000 r9:00000000 r8:cf270020 r7:cf831000 r6:cf830800 r5:ce
a3e8bc
[24497.176940]  r4:cf9ec680
[24497.179505] [<c0039504>] (process_one_work) from [<c003a1dc>] (worker_thread+0x
31c/0x50c)
[24497.187713]  r10:cf270000 r9:00000088 r8:cf270020 r7:cf9ec698 r6:cf830814 r5:cf
830800
[24497.195639]  r4:cf9ec680
[24497.198213] [<c0039ec0>] (worker_thread) from [<c003dbfc>] (kthread+0xf4/0xf8)
[24497.205464]  r10:00000000 r9:00000000 r8:00000000 r7:c0039ec0 r6:cf9ec680 r5:00
000000
[24497.213390]  r4:cf148ec0
[24497.215962] [<c003db08>] (kthread) from [<c0008f18>] (ret_from_fork+0x14/0x3c)
[24497.223213]  r7:00000000 r6:00000000 r5:c003db08 r4:cf148ec0

Vanav, 2015-05-20:
Other people confirm that kernel version changes nothing.

`webdestroya`:
> Just got a CPU hang after 1d 20h. You can see rcu_sched self-detected stall on CPU about 10 lines into the syslog below. I am running 4.0.2, r45669. I can't seem to get beyond 2 days of uptime. (I had this lockup previously the day I left for vacation)
```
Thu May 14 17:38:00 2015 daemon.info hostapd: wlan1: STA c8:bc:c8:xx:xx:xx WPA: group key handshake completed (RSN)
Thu May 14 17:39:02 2015 daemon.info hostapd: wlan0: STA 8c:29:37:xx:xx:xx WPA: group key handshake completed (RSN)
Thu May 14 17:39:03 2015 daemon.info hostapd: wlan0: STA ac:fd:ec:xx:xx:xx WPA: group key handshake completed (RSN)
Thu May 14 17:40:00 2015 cron.info crond[1185]: USER root pid 12899 cmd /sbin/fan_ctrl.sh
Thu May 14 17:45:00 2015 cron.info crond[1185]: USER root pid 13038 cmd /sbin/fan_ctrl.sh
Thu May 14 17:48:00 2015 daemon.info hostapd: wlan1: STA c8:bc:c8:xx:xx:xx WPA: group key handshake completed (RSN)
Thu May 14 17:49:02 2015 daemon.info hostapd: wlan0: STA 8c:29:37:xx:xx:xx WPA: group key handshake completed (RSN)
Thu May 14 17:49:02 2015 daemon.info hostapd: wlan0: STA ac:fd:ec:xx:xx:xx WPA: group key handshake completed (RSN)
Thu May 14 17:50:00 2015 cron.info crond[1185]: USER root pid 13044 cmd /sbin/fan_ctrl.sh
Thu May 14 17:51:22 2015 kern.err kernel: [162837.503596] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=310369 c=310368 q=72)
Thu May 14 17:51:22 2015 kern.info kernel: [162837.512926] Task dump for CPU 0:
Thu May 14 17:51:22 2015 kern.info kernel: [162837.516265] kworker/u4:2    R running      0   325      2 0x00000002
Thu May 14 17:51:22 2015 kern.info kernel: [162837.522794] Workqueue: phy0 mwl_watchdog_ba_events [mwlwifi]
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.528580] Backtrace:
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.531157] [<c001acb0>] (dump_backtrace) from [<c001b018>] (show_stack+0x18/0x1c)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.538849]  r6:c042fb40 r5:c042fc00 r4:cfa08000 r3:00000006
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.544704] [<c001b000>] (show_stack) from [<c0045d74>] (sched_show_task+0xcc/0x100)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.552574] [<c0045ca8>] (sched_show_task) from [<c00480c0>] (dump_cpu_task+0x40/0x44)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.560613]  r5:c042fc00 r4:00000000
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.564354] [<c0048080>] (dump_cpu_task) from [<c0061694>] (rcu_dump_cpu_stacks+0x70/0xb4)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.572735]  r4:00000000 r3:00000001
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.576468] [<c0061624>] (rcu_dump_cpu_stacks) from [<c0064500>] (rcu_check_callbacks+0x29c/0x79c)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.585553]  r9:c042fb40 r8:c04284f8 r7:c042fb40 r6:0f9b6000 r5:c0428dc8 r4:c0424840
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.593504] [<c0064264>] (rcu_check_callbacks) from [<c006673c>] (update_process_times+0x40/0x6c)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.602502]  r10:c0450d08 r9:c0450d08 r8:c0450c80 r7:cfb59800 r6:cfdd9428 r5:00000000
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.610536]  r4:cfa08000
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.613197] [<c00666fc>] (update_process_times) from [<c0074cd8>] (tick_sched_timer+0x210/0x254)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.622108]  r5:00009419 r4:0f31a860
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.625845] [<c0074ac8>] (tick_sched_timer) from [<c0067124>] (__run_hrtimer.isra.35+0xa8/0x144)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.634756]  r10:00000001 r9:cfdd9220 r8:00000000 r7:cfdd9220 r6:cfdd9220 r5:cfdd9258
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.642779]  r4:cfdd9428
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.645450] [<c006707c>] (__run_hrtimer.isra.35) from [<c0067794>] (hrtimer_interrupt+0x118/0x28c)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.654535]  r7:cfdd9220 r6:cfdd9230 r5:00009419 r4:0f319d98
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.660381] [<c006767c>] (hrtimer_interrupt) from [<c025cfa8>] (armada_370_xp_timer_interrupt+0x4c/0x54)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.669989]  r10:c0468e60 r9:000003ff r8:cfddcd80 r7:00000010 r6:cf80a380 r5:c0433524
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.678022]  r4:cfddcd80
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.680681] [<c025cf5c>] (armada_370_xp_timer_interrupt) from [<c005d288>] (handle_percpu_devid_irq+0x74/0x8c)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.690810]  r4:cf803c40 r3:c025cf5c
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.694545] [<c005d214>] (handle_percpu_devid_irq) from [<c00595b8>] (generic_handle_irq+0x34/0x44)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.703718]  r8:cf805000 r7:00000001 r6:00000000 r5:00000000 r4:00000010 r3:c005d214
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.711663] [<c0059584>] (generic_handle_irq) from [<c00598d0>] (__handle_domain_irq+0xa0/0xb4)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.720487]  r4:c04237b0 r3:0000005f
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.724223] [<c0059830>] (__handle_domain_irq) from [<c00086bc>] (armada_370_xp_handle_irq+0x64/0xe0)
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.733561]  r8:cfb59800 r7:00000001 r6:c0468e78 r5:20000153 r4:c0011ddc r3:cfb59800
[... cut ...]
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.771787] 9840: 20000153 ffffffff
Thu May 14 17:51:22 2015 kern.warn kernel: [162837.775391]  r10:00000001 r9:00f7e88f r8:00000000 r7:cfb59834 r6:ffffffff r5:20000153
```
Source: https://forum.openwrt.org/viewtopic.php?pid=276340#p276340

`lifehacksback`:
> My "stable" lasted 4 days before the hang so 4.0.2 is sadly still having the issue (as pointed out before there were no fixed introduced with 4.0.2)
Source: https://forum.openwrt.org/viewtopic.php?pid=276521#p276521

yuhhaurlin, 2015-05-21:
Thanks for your information.

gufus, 2015-05-22:
@yuhhaurlin 

BTY, I've seen that error mss before (it's back again}

https://forum.openwrt.org/viewtopic.php?pid=261726#p261726

ta7105, 2015-05-26:
Experienced a freeze here as well:
<pre><code>May 16 00:21:23 192.168.12.1 kernel: [1470059.151316] MSSFIX(wan): IN=br-lan OUT=pppoe-wan MAC=b4:75:0e:f7:22:0b:54:ee:75:04:6d:aa:08:00 SRC=192.168.12.101 DST=203.130.48.18 LEN=52 TOS=0x00 PREC=0x00 TTL=127 ID=3549 DF PROTO=TCP SPT=13266 DPT=80 WINDOW=8192 RES=0x00 SYN URGP=0 MARK=0x33
May 16 00:21:24 192.168.12.1 kernel: [1470059.540368] MSSFIX(wan): IN=br-lan OUT=pppoe-wan MAC=b4:75:0e:f7:22:0b:30:75:12:29:ff:e2:08:00 SRC=192.168.12.205 DST=216.58.209.168 LEN=60 TOS=0x00 PREC=0x00 TTL=63 ID=46417 DF PROTO=TCP SPT=45467 DPT=443 WINDOW=14600 RES=0x00 SYN URGP=0 MARK=0x33
May 16 00:21:29 192.168.12.1 pppd[1776]: No response to 5 echo-requests
May 16 00:21:29 192.168.12.1 pppd[1776]: Serial link appears to be disconnected.
May 16 00:21:29 192.168.12.1 pppd[1776]: Connect time 851.6 minutes.
May 16 00:21:29 192.168.12.1 pppd[1776]: Sent 3619708362 bytes, received 940543620 bytes.
May 16 00:22:25 192.168.12.1 kernel: [1470119.618170] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=29412057 c=29412056 q=1022)
May 16 00:22:25 192.168.12.1 kernel: [1470119.628107] Task dump for CPU 0:
May 16 00:22:25 192.168.12.1 kernel: [1470119.631533] kworker/u4:1    R running      0  1265      2 0x00000002
May 16 00:22:25 192.168.12.1 kernel: [1470119.638199] Workqueue: phy0 ieee80211_iface_work [mac80211]
May 16 00:22:25 192.168.12.1 kernel: [1470119.643987] Backtrace:
May 16 00:22:25 192.168.12.1 kernel: [1470119.646652] [<c001adb4>] (dump_backtrace) from [<c001b11c>] (show_stack+0x18/0x1c)
May 16 00:22:25 192.168.12.1 kernel: [1470119.654432]  r6:c0422d80 r5:c0422e40 r4:cfadd800 r3:00000006
May 16 00:22:25 192.168.12.1 kernel: [1470119.660377] [<c001b104>] (show_stack) from [<c0044b94>] (sched_show_task+0xc8/0xfc)
May 16 00:22:25 192.168.12.1 kernel: [1470119.668260] [<c0044acc>] (sched_show_task) from [<c0046d88>] (dump_cpu_task+0x40/0x44)
May 16 00:22:25 192.168.12.1 kernel: [1470119.676382]  r5:c0422e40 r4:00000000
May 16 00:22:25 192.168.12.1 kernel: [1470119.680207] [<c0046d48>] (dump_cpu_task) from [<c005f30c>] (rcu_dump_cpu_stacks+0x70/0xb4)
May 16 00:22:25 192.168.12.1 kernel: [1470119.688682]  r4:00000000 r3:00000001
May 16 00:22:25 192.168.12.1 kernel: [1470119.692496] [<c005f29c>] (rcu_dump_cpu_stacks) from [<c0061ea4>] (rcu_check_callbacks+0x24c/0x668)
May 16 00:22:25 192.168.12.1 kernel: [1470119.701668]  r9:c0422d80 r8:c0422d80 r7:0f9c3000 r6:000003fe r5:c0415214 r4:cfdd8214
May 16 00:22:25 192.168.12.1 kernel: [1470119.709716] [<c0061c58>] (rcu_check_callbacks) from [<c0063f4c>] (update_process_times+0x48/0x68)
May 16 00:22:25 192.168.12.1 kernel: [1470119.718802]  r10:c0444970 r9:c0444970 r8:c04448f0 r7:00000000 r6:00000000 r5:cfadd800
May 16 00:22:25 192.168.12.1 kernel: [1470119.726910]  r4:c0d7c000
May 16 00:22:25 192.168.12.1 kernel: [1470119.729671] [<c0063f04>] (update_process_times) from [<c0072008>] (tick_sched_timer+0x210/0x254)
May 16 00:22:25 192.168.12.1 kernel: [1470119.738669]  r7:c0d7d660 r6:cfdd8570 r5:0005390b r4:38910a26
May 16 00:22:25 192.168.12.1 kernel: [1470119.744599] [<c0071df8>] (tick_sched_timer) from [<c0064cb8>] (__run_hrtimer.isra.35+0xa8/0x144)
May 16 00:22:25 192.168.12.1 kernel: [1470119.753597]  r10:cfdd8368 r9:00000000 r8:cfdd8378 r7:00000001 r6:cfdd8368 r5:cfdd83a0
May 16 00:22:25 192.168.12.1 kernel: [1470119.761720]  r4:cfdd8570
May 16 00:22:25 192.168.12.1 kernel: [1470119.764465] [<c0064c10>] (__run_hrtimer.isra.35) from [<c0064f68>] (hrtimer_interrupt+0x138/0x2ac)
May 16 00:22:25 192.168.12.1 kernel: [1470119.773637]  r7:00000001 r6:cfdd8368 r5:0005390b r4:3890fe6e
May 16 00:22:25 192.168.12.1 kernel: [1470119.779581] [<c0064e30>] (hrtimer_interrupt) from [<c0254944>] (armada_370_xp_timer_interrupt+0x4c/0x54)
May 16 00:22:25 192.168.12.1 kernel: [1470119.789277]  r10:c045cb28 r9:000003ff r8:cfddbc40 r7:00000010 r6:cf80aa00 r5:c0426f2c
May 16 00:22:25 192.168.12.1 kernel: [1470119.797385]  r4:cfddbc40
May 16 00:22:25 192.168.12.1 kernel: [1470119.800143] [<c02548f8>] (armada_370_xp_timer_interrupt) from [<c005b9a4>] (handle_percpu_devid_irq+0x74/0x8c)
May 16 00:22:25 192.168.12.1 kernel: [1470119.810361]  r4:cf803c40 r3:c02548f8
May 16 00:22:25 192.168.12.1 kernel: [1470119.814177] [<c005b930>] (handle_percpu_devid_irq) from [<c0057d8c>] (generic_handle_irq+0x34/0x44)
May 16 00:22:25 192.168.12.1 kernel: [1470119.823435]  r8:cf805000 r7:00000001 r6:c0d7dd68 r5:00000000 r4:00000010 r3:c005b930
May 16 00:22:25 192.168.12.1 kernel: [1470119.831487] [<c0057d58>] (generic_handle_irq) from [<c0058094>] (__handle_domain_irq+0x90/0xa4)
May 16 00:22:25 192.168.12.1 kernel: [1470119.840398]  r4:c04158f8 r3:00000061
May 16 00:22:25 192.168.12.1 kernel: [1470119.844209] [<c0058004>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
May 16 00:22:25 192.168.12.1 kernel: [1470119.853641]  r8:c0d7d660 r7:00000001 r6:c045cb3c r5:20000153 r4:c0011da4 r3:c0d7d660
May 16 00:22:25 192.168.12.1 kernel: [1470119.861688] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
May 16 00:22:25 192.168.12.1 kernel: [1470119.870254] Exception stack(0xc0d7d660 to 0xc0d7d6a8)
May 16 00:22:25 192.168.12.1 kernel: [1470119.875515] d660: cea7a838 00000000 0000c2d8 0000c2d7 ce5f7280 00000000 c074de20 00000004
May 16 00:22:25 192.168.12.1 kernel: [1470119.883913] d680: 00000000 08c2a2f8 00000001 c0d7d6b4 c0d7d6b8 c0d7d6a8 bf32d658 c0011da4
[... cut ...]
May 16 00:25:25 192.168.12.1 kernel: [1470300.172045]  r9:cf3cb000 r8:cea1afc0 r7:ca9191c0 r6:c041c334 r5:cf3cb000 r4:ce5f7280
May 16 00:25:25 192.168.12.1 kernel: [1470300.180080] [<c0279288>] (dev_hard_start_xmit) from [<c0279970>] (__dev_queue_xmit+0x3cc/0x524)
May 16 00:25:25 192.168.12.1 kernel: [1470300.188991]  r10:ce977800 r9:ce86e478 r8:00000000 r7:ce8d5900 r6:cf3cb000 r5:c0415e10
May 16 00:25:25 192.168.12.1 kernel: [1470300.197112]  r4:ce5f7280
May 16 00:25:25 192.168.12.1 kernel: [1470300.199858] [<c02795a4>] (__dev_queue_xmit) from [<c0279adc>] (dev_queue_xmit+0x14/0x18)

</code></pre>

Vanav, 2015-06-13:
It looks like this issue affects latest stock firmware too. Reported multiple random halts for version 1.1.9 [1] and multiple random reboots [2].

[1] https://forum.openwrt.org/viewtopic.php?pid=279651#p279651
[2] https://forum.openwrt.org/viewtopic.php?pid=279593#p279593

Nihtilan, 2015-06-19:
@yuhhaurlin   not sure if this helps or not, but one of the DD-WRT devs said this is related to nvram corruption. See link: http://www.dd-wrt.com/phpBB2/viewtopic.php?t=256298&start=585 (3rd post from bottom)

tusc, 2015-07-14:
@yuhhaurlin

Any update on this? I just updated to trunk today and within 5 hours I get a stall message:

|       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 CHAOS CALMER (Bleeding Edge, r46364)
 -----------------------------------------------------
  * 1 1/2 oz Gin            Shake with a glassful
  * 1/4 oz Triple Sec       of broken ice and pour
  * 3/4 oz Lime Juice       unstrained into a goblet.
  * 1 1/2 oz Orange Juice
  * 1 tsp. Grenadine Syrup
 -----------------------------------------------------
root@wrt1900ac:~# uname -a
Linux wrt1900ac 4.0.8 #1 SMP Tue Jul 14 10:26:38 CDT 2015 armv7l GNU/Linux

Tue Jul 14 16:11:10 2015 kern.err kernel: [ 7791.964027] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=82522 c=82521 q=18045)
Tue Jul 14 16:11:10 2015 kern.info kernel: [ 7791.973360] Task dump for CPU 0:
Tue Jul 14 16:11:10 2015 kern.info kernel: [ 7791.976611] luci            R running      0 21964   2047 0x00000003
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7791.983024] Backtrace:
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7791.985526] [<c001ac90>] (dump_backtrace) from [<c001aff8>] (show_stack+0x18/0x1c)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7791.993124]  r6:c0437b40 r5:c0437c00 r4:ceb79c00 r3:00000000
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7791.998892] [<c001afe0>] (show_stack) from [<c0045d14>] (sched_show_task+0xcc/0x100)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.006690] [<c0045c48>] (sched_show_task) from [<c00480ec>] (dump_cpu_task+0x40/0x44)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.014647]  r5:c0437c00 r4:00000000
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.018289] [<c00480ac>] (dump_cpu_task) from [<c0061770>] (rcu_dump_cpu_stacks+0x70/0xb4)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.026590]  r4:00000000 r3:00000001
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.030229] [<c0061700>] (rcu_dump_cpu_stacks) from [<c00645dc>] (rcu_check_callbacks+0x29c/0x79c)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.039228]  r9:c0437b40 r8:c04304f8 r7:c0437b40 r6:0f9ad000 r5:c0430dc8 r4:c042d840
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.047106] [<c0064340>] (rcu_check_callbacks) from [<c0066818>] (update_process_times+0x40/0x6c)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.056018]  r10:c0458d48 r9:c0458d48 r8:c0458cc0 r7:ce1b77f0 r6:cfdd9428 r5:00000000
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.063954]  r4:ceb79c00
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.066542] [<c00667d8>] (update_process_times) from [<c0074e3c>] (tick_sched_timer+0x224/0x268)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.075368]  r5:00000716 r4:2f0c3ec5
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.079007] [<c0074c18>] (tick_sched_timer) from [<c0067248>] (__run_hrtimer.isra.35+0xa8/0x144)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.087831]  r10:00000001 r9:cfdd9220 r8:00000000 r7:cfdd9220 r6:cfdd9220 r5:cfdd9258
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.095779]  r4:cfdd9428
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.098353] [<c00671a0>] (__run_hrtimer.isra.35) from [<c00678b8>] (hrtimer_interrupt+0x118/0x28c)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.107351]  r7:cfdd9220 r6:cfdd9230 r5:00000716 r4:2f0c3295
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.113109] [<c00677a0>] (hrtimer_interrupt) from [<c0260b18>] (armada_370_xp_timer_interrupt+0x4c/0x54)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.122630]  r10:c0470ea8 r9:000003ff r8:cfddcd80 r7:00000010 r6:cf80a380 r5:c043b530
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.130578]  r4:cfddcd80
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.133151] [<c0260acc>] (armada_370_xp_timer_interrupt) from [<c005d364>] (handle_percpu_devid_irq+0x74/0x8c)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.143194]  r4:cf803c40 r3:c0260acc
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.146850] [<c005d2f0>] (handle_percpu_devid_irq) from [<c0059694>] (generic_handle_irq+0x34/0x44)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.155937]  r8:cf805000 r7:00000001 r6:00000000 r5:00000000 r4:00000010 r3:c005d2f0
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.163800] [<c0059660>] (generic_handle_irq) from [<c00599ac>] (__handle_domain_irq+0xa0/0xb4)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.172537]  r4:c042c7b0 r3:0000005f
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.176186] [<c005990c>] (__handle_domain_irq) from [<c00086bc>] (armada_370_xp_handle_irq+0x64/0xe0)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.185447]  r8:ce1b77f0 r7:00000001 r6:c0470ec0 r5:20000013 r4:bf3677c4 r3:ce1b77f0
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.193321] [<c0008658>] (armada_370_xp_handle_irq) from [<c00097c0>] (__irq_svc+0x40/0x54)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.201713] Exception stack(0xce1b77f0 to 0xce1b7838)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.206804] 77e0:                                     00000000 00001e33 00000000 00000000
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.215034] 7800: ccb64800 ce1b78d0 ccb64800 ceb0cb40 ce8f94c0 ccb64de4 10078002 ce1b7884
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.223251] 7820: 00000000 ce1b7838 0000e523 bf3677c4 20000013 ffffffff
[... cut ...]
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.432741]  r10:00000000 r9:ce1b6000 r8:c0009044 r7:00000128 r6:beced81c r5:00000000
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.440686]  r4:cf67f540
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.443261] [<c02703e0>] (__sys_sendmsg) from [<c027045c>] (SyS_sendmsg+0x10/0x14)
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.450867]  r6:b6fc3188 r5:00000000 r4:00000000
Tue Jul 14 16:11:10 2015 kern.warn kernel: [ 7792.455579] [<c027044c>] (SyS_sendmsg) from [<c0008ea0>] (ret_fast_syscall+0x0/0x3c)

kylejvrsa, 2015-07-15:
Hi,

FYI

Running Openwrt RC2 on my wrt1900ac, I intermittently experience this issue, it actually just happened. sometimes it's once a day sometimes it happens a couple of times a day. I have a cronjob which reboots the router everyday at 05:00 and even with the reboot the router still hangs intermittently. Is there anything I can do to help?

Thank you,

kevle, 2015-07-15:
Built from the trunk from 2 days ago I've run into this issue as well. Hardware version 1. It's apparently triggered by calling "iwinfo" and "iw dev wlan0 station dump" also "wifi" , "wifi down", "wifi up" seem to be problematic.

Here are my logs:

```
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.574066] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=153912 c=153911 q=5077)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.583482] Task dump for CPU 0:
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.583496] hostapd         R running      0  1430      1 0x00000002
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.583514] Backtrace: 
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.586014] [<c001af04>] (dump_backtrace) from [<c001b288>] (show_stack+0x18/0x1c)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.593611]  r6:c041edc0 r5:c041ee80 r4:cf9e2000 r3:dc8cb012
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.599380] [<c001b270>] (show_stack) from [<c0044fb8>] (sched_show_task+0xc8/0xfc)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.607089] [<c0044ef0>] (sched_show_task) from [<c0047250>] (dump_cpu_task+0x40/0x44)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.615045]  r5:c041ee80 r4:00000000
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.618687] [<c0047210>] (dump_cpu_task) from [<c005f95c>] (rcu_dump_cpu_stacks+0x70/0xb4)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.626987]  r4:00000000 r3:00000001
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.630627] [<c005f8ec>] (rcu_dump_cpu_stacks) from [<c00624f4>] (rcu_check_callbacks+0x24c/0x668)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.639624]  r9:c041edc0 r8:c041edc0 r7:0f9c7000 r6:000013d5 r5:c0412214 r4:cfdd9214
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.647501] [<c00622a8>] (rcu_check_callbacks) from [<c006459c>] (update_process_times+0x48/0x68)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.656413]  r10:c0440680 r9:c0440680 r8:c0440600 r7:00000000 r6:00000000 r5:cf9e2000
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.664358]  r4:ce286000
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.666937] [<c0064554>] (update_process_times) from [<c00727f0>] (tick_sched_timer+0x224/0x268)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.675760]  r7:ce287830 r6:cfdd9570 r5:000003ee r4:69ac6ae4
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.681516] [<c00725cc>] (tick_sched_timer) from [<c0065358>] (__run_hrtimer.isra.35+0xa8/0x144)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.690340]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.698289]  r4:cfdd9570
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.700860] [<c00652b0>] (__run_hrtimer.isra.35) from [<c0065608>] (hrtimer_interrupt+0x138/0x2ac)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.709857]  r7:00000001 r6:cfdd9368 r5:000003ee r4:69ac5dec
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.715627] [<c00654d0>] (hrtimer_interrupt) from [<c0255424>] (armada_370_xp_timer_interrupt+0x4c/0x54)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.725147]  r10:c0458828 r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c0422f60
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.733083]  r4:cfddcc40
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.735669] [<c02553d8>] (armada_370_xp_timer_interrupt) from [<c005bf88>] (handle_percpu_devid_irq+0x74/0x8c)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.745712]  r4:cf803c40 r3:c02553d8
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.749353] [<c005bf14>] (handle_percpu_devid_irq) from [<c0058370>] (generic_handle_irq+0x34/0x44)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.758438]  r8:cf805000 r7:00000001 r6:00000000 r5:00000000 r4:00000010 r3:c005bf14
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.766315] [<c005833c>] (generic_handle_irq) from [<c0058678>] (__handle_domain_irq+0x90/0xa4)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.775053]  r4:c04128f8 r3:0000005d
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.778689] [<c00585e8>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.787947]  r8:ce287830 r7:00000001 r6:c045883c r5:20000013 r4:bf152ab8 r3:ce287830
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.795823] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.804215] Exception stack(0xce287830 to 0xce287878)
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.809297] 7820:                                     ce01e9c0 ce287b30 00000000 00000000
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.817521] 7840: ce01e800 001de509 ce287900 ce540b40 cf2d64c0 ce01ede4 0004572c ce2878c4
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.825748] 7860: ce01eb40 ce287878 ce01e9d8 bf152ab8 20000013 ffffffff
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.832390]  r10:0004572c r9:ce01ede4 r8:cf2d64c0 r7:ce287864 r6:ffffffff r5:20000013
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.840336]  r4:bf152ab8
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.842983] [<bf152688>] (sta_set_sinfo [mac80211]) from [<bf1656fc>] (ieee80211_get_station+0x48/0x120 [mac80211])
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.853462]  r10:cf900000 r9:cf32e010 r8:00000000 r7:ce287900 r6:cf32e020 r5:00000000
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.861409]  r4:ce540f3c
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.864030] [<bf1656b4>] (ieee80211_get_station [mac80211]) from [<bf1248d4>] (nl80211_get_station+0x60/0x110 [cfg80211])
Jul 15 22:14:07 wrt1900ac kernel: [ 4322.875030]  r7:cf32e020 r6:cf2d6000 r5:ce540000 r4:ce287c74
[... cut ...]
Jul 15 22:59:21 wrt1900ac kernel: [  422.104984]  r10:00000000 r9:cde12000 r8:c0009024 r7:00000128 r6:00000000 r5:be9feb78
Jul 15 22:59:21 wrt1900ac kernel: [  422.112930]  r4:cf556840
Jul 15 22:59:21 wrt1900ac kernel: [  422.115516] [<c0263d6c>] (__sys_sendmsg) from [<c0263df8>] (SyS_sendmsg+0x10/0x14)
Jul 15 22:59:21 wrt1900ac kernel: [  422.123113]  r6:b6f7ea48 r5:00000000 r4:00000000
Jul 15 22:59:21 wrt1900ac kernel: [  422.127817] [<c0263de8>] (SyS_sendmsg) from [<c0008e80>] (ret_fast_syscall+0x0/0x38)
```

DavidMcWRT, 2015-07-27:
@Vanav 

see the new firmware bin file posted in issue #20.

Have tested on WRT1900AC v1 with OpenWRT CC-RC3 and the new bin file, and issue #20 seems to be solved for me, at least in my limited experimentation.

Would be interested to find out if this issue #21 still persists for you after the #20 fix.

tusc, 2015-07-28:
Now that issue #20 is resolved, any plans to work on this one? I just got another stalled message on the  ieee80211_iface_work process. This is on a recent trunk with the new wifi firmware 7.2.8.6. 

```
 -----------------------------------------------------
 CHAOS CALMER (Bleeding Edge, r46441)
 -----------------------------------------------------
  * 1 1/2 oz Gin            Shake with a glassful
  * 1/4 oz Triple Sec       of broken ice and pour
  * 3/4 oz Lime Juice       unstrained into a goblet.
  * 1 1/2 oz Orange Juice
  * 1 tsp. Grenadine Syrup
 -----------------------------------------------------
root@wrt1900ac:/# uname -a
Linux wrt1900ac 4.0.9 #2 SMP Sat Jul 25 17:46:35 CDT 2015 armv7l GNU/Linux
root@wrt1900ac:/#

[71974.327601] ieee80211 phy0: interface: 3, change: 0x100
[71974.333013] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[71974.355015] ieee80211 phy0: interface: 3, change: 0x3e
[71980.484386] ieee80211 phy0: interface: 3, change: 0x100
[71980.489943] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[71980.511761] ieee80211 phy0: interface: 3, change: 0x3e
[74219.966804] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=938310 c=938309 q=90)
[74219.976048] Task dump for CPU 0:
[74219.979300] kworker/u4:1    R running      0 31952      2 0x00000002
[74219.985796] Workqueue: phy1 ieee80211_iface_work [mac80211]
[74219.991409] Backtrace:
[74219.993902] [<c001add4>] (dump_backtrace) from [<c001b15c>] (show_stack+0x18/0x1c)
[74220.001508]  r6:c043bb40 r5:c043bc00 r4:cf993000 r3:dc8ba209
[74220.007280] [<c001b144>] (show_stack) from [<c0046068>] (sched_show_task+0xcc/0x100)
[74220.015063] [<c0045f9c>] (sched_show_task) from [<c0048440>] (dump_cpu_task+0x40/0x44)
[74220.023016]  r5:c043bc00 r4:00000000
[74220.026656] [<c0048400>] (dump_cpu_task) from [<c0061b30>] (rcu_dump_cpu_stacks+0x70/0xb4)
[74220.034957]  r4:00000000 r3:00000001
[74220.038607] [<c0061ac0>] (rcu_dump_cpu_stacks) from [<c006499c>] (rcu_check_callbacks+0x29c/0x79c)
[74220.047606]  r9:c043bb40 r8:c04344f8 r7:c043bb40 r6:0f9aa000 r5:c0434dc8 r4:c0430840
[74220.055470] [<c0064700>] (rcu_check_callbacks) from [<c0066bd8>] (update_process_times+0x40/0x6c)
[74220.064380]  r10:c045ccf8 r9:c045ccf8 r8:c045cc70 r7:cbcd7748 r6:cfdd9428 r5:00000000
[74220.072328]  r4:cf993000
[74220.074903] [<c0066b98>] (update_process_times) from [<c0075308>] (tick_sched_timer+0x224/0x268)
[74220.083726]  r5:00004380 r4:b1ed7d85
[74220.087376] [<c00750e4>] (tick_sched_timer) from [<c0067608>] (__run_hrtimer.isra.35+0xa8/0x144)
[74220.096191]  r10:00000001 r9:cfdd9220 r8:00000000 r7:cfdd9220 r6:cfdd9220 r5:cfdd9258
[74220.104136]  r4:cfdd9428
[74220.106707] [<c0067560>] (__run_hrtimer.isra.35) from [<c0067c78>] (hrtimer_interrupt+0x118/0x28c)
[74220.115705]  r7:cfdd9220 r6:cfdd9230 r5:00004380 r4:b1ed726d
[74220.121475] [<c0067b60>] (hrtimer_interrupt) from [<c0262468>] (armada_370_xp_timer_interrupt+0x4c/0x54)
[74220.130996]  r10:c0474e60 r9:000003ff r8:cfddcd80 r7:00000010 r6:cf80a380 r5:c043f524
[74220.138944]  r4:cfddcd80
[74220.141516] [<c026241c>] (armada_370_xp_timer_interrupt) from [<c005d6b8>] (handle_percpu_devid_irq+0x74/0x8c)
[74220.151558]  r4:cf803c40 r3:c026241c
[74220.155194] [<c005d644>] (handle_percpu_devid_irq) from [<c00599e8>] (generic_handle_irq+0x34/0x44)
[74220.164278]  r8:cf805000 r7:00000001 r6:cbcd7d78 r5:00000000 r4:00000010 r3:c005d644
[74220.172150] [<c00599b4>] (generic_handle_irq) from [<c0059d00>] (__handle_domain_irq+0xa0/0xb4)
[74220.180888]  r4:c042f7b0 r3:0000005f
[74220.184522] [<c0059c60>] (__handle_domain_irq) from [<c00086bc>] (armada_370_xp_handle_irq+0x64/0xe0)
[74220.193781]  r8:cbcd7748 r7:00000001 r6:c0474e78 r5:20000153 r4:c0011e74 r3:cbcd7748
[74220.201656] [<c0008658>] (armada_370_xp_handle_irq) from [<c00097c0>] (__irq_svc+0x40/0x54)
[74220.210048] Exception stack(0xcbcd7748 to 0xcbcd7790)
[74220.215132] 7740:                   ce74ee78 00000000 0000da41 0000da40 00000060 ce5b8d00
[74220.223356] 7760: 00000000 cc368dd8 00000000 00000001 cd6e0dc8 cbcd779c cbcd77a0 cbcd7790
[74220.231580] 7780: bf2ba810 c0011e74 20000153 ffffffff
[74220.236655]  r10:cd6e0dc8 r9:00000001 r8:00000000 r7:cbcd777c r6:ffffffff r5:20000153
[74220.244600]  r4:c0011e74
[74220.247229] [<c0011e40>] (_raw_spin_lock) from [<bf2ba810>] (mwl_tx_xmit+0x200/0x8f0 [mwlwifi])
[74220.255989] [<bf2ba610>] (mwl_tx_xmit [mwlwifi]) from [<bf2b5738>] (mwl_mac80211_tx+0x50/0x54 [mwlwifi])
[74220.265508]  r10:cf0fac78 r9:00000000 r8:00000002 r7:00000000 r6:ce74cb60 r5:ce5b8d00
[74220.273454]  r4:ce5b8d00
[74220.276086] [<bf2b56e8>] (mwl_mac80211_tx [mwlwifi]) from [<bf263ac8>] (ieee80211_nullfunc_get+0x11c0/0x1380 [mac80211])
[74220.287000]  r4:cbcd78c0 r3:bf2b56e8
[... 87 lines cut ...]
[74220.856986] [<c003ec38>] (kthread) from [<c0008f48>] (ret_from_fork+0x14/0x2c)
[74220.864238]  r7:00000000 r6:00000000 r5:c003ec38 r4:cdcd21c0
[74224.437081] ieee80211 phy0: interface: 3, change: 0x100
[74224.442363] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[74224.464530] ieee80211 phy0: interface: 3, change: 0x3e
[74235.156808] INFO: rcu_bh detected stalls on CPUs/tasks: { 0} (detected by 1, t=6002 jiffies, g=1909, c=1908, q=4)
[74235.167175] Task dump for CPU 0:
[74235.170420] kworker/u4:1    R running      0 31952      2 0x00000002
[74235.176927] Workqueue: phy1 ieee80211_iface_work [mac80211]
[74235.182531] Backtrace:
[74235.185065] [<cbcd6014>] (0xcbcd6014) from [<cd6fe218>] (0xcd6fe218)
[... cut ...]
[74400.023799] [<c001add4>] (dump_backtrace) from [<c001b15c>] (show_stack+0x18/0x1c)
[74400.031404]  r6:c043bb40 r5:c043bc00 r4:cf993000 r3:dc8ba209
[74400.037177] [<c001b144>] (show_stack) from [<c0046068>] (sched_show_task+0xcc/0x100)
[74400.044959] [<c0045f9c>] (sched_show_task) from [<c0048440>] (dump_cpu_task+0x40/0x44)
[74400.052910]  r5:c043bc00 r4:00000000
[74400.056556] [<c0048400>] (dump_cpu_task) from [<c0061b30>] (rcu_dump_cpu_stacks+0x70/0xb4)
[74400.064857]  r4:00000000 r3:00000001
```

RedShamilton, 2015-08-02:
I have been getting these since I bought the WRT1900ACv1. I actually got the same behavior with the stock firmware where the WiFi stops working and only a hard reset fixes the issue. I installed OpenWRT in hopes that this would go away (and for some of my PhD research related to WiFi) but the problem persisted. I have just recently installed the latest version of OpenWRT CC RC3 and still observe this behavior. Though I haven't taken the time to log everything and trace the root cause, it seems to me that this occurs when I use the 802.11ac radio and transfer a decent amount of data at ac's high-throughputs. As a result, I have increased the fans script interval to a shorter period to see if this was a heat related issue. This hasn't seem to help the issue. I have also written a cron job that reboots the server once every other day. This may be masking the issue, also I think that there are times when the system completely stalls and so the script can't run.

I am going to setup persistent logging to a syslog server so I can maybe better understand this issue and help resolve it.  If anyone has any suggestions for additional logging to enable I would be interested to hear them.

kevle, 2015-08-02:
@RedShamilton it looks like the latest driver is not in RC3 yet [if I read this correctly](https://forum.openwrt.org/viewtopic.php?pid=285680#p285680).

RedShamilton, 2015-08-02:
@kevle yes, according to this post it does seem that way. Do you have a link to an erratic of fixes this latest driver fixes? Also, I have begun collecting logs now on my syslog server so maybe I can see what is happening as well. However, if this latest driver does indeed fix this issue, I guess there is no point in my putting in effort to discover what is happening on the old driver version.

fecaleagle, 2015-08-02:
The latest driver seems to address #21, but I don't believe that it addresses #20.  I was never experiencing full halts on a WRT1200AC, only issue #21 after dissassociating.

NemoAlex, 2015-08-09:
I have tried the latest driver, basically works fine. Better than the version before.
Everything is good when using 2.4Ghz.
But still have some issues on 5Ghz:
1. Samba server is crazy slow. But downloading files from internet still fast. Reboot could solve this problem for a while.
2. Sometimes SSH to router stucks. Not sure if 2.4Ghz has this problem too.

I have tried to use different Channel and Transmit Power, same problems.

tusc, 2015-08-11:
@RedShamilton,

I posted this question on the wrt1900ac as I am in the same boat as you, always experiencing a lookup regardless of the build: https://forum.openwrt.org/viewtopic.php?pid=287177#p287177

mjevans, 2015-08-23:
I was finally able to view my version of this occurrence (netconsole would have been very helpful) using dmesg -c and a loop via ssh...

Is this the same issue or a different one?

 CHAOS CALMER (Bleeding Edge, r46584)
Linux OpenWrt 3.18.20 #1 SMP Thu Aug 13 13:49:40 UTC 2015 armv7l GNU/Linux

[514965.250811] ieee80211 phy0: interface: 3, change: 0x100
[514965.250854] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[514965.272946] ieee80211 phy0: interface: 3, change: 0x3e
[514965.283735] ieee80211 phy0: interface: 3, change: 0x100
[514965.283814] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[514965.305906] ieee80211 phy0: interface: 3, change: 0x3e
[515105.256865] ieee80211 phy0: interface: 3, change: 0x100
[515105.256910] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[515105.278939] ieee80211 phy0: interface: 3, change: 0x3e
[515105.289256] ieee80211 phy0: interface: 3, change: 0x100
[515105.289299] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[515105.311323] ieee80211 phy0: interface: 3, change: 0x3e
[515116.989864] ieee80211 phy0: interface: 3, change: 0x100
[515116.989908] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[515117.011960] ieee80211 phy0: interface: 3, change: 0x3e
[515117.022184] ieee80211 phy0: interface: 3, change: 0x100
[515117.022229] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[515117.044293] ieee80211 phy0: interface: 3, change: 0x3e
[515203.803030] ieee80211 phy0: result error
[515204.815312] ieee80211 phy0: result error
[515332.012543] ieee80211 phy0: result error
[515359.415586] ieee80211 phy0: result error
[515360.427144] ieee80211 phy0: result error
[515425.526685] ieee80211 phy0: result error
[515426.539297] ieee80211 phy0: result error
[515685.451446] ieee80211 phy0: result error
[515686.486868] ieee80211 phy0: result error
[516551.585348] ieee80211 phy0: result error
[516552.596039] ieee80211 phy0: result error
[516553.606379] ieee80211 phy0: result error
[517117.766091]  r9:c0466700 r8:ce034700 r7:000005dc r6:ce034700 r5:cdc19900 r4:ce034700
[517117.774159] [<c02bf07c>] (ip_forward_finish) from [<c02bf4c0>] (ip_forward+0x378/0x47c)
[517117.782291]  r4:ce613890 r3:00000002
[517117.786015] [<c02bf148>] (ip_forward) from [<c02bd81c>] (ip_rcv_finish+0x3b8/0x3f4)
[517117.793795]  r8:ce034700 r7:00000000 r6:ce034700 r5:cd70b100 r4:cd70b150
[517117.800708] [<c02bd464>] (ip_rcv_finish) from [<c02bddf4>] (ip_rcv+0x3e0/0x4a4)
[517117.808130]  r7:00000000 r6:03090933 r5:cd70b100 r4:00000050
[517117.813982] [<c02bda14>] (ip_rcv) from [<c028ca2c>] (__netif_receive_skb_core+0x6cc/0x768)
[517117.822372]  r10:c044a360 r9:00000000 r8:cf9e6800 r7:00000008 r6:c044a34c r5:c044b79c
[517117.830405]  r4:ce034700
[517117.833064] [<c028c360>] (__netif_receive_skb_core) from [<c028cb3c>] (__netif_receive_skb+0x74/0x88)
[517117.842410]  r10:cfddaa30 r9:cfddaa24 r8:00000040 r7:cfdda9d4 r6:00000000 r5:00000000
[517117.850445]  r4:cfddaa34
[517117.853104] [<c028cac8>] (__netif_receive_skb) from [<c028ddbc>] (process_backlog+0x90/0x164)
[517117.861752]  r5:00000000 r4:cfddaa34
[517117.865475] [<c028dd2c>] (process_backlog) from [<c028dc34>] (net_rx_action+0x88/0x180)
[517117.873602]  r10:cfdda9c8 r9:c0448100 r8:cfdda9c0 r7:0000012b r6:00000040 r5:cfddaa34

mjevans, 2015-08-28:
A better crash happened last night.
```
[302255.844306] ieee80211 phy0: interface: 3, change: 0x3e
[302322.195954] ieee80211 phy0: interface: 3, change: 0x100
[302322.195998] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[302322.217889] ieee80211 phy0: interface: 3, change: 0x3e
[302322.228261] ieee80211 phy0: interface: 3, change: 0x100
[302322.228312] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[302322.250058] ieee80211 phy0: interface: 3, change: 0x3e
[302344.879603] ieee80211 phy0: interface: 3, change: 0x100
[302344.879649] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[302344.901250] ieee80211 phy0: interface: 3, change: 0x3e
[302344.911752] ieee80211 phy0: interface: 3, change: 0x100
[302344.911806] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[302344.933642] ieee80211 phy0: interface: 3, change: 0x3e
[302412.200671] ieee80211 phy0: interface: 3, change: 0x100
[302412.200715] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[302412.222958] ieee80211 phy0: interface: 3, change: 0x3e
[302412.233408] ieee80211 phy0: interface: 3, change: 0x100
[302412.233455] ieee80211 phy0: wmm:26, rsn:24, rsn48:22, ht:62, vht:0
[302412.255194] ieee80211 phy0: interface: 3, change: 0x3e
[317299.065932] ieee80211 phy0: result error
[317300.093774] ieee80211 phy0: result error
[321428.498168] INFO: rcu_sched self-detected stall on CPU { 0} (t=6000 jiffies g=2908553 c=2908552 q=831)
[321428.507758] Task dump for CPU 0:
[321428.511096] kworker/u4:0 R running 0 6 2 0x00000002
[321428.517666] Workqueue: phy0 ieee80211_iface_work [mac80211]
[321428.523365] Backtrace: 
[321428.525942] [<c001af94>] (dump_backtrace) from [<c001b318>] (show_stack+0x18/0x1c)
[321428.533634] r6:c0451000 r5:c04510c0 r4:cf830400 r3:dc8cb014
[321428.539492] [<c001b300>] (show_stack) from [<c004521c>] (sched_show_task+0xc8/0xfc)
[321428.547274] [<c0045154>] (sched_show_task) from [<c00474b4>] (dump_cpu_task+0x40/0x44)
[321428.555312] r5:c04510c0 r4:00000000
[321428.559052] [<c0047474>] (dump_cpu_task) from [<c005fbc4>] (rcu_dump_cpu_stacks+0x70/0xb4)
[321428.567432] r4:00000000 r3:00000001
[321428.571167] [<c005fb54>] (rcu_dump_cpu_stacks) from [<c006275c>] (rcu_check_callbacks+0x24c/0x668)
[321428.580252] r9:c0451000 r8:c0451000 r7:0f995000 r6:0000033f r5:c0444214 r4:cfdd9214
[321428.588203] [<c0062510>] (rcu_check_callbacks) from [<c0064804>] (update_process_times+0x48/0x68)
[321428.597202] r10:c0473b70 r9:c0473b70 r8:c0473af0 r7:00000000 r6:00000000 r5:cf830400
[321428.605234] r4:cf84a000
[321428.607895] [<c00647bc>] (update_process_times) from [<c0072a58>] (tick_sched_timer+0x224/0x268)
[321428.616806] r7:cf84b2a0 r6:cfdd9570 r5:00012455 r4:2d7273ef
[321428.622662] [<c0072834>] (tick_sched_timer) from [<c00655c0>] (run_hrtimer.isra.35+0xa8/0x144)
[321428.631573] r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
[321428.639608] r4:cfdd9570
[321428.642267] [<c0065518>] (run_hrtimer.isra.35) from [<c0065870>] (hrtimer_interrupt+0x138/0x2ac)
[321428.651351] r7:00000001 r6:cfdd9368 r5:00012455 r4:2d72680f
[321428.657195] [<c0065738>] (hrtimer_interrupt) from [<c0269d7c>] (armada_370_xp_timer_interrupt+0x4c/0x54)
[321428.666802] r10:c048be98 r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c04556e8
[321428.674835] r4:cfddcc40
[321428.677495] [<c0269d30>] (armada_370_xp_timer_interrupt) from [<c005c1ec>] (handle_percpu_devid_irq+0x74/0x8c)
[321428.687624] r4:cf803c40 r3:c0269d30
[321428.691364] [<c005c178>] (handle_percpu_devid_irq) from [<c00585d4>] (generic_handle_irq+0x34/0x44)
[321428.700537] r8:cf805000 r7:00000001 r6:cf84bd58 r5:00000000 r4:00000010 r3:c005c178
[321428.708498] [<c00585a0>] (generic_handle_irq) from [<c00588dc>] (handle_domain_irq+0x90/0xa4)
[321428.717315] r4:c04448f8 r3:0000005e
[321428.721050] [<c005884c>] (handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
[321428.730396] r8:cf84b2a0 r7:00000001 r6:c048beac r5:20000153 r4:c0011ecc r3:cf84b2a0
[321428.738353] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (irq_svc+0x40/0x54)
[321428.746824] Exception stack(0xcf84b2a0 to 0xcf84b2e8)
[321428.752002] b2a0: cea22e78 00000000 00008664 00008663 00000000 cf9991c0 00000000 cd50c5d8
[321428.760319] b2c0: 00000004 00000001 cd5cddc8 cf84b2f4 cf84b2f8 cf84b2e8 bf43a80c c0011ecc
[321428.768626] b2e0: 20000153 ffffffff
[321428.772222] r10:cd5cddc8 r9:00000001 r8:00000004 r7:cf84b2d4 r6:ffffffff r5:20000153
[321428.780253] r4:c0011ecc
[321428.782942] [<c0011e98>] (_raw_spin_lock) from [<bf43a80c>] (mwl_tx_xmit+0x200/0x8f0 [mwlwifi])
[321428.791796] [<bf43a60c>] (mwl_tx_xmit [mwlwifi]) from [<bf435738>] (mwl_mac80211_tx+0x50/0x54 [mwlwifi])
[321428.801406] r10:cf2e8c78 r9:00000000 r8:00000002 r7:00000000 r6:cea20b60 r5:cf9991c0
[321428.809436] r4:cf9991c0
[321428.812151] [<bf4356e8>] (mwl_mac80211_tx [mwlwifi]) from [<bf3da998>] (ieee80211_nullfunc_get+0x11c0/0x1380 [mac80211])
[... 14 lines cut ...]
[321428.939024] [<c028eee0>] (dev_hard_start_xmit) from [<c02a93c4>] (sch_direct_xmit+0xb4/0x218)
[321428.947677] r10:cf2e8000 r9:cf9991c0 r8:cdd17868 r7:cfb7aa00 r6:cdd17800 r5:cdd17800
[321428.955711] r4:00000000
[321428.958381] [<c02a9310>] (sch_direct_xmit) from [<c028f438>] (dev_queue_xmit+0x23c/0x524)
[321428.966850] r10:00000000 r9:cdd17868 r8:00000000 r7:cfb7aa00 r6:cf2e8000 r5:cdd17800
[321428.974880] r4:cf9991c0
[321428.977539] [<c028f1fc>] (dev_queue_xmit) from [<c028f734>] (dev_queue_xmit+0x14/0x18)
[... cut ...]
[321429.041198] [<c0322a28>] (br_deliver) from [<c0320ba4>] (br_dev_xmit+0x234/0x250)
[321429.048818] [<c0320970>] (br_dev_xmit) from [<c028f150>] (dev_hard_start_xmit+0x270/0x31c)
[321429.057200] r9:cd02f278 r8:00000000 r7:cf897700 r6:cf3d8000 r5:cf3d8000 r4:cf9991c0
[321429.065159] [<c028eee0>] (dev_hard_start_xmit) from [<c028f5c8>] (dev_queue_xmit+0x3cc/0x524)
[321429.073983] r10:00000000 r9:cd02f278 r8:00000000 r7:cf897700 r6:cf3d8000 r5:c0444e10
[321429.082018] r4:cf9991c0
[321429.084676] [<c028f1fc>] (dev_queue_xmit) from [<c028f734>] (dev_queue_xmit+0x14/0x18)
```

davidc502, 2015-09-03:
It took 16 days for a CPU stall, but captured both of the logs this time.  It was the type of crash that causes the entire router to freeze... No Wifi, can't even ping the gateway. If someone wants more of the logs let me know.

From the Kernel Log.

[1139506.644751] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=7831721 c=7831720 q=309)
[1139506.654430] Task dump for CPU 0:
[1139506.657855] kworker/u4:1    R running      0 32761      2 0x00000002
[1139506.664500] Workqueue: phy1 ieee80211_ba_session_work [mac80211]
[1139506.670722] Backtrace:
[1139506.673387] [<c001ae14>] (dump_backtrace) from [<c001b17c>] (show_stack+0x18/0x1c)
[1139506.681167]  r6:c044cfc0 r5:c044d080 r4:ce64f400 r3:00000006
[1139506.687112] [<c001b164>] (show_stack) from [<c0044e64>] (sched_show_task+0xc8/0xfc)
[1139506.694996] [<c0044d9c>] (sched_show_task) from [<c00470fc>] (dump_cpu_task+0x40/0x44)
[1139506.703119]  r5:c044d080 r4:00000000
[1139506.706944] [<c00470bc>] (dump_cpu_task) from [<c005f79c>] (rcu_dump_cpu_stacks+0x70/0xb4)
[1139506.715419]  r4:00000000 r3:00000001
[1139506.719233] [<c005f72c>] (rcu_dump_cpu_stacks) from [<c0062334>] (rcu_check_callbacks+0x24c/0x668)
[1139506.728406]  r9:c044cfc0 r8:c044cfc0 r7:0f99a000 r6:00000135 r5:c043f214 r4:cfdd9214
[1139506.736456] [<c00620e8>] (rcu_check_callbacks) from [<c00643dc>] (update_process_times+0x48/0x68)
[1139506.745543]  r10:c046fa70 r9:c046fa70 r8:c046f9f0 r7:00000000 r6:00000000 r5:ce64f400
[1139506.753651]  r4:ce634000
[1139506.756412] [<c0064394>] (update_process_times) from [<c007250c>] (tick_sched_timer+0x224/0x268)
[1139506.765411]  r7:ce635648 r6:cfdd9570 r5:00040c59 r4:eb87c32c
[1139506.771341] [<c00722e8>] (tick_sched_timer) from [<c0065198>] (__run_hrtimer.isra.35+0xa8/0x144)
[1139506.780339]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
[1139506.788460]  r4:cfdd9570
[1139506.791206] [<c00650f0>] (__run_hrtimer.isra.35) from [<c0065448>] (hrtimer_interrupt+0x138/0x2ac)
[1139506.800377]  r7:00000001 r6:cfdd9368 r5:00040c59 r4:eb87b7ec
[1139506.806320] [<c0065310>] (hrtimer_interrupt) from [<c02653e8>] (armada_370_xp_timer_interrupt+0x4c/0x54)
[1139506.816016]  r10:c0487d58 r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c04514f8
[1139506.824125]  r4:cfddcc40
[1139506.826885] [<c026539c>] (armada_370_xp_timer_interrupt) from [<c005be34>] (handle_percpu_devid_irq+0x74/0x8c)
[1139506.837103]  r4:cf803c40 r3:c026539c
[1139506.840919] [<c005bdc0>] (handle_percpu_devid_irq) from [<c005821c>] (generic_handle_irq+0x34/0x44)
[1139506.850179]  r8:cf805000 r7:00000001 r6:ce635d78 r5:00000000 r4:00000010 r3:c005bdc0
[1139506.858230] [<c00581e8>] (generic_handle_irq) from [<c0058524>] (__handle_domain_irq+0x90/0xa4)
[1139506.867143]  r4:c043f8f8 r3:0000005e
[1139506.870955] [<c0058494>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
[1139506.880388]  r8:ce635648 r7:00000001 r6:c0487d6c r5:20000153 r4:c0011e14 r3:ce635648
[1139506.888437] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)

This was from the system log.

Wed Sep  2 21:51:22 2015 kern.err kernel: [1139506.644751] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=7831721 c=7831720 q=309)
Wed Sep  2 21:51:22 2015 kern.info kernel: [1139506.654430] Task dump for CPU 0:
Wed Sep  2 21:51:22 2015 kern.info kernel: [1139506.657855] kworker/u4:1    R running      0 32761      2 0x00000002
Wed Sep  2 21:51:22 2015 kern.info kernel: [1139506.664500] Workqueue: phy1 ieee80211_ba_session_work [mac80211]
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.670722] Backtrace:
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.673387] [<c001ae14>] (dump_backtrace) from [<c001b17c>] (show_stack+0x18/0x1c)
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.681167]  r6:c044cfc0 r5:c044d080 r4:ce64f400 r3:00000006
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.687112] [<c001b164>] (show_stack) from [<c0044e64>] (sched_show_task+0xc8/0xfc)
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.694996] [<c0044d9c>] (sched_show_task) from [<c00470fc>] (dump_cpu_task+0x40/0x44)
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.703119]  r5:c044d080 r4:00000000
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.706944] [<c00470bc>] (dump_cpu_task) from [<c005f79c>] (rcu_dump_cpu_stacks+0x70/0xb4)
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.715419]  r4:00000000 r3:00000001
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.719233] [<c005f72c>] (rcu_dump_cpu_stacks) from [<c0062334>] (rcu_check_callbacks+0x24c/0x668)
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.728406]  r9:c044cfc0 r8:c044cfc0 r7:0f99a000 r6:00000135 r5:c043f214 r4:cfdd9214
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.736456] [<c00620e8>] (rcu_check_callbacks) from [<c00643dc>] (update_process_times+0x48/0x68)
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.745543]  r10:c046fa70 r9:c046fa70 r8:c046f9f0 r7:00000000 r6:00000000 r5:ce64f400
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.753651]  r4:ce634000
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.756412] [<c0064394>] (update_process_times) from [<c007250c>] (tick_sched_timer+0x224/0x268)
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.765411]  r7:ce635648 r6:cfdd9570 r5:00040c59 r4:eb87c32c
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.771341] [<c00722e8>] (tick_sched_timer) from [<c0065198>] (__run_hrtimer.isra.35+0xa8/0x144)
[... cut ...]
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.870955] [<c0058494>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.880388]  r8:ce635648 r7:00000001 r6:c0487d6c r5:20000153 r4:c0011e14 r3:ce635648
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.888437] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
Wed Sep  2 21:51:22 2015 kern.warn kernel: [1139506.897002] Exception stack(0xce635648 to 0xce635690)

ozbenh, 2015-09-05:
Has anybody tried building the kernel with lockdep, spinlock debugging etc.. enabled ?

 it *looks* like it might be deadlocking on a driver internal spinlock in the tx path (mwl_tx_xmit), possibly as a result of interrupting an existing lock holder (ie, driver locking bugs ?) though it's not clear as the driver doesn't seem to show up in the call chain. Maybe an A->B/B->A deadlock with the other processor.

notnyt, 2015-09-05:
@ozbenh They redid a bunch of that code in 10.3.0.8.  Basically, waiting for the next version for more testing :(

mjevans, 2015-09-11:
Instead of an RC a full release

OpenWrt Chaos Calmer 15.05 / LuCI Master (git-15.233.47308-791ca8b)

Let's see if this resolves the WiFi crash issues...

tusc, 2015-09-11:
@mjevans,

don't hold your breath. Nothing has changed with the wifi driver. The final release of Chaos Calmer has no bearing on the state of the wifi driver.

davidc502, 2015-09-19:
Router locked up twice in a span of 12 hours... I've had 5 lock ups since going to CC. Thinking about reverting back to RC3.

Another complete lockup --- just happened.

Fri Sep 18 16:10:23 2015 kern.err kernel: [208175.526942] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=1751521 c=1751520 q=355)
Fri Sep 18 16:10:23 2015 kern.info kernel: [208175.536529] Task dump for CPU 0:
Fri Sep 18 16:10:23 2015 kern.info kernel: [208175.539868] kworker/u4:1    R running      0    15      2 0x00000002
Fri Sep 18 16:10:23 2015 kern.info kernel: [208175.546395] Workqueue: phy0 0xbf1af090 [mac80211@bf19f000+0x54aac]
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.552704] Backtrace:
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.555266] Function entered at [<c001ad40>] from [<c001b09c>]
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.561217]  r6:c03e2fc0 r5:c03e3080 r4:cf886000 r3:00000006
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.567055] Function entered at [<c001b084>] from [<c0044d84>]
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.573000] Function entered at [<c0044cbc>] from [<c004701c>]
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.578950]  r5:c03e3080 r4:00000000
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.582659] Function entered at [<c0046fdc>] from [<c005f658>]
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.588610]  r4:00000000 r3:00000001
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.592319] Function entered at [<c005f5e8>] from [<c00621f0>]
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.598270]  r9:c03e2fc0 r8:c03e2fc0 r7:0fa04000 r6:00000163 r5:c03d5214 r4:cfdd9214
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.606205] Function entered at [<c0061fa4>] from [<c0064298>]
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.612156]  r10:c04058f0 r9:c04058f0 r8:c0405870 r7:00000000 r6:00000000 r5:cf886000
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.620189]  r4:cf892000
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.622835] Function entered at [<c0064250>] from [<c0072398>]
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.628786]  r7:cf893658 r6:cfdd9570 r5:0000bd54 r4:91bd5be7
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.634614] Function entered at [<c0072174>] from [<c0065054>]
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.640566]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
Fri Sep 18 16:10:23 2015 kern.warn kernel: [208175.648598]  r4:cfdd9570

kylejvrsa, 2015-10-22:
@yuhhaurlin 

I disabled WMM Mode on al the SSIDs and had 22 days uptime then I switched on WMM mode again and within a day the router locked up

I have many different clients connected: Macbooks, Imacs, Linux laptops, android phones, iphones etc

I usually have around 40 clients connected at any one time during the day.

thagabe, 2015-10-22:
@kylejvrsa Stock or Openwrt/DD-WRT?

yuhhaurlin, 2015-10-23:
@kylejvrsa How about version 10.3.0.10?

kylejvrsa, 2015-10-23:
@thagabe

I am running CC final and I can definitely say that it has something to do with WMM mode.

@yuhhaurlin 

As I am running CC Final it is 10.3.0.3 driver that I am using. Unfortunately the router is somewhat in a production environment so I am unable to test the latest driver as I have read that other seem to still have issues with it and I only have the one router. However disabling WMM mode is a huge speed drop in wifi, I have traded of speed for stability and the people are happy. When WMM mode is enabled it doesn't take long to crash and what I have noted is that it's the people that let their devices go into sleep/hibernate that are having the most issues with WMM (when coming back from sleep/hibernate mode they seem to be connected for a little while and lose connection but device still says connected...a wifi on and off on end device gets them connected again)  and might also be the ones causing the router to lock up...but I am speculating here now...

yuhhaurlin, 2015-11-12:
Does anyone still encounter this problem with 10.3.0.14?

kylejvrsa, 2015-11-12:
@yuhhaurlin I can't test this as I am running CC Final on wrt1900ac v1 need to get a driver compiled for wrt1900ac with cc final. Will see if I can get hold of one and revert back on my findings :)

DavidMcWRT, 2015-11-12:
@kylejvrsa 
In case you've not seen it yet, someone has posted it for you on the openwrt forum:
https://forum.openwrt.org/viewtopic.php?pid=299537#p299537

gufus, 2015-11-12:
@yuhhaurlin 

Using username "root".
Unauthorized access is prohibited!
Security is enabled, and your IP address has been logged.
Authenticating with public key "rsa-key-20120810"

BusyBox v1.24.1 (2015-11-10 00:30:38 CET) built-in shell (ash)

root@AC-DD-WRT:~# uptime
 15:27:29 up 1 day, 23:29,  load average: 0.00, 0.02, 0.04
root@AC-DD-WRT:~#

using 10.3.0.14

Firmware Version
DD-WRT v3.0-r28112 std (11/10/15)

No problems here :+1:

tusc, 2015-11-12:
@yuhhaurlin,

I think I have been the most vocal with his bug as I have posted several kernel logs on this issue.

I can say with confidence this issue appears to be behind us (knocks on wood). I've been running for over 6 days without an issue since using the new mwlwifi driver. Previously I would get stalls every day.

DavidMcWRT, 2015-11-12:
@tusc 
please verify which driver version (and openwrt build and hardware revision) so that we can be clear which version is OK.

tusc, 2015-11-13:
It was on 10.3.0.13 with Openwrt trunk on 4.1.12. I have since upgraded to 10.3.0.14.

DavidMcWRT, 2015-11-13:
thanks - def. useful to know.

Would be good to also compare people's experiences with a 3.18.2x kernel for it there's a 15.05.1 release of CC.

I also note that a recent driver was submitted upstream to the kernel devs and comments were made about some of the locks and locks types used and nearness of TX and RX variables, although of which might be contributing to various issues (and maybe the irq one too), so progress at least is being made - credit to Marvell.

closed 2015-11-24 by yuhhaurlin

RedShamilton, 2015-11-24:
This was just closed, was the issue resolved? If so, in what version of the driver/kernel?

yuhhaurlin, 2015-11-24:
It looks like no one reported this problem on 10.3.0.14, I close it. If anyone still finds this problem, he can reopen it.

PHLAK, 2015-12-01:
I'm still running 15.05 final and experiencing this issue.  Is it possible to upgrade only the kmod-mwlwifi to v10.3.0.14 without doing any other updates?

NemoAlex, 2015-12-01:
@PHLAK Sure. Use [this file](https://github.com/NemoAlex/openwrt-wrt1900ac-docs/blob/master/ipks/chaos_calmer-15.05-mvebu/kmod-mwlwifi_3.18.20%2B10.3.0.14-20151130-1_mvebu.ipk)
Just opkg install it.
Or you can build it by yourself. Check the usage of [SDK](https://wiki.openwrt.org/doc/howto/obtain.firmware.sdk).

PHLAK, 2015-12-01:
@NemoAlex My opkg repos are still pointing at the 15.05 release URLs which still contains v10.0.0.3 and if I update my repo URL I'll get more than just this package (which is undesirable).

    root@OpenWrt:~# opkg list | grep kmod-mwlwifi
    kmod-mwlwifi - 3.18.20+10.3.0.3-20150804-1 - Marvell 88W8864 wireless driver

---

    root@OpenWrt:~# cat /etc/opkg.conf 
    dest root /
    dest ram /tmp
    lists_dir ext /var/opkg-lists
    option overlay_root /overlay
    src/gz chaos_calmer_base http://downloads.openwrt.org/chaos_calmer/15.05/mvebu/generic/packages/base
    src/gz chaos_calmer_luci http://downloads.openwrt.org/chaos_calmer/15.05/mvebu/generic/packages/luci
    src/gz chaos_calmer_packages http://downloads.openwrt.org/chaos_calmer/15.05/mvebu/generic/packages/packages
    src/gz chaos_calmer_routing http://downloads.openwrt.org/chaos_calmer/15.05/mvebu/generic/packages/routing
    src/gz chaos_calmer_telephony http://downloads.openwrt.org/chaos_calmer/15.05/mvebu/generic/packages/telephony
    src/gz chaos_calmer_management http://downloads.openwrt.org/chaos_calmer/15.05/mvebu/generic/packages/management
    option check_signature 1

fecaleagle, 2015-12-01:
@PHLAK You can just extract the contents of the IPK file and copy the contents manually.  That's what I did, b/c I'm still on RC3.

glyph, 2015-12-25:
@PHLAK - any luck with the new driver? I'd had pretty good luck so far, but had a couple of hard crashes in the last few days which make me think I might be suffering from this as well; wondering if I should give that a try.

It's worth noting that non-committers can't reopen issues, so it's stuck closed.

PHLAK, 2015-12-31:
@glyph With the new driver I don't get any more hard lock ups that require restarting the router.  However, it seems like I get "soft" crashes occasionally where my devices stop having access to the network/internet even though they never drop their WiFi connection.  This usually lasts anywhere from 10 seconds to a minute or two then things go back to normal without any action on my part.  It's not perfect, but it's better than where I was before.

fecaleagle, 2015-12-31:
@PHLAK This, exactly!  I'm on a WRT1200AC, and I never experienced hard crashes, but I have always received these "soft crashes" and still do on v10.3.0.14.  I always reboot the router when we catch it, because for us, it seems to last longer than a minute or so, but connected devices stay connected to Wifi with no network connectivity.  Disconnected devices are at this point unable to connect.  We catch it once every two days or so, and reboot the router to resolve it.  I'm curious what logs would be liable to catch anything useful, because I feel that this should be reported.  I've been lazy and just kept hoping someone else would bring it up, so thank you.  If anybody can identify helpful logs, I'd be happy to post them and a new issue.

PHLAK, 2015-12-31:
I'll grab some logs as well if someone knows what I should grab.

jbeagley52, 2016-01-04:
```
BusyBox v1.23.2 (2015-10-08 13:37:22 CEST) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 CHAOS CALMER (Chaos Calmer, r47165)
 -----------------------------------------------------
  * 1 1/2 oz Gin            Shake with a glassful
  * 1/4 oz Triple Sec       of broken ice and pour
  * 3/4 oz Lime Juice       unstrained into a goblet.
  * 1 1/2 oz Orange Juice
  * 1 tsp. Grenadine Syrup
 -----------------------------------------------------
root@chaos:~# uname -a
Linux chaos 3.18.21 #1 SMP Thu Oct 8 13:41:29 CEST 2015 armv7l GNU/Linux
root@chaos:~# cat /tmp/sysinfo/board_name 
armada-385-linksys-caiman
root@chaos:~# cat /tmp/sysinfo/model 
Linksys WRT1200AC
```

I'm not sure if this is the same problem or not, but at least one client on my network seems to "hang" and will only transfer at ridiculously slow speeds, effectively making it impossible to use without rebooting. I can't seem to reproduce the issue by doing anything in particular. I just have to wait for it to happen.

```
[   18.592311] ieee80211 phy1: Channel: 5: 0x0 0x0 0xf
[   18.597219] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0 
[   18.602936] ieee80211 phy1: Channel: 6: 0x0 0x0 0xf
[   18.607855] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0 
[   18.613567] ieee80211 phy1: Channel: 7: 0x0 0x0 0xf
[   18.618491] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0 
[   18.624205] ieee80211 phy1: Channel: 8: 0x0 0x0 0xf
[   18.629112] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0 
[   18.634821] ieee80211 phy1: Channel: 9: 0x0 0x0 0xf
[   18.639729] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0 
[   18.645442] ieee80211 phy1: Channel: 10: 0x0 0x0 0xf
[   18.650442] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0 
[   18.656154] ieee80211 phy1: Channel: 11: 0x0 0x0 0xf
[   18.661162] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0 
[   18.666878] ieee80211 phy0: phy0: AU
[   18.670476] ieee80211 phy0: Channel: 36: 0x0 0x0 0xf
[   18.675494] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[   18.682594] ieee80211 phy0: Channel: 40: 0x0 0x0 0xf
[   18.687588] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[   18.694691] ieee80211 phy0: Channel: 44: 0x0 0x0 0xf
[   18.699685] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[   18.706786] ieee80211 phy0: Channel: 48: 0x0 0x0 0xf
[   18.711800] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[   18.718884] ieee80211 phy0: Channel: 52: 0x0 0x0 0xf
[   18.723897] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[   18.731005] ieee80211 phy0: Channel: 56: 0x0 0x0 0xf
[   18.735999] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[   18.743109] ieee80211 phy0: Channel: 60: 0x0 0x0 0xf
[   18.748105] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[   18.755215] ieee80211 phy0: Channel: 64: 0x0 0x0 0xf
[   18.760224] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[   18.767333] ieee80211 phy0: Channel: 100: 0x0 0x0 0xf
[   18.772446] ieee80211 phy0: 17 17 17 17 17 17 17 15 17 17 17 14 17 17 17 14 
[   18.779567] ieee80211 phy0: Channel: 104: 0x0 0x0 0xf
[   18.784676] ieee80211 phy0: 17 17 17 17 17 17 17 15 17 17 17 14 17 17 17 14 
[   18.791776] ieee80211 phy0: Channel: 108: 0x0 0x0 0xf
[   18.796862] ieee80211 phy0: 17 17 17 17 17 17 17 15 17 17 17 14 17 17 17 14 
[   18.803982] ieee80211 phy0: Channel: 112: 0x0 0x0 0xf
[   18.809064] ieee80211 phy0: 17 17 17 17 17 17 17 15 17 17 17 14 17 17 17 14 
[   18.816192] ieee80211 phy0: Channel: 116: 0x0 0x0 0xf
[   18.821291] ieee80211 phy0: 17 17 17 17 17 17 17 15 17 17 17 14 17 17 17 14 
[   18.828392] ieee80211 phy0: Channel: 120: 0x0 0x0 0xf
[... 99 lines cut ...]
[ 3403.023432] ieee80211 phy1: no memory
[ 3409.589127] ieee80211 phy1: no memory
[ 3409.592821] ieee80211 phy1: no memory
[ 3409.596517] ieee80211 phy1: no memory
[ 3409.600199] ieee80211 phy1: no memory
[ 3409.604000] ieee80211 phy1: no memory
[ 3409.607703] ieee80211 phy1: no memory
[ 3409.611391] ieee80211 phy1: no memory
[ 3409.615076] ieee80211 phy1: no memory
[ 3409.618766] ieee80211 phy1: no memory
[ 3409.622451] ieee80211 phy1: no memory
[ 3409.626140] ieee80211 phy1: no memory
[ 3409.629827] ieee80211 phy1: no memory
[ 3409.633531] ieee80211 phy1: no memory
[ 3409.637220] ieee80211 phy1: no memory
[ 3409.640905] ieee80211 phy1: no memory
[ 3409.644588] ieee80211 phy1: no memory
[ 3409.648277] ieee80211 phy1: no memory
[ 3409.651960] ieee80211 phy1: no memory
[ 3409.655649] ieee80211 phy1: no memory
[ 3409.659355] ieee80211 phy1: no memory
```
As of today, I've set vm.min_size_kbytes=16384, vm.swappiness=100 and vm.drop_caches=3, which seem to have limited the problems to just 2.4GHz whereas before I was experiencing the problem on both bands.

NB: I have a USB drive with a 1GB swap partition that I've never seen the router use, despite the "no memory" errors.

yuhhaurlin, 2016-01-04:
@jbeagley52 Please test with latest driver. If you have memory problem, please check issue #52.