kaloz/mwlwifi #22: 'Hide ESSID' doesn't Hide the ESSID (OpenWRT/WRT1900AC)

Malathion, 2015-05-23:
Occurs on multiple builds of OpenWRT and it does not appear to be a Luci issue. Now using r45711. First thing I did on a fresh install of generic was disable the SSIDs on both radios. I still see the broadcasted SSIDs, even after a powercycle.
==============================================
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
        option country 'US'
        option channel 'auto'
        option htmode 'HT40'
        option txpower '20'

config wifi-iface
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid '****'
        option encryption 'psk2+ccmp'
        option key '*****'
        option hidden '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'soc/soc:pcie-controller/pci0000:00/0000:00:03.0/0000:03:00.0'
        option htmode 'VHT80'
        option country 'US'
        option txpower '17'

config wifi-iface
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid '*****'
        option encryption 'psk2+ccmp'
        option key '*****'
        option hidden '1'
==============================================
root@wrt1900ac:~# wifi status
{
        "radio0": {
                "up": true,
                "pending": false,
                "autostart": true,
                "disabled": false,
                "config": {
                        "hwmode": "11g",
                        "path": "soc\/soc:pcie-controller\/pci0000:00\/0000:00:02.0\/0000:02:00.0",
                        "country": "US",
                        "channel": "auto",
                        "htmode": "HT40",
                        "txpower": 20
                },
                "interfaces": [
                        {
                                "section": "@wifi-iface[0]",
                                "ifname": "wlan0",
                                "config": {
                                        "mode": "ap",
                                        "ssid": "*****",
                                        "encryption": "psk2+ccmp",
                                        "key": "*****",
                                        "hidden": true,
                                        "network": [
                                                "lan"
                                        ]
                                }
                        }
                ]
        },
        "radio1": {
                "up": true,
                "pending": false,
                "autostart": true,
                "disabled": false,
                "config": {
                        "channel": "36",
                        "hwmode": "11a",
                        "path": "soc\/soc:pcie-controller\/pci0000:00\/0000:00:03.0\/0000:03:00.0",
                        "htmode": "VHT80",
                        "country": "US",
                        "txpower": 17
                },
                "interfaces": [
                        {
                                "section": "@wifi-iface[1]",
                                "ifname": "wlan1",
                                "config": {
                                        "mode": "ap",
                                        "ssid": "*****",
                                        "encryption": "psk2+ccmp",
                                        "key": "*****",
                                        "hidden": true,
                                        "network": [
                                                "lan"
                                        ]
                                }
                        }
                ]
        }
}
==============================================
root@wrt1900ac:~# cat /tmp/run/hostapd-phy0.conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
hw_mode=g
channel=acs_survey

ieee80211n=1
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40]

interface=wlan0
ctrl_interface=/var/run/hostapd
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=1
uapsd_advertisement_enabled=1
wpa_passphrase=*****
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=*****
bridge=br-lan
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=00:25:9c:******

root@wrt1900ac:~# cat /tmp/run/hostapd-phy1.conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
ieee80211h=1
hw_mode=a
channel=36

ieee80211n=1
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC1][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

interface=wlan1
ctrl_interface=/var/run/hostapd
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=1
uapsd_advertisement_enabled=1
wpa_passphrase=******
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=*****
bridge=br-lan
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=00:25:9c:******

pdfruth, 2015-05-24:
If it helps anyone... I was able to get ssid hiding to work by making this one-line code change to mwl_mac80211.c
I changed line 870 from;
if ((info->ssid[0] != '\0') && (info->ssid_len != 0))
to;
if ((info->ssid[0] != '\0') && (info->ssid_len != 0) && !info->hidden_ssid)

yuhhaurlin, 2015-05-25:
Thanks. The fix is committed.

closed 2015-05-26 by yuhhaurlin