kaloz/mwlwifi #23: Poor Wi-Fi Performance

jguerin, 2015-05-31:
I seem to be able to get 1300Mbps link rate with this build, whereas it was only 600Mbps on the Linksys stock firmware (that's good!).

However, actual transfer speeds seem lower than expected on my MacBook. From the tests that Ars Technica [did](http://arstechnica.com/apple/2013/10/retina-round-two-apples-15-inch-2013-retina-macbook-pro-reviewed/3/), it should be able to hit 600Mbps.

I ran iperf transfers from my Router to my Macbook: `iperf3 -c localhost.macbook -i 1 -t 30 -b 900M -w 256k`

I firstly did a 900Mbps test on Wi-Fi:
```
[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-1.00   sec  30.9 MBytes   259 Mbits/sec                  
[  5]   1.00-2.00   sec  42.0 MBytes   353 Mbits/sec                  
[  5]   2.00-3.00   sec  41.0 MBytes   344 Mbits/sec                  
[  5]   3.00-4.00   sec  42.8 MBytes   359 Mbits/sec                  
[  5]   4.00-5.00   sec  44.1 MBytes   370 Mbits/sec                  
[  5]   5.00-6.00   sec  44.1 MBytes   370 Mbits/sec  
```
I got a baseline going via Ethernet, and it seemed to hit 900Mbps:
```
[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-1.00   sec  71.9 MBytes   603 Mbits/sec                  
[  5]   1.00-2.00   sec   108 MBytes   905 Mbits/sec                  
[  5]   2.00-3.00   sec   108 MBytes   907 Mbits/sec                  
[  5]   3.00-4.00   sec   108 MBytes   907 Mbits/sec                  
[  5]   4.00-5.00   sec   109 MBytes   912 Mbits/sec
```

There doesn't seem to be any frame-level failures or retransmissions going on:
```
Station mac:add:ress (on wlan1)
    inactive time:    0 ms
    rx bytes:    15520528
    rx packets:    180106
    tx bytes:    2128436620
    tx packets:    1394173
    tx retries:    0
    tx failed:    0
    signal:      -42 dBm
    signal avg:    -42 dBm
    tx bitrate:    1053.0 MBit/s VHT-MCS 8 80MHz VHT-NSS 3
    rx bitrate:    1300.0 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 3
    authorized:    yes
    authenticated:    yes
    preamble:    long
    WMM/WME:    yes
    MFP:        no
    TDLS peer:    no
```
Are these speeds expected? Is there anything else that I can look into to troubleshoot if not?

yuhhaurlin, 2015-06-01:
Enhancement of Tx throughput is ongoing.

jguerin, 2015-06-01:
@yuhhaurlin do you need any more info to help with this? Do you need help validating the throughput changes?

yuhhaurlin, 2015-06-01:
No. Thanks.

closed 2015-06-01 by jguerin