kaloz/mwlwifi #24: Don't list 160MHz channel width availability in Luci if the hardware doesn't support it

jguerin, 2015-05-31:
Not sure whether this should become a bug against Luci, but the Wi-Fi config allows a user to select 160MHz on a WRT1900AC, even though the hardware itself doesn't support it. Is this something that the driver/package advertises to the interface? It seems to correctly report its capabilities in `iw`:
```
root@wrt1900ac-1:~# iw phy1 info
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2247 bytes
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
        Band 2:
..
                HT TX/RX MCS rate indexes supported: 0-23, 32
                VHT Capabilities (0x33981932):
                        Max MPDU length: 11454
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        SU Beamformer
                        SU Beamformee
```

yuhhaurlin, 2015-06-01:
Driver does not report capability of 160 MHz or 80+80 MHz.

jguerin, 2015-06-01:
@yuhhaurlin I realised this, my question is why does the Luci interface think that it does?

yuhhaurlin, 2015-06-01:
I have no idea about that.

jguerin, 2015-06-01:
@kaloz any idea...?

gufus, 2015-06-01:
It's a luci bug

https://github.com/kaloz/mwlwifi/issues/18

jguerin, 2015-06-01:
Roger, closing this one then.

closed 2015-06-01 by jguerin