kaloz/mwlwifi #25: 2.4GHz radio is configured to 40MHz channel width, but reports it at only 20MHz

jguerin, 2015-05-31:
I've got my 2.4GHz radio configured to be 40MHz & N mode, but it's reporting it only at 20MHz. Additionally, the mode appears to be set to 11g only.

```
root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
	option country 'US'
	option channel 'auto'
	option htmode 'HT40'
	option txpower '20'

config wifi-iface
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'xx'
	option encryption 'psk2'
	option key 'xx'
```
```
root@OpenWrt:~# iw dev
phy#0
	Interface wlan0
		ifindex 12
		wdev 0x2
		addr xx
		ssid xx
		type AP
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
```
```
root@OpenWrt:~# iw phy0 info
Wiphy phy0
	max # scan SSIDs: 4
	max scan IEs length: 2242 bytes
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
	Band 1:
		Capabilities: 0x6f
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 3839 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 4 usec (0x05)
		HT TX/RX MCS rate indexes supported: 0-23, 32
		VHT Capabilities (0x33981932):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			SU Beamformer
			SU Beamformee
			MU Beamformer
			MU Beamformee
			RX antenna pattern consistency
			TX antenna pattern consistency
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 0 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 0 Mbps
		Frequencies:
			* 2412 MHz [1] (30.0 dBm)
			* 2417 MHz [2] (30.0 dBm)
			* 2422 MHz [3] (30.0 dBm)
			* 2427 MHz [4] (30.0 dBm)
			* 2432 MHz [5] (30.0 dBm)
			* 2437 MHz [6] (30.0 dBm)
			* 2442 MHz [7] (30.0 dBm)
			* 2447 MHz [8] (30.0 dBm)
			* 2452 MHz [9] (30.0 dBm)
			* 2457 MHz [10] (30.0 dBm)
			* 2462 MHz [11] (30.0 dBm)
			* 2467 MHz [12] (disabled)
			* 2472 MHz [13] (disabled)
			* 2484 MHz [14] (disabled)
	valid interface combinations:
		 * #{ AP } <= 16, #{ managed } <= 1,
		   total <= 16, #channels <= 1
	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
```

yuhhaurlin, 2015-06-01:
Please do not use auto channel now.

jguerin, 2015-06-01:
Forced channel to 11, it's still reporting a 20MHz channel width:
```
phy#0
	Interface wlan0
		ifindex 17
		wdev 0x3
		addr xx
		ssid xx
		type AP
		channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
```

yuhhaurlin, 2015-06-01:
Please check if you get this kind of message "20/40 MHz operation not permitted on channel pri=1 sec=5 based on overlapping BSSes".

jguerin, 2015-06-01:
Yup, found it.
```
daemon.notice netifd: radio0 (16193): 20/40 MHz operation not permitted on channel pri=6 sec=10 based on overlapping BSSes
```

Does this mean the area is too crowded to support 40MHz channels?

jguerin, 2015-06-01:
Closing as by design. Would be nice to have better UI to address this, but I doubt it'll ever get fixed.

closed 2015-06-01 by jguerin

raven-au, 2015-06-01:
Yeah, there's no b/g coexistence option to disable as seen on some other firmware implementations. Disabling that is usually enough. Oddly, selecting "auto" in stock WRT1900AC firmware is enough to do this too.

jguerin, 2015-06-01:
No worries, at least the mystery is solved. I mostly use 5GHz anyway.

notnyt, 2015-06-01:
use option noscan

AndyX-Net, 2018-01-19:
Enter the following command in SSH console or webshell ：
uci set wireless.radio0.noscan=1
uci commit
reboot

cowwoc, 2018-08-01:
In the web interface, Wireless -> Advanced Settings, there is a checkbox "Force 40MHz mode". This worked for me.