kaloz/mwlwifi #26: Frequent disconnections for new activation requests

mirsev, 2015-06-06:
... disconnecting for new activation request ...

Hi, recently I have bought Linksys WRT1200AC router to replace my old WRT54GL. First week I ran it with latest stock firmware to see how it works and to monitor the connection quality from my notebook running Ubuntu  14.04.2 LTS with rtl8188ee WiFi chipset. It was OK in general but I have noticed a lot of messages in /var/log/syslog in my notebook which start with 'disconnecting for new activation request'.

When it happens on the notebook, there is no big problem. But I also have a TV with Samsung BD C6900 player which is connected to Internet over WiFi and at the same time when the message mentioned above appears in the notebook system log, video stream on TV freezes and starts buffering which means that old connection is broken and player needs to connect again.

That was quite annoying, so I decided to try OpenWRT firmware with a hope that these new activation requests can be controlled somehow to keep video stream running without interruption. I have downloaded and installed Kaloz's firmware, so the router is running now OpenWrt Chaos Calmer r45573 / LuCI (git-15.090.50849-576e235). I have configured the router to setup my home WiFi network with WPA2 PSK (CCMP) encryption and started to monitor the connection status with my notebook. Unfortunately, new activation requests appear too frequently like it was with the stock firmware. Here is an example:

# grep 'disconnecting for new activation request' /var/log/syslog

Jun  6 08:00:24 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 08:16:54 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 08:38:55 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 08:55:25 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 09:00:55 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 09:17:25 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 10:17:56 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 10:29:01 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 10:45:31 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 10:56:31 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 11:24:06 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 11:46:10 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 12:02:44 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.
Jun  6 12:19:18 qpoint NetworkManager[16604]: <info> (wlan0): disconnecting for new activation request.

With old Linksys WRT54GL router running Tomato the connection was much more stable.

rpallai, 2015-09-03:
Similar problem here, client getting disconnected every 5-15 minutes.

Messages on the client side are different probably due to different distribution used (Fedora 22). This combined with #34 (long traffic stalling) could lead to permanent client disconnection.

Connection of this client to my WRT160NL AP is very stable.

```
Sep 03 03:58:04 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 03:58:04 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 03:58:04 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 03:58:05 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: authenticated
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 03:58:05 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: associated
Sep 03 03:58:05 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> associated
Sep 03 03:58:05 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
Sep 03 03:58:53 dh.darknet kernel: ath: phy0: Failed to stop TX DMA, queues=0x008!
Sep 03 03:59:16 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 03:59:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 03:59:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 03:59:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: authenticated
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 03:59:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: associated
Sep 03 03:59:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> associated
Sep 03 03:59:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
Sep 03 04:02:09 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 04:02:09 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 04:02:09 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 04:02:10 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: authenticated
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: associated
Sep 03 04:02:10 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associated
Sep 03 04:02:10 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
Sep 03 04:08:15 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 04:08:15 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 04:08:15 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 04:08:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: authenticated
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: associated
Sep 03 04:08:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associated
Sep 03 04:08:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
```

rpallai, 2015-09-03:
Logs with 10.3.0.3:

```
Sep 03 16:56:06 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 16:56:06 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 16:56:06 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 16:56:07 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 16:56:07 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 16:56:07 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 16:56:07 dh.darknet kernel: wlp2s0: authenticated
Sep 03 16:56:07 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 16:56:07 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating
Sep 03 16:56:07 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 16:56:07 dh.darknet kernel: wlp2s0: associated
Sep 03 16:56:07 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> 4-way handshake
Sep 03 16:56:07 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: 4-way handshake -> completed
Sep 03 17:18:39 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 17:18:39 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 17:18:39 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 17:18:40 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 17:18:40 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 17:18:40 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 17:18:40 dh.darknet kernel: wlp2s0: authenticated
Sep 03 17:18:40 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 17:18:40 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 17:18:40 dh.darknet kernel: wlp2s0: associated
Sep 03 17:18:40 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associated
Sep 03 17:18:40 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
Sep 03 17:24:52 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 17:24:52 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 17:24:52 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 17:24:53 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 17:24:53 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 17:24:53 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 17:24:53 dh.darknet kernel: wlp2s0: authenticated
Sep 03 17:24:53 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 17:24:53 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating
Sep 03 17:24:53 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 17:24:53 dh.darknet kernel: wlp2s0: associated
Sep 03 17:24:53 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> associated
Sep 03 17:24:53 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
Sep 03 17:32:37 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 17:32:37 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 17:32:37 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 17:32:38 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 17:32:38 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 17:32:38 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 17:32:38 dh.darknet kernel: wlp2s0: authenticated
Sep 03 17:32:38 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 17:32:38 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating
Sep 03 17:32:38 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 17:32:38 dh.darknet kernel: wlp2s0: associated
Sep 03 17:32:38 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> associated
Sep 03 17:32:38 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed

Thu Sep  3 16:55:04 2015 daemon.notice netifd: wan (1368): Lease of * obtained, lease time 600
Thu Sep  3 16:56:07 2015 daemon.info hostapd: wlan1: STA a0:f3:c1:f8:9b:e0 IEEE 802.11: authenticated
Thu Sep  3 16:56:07 2015 daemon.info hostapd: wlan1: STA a0:f3:c1:f8:9b:e0 IEEE 802.11: associated (aid 1)
[... cut ...]
wlan1 (phy #1): mgmt TX status (cookie dda99340): acked
wlan1 (phy #1): mgmt TX status (cookie de8b9040): acked
wlan1: del station a0:f3:c1:f8:9b:e0
wlan1: new station a0:f3:c1:f8:9b:e0
wlan1 (phy #1): mgmt TX status (cookie dee38880): acked
wlan1 (phy #1): mgmt TX status (cookie de8c5100): acked
wlan1 (phy #1): mgmt TX status (cookie de33cd00): acked
wlan1: del station a0:f3:c1:f8:9b:e0
wlan1: new station a0:f3:c1:f8:9b:e0
```

* a0:f3:c1:f8:9b:e0 is a downloading client
* 30:a8:db:99:76:64 is a mobile phone, mainly idle

rpallai, 2015-09-04:
The latest stock firmware is also affected but events are rarer and I have not seen permanent disconnection.

Client logs with stock firmware:
```
Sep 03 23:33:16 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)                              
Sep 03 23:33:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected  
Sep 03 23:33:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning   
Sep 03 23:33:17 dh.darknet kernel: wlp2s0: authenticate with c0:56:27:73:40:44                                             
Sep 03 23:33:17 dh.darknet kernel: wlp2s0: send auth to c0:56:27:73:40:44 (try 1/3)                                        
Sep 03 23:33:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating 
Sep 03 23:33:17 dh.darknet kernel: wlp2s0: authenticated                                                                   
Sep 03 23:33:17 dh.darknet kernel: wlp2s0: associating with AP with corrupt probe response                                 
Sep 03 23:33:17 dh.darknet kernel: wlp2s0: associate with c0:56:27:73:40:44 (try 1/3)                                      
Sep 03 23:33:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating                                                                                                                         
Sep 03 23:33:17 dh.darknet kernel: wlp2s0: RX AssocResp from c0:56:27:73:40:44 (capab=0x411 status=0 aid=1)                
Sep 03 23:33:17 dh.darknet kernel: wlp2s0: associated                                                                      
Sep 03 23:33:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> 4-way handshake                                                                                                                        
Sep 03 23:33:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: 4-way handshake -> completed                                                                                                                          
Sep 03 23:36:17 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)                              
Sep 03 23:36:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected  
Sep 03 23:36:18 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning   
Sep 03 23:36:18 dh.darknet kernel: ath: phy0: Failed to stop TX DMA, queues=0x008!                                         
Sep 03 23:36:18 dh.darknet kernel: wlp2s0: authenticate with c0:56:27:73:40:44                                             
Sep 03 23:36:18 dh.darknet kernel: wlp2s0: send auth to c0:56:27:73:40:44 (try 1/3)                                        
Sep 03 23:36:18 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating 
Sep 03 23:36:18 dh.darknet kernel: wlp2s0: authenticated                                                                   
Sep 03 23:36:18 dh.darknet kernel: wlp2s0: associating with AP with corrupt probe response                                 
Sep 03 23:36:18 dh.darknet kernel: wlp2s0: associate with c0:56:27:73:40:44 (try 1/3)                                      
Sep 03 23:36:18 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating                                                                                                                         
Sep 03 23:36:19 dh.darknet kernel: wlp2s0: RX AssocResp from c0:56:27:73:40:44 (capab=0x411 status=0 aid=1)                
Sep 03 23:36:19 dh.darknet kernel: wlp2s0: associated                                                                      
Sep 03 23:36:19 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> 4-way handshake                                                                                                                        
Sep 03 23:36:19 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: 4-way handshake -> completed                                                                                                                          
Sep 04 00:46:29 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)                              
Sep 04 00:46:29 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected  
Sep 04 00:46:29 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning   
Sep 04 00:46:30 dh.darknet kernel: wlp2s0: authenticate with c0:56:27:73:40:44                                             
Sep 04 00:46:30 dh.darknet kernel: wlp2s0: send auth to c0:56:27:73:40:44 (try 1/3)                                        
Sep 04 00:46:30 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating 
Sep 04 00:46:30 dh.darknet kernel: wlp2s0: authenticated                                                                   
Sep 04 00:46:30 dh.darknet kernel: wlp2s0: associating with AP with corrupt probe response                                 
[... cut ...]
Sep 04 02:44:28 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating
Sep 04 02:44:28 dh.darknet kernel: wlp2s0: RX AssocResp from c0:56:27:73:40:44 (capab=0x411 status=0 aid=1)
Sep 04 02:44:28 dh.darknet kernel: wlp2s0: associated
Sep 04 02:44:28 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> 4-way handshake
Sep 04 02:44:28 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: 4-way handshake -> completed
```

rpallai, 2015-09-05:
Seems like this issue depends on device used at the client side.

I've stressed the stock firmware for many hours and there is no disconnection with RTL8812AE at 2.4 Ghz driven by Fedora.

yuhhaurlin, 2015-11-12:
Does anyone still encounter this problem with 10.3.0.14?

closed 2015-11-24 by yuhhaurlin