kaloz/mwlwifi #27: Luci shows wrong width rate in 5ghz

gufus, 2015-06-07:
Luci shows wrong width rate in 5ghz

openwrt Cc-RCI
kmod-mwlwifi 	3.18.11+10.2.8.5.p0-20150422-1

Linksys WRT1900AC (Mamba)
Security is enabled, and your IP address has been logged.

root@AC1900M:~# iw dev
phy#1
        Interface wlan1
                ifindex 39
                wdev 0x100000019
                addr 00:25:9c:13:0b:a1
                ssid gypsynet_5ghz
                type AP
                channel 161 (5805 MHz), width: 80 MHz, center1: 5775 MHz
phy#0
        Interface wlan0
                ifindex 13
                wdev 0x2
                addr 00:25:9c:13:0b:a0
                ssid gypsynet
                type AP
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
root@AC1900M:~#

See here http://www.gypsy-designs.com/661.jpg

yuhhaurlin, 2015-06-08:
Please use "iw wlan1 station dump" to check if driver reported correct rate information. If yes, please report this problem to LuCI.

gufus, 2015-06-09:
Sorry for the slow response, it's on my to-do list. I'll post the info once it's done.

gufus, 2015-06-12:
Yes, it's correct.

I'll close the issue..

Using username "root".
Authenticating with public key "rsa-key-20120810"

BusyBox v1.23.2 (2015-04-22 10:08:13 CEST) built-in shell (ash)

Linksys WRT1900AC (Mamba)
Security is enabled, and your IP address has been logged.

root@AC1900M:~# iw wlan1 station dump
Station 74:d0:2b:d0:1c:94 (on wlan1)
        inactive time:  10 ms
        rx bytes:       652149
        rx packets:     2534
        tx bytes:       2548312
        tx packets:     2743
        tx retries:     0
        tx failed:      0
        signal:         -70 dBm
        signal avg:     -70 dBm
        tx bitrate:     702.0 MBit/s VHT-MCS 8 80MHz VHT-NSS 2
        rx bitrate:     585.0 MBit/s VHT-MCS 7 80MHz VHT-NSS 2
        authorized:     yes
        authenticated:  yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
root@AC1900M:~#

closed 2015-06-12 by gufus