kaloz/mwlwifi #28: lan/wan no access after change a client's band

DavidMcWRT, 2015-06-23:
This may or may not be the same as some if of the older issue reports.

Have laptop connected to 5GHz (N).

Have android connected to 5GHz (N).

Start an on-going ping on the laptop, everything OK.

Switch android to 2.4GHz - pings time out, no router login access, no Internet access.

Switch android back to 5GHz and everything resumes normally.

Trunk r45950, uptime 9d 14h 2m 13s

Vanav, 2015-06-23:
If you connect android first, then connect laptop, then switch android to 2.4GHz - and laptop has ping timeout, but when you switch android back to 5 GHz, laptop ping resumes - then this is issue #20.

DavidMcWRT, 2015-06-23:
Yes, that was the sequence.

Note, however, that all devices are N rather than AC or something else, and indeed I also have my (WRT1900AC/OpenWRT) 5GHz radio set to N rather than AC (since no AC devices).

DavidMcWRT, 2015-07-27:
since this does appear to be a dup of issue #20 and since the new firmware bin file posted today seems to have fixed it for me, I am closing this issue.

closed 2015-07-27 by DavidMcWRT