kaloz/mwlwifi #29: After reboot, WAN SSH fails unless DropBear is restarted

JW0914, 2015-06-24:
After reboot, remote WAN SSH fails unless DropBear is restarted, whereas LAN SSH has no issues.

(I'm not sure what information is needed, so please let me know what other information I need to pull and post.)

K.V.: 3.18.16
F.V.: r46118 / LuCI (git-15.168.50780-bae48b6)
DropBear V.: 2015.67-1

LAN SSH is set to it's own port and set to LAN only
WAN SSH is set to it's own port and set to WAN only

From looking at the running processes, DropBear is starting a process for the LAN SSH port, but is not starting the subsequent process for the WAN SSH port.  As a workaround, I'll simply add a command to restart DropBear after a reboot.

kaloz, 2015-06-25:
This has nothing to do with the wireless driver. Generic OpenWrt bugs should go to https://dev.openwrt.org

closed 2015-06-25 by kaloz

JW0914, 2015-06-25:
My bad... When I had asked for the link for bug reports, you gave me this one, so I assumed it was for all bug reports