kaloz/mwlwifi #30: wifi 2.4 or 5ghz Slowwwwwww

jdthomas4181, 2015-07-21:
Ok i just installed openwrt-15.05-rc3-mvebu-armada-xp-linksys-mamba-squashfs-factory on this wrt1900ac v1   and now access to the internet via wireless is excruciatingly slow....

backgroud of my network DHCP and DNS on Windows server 2012, internet router is Pfsense. All im using the WRT1900AC for is a wifi access point. doesnt matter if i try and use 5ghz or 2.4ghz the result is the same 30-60 seconds for websites loading on my phone..not as bad with laptop however its still bad...speedtest.net shows speeds of 20mb down 5 up  i have a 30mb internet connection....however the issue isnt entirely with the speed as it is the ping time which shows anywhere from 65 to 95ms.
With stock firmware it was 14 to 16ms and would download at full speed...the reason im using openwrt is because the factory firmware doesnt support Class A networks.

JW0914, 2015-07-21:
Just an fyi... in order for any dev to troubleshoot your issue, actual information from the device is required.  This includes logs, config files, etc.

jdthomas4181, 2015-07-22:
now the wireless 5ghz doesnt even work, says its not associated.

heres my log

ue Jul 21 19:40:00 2015 cron.info crond[995]: USER root pid 3734 cmd /sbin/fan_ctrl.sh
Tue Jul 21 19:45:00 2015 cron.info crond[995]: USER root pid 3970 cmd /sbin/fan_ctrl.sh
Tue Jul 21 19:48:15 2015 daemon.info hostapd: wlan0: STA f4:f1:e1:02:e3:2b WPA: group key handshake completed (RSN)
Tue Jul 21 19:48:57 2015 daemon.info hostapd: wlan0: STA f4:f1:e1:02:e3:2b IEEE 802.11: disassociated
Tue Jul 21 19:48:58 2015 daemon.info hostapd: wlan0: STA f4:f1:e1:02:e3:2b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Jul 21 19:50:00 2015 cron.info crond[995]: USER root pid 4196 cmd /sbin/fan_ctrl.sh
Tue Jul 21 19:55:00 2015 cron.info crond[995]: USER root pid 4422 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:00:00 2015 cron.info crond[995]: USER root pid 4648 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:05:00 2015 cron.info crond[995]: USER root pid 4874 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:10:00 2015 cron.info crond[995]: USER root pid 5100 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:15:00 2015 cron.info crond[995]: USER root pid 5326 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:20:00 2015 cron.info crond[995]: USER root pid 5541 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:25:00 2015 cron.info crond[995]: USER root pid 5766 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:30:00 2015 cron.info crond[995]: USER root pid 5992 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:35:00 2015 cron.info crond[995]: USER root pid 6218 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:40:00 2015 cron.info crond[995]: USER root pid 6444 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:45:00 2015 cron.info crond[995]: USER root pid 6670 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:50:00 2015 cron.info crond[995]: USER root pid 6896 cmd /sbin/fan_ctrl.sh
Tue Jul 21 20:55:00 2015 cron.info crond[995]: USER root pid 7122 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:00:00 2015 cron.info crond[995]: USER root pid 7349 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:05:00 2015 cron.info crond[995]: USER root pid 7564 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:10:00 2015 cron.info crond[995]: USER root pid 7790 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:15:00 2015 cron.info crond[995]: USER root pid 8017 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:20:00 2015 cron.info crond[995]: USER root pid 8243 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:22:43 2015 daemon.notice netifd: wan (1386): Sending renew...
Tue Jul 21 21:22:43 2015 daemon.notice netifd: wan (1386): Lease of 74.141.176.224 obtained, lease time 86400
Tue Jul 21 21:22:43 2015 user.notice firewall: Reloading firewall due to ifupdate of wan (eth1)
Tue Jul 21 21:25:00 2015 cron.info crond[995]: USER root pid 8490 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:30:00 2015 cron.info crond[995]: USER root pid 8716 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:35:00 2015 cron.info crond[995]: USER root pid 8942 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:40:00 2015 cron.info crond[995]: USER root pid 9167 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:45:00 2015 cron.info crond[995]: USER root pid 9393 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:50:00 2015 cron.info crond[995]: USER root pid 9608 cmd /sbin/fan_ctrl.sh
Tue Jul 21 21:55:00 2015 cron.info crond[995]: USER root pid 9834 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:00:00 2015 cron.info crond[995]: USER root pid 10060 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:00:30 2015 daemon.info dnsmasq[1472]: read /etc/hosts - 1 addresses
Tue Jul 21 22:00:30 2015 daemon.info dnsmasq[1472]: read /tmp/hosts/odhcpd - 0 addresses
Tue Jul 21 22:00:30 2015 daemon.info dnsmasq[1472]: read /tmp/hosts/dhcp - 1 addresses
Tue Jul 21 22:00:58 2015 daemon.warn odhcpd[974]: DHCPV6 SOLICIT IA_NA from 000100011d307f4d000c29ac8ad6 on br-lan: ok 2605:a000:160b:c0a8::3ec/128 fde9:a0af:c66f::3ec/128 
Tue Jul 21 22:00:59 2015 daemon.warn odhcpd[974]: DHCPV6 REQUEST IA_NA from 000100011d307f4d000c29ac8ad6 on br-lan: ok 2605:a000:160b:c0a8::3ec/128 fde9:a0af:c66f::3ec/128 
Tue Jul 21 22:00:59 2015 daemon.info dnsmasq[1472]: read /etc/hosts - 1 addresses
Tue Jul 21 22:00:59 2015 daemon.info dnsmasq[1472]: read /tmp/hosts/odhcpd - 1 addresses
Tue Jul 21 22:00:59 2015 daemon.info dnsmasq[1472]: read /tmp/hosts/dhcp - 1 addresses
Tue Jul 21 22:05:00 2015 cron.info crond[995]: USER root pid 10292 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:10:00 2015 cron.info crond[995]: USER root pid 10518 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:15:00 2015 cron.info crond[995]: USER root pid 10745 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:20:00 2015 cron.info crond[995]: USER root pid 10960 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:25:00 2015 cron.info crond[995]: USER root pid 11186 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:30:00 2015 cron.info crond[995]: USER root pid 11412 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:35:00 2015 cron.info crond[995]: USER root pid 11638 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:40:00 2015 cron.info crond[995]: USER root pid 11864 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:45:00 2015 cron.info crond[995]: USER root pid 12090 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:50:00 2015 cron.info crond[995]: USER root pid 12316 cmd /sbin/fan_ctrl.sh
Tue Jul 21 22:55:00 2015 cron.info crond[995]: USER root pid 12542 cmd /sbin/fan_ctrl.sh
Tue Jul 21 23:00:00 2015 cron.info crond[995]: USER root pid 12767 cmd /sbin/fan_ctrl.sh
Tue Jul 21 23:05:00 2015 cron.info crond[995]: USER root pid 12993 cmd /sbin/fan_ctrl.sh
Tue Jul 21 23:10:00 2015 cron.info crond[995]: USER root pid 13219 cmd /sbin/fan_ctrl.sh
Tue Jul 21 23:15:00 2015 cron.info crond[995]: USER root pid 13434 cmd /sbin/fan_ctrl.sh
[... cut ...]
Wed Jul 22 08:20:00 2015 cron.info crond[995]: USER root pid 5593 cmd /sbin/fan_ctrl.sh
Wed Jul 22 08:25:00 2015 cron.info crond[995]: USER root pid 5819 cmd /sbin/fan_ctrl.sh
Wed Jul 22 08:30:00 2015 cron.info crond[995]: USER root pid 6045 cmd /sbin/fan_ctrl.sh
Wed Jul 22 08:35:00 2015 cron.info crond[995]: USER root pid 6271 cmd /sbin/fan_ctrl.sh
Wed Jul 22 08:40:00 2015 cron.info crond[995]: USER root pid 6497 cmd /sbin/fan_ctrl.sh
Wed Jul 22 08:45:00 2015 cron.info crond[995]: USER root pid 6724 cmd /sbin/fan_ctrl.sh

yuhhaurlin, 2015-07-23:
Please do not use Auto channel.

yuhhaurlin, 2015-07-23:
Except for that, the driver should work well.

jdthomas4181, 2015-07-24:
ive taken it off auto, its still very slow, and what can cause the 5ghz to stay offline with the message 	Signal: 0 dBm / Noise: 0 dBm
0%	SSID: OpenWrt | Mode: Unknown
Wireless is disabled or not associated

its not disabled!

notnyt, 2015-07-25:
Nobody else is experiencing this issue, and you haven't posted your config, so I'm not sure what support you're expecting to get.  I'd suggest posting on the forum, as this does not appear to be a driver issue.

jdthomas4181, 2015-07-25:
OK I'll look bout posting config when I return at 5am....

jdthomas4181, 2015-07-25:
ok so how do i look at my config files?

JW0914, 2015-07-25:
As @notnyt stated, this is not a driver issue and needs to be posted on the WRT1900 thread page: https://forum.openwrt.org/viewtopic.php?id=50173&p=266

Please include the following in your post on the thread page:
 * Wireless Config `cat /etc/config/wireless` (Remove plaintext password/key)

jdthomas4181, 2015-07-26:
well i re flashed the firmware i got the 5ghz wireless working again...speed seems fine on my laptop now however with android devices its still slow... i will post of the forum you may close this

closed 2015-07-26 by jdthomas4181