kaloz/mwlwifi #31: Monitor Mode Kernel Panic

XdengaX, 2015-07-30:
Im running a Linksys WRT1900ac (V1 and V2) on the latest code off the OpenWrt repository.
I've put given the ban (2.4) card two interfaces. One AP mode and one Monitor mode.
When it is running with a Monitor mode interface, it kernel panics every few minutes and restarts.
When the router is running with just an AP mode interface, none of these errors happen.
Following the backtrace, skb_dequeue seems to be called from mwl_tx_done in tx.c.

[   78.913339] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   78.921466] pgd = c0004000
[   78.924182] [00000000] *pgd=00000000
[   78.927782] Internal error: Oops: 817 [#1] SMP ARM
[   78.932589] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_RE
JECT ipt_MASQUERADE armada_thermal xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_commen
t xt_TCPMSS xt_REDIRECT xt_LOG xt_CT thermal_sys slhc nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ip
v6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack mvsdio iptable_raw iptable_mangle iptable_filter ip_tables crc_cc
itt tmp421 pwm_fan hwmon i2c_mv64xxx i2c_dev mwlwifi mac80211 cfg80211 compat ledtrig_usbdev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf
_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables regmap_spi ipv6 arc4 crypto_blkcipher mmc_block usb_stora
ge mmc_core leds_tlc59116 regmap_i2c i2c_core xhci_plat_hcd xhci_pci xhci_hcd ahci_mvebu sata_mv ahci ehci_orion ehci_platform ehci_hc
d sd_mod ahci_platform libahci_platform libahci libata scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[   79.025190] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W      3.18.19 #1
[   79.032354] task: c041d3d8 ti: c0416000 task.ti: c0416000
[   79.037777] PC is at skb_dequeue+0x54/0x70
[   79.041888] LR is at skb_dequeue+0x1c/0x70
[   79.046000] pc : [<c026afc0>]    lr : [<c026af88>]    psr: a0000193
[   79.046000] sp : c0417db0  ip : 00000000  fp : c0417dcc
[   79.057525] r10: de6b14c4  r9 : de6b0b60  r8 : 03800c00
[   79.062769] r7 : dd87fc58  r6 : de6b2468  r5 : de6b245c  r4 : de374e80
[   79.069322] r3 : 00000000  r2 : de9107c0  r1 : 20000193  r0 : 20000193
[   79.075875] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[   79.083301] Control: 10c5387d  Table: 1e35804a  DAC: 00000015
[   79.089068] Process swapper/0 (pid: 0, stack limit = 0xc0416238)
[   79.095098] Stack: (0xc0417db0 to 0xc0418000)
[   79.099472] 7da0:                                     de9104ac de910400 00000000 de6b14ec
[   79.107683] 7dc0: c0417e34 c0417dd0 bf2780d8 c026af78 00000000 de6b0b60 00000188 c041e318
[   79.115894] 7de0: 00000000 00000000 00000000 20000113 de6b14c0 de6b1fa8 de6b1830 de6b245c
[   79.124105] 7e00: c041e318 c0419048 c0029df4 de6b24bc de6b24c0 00000000 c043a7c0 c041413c
[   79.132316] 7e20: c0416000 00000006 c0417e5c c0417e38 c00297b4 bf277f08 00000000 c0418098
[   79.140527] 7e40: c0418080 40000006 c0416000 00000100 c0417ebc c0417e60 c0029980 c0029724
[   79.148738] 7e60: c0425060 df4081c0 c0425060 00200000 c037c444 c0416020 ffffa9a5 c0418100
[   79.156949] 7e80: c043a7c0 0000000a c04141b0 c0418080 df4081c0 c04148f8 00000000 00000000
[   79.165160] 7ea0: 00000001 df407000 00000000 c04190d4 c0417ed4 c0417ec0 c0029d30 c0029888
[   79.173371] 7ec0: 00000100 c04148f8 c0417efc c0417ed8 c005867c c0029c9c c0417f20 e080210c
[   79.181582] 7ee0: c0419cb0 c0417f20 e0802100 c0416000 c0417f1c c0417f00 c000875c c00585f4
[   79.189793] 7f00: c00188bc 60000013 ffffffff c0417f54 c0417f74 c0417f20 c00097a0 c0008720
[   79.198004] 7f20: ffffffed 00000000 0004c61e c000a280 c0416038 00000000 00000000 c037c438
[   79.206215] 7f40: c0416000 00000000 c04190d4 c0417f74 c0417f78 c0417f68 c00188b8 c00188bc
[   79.214426] 7f60: 60000013 ffffffff c0417fa4 c0417f78 c0051ac0 c001888c c0417f94 c0415380
[   79.222637] 7f80: c041cdff c0415380 c0414a40 ffffffed c000ede8 c0419000 c0417fb4 c0417fa8
[   79.230848] 7fa0: c000d264 c0051984 c0417ff4 c0417fb8 c03e4dfc c000d1f4 ffffffff ffffffff
[   79.239059] 7fc0: c03e46dc 00000000 00000000 c040c4f8 c0439fe4 c0419080 c040c4f4 c041c214
[   79.247270] 7fe0: 0000406a 414fc091 00000000 c0417ff8 00008074 c03e4988 00000000 00000000
[   79.255477] Backtrace:
[   79.257951] [<c026af6c>] (skb_dequeue) from [<bf2780d8>] (mwl_tx_done+0x1dc/0xa4c [mwlwifi])
[   79.266423]  r6:de6b14ec r5:00000000 r4:de910400 r3:de9104ac
[   79.272151] [<bf277efc>] (mwl_tx_done [mwlwifi]) from [<c00297b4>] (tasklet_action+0x9c/0x108)
[   79.280797]  r10:00000006 r9:c0416000 r8:c041413c r7:c043a7c0 r6:00000000 r5:de6b24c0
[   79.288707]  r4:de6b24bc
[   79.291261] [<c0029718>] (tasklet_action) from [<c0029980>] (__do_softirq+0x104/0x234)
[   79.299208]  r9:00000100 r8:c0416000 r7:40000006 r6:c0418080 r5:c0418098 r4:00000000
[   79.307035] [<c002987c>] (__do_softirq) from [<c0029d30>] (irq_exit+0xa0/0xb0)
[   79.314284]  r10:c04190d4 r9:00000000 r8:df407000 r7:00000001 r6:00000000 r5:00000000
[   79.322194]  r4:c04148f8
[   79.324751] [<c0029c90>] (irq_exit) from [<c005867c>] (__handle_domain_irq+0x94/0xa4)
[   79.332611]  r4:c04148f8 r3:00000100
[   79.336222] [<c00585e8>] (__handle_domain_irq) from [<c000875c>] (gic_handle_irq+0x48/0x64)
[   79.344605]  r8:c0416000 r7:e0802100 r6:c0417f20 r5:c0419cb0 r4:e080210c r3:c0417f20
[   79.352432] [<c0008714>] (gic_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
[   79.359945] Exception stack(0xc0417f20 to 0xc0417f68)
[   79.365017] 7f20: ffffffed 00000000 0004c61e c000a280 c0416038 00000000 00000000 c037c438
[   79.373229] 7f40: c0416000 00000000 c04190d4 c0417f74 c0417f78 c0417f68 c00188b8 c00188bc
[   79.381438] 7f60: 60000013 ffffffff
[   79.384938]  r7:c0417f54 r6:ffffffff r5:60000013 r4:c00188bc
[   79.390662] [<c0018880>] (arch_cpu_idle) from [<c0051ac0>] (cpu_startup_entry+0x148/0x2a8)
[   79.398963] [<c0051978>] (cpu_startup_entry) from [<c000d264>] (rest_init+0x7c/0x80)
[   79.406735]  r7:c0419000
[   79.409289] [<c000d1e8>] (rest_init) from [<c03e4dfc>] (start_kernel+0x480/0x48c)
[   79.416807] [<c03e497c>] (start_kernel) from [<00008074>] (0x8074)
[   79.423013] Code: e894000c e584c000 e584c004 e5823004 (e5832000)
[   79.429131] ---[ end trace bc2e7cc8914e0b69 ]---
[   79.433765] Kernel panic - not syncing: Fatal exception in interrupt
[   79.440146] CPU1: stopping
[   79.442866] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D W      3.18.19 #1
[   79.450029] Backtrace:
[   79.452498] [<c001af04>] (dump_backtrace) from [<c001b288>] (show_stack+0x18/0x1c)
[   79.460097]  r6:00000000 r5:00000000 r4:c0424d3c r3:dc8cb013
[   79.465824] [<c001b270>] (show_stack) from [<c018093c>] (dump_stack+0x8c/0x9c)
[   79.473078] [<c01808b0>] (dump_stack) from [<c001d410>] (handle_IPI+0xe8/0x174)
[   79.480415]  r5:c04148f8 r4:00000001
[   79.484025] [<c001d328>] (handle_IPI) from [<c0008774>] (gic_handle_irq+0x60/0x64)
[   79.491624]  r6:df45df58 r5:c0419cb0 r4:e080210c r3:df45df58
[   79.497347] [<c0008714>] (gic_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
[   79.504859] Exception stack(0xdf45df58 to 0xdf45dfa0)
[   79.509930] df40:                                                       ffffffed 00000000
[   79.518142] df60: 00020796 c000a280 df45c030 00000000 00000000 c037c438 df45c000 00000000
[   79.526354] df80: c04190d4 df45dfac df45dfb0 df45dfa0 c00188b8 c00188bc 60000013 ffffffff
[   79.534563]  r7:df45df8c r6:ffffffff r5:60000013 r4:c00188bc
[   79.540287] [<c0018880>] (arch_cpu_idle) from [<c0051ac0>] (cpu_startup_entry+0x148/0x2a8)
[   79.548587] [<c0051978>] (cpu_startup_entry) from [<c001d0f8>] (secondary_start_kernel+0x144/0x14c)
[   79.557668]  r7:c043a040
[   79.560226] [<c001cfb4>] (secondary_start_kernel) from [<00008804>] (0x8804)
[   79.567302]  r5:00000015 r4:1f44406a
[   79.570910] Rebooting in 3 seconds..

XdengaX, 2015-08-02:
Recompiled a new image with the new 7.2.8.6 firmware and it is still kernel panicking and restarts after a few minutes with a monitor mode interface in place.

[  140.799500] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[  140.807628] pgd = c0004000
[  140.810344] [00000000] *pgd=00000000
[  140.813943] Internal error: Oops: 817 [#1] SMP ARM
[  140.818751] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT 
ipt_MASQUERADE armada_thermal xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMS
S xt_REDIRECT xt_LOG xt_CT thermal_sys slhc nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ip
v4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack mvsdio iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt tmp421 pwm_fan h
wmon i2c_mv64xxx i2c_dev mwlwifi mac80211 cfg80211 compat ledtrig_usbdev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw 
ip6table_mangle ip6table_filter ip6_tables x_tables regmap_spi ipv6 arc4 crypto_blkcipher mmc_block usb_storage mmc_core leds_tlc59116 regm
ap_i2c i2c_core xhci_plat_hcd xhci_pci xhci_hcd ahci_mvebu sata_mv ahci ehci_orion ehci_platform ehci_hcd sd_mod ahci_platform libahci_plat
form libahci libata scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[  140.911375] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W      3.18.19 #1
[  140.918540] task: c041b398 ti: c0414000 task.ti: c0414000
[  140.923963] PC is at skb_dequeue+0x54/0x70
[  140.928074] LR is at skb_dequeue+0x1c/0x70
[  140.932186] pc : [<c026877c>]    lr : [<c0268744>]    psr: a0000193
[  140.932186] sp : c0415db0  ip : 00000000  fp : c0415dcc
[  140.943710] r10: de6b1844  r9 : de6b0b40  r8 : 03800c00
[  140.948955] r7 : df5f2e60  r6 : de6b27e8  r5 : de6b27dc  r4 : def5aa00
[  140.955508] r3 : 00000000  r2 : de8774c0  r1 : 20000193  r0 : 20000193
[  140.962062] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[  140.969487] Control: 10c5387d  Table: 1da3c04a  DAC: 00000015
[  140.975254] Process swapper/0 (pid: 0, stack limit = 0xc0414238)
[  140.981284] Stack: (0xc0415db0 to 0xc0416000)
[  140.985658] 5da0:                                     de85292c de852880 00000000 de6b186c
[  140.993870] 5dc0: c0415e34 c0415dd0 bf26f0ac c0268734 00000000 00000000 00000188 00000000
[  141.002081] 5de0: c0417084 c0416080 c0415e0c 20000113 de6b1840 de6b2328 de6b1bb0 de6b27dc
[  141.010292] 5e00: c041c2d0 c0417044 c0029c64 de6b283c de6b2840 00000000 c0438780 c041113c
[  141.018503] 5e20: c0414000 00000006 c0415e5c c0415e38 c0029624 bf26eedc 00000000 c0416098
[  141.026715] 5e40: c0416080 40000006 c0414000 00000100 c0415ebc c0415e60 c00297f0 c0029594
[  141.034927] 5e60: c0423020 df4081c0 c0423020 00200000 c037939c c0414020 ffffc1d1 c0416100
[  141.043137] 5e80: c0438780 0000000a c04111b0 c0416080 df4081c0 c04118f8 00000000 00000000
[  141.051348] 5ea0: 00000001 df407000 00000000 c04170cc c0415ed4 c0415ec0 c0029ba0 c00296f8
[  141.059560] 5ec0: 00000100 c04118f8 c0415efc c0415ed8 c0058304 c0029b0c c0415f20 e080210c
[  141.067771] 5ee0: c0417ca8 c0415f20 e0802100 c0414000 c0415f1c c0415f00 c000875c c005827c
[  141.075982] 5f00: c00187c0 60000013 ffffffff c0415f54 c0415f74 c0415f20 c00097a0 c0008720
[  141.084193] 5f20: ffffffed 00000000 000529f8 c000a280 c0414038 00000000 00000000 c0379388
[  141.092405] 5f40: c0414000 00000000 c04170cc c0415f74 c0415f78 c0415f68 c00187bc c00187c0
[  141.100616] 5f60: 60000013 ffffffff c0415fa4 c0415f78 c0051748 c0018790 c0415f94 c0412380
[  141.108827] 5f80: c041adbf c0412380 c0411a40 ffffffed c000edc8 c0417000 c0415fb4 c0415fa8
[  141.117038] 5fa0: c000d264 c005160c c0415ff4 c0415fb8 c03e1db4 c000d1f4 ffffffff ffffffff
[  141.125249] 5fc0: c03e16dc 00000000 00000000 c04092f8 c0437fa4 c041707c c04092f4 c041a1d4
[  141.133461] 5fe0: 0000406a 414fc091 00000000 c0415ff8 00008074 c03e1988 00000000 00000000
[  141.141669] Backtrace:
[  141.144145] [<c0268728>] (skb_dequeue) from [<bf26f0ac>] (mwl_tx_done+0x1dc/0xa4c [mwlwifi])
[  141.152617]  r6:de6b186c r5:00000000 r4:de852880 r3:de85292c
[  141.158348] [<bf26eed0>] (mwl_tx_done [mwlwifi]) from [<c0029624>] (tasklet_action+0x9c/0x108)
[  141.166993]  r10:00000006 r9:c0414000 r8:c041113c r7:c0438780 r6:00000000 r5:de6b2840
[  141.174906]  r4:de6b283c
[  141.177461] [<c0029588>] (tasklet_action) from [<c00297f0>] (__do_softirq+0x104/0x234)
[  141.185408]  r9:00000100 r8:c0414000 r7:40000006 r6:c0416080 r5:c0416098 r4:00000000
[  141.193237] [<c00296ec>] (__do_softirq) from [<c0029ba0>] (irq_exit+0xa0/0xb0)
[  141.200486]  r10:c04170cc r9:00000000 r8:df407000 r7:00000001 r6:00000000 r5:00000000
[  141.208399]  r4:c04118f8
[  141.210955] [<c0029b00>] (irq_exit) from [<c0058304>] (__handle_domain_irq+0x94/0xa4)
[  141.218814]  r4:c04118f8 r3:00000100
[  141.222424] [<c0058270>] (__handle_domain_irq) from [<c000875c>] (gic_handle_irq+0x48/0x64)
[  141.230807]  r8:c0414000 r7:e0802100 r6:c0415f20 r5:c0417ca8 r4:e080210c r3:c0415f20
[  141.238635] [<c0008714>] (gic_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
[  141.246147] Exception stack(0xc0415f20 to 0xc0415f68)
[  141.251219] 5f20: ffffffed 00000000 000529f8 c000a280 c0414038 00000000 00000000 c0379388
[  141.259430] 5f40: c0414000 00000000 c04170cc c0415f74 c0415f78 c0415f68 c00187bc c00187c0
[  141.267639] 5f60: 60000013 ffffffff 
[... cut ...]
[  141.285167] [<c0051600>] (cpu_startup_entry) from [<c000d264>] (rest_init+0x7c/0x80)
[  141.292941]  r7:c0417000
[  141.295495] [<c000d1e8>] (rest_init) from [<c03e1db4>] (start_kernel+0x438/0x444)
[  141.303013] [<c03e197c>] (start_kernel) from [<00008074>] (0x8074)
[  141.309219] Code: e894000c e584c000 e584c004 e5823004 (e5832000)
[  141.315337] ---[ end trace 2f86ba3238a27d97 ]---
[  141.319971] Kernel panic - not syncing: Fatal exception in interrupt
[  142.437598] SMP: failed to stop secondary CPUs
[  142.442059] Rebooting in 3 seconds..

XdengaX, 2015-08-02:
Part of the output has been chopped off. It can be seen in its entirety here http://pastebin.com/tfv5nHvw

yuhhaurlin, 2015-11-12:
Does anyone still encounter this problem with 10.3.0.14?

closed 2015-11-24 by yuhhaurlin