kaloz/mwlwifi #32: WDS does not appear to work.

onslauth, 2015-07-31:
Hi,

I have purchased two Linksys WRT1900AC devices, and would like to connect them together using WDS.

I have installed the latest version of the OpenWRT firmware on both devices:
```
root@Chaos-ni:/etc# cat openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='15.05-rc3'
DISTRIB_REVISION='r46163'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05-rc3'
DISTRIB_TAINTS=''
root@Chaos-ni:/etc#
```

```
root@Chaos-ni:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='15.05-rc3'
DISTRIB_REVISION='r46163'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05-rc3'
DISTRIB_TAINTS=''
root@Chaos-ni:~# 
```

I have configured the devices as follows
### AP
#### Network config
```
root@Chaos-ichi:~# cat /etc/config/network 

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd17:4a8c:691d::/48'

config interface 'lan'
	option force_link '1'
	option type 'bridge'
	option proto 'static'
	option ip6assign '60'
	option ipaddr '10.0.1.1'
	option netmask '255.0.0.0'
	option delegate '0'
	option _orig_ifname 'eth0'
	option _orig_bridge 'true'
	option ifname 'eth0'
	option gateway '10.6.6.6'
	option dns '8.8.8.8'
	option stp '1'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0 1 2 3 5'

config switch_vlan
[... 262 lines cut ...]
23:10:19.153513 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
23:10:19.321603 ARP, Request who-has 169.254.251.46 tell 10.0.6.66, length 46
23:10:20.155524 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
23:10:21.153512 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
23:10:22.153512 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
23:10:23.156199 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
23:10:24.153512 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
23:10:25.153512 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
23:10:26.156883 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
23:10:26.469883 STP 802.1d, Config, Flags [none], bridge-id 7fff.c0:56:27:5f:8a:85.8002, length 35
23:10:27.153514 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
23:10:28.153507 ARP, Request who-has 10.0.1.2 tell 10.0.1.1, length 28
```

### Client side
```
23:50:42.577981 STP 802.1d, Config, Flags [none], bridge-id 7fff.c0:56:27:5f:8a:3d.8002, length 35
23:50:44.577969 STP 802.1d, Config, Flags [none], bridge-id 7fff.c0:56:27:5f:8a:3d.8002, length 35
23:50:46.578018 STP 802.1d, Config, Flags [none], bridge-id 7fff.c0:56:27:5f:8a:3d.8002, length 35
23:50:48.577976 STP 802.1d, Config, Flags [none], bridge-id 7fff.c0:56:27:5f:8a:3d.8002, length 35
```

I am happy to provide any other details you require, and to test anything you would like to try.

Thanks

yuhhaurlin, 2015-11-24:
Mesh would be supported. I will close this item.

closed 2015-11-24 by yuhhaurlin

onslauth, 2015-11-26:
Hi, 

What do you mean by 'mesh would be supported' ? Is this a new technology currently available in OpenWRT?

mrefky, 2016-04-11:
I just buy WRT1200AC to be used in a WDS wifi repeater configuration, this is too bad not to have WDS supported ....

fghaas, 2016-12-30:
I'm being bitten by this exact issue, though I wouldn't say it's limited to just ARP requests — rather it seems that no layer 2 broadcasts make it back to the "client side" at all. This also applies to DHCP offers. The "client side" properly forwards the discover packet, the "AP side" responds with an offer, the offer never makes it to the client.

@onslauth, did you ever get to the bottom of this?

onslauth, 2017-01-18:
@fghaas Nah, I haven't actually used the devices since I bought them, I originally bought them in the hope of using them for WDS, and have not played with them since then.

fghaas, 2017-01-18:
@onslauth For what it's worth, I got it to work by doing two things:

- Upgrade to LEDE (which currently runs on a 4.4.39 kernel)
- Restricting my setup to *one* WDS-enabled ESSID.

It appears that if more than one ESSID is defined with WDS in AP mode on the router, then the access points can only join one of them in STA mode. (It's OK to have other ESSIDs defined on the router, you just apparently can't scale those across multiple access points.) I'm not sure if this is an actual driver problem, or if it's something where the network stack on OpenWRT/LEDE simply uses the driver incorrectly.

onslauth, 2017-01-18:
@fghaas OMG really? That's wonderful, I will spend the evening sorting this out.

When you say LEDE, what do you mean? Is that the new code name for OpenWRT release?

fghaas, 2017-01-18:
[LEDE](https://lede-project.org/) is the Linux Embedded Development Environment, you can find downloads for the `mvebu` target (WRT1900AC/ACv2/ACS) are [here](https://downloads.lede-project.org/snapshots/targets/mvebu/generic/).

onslauth, 2017-01-18:
@fghaas Hey, so let me get this straight, I was re-reading your previous comment, and if I understand you correctly, you have linked the two devices using WDS, but then have created different ESSIDs on each device for clients to join?

onslauth, 2017-01-18:
@fghaas Ah, that sucks, I have the V1 edition. I see that on the [OpenWRT Forum](https://wiki.openwrt.org/toh/linksys/wrt1x00ac_series) they had links for LEDE for v1, but they links seem to be 404'd

fghaas, 2017-01-18:
No, the ESSIDs and WPA2 PSKs are identical. The router has the network defined with `wds 1` and `mode ap`. The APs define the same network with `wds 1` and `mode sta` (that's the link to the router), and again with `wds 0` and `mode ap` (that's what Wi-Fi clients connect to).

Also, make sure that your APs explicitly set the router's BSSID. If you configure your networks with LuCI then this happens automatically; if you shell into your system then you must set it in `/etc/config/wireless`.

The LEDE and OpenWRT wikis have more information.

fghaas, 2017-01-18:
Also, check https://downloads.lede-project.org/snapshots/targets/mvebu/generic/ again. The `wrt1900ac` links should be for you. No?

onslauth, 2017-01-23:
@fghaas Thanks for all the help. I updated my routers to the latest LEDE build, and used LUCI to configure it and get it working.

Everything is working great, and thanks for letting me know about the update.

fghaas, 2017-01-23:
Sweet! Glad to hear those boxes aren't collecting dust anymore. :)