kaloz/mwlwifi #33: please remove kernel debug messages

DavidMcWRT, 2015-08-04:
Please can a revised driver be released removing (or at the very least, reducing) the kernel debug messages - at least for CC [Final].

[294174.688370] ieee80211 phy1: interface: 3, change: 0x36
[303998.722569] ieee80211 phy1: interface: 3, change: 0x100
[303998.727999] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[303998.749359] ieee80211 phy1: interface: 3, change: 0x36
[329565.546830] ieee80211 phy1: interface: 3, change: 0x100
[329565.552291] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[329565.573764] ieee80211 phy1: interface: 3, change: 0x36
[329601.516977] ieee80211 phy0: interface: 3, change: 0x100
[329601.522428] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[329601.543923] ieee80211 phy0: interface: 3, change: 0x3e
[330045.236466] ieee80211 phy1: interface: 3, change: 0x100
[330045.241842] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[330045.263376] ieee80211 phy1: interface: 3, change: 0x36
[331218.023842] ieee80211 phy0: interface: 3, change: 0x100
[331218.029237] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[331218.050776] ieee80211 phy0: interface: 3, change: 0x3e
[331219.590326] ieee80211 phy1: interface: 3, change: 0x100
[331219.595790] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[331219.617294] ieee80211 phy1: interface: 3, change: 0x36
[332652.975943] ieee80211 phy1: interface: 3, change: 0x100
[332652.981368] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[332653.002773] ieee80211 phy1: interface: 3, change: 0x36
[352344.946098] ieee80211 phy1: interface: 3, change: 0x100
[352344.951489] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[352344.973210] ieee80211 phy1: interface: 3, change: 0x36
[357890.249390] ieee80211 phy1: interface: 3, change: 0x100
[357890.254795] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[357890.276301] ieee80211 phy1: interface: 3, change: 0x36
[386019.703305] ieee80211 phy1: interface: 3, change: 0x100
[386019.708821] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[386019.730303] ieee80211 phy1: interface: 3, change: 0x36
[386927.897351] ieee80211 phy1: interface: 3, change: 0x100
[386927.902781] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[386927.924145] ieee80211 phy1: interface: 3, change: 0x36
[395046.092282] ieee80211 phy0: interface: 3, change: 0x100
[395046.097744] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[395046.119211] ieee80211 phy0: interface: 3, change: 0x3e
[395447.753753] ieee80211 phy0: interface: 3, change: 0x100
[395447.759128] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[395447.780521] ieee80211 phy0: interface: 3, change: 0x3e
[395595.651746] ieee80211 phy0: interface: 3, change: 0x100
[395595.657176] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[395595.678538] ieee80211 phy0: interface: 3, change: 0x3e
[397091.251543] ieee80211 phy1: interface: 3, change: 0x100
[397091.256980] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[397091.278556] ieee80211 phy1: interface: 3, change: 0x36
[397092.079116] ieee80211 phy0: interface: 3, change: 0x100
[397092.084548] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[397092.105913] ieee80211 phy0: interface: 3, change: 0x3e
[397149.269126] ieee80211 phy0: interface: 3, change: 0x100
[397149.274583] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[397149.296077] ieee80211 phy0: interface: 3, change: 0x3e
[397247.779537] ieee80211 phy0: interface: 3, change: 0x100
[397247.784931] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[397247.806485] ieee80211 phy0: interface: 3, change: 0x3e
[443258.322925] ieee80211 phy1: interface: 3, change: 0x100
[443258.328367] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[443258.349937] ieee80211 phy1: interface: 3, change: 0x36
[443262.219158] ieee80211 phy1: interface: 3, change: 0x100
[443262.224664] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[443262.246189] ieee80211 phy1: interface: 3, change: 0x36
[443265.924133] ieee80211 phy1: interface: 3, change: 0x100
[443265.929584] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[443265.951029] ieee80211 phy1: interface: 3, change: 0x36
[455140.216034] ieee80211 phy1: interface: 3, change: 0x100
[455140.221547] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[455140.242991] ieee80211 phy1: interface: 3, change: 0x36
[461625.896051] ieee80211 phy0: interface: 3, change: 0x100
[... 61 lines cut ...]
[602624.798306] ieee80211 phy0: interface: 3, change: 0x3e
[603054.421603] ieee80211 phy1: interface: 3, change: 0x100
[603054.427259] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[603054.449622] ieee80211 phy1: interface: 3, change: 0x36
[603252.601849] ieee80211 phy0: interface: 3, change: 0x100
[603252.608100] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[603252.634808] ieee80211 phy0: interface: 3, change: 0x3e
[603254.192468] ieee80211 phy1: interface: 3, change: 0x100
[603254.198579] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[603254.223611] ieee80211 phy1: interface: 3, change: 0x36
[617528.367011] ieee80211 phy1: interface: 3, change: 0x100
[617528.372388] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[617528.393781] ieee80211 phy1: interface: 3, change: 0x36
[637950.291900] ieee80211 phy1: interface: 3, change: 0x100
[637950.297403] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[637950.318964] ieee80211 phy1: interface: 3, change: 0x36
[638187.868693] ieee80211 phy1: interface: 3, change: 0x100
[638187.874124] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[638187.895498] ieee80211 phy1: interface: 3, change: 0x36
[640286.627676] ieee80211 phy1: interface: 3, change: 0x100
[640286.633042] ieee80211 phy1: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[640286.654589] ieee80211 phy1: interface: 3, change: 0x36
[643510.688662] ieee80211 phy0: interface: 3, change: 0x100
[643510.694094] ieee80211 phy0: wmm:26, rsn:0, rsn48:22, ht:62, vht:0
[643510.715765] ieee80211 phy0: interface: 3, change: 0x3e

DavidMcWRT, 2015-08-04:
@kaloz @yuhhaurlin

yuhhaurlin, 2015-08-04:
Can I remove it when I release next version?

DavidMcWRT, 2015-08-04:
@yuhhaurlin @kaloz 

if you're planning on next version before CC Final then yes ;)

I just don't think all debug stuff is good for what's going to be a production release.

yuhhaurlin, 2015-08-04:
When will be the due date for CC Final?

DavidMcWRT, 2015-08-04:
@yuhhaurlin @kaloz 

"When it's ready" ;)

The CC RCs have been 3 to 4 weeks apart, and last one was about 3 weeks ago.  Don't know if there will be an RC4 before Final.

There wasn't an RC4 with BB.  The gap between BB RC3 and BB Final was just under 7 weeks though.

Of course, past no indication of the future.

Kaloz may have a better idea of when.

kaloz, 2015-08-04:
Well, soon enough. @yuhhaurlin : what's your ETA on the new version?

yuhhaurlin, 2015-08-04:
Next week.

yuhhaurlin, 2015-08-04:
I think I will mark off these code now.

yuhhaurlin, 2015-08-04:
Done.

DavidMcWRT, 2015-08-04:
@yuhhaurlin @kaloz 

Thanks for doing this.

Hope you can still push for the new version too and that it might get into RC4/Final if possible.

Appreciate all your work on this firmware in general too - thanks.

closed 2015-08-04 by DavidMcWRT