kaloz/mwlwifi #34: performance trouble with 10.3.0.8

notnyt, 2015-08-19:
When upgrading to 59e0d3d167a5bc2c03010df4ed54cc0653b5ae58 from aaf953e1baf9cd65979103c4fda98c0bd48f4759 wifi perormance on both 2.4ghz and 5ghz was severely impacted.  Reverting to the previous version resolved the issue.

The affected client was a 2008 unibody macbook.

An LG G4 phone seemed unaffected.

The driver was outputting the following
````
[  409.130865] ieee80211 phy1: check ba result error
[  410.164653] ieee80211 phy1: check ba result error
[  411.199304] ieee80211 phy1: check ba result error
[  412.234463] ieee80211 phy1: check ba result error
[  413.263482] ieee80211 phy1: error code: -22
[  445.911823] ieee80211 phy1: check ba result error
[  447.935847] ieee80211 phy1: check ba result error
[  449.958488] ieee80211 phy1: check ba result error
````

yuhhaurlin, 2015-08-20:
Did the client have problem with 10.3.0.3?

yuhhaurlin, 2015-08-20:
Sorry. It is all right for 10.3.0.3. Please wait for next release to see if you still have this kind of problem.

damianyang, 2015-08-20:
I am having same issues with MacBook Pro (Mid 2012 Retina) client on WRT1200AC with this driver. I am currently running in legacy mode (11g) to avoid this issue.

notnyt, 2015-08-27:
@yuhhaurlin  any eta on the next release?

yuhhaurlin, 2015-08-31:
10.3.0.9 is under development. Except for this fix, it will enhance Tx throughput of mwlwifi driver to be as good as proprietary driver. Debugfs will also be added to this version of code. Once if done, I will synchronize it to here.

Note: With support of AMSDU, Rx throughput of mwlwifi driver is almost the same as proprietary driver, but Tx throughput is still not as good as proprietary driver.

tusc, 2015-08-31:
will 10.3.0.9 address issue #21? This is the biggest issue right now with the wireless driver and I would categorize performance as secondary over stability.

kaloz, 2015-08-31:
@yuhhaurlin : could you add support for mac80211's chainmask get/set functions, too? It would allow custom configs as well as valid info in "iw phy phy0 info".

thagabe, 2015-08-31:
@yuhhaurlin I agree with @tusc. My router heats up quite a bit (mainly because it is summer here) and I have noticed stability and performance issues. If i recall correctly the wireless chip was the one that caused the halt in the system. I have tried to keep my system cooler by rewriting the default fan control but i have found limited success. Stability should take center stage however performance is always a welcomed change. I think it's too late to refocus the attention to stability on 10.3.0.9 but for the next release stability should be taken more into consideration as well as stomping out the problem with weird performance inconsistencies. Thank you for the update.

TireMeat, 2015-08-31:
(sorry for the "set IRQ" bold thing ... unintentional)

tusc, 2015-08-31:
@TireMeat,

when you say hang did you the message "INFO: rcu_sched self-detected stall on CPU...." in your syslog? This is the issue that is reported in #21

TireMeat, 2015-08-31:
The hang was the same as described in #21 except that wireless did not connect/disconnect while my file transfer tests were running.   I'll confirm tonight and let you know.

TireMeat, 2015-09-01:
@tusc,

I tried (and succeeded) in hanging the router in various ways, but have not seen the #21 message in my syslog.   I'm having it syslog remotely to a server wired directly into the 1900's switch using static IPs on both 1900 and the server, which works fine -- though I'm thinking to make the serial header to see any console logs instead of relying on syslog.

Incidentally, while LAN_SpeedTest utility didn't lock up the router all day while I was at work, downloading a torrent on my PC through the 1900 via 5G ac WiFi did so in maybe less than a minute.

rpallai, 2015-09-02:
IMHO "error code: -22" is not just a performance trouble but even stability. My WRT1200AC dropped the connection of a client after 1-2 hours downloading with this error.

I've 42MB of 512MB RAM used and seen much higher data rates on this router than this download, it's not IRQ balance or memory pressure problem.

mwlwifi 10.3.0.8 in CHAOS CALMER (Bleeding Edge, r46760)

Log messages:
```
Wed Sep  2 19:38:24 2015 kern.err kernel: [22476.624445] ieee80211 phy1: check ba result error
Wed Sep  2 19:38:25 2015 kern.err kernel: [22477.649258] ieee80211 phy1: check ba result error
Wed Sep  2 19:38:26 2015 kern.err kernel: [22478.671713] ieee80211 phy1: check ba result error
Wed Sep  2 19:38:27 2015 kern.err kernel: [22479.694064] ieee80211 phy1: check ba result error
Wed Sep  2 19:38:28 2015 kern.err kernel: [22480.716321] ieee80211 phy1: check ba result error
Wed Sep  2 19:38:29 2015 kern.err kernel: [22481.733487] ieee80211 phy1: error code: -22
Wed Sep  2 19:38:29 2015 daemon.info hostapd: wlan1: STA a0:f3:c1:f8:9b:e0 IEEE 802.11: authenticated
Wed Sep  2 19:38:29 2015 daemon.info hostapd: wlan1: STA a0:f3:c1:f8:9b:e0 IEEE 802.11: authenticated
Wed Sep  2 19:38:30 2015 daemon.info hostapd: wlan1: STA a0:f3:c1:f8:9b:e0 IEEE 802.11: authenticated
Wed Sep  2 19:38:30 2015 daemon.info hostapd: wlan1: STA a0:f3:c1:f8:9b:e0 IEEE 802.11: authenticated
Wed Sep  2 19:38:31 2015 daemon.info hostapd: wlan1: STA a0:f3:c1:f8:9b:e0 IEEE 802.11: authenticated

Sep 02 19:38:25 dh.darknet NetworkManager[1273]: <warn>  Connection disconnected (reason -4)
Sep 02 19:38:25 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 02 19:38:25 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 02 19:38:26 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 02 19:38:26 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 02 19:38:26 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 02 19:38:26 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 2/3)
Sep 02 19:38:26 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 3/3)
Sep 02 19:38:27 dh.darknet kernel: wlp2s0: authentication with 00:25:9c:13:b0:8b timed out
Sep 02 19:38:27 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: authenticating -> disconnected
Sep 02 19:38:27 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 02 19:38:28 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 02 19:38:28 dh.darknet kernel: wlp2s0: direct probe to 00:25:9c:13:b0:8b (try 1/3)
Sep 02 19:38:28 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 02 19:38:28 dh.darknet kernel: wlp2s0: direct probe to 00:25:9c:13:b0:8b (try 2/3)
Sep 02 19:38:28 dh.darknet kernel: wlp2s0: direct probe to 00:25:9c:13:b0:8b (try 3/3)
Sep 02 19:38:28 dh.darknet kernel: wlp2s0: authentication with 00:25:9c:13:b0:8b timed out
Sep 02 19:38:28 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: authenticating -> disconnected
Sep 02 19:38:29 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 02 19:38:30 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 02 19:38:30 dh.darknet kernel: wlp2s0: direct probe to 00:25:9c:13:b0:8b (try 1/3)
Sep 02 19:38:30 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 02 19:38:30 dh.darknet kernel: wlp2s0: 00:25:9c:13:b0:8b unexpected authentication state: alg 0 (expected 0) transact 2 (expected 0)
Sep 02 19:38:30 dh.darknet kernel: wlp2s0: 00:25:9c:13:b0:8b unexpected authentication state: alg 0 (expected 0) transact 2 (expected 0)
Sep 02 19:38:30 dh.darknet kernel: wlp2s0: 00:25:9c:13:b0:8b unexpected authentication state: alg 0 (expected 0) transact 2 (expected 0)
Sep 02 19:38:30 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 2/3)
Sep 02 19:38:31 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 3/3)
Sep 02 19:38:31 dh.darknet kernel: wlp2s0: authentication with 00:25:9c:13:b0:8b timed out
Sep 02 19:38:31 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: authenticating -> disconnected
Sep 02 19:38:31 dh.darknet NetworkManager[1273]: <info>  Connectivity check for uri 'https://fedoraproject.org/static/hotspot.txt' failed with 'Peer failed to perform TLS handshake'.
Sep 02 19:38:36 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 02 19:38:41 dh.darknet NetworkManager[1273]: <warn>  (wlp2s0): link timed out.
Sep 02 19:38:41 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): device state change: activated -> failed (reason 'ssid-not-found') [100 120 53]
Sep 02 19:38:41 dh.darknet NetworkManager[1273]: <warn>  (wlp2s0): Activation: failed for connection 'OpenWrt2 (test)'
Sep 02 19:38:41 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): device state change: failed -> disconnected (reason 'none') [120 30 0]
Sep 02 19:38:41 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): deactivating device (reason 'none') [0]
Sep 02 19:38:41 dh.darknet kernel: IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
Sep 02 19:38:41 dh.darknet NetworkManager[1273]: <warn>  Couldn't disconnect supplicant interface: This interface is not connected.
Sep 02 19:38:41 dh.darknet NetworkManager[1273]: <info>  (wlp2s0): supplicant interface state: scanning -> inactive
```

yuhhaurlin, 2015-09-03:
@ rpallai
Did you encounter the same problem on 10.3.0.3?

notnyt, 2015-09-03:
I get a bunch of error code: -22 on 10.3.0.3
````
[289503.844760] ieee80211 phy1: result error
[289504.865325] ieee80211 phy1: result error
[289505.880306] ieee80211 phy1: result error
[289506.896666] ieee80211 phy1: result error
[289507.911764] ieee80211 phy1: result error
[289508.927153] ieee80211 phy1: error code: -22
[289509.954501] ieee80211 phy1: result error
[289510.963250] ieee80211 phy1: result error
[289511.971403] ieee80211 phy1: result error
[289512.979514] ieee80211 phy1: result error
[289513.987828] ieee80211 phy1: result error
[289514.993881] ieee80211 phy1: error code: -22
[289559.213291] ieee80211 phy1: result error
[289560.234195] ieee80211 phy1: result error
[289561.248002] ieee80211 phy1: result error
[289562.269722] ieee80211 phy1: result error
[289563.294217] ieee80211 phy1: result error
[289564.377248] ieee80211 phy1: error code: -22
[289564.523108] ieee80211 phy1: result error
[289565.578803] ieee80211 phy1: result error
[289566.651762] ieee80211 phy1: result error
[289567.749378] ieee80211 phy1: result error
[289568.865748] ieee80211 phy1: result error
[289569.916856] ieee80211 phy1: error code: -22
[289570.472904] ieee80211 phy1: result error
````

yuhhaurlin, 2015-09-03:
@ notnyt

What client did you use?

yuhhaurlin, 2015-09-03:
@ notnyt

The problem also happened after running for a while?

yuhhaurlin, 2015-09-03:
Tx throughput is enhanced to be as good as proprietary driver now. One problem of 10.3.0.8 is that all packets in AMSDU should be acknowledged to mac80211.

About issue #21 and check ba error, we will try to fix it. Thanks.

notnyt, 2015-09-03:
@yuhhaurlin Problem can happen right away, doesn't seem to be time dependent.  It might be a 2008 unibody macbook running osx triggering that, but not sure.

````
Mon Aug 31 11:09:16 2015 daemon.info hostapd: wlan0: STA 00:23:12:54:04:63 IEEE 802.11: disassociated
Mon Aug 31 11:09:17 2015 daemon.info hostapd: wlan1: STA 00:23:12:54:04:63 IEEE 802.11: authenticated
Mon Aug 31 11:09:17 2015 daemon.info hostapd: wlan1: STA 00:23:12:54:04:63 IEEE 802.11: associated (aid 1)
Mon Aug 31 11:09:17 2015 daemon.info hostapd: wlan1: STA 00:23:12:54:04:63 WPA: pairwise key handshake completed (RSN)
Mon Aug 31 11:09:17 2015 daemon.info hostapd: wlan0: STA 00:23:12:54:04:63 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Aug 31 11:09:57 2015 kern.err kernel: [289503.844760] ieee80211 phy1: result error
Mon Aug 31 11:09:58 2015 kern.err kernel: [289504.865325] ieee80211 phy1: result error
Mon Aug 31 11:09:59 2015 kern.err kernel: [289505.880306] ieee80211 phy1: result error
Mon Aug 31 11:10:00 2015 kern.err kernel: [289506.896666] ieee80211 phy1: result error
Mon Aug 31 11:10:01 2015 kern.err kernel: [289507.911764] ieee80211 phy1: result error
Mon Aug 31 11:10:02 2015 kern.err kernel: [289508.927153] ieee80211 phy1: error code: -22
Mon Aug 31 11:10:03 2015 kern.err kernel: [289509.954501] ieee80211 phy1: result error
Mon Aug 31 11:10:04 2015 kern.err kernel: [289510.963250] ieee80211 phy1: result error
Mon Aug 31 11:10:05 2015 kern.err kernel: [289511.971403] ieee80211 phy1: result error
Mon Aug 31 11:10:06 2015 kern.err kernel: [289512.979514] ieee80211 phy1: result error
Mon Aug 31 11:10:07 2015 kern.err kernel: [289513.987828] ieee80211 phy1: result error
Mon Aug 31 11:10:08 2015 kern.err kernel: [289514.993881] ieee80211 phy1: error code: -22
Mon Aug 31 11:10:52 2015 kern.err kernel: [289559.213291] ieee80211 phy1: result error
Mon Aug 31 11:10:54 2015 kern.err kernel: [289560.234195] ieee80211 phy1: result error
Mon Aug 31 11:10:55 2015 kern.err kernel: [289561.248002] ieee80211 phy1: result error
Mon Aug 31 11:10:56 2015 kern.err kernel: [289562.269722] ieee80211 phy1: result error
Mon Aug 31 11:10:57 2015 kern.err kernel: [289563.294217] ieee80211 phy1: result error
Mon Aug 31 11:10:58 2015 kern.err kernel: [289564.377248] ieee80211 phy1: error code: -22
Mon Aug 31 11:10:58 2015 kern.err kernel: [289564.523108] ieee80211 phy1: result error
Mon Aug 31 11:10:59 2015 kern.err kernel: [289565.578803] ieee80211 phy1: result error
Mon Aug 31 11:11:00 2015 kern.err kernel: [289566.651762] ieee80211 phy1: result error
Mon Aug 31 11:11:01 2015 kern.err kernel: [289567.749378] ieee80211 phy1: result error
Mon Aug 31 11:11:02 2015 kern.err kernel: [289568.865748] ieee80211 phy1: result error
Mon Aug 31 11:11:03 2015 kern.err kernel: [289569.916856] ieee80211 phy1: error code: -22
Mon Aug 31 11:11:04 2015 kern.err kernel: [289570.472904] ieee80211 phy1: result error
````

damianyang, 2015-09-03:
I had same error messages (running trunk on WRT1200AC) with MacBook pro
2012 Retina. It initially started with just result error back with 10.3.0.3
then it changed to check ba result error when using 10.3.0.8.

@yuhhaurlin <https://github.com/yuhhaurlin> Problem can happen right away,
doesn't seem to be time dependent. It might be a 2008 unibody macbook
running osx triggering that, but not sure.

Mon Aug 31 11:09:16 2015 daemon.info hostapd: wlan0: STA
00:23:12:54:04:63 IEEE 802.11: disassociated
Mon Aug 31 11:09:17 2015 daemon.info hostapd: wlan1: STA
00:23:12:54:04:63 IEEE 802.11: authenticated
Mon Aug 31 11:09:17 2015 daemon.info hostapd: wlan1: STA
00:23:12:54:04:63 IEEE 802.11: associated (aid 1)
Mon Aug 31 11:09:17 2015 daemon.info hostapd: wlan1: STA
00:23:12:54:04:63 WPA: pairwise key handshake completed (RSN)
Mon Aug 31 11:09:17 2015 daemon.info hostapd: wlan0: STA
00:23:12:54:04:63 IEEE 802.11: deauthenticated due to inactivity
(timer DEAUTH/REMOVE)
Mon Aug 31 11:09:57 2015 kern.err kernel: [289503.844760] ieee80211
phy1: result error
Mon Aug 31 11:09:58 2015 kern.err kernel: [289504.865325] ieee80211
phy1: result error
Mon Aug 31 11:09:59 2015 kern.err kernel: [289505.880306] ieee80211
phy1: result error
Mon Aug 31 11:10:00 2015 kern.err kernel: [289506.896666] ieee80211
phy1: result error
Mon Aug 31 11:10:01 2015 kern.err kernel: [289507.911764] ieee80211
phy1: result error
Mon Aug 31 11:10:02 2015 kern.err kernel: [289508.927153] ieee80211
phy1: error code: -22
Mon Aug 31 11:10:03 2015 kern.err kernel: [289509.954501] ieee80211
phy1: result error
Mon Aug 31 11:10:04 2015 kern.err kernel: [289510.963250] ieee80211
phy1: result error
Mon Aug 31 11:10:05 2015 kern.err kernel: [289511.971403] ieee80211
phy1: result error
Mon Aug 31 11:10:06 2015 kern.err kernel: [289512.979514] ieee80211
phy1: result error
Mon Aug 31 11:10:07 2015 kern.err kernel: [289513.987828] ieee80211
phy1: result error
Mon Aug 31 11:10:08 2015 kern.err kernel: [289514.993881] ieee80211
phy1: error code: -22
Mon Aug 31 11:10:52 2015 kern.err kernel: [289559.213291] ieee80211
phy1: result error
Mon Aug 31 11:10:54 2015 kern.err kernel: [289560.234195] ieee80211
phy1: result error
Mon Aug 31 11:10:55 2015 kern.err kernel: [289561.248002] ieee80211
phy1: result error
Mon Aug 31 11:10:56 2015 kern.err kernel: [289562.269722] ieee80211
phy1: result error
Mon Aug 31 11:10:57 2015 kern.err kernel: [289563.294217] ieee80211
phy1: result error
Mon Aug 31 11:10:58 2015 kern.err kernel: [289564.377248] ieee80211
phy1: error code: -22
Mon Aug 31 11:10:58 2015 kern.err kernel: [289564.523108] ieee80211
phy1: result error
Mon Aug 31 11:10:59 2015 kern.err kernel: [289565.578803] ieee80211
phy1: result error
Mon Aug 31 11:11:00 2015 kern.err kernel: [289566.651762] ieee80211
phy1: result error
Mon Aug 31 11:11:01 2015 kern.err kernel: [289567.749378] ieee80211
phy1: result error
Mon Aug 31 11:11:02 2015 kern.err kernel: [289568.865748] ieee80211
phy1: result error
Mon Aug 31 11:11:03 2015 kern.err kernel: [289569.916856] ieee80211
phy1: error code: -22
Mon Aug 31 11:11:04 2015 kern.err kernel: [289570.472904] ieee80211
phy1: result error

yuhhaurlin, 2015-09-03:
Thanks. 10.3.0.8 will give more clear ba related error messages.

rpallai, 2015-09-03:
@ yuhhaurlin

10.3.0.3 has another issue that prevents longer stress testing: each tcp stream begins dying after the initial burst after 40-50 minutes of downloading.

Connection stability of the client is not better on 10.3.0.3, reconnection happens in every 5-10 minutes as with 10.3.0.8. Most disconnections has no related error log entry in the AP log, happens for unknown reasons.
On the other hand connection of this client to my WRT160NL AP is very stable.

10.3.0.8 is free from the tcp dying issue so I can stress it longer, I think this is why I don't encounter exactly the same problem on 10.3.0.3.

```
Sep 03 03:58:04 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 03:58:04 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 03:58:04 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 03:58:05 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: authenticated
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 03:58:05 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 03:58:05 dh.darknet kernel: wlp2s0: associated
Sep 03 03:58:05 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> associated
Sep 03 03:58:05 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
Sep 03 03:58:53 dh.darknet kernel: ath: phy0: Failed to stop TX DMA, queues=0x008!
Sep 03 03:59:16 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 03:59:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 03:59:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 03:59:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: authenticated
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 03:59:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associating
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 03:59:17 dh.darknet kernel: wlp2s0: associated
Sep 03 03:59:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associating -> associated
Sep 03 03:59:17 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
Sep 03 04:00:08 dh.darknet kernel: audit_printk_skb: 9 callbacks suppressed
Sep 03 04:02:09 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 04:02:09 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 04:02:09 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 04:02:10 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: authenticated
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 04:02:10 dh.darknet kernel: wlp2s0: associated
Sep 03 04:02:10 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associated
Sep 03 04:02:10 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
Sep 03 04:08:15 dh.darknet NetworkManager[17950]: <warn>  Connection disconnected (reason -4)
Sep 03 04:08:15 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: completed -> disconnected
Sep 03 04:08:15 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: disconnected -> scanning
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: authenticate with 00:25:9c:13:b0:8b
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: send auth to 00:25:9c:13:b0:8b (try 1/3)
Sep 03 04:08:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: scanning -> authenticating
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: authenticated
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: associate with 00:25:9c:13:b0:8b (try 1/3)
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: RX AssocResp from 00:25:9c:13:b0:8b (capab=0x431 status=0 aid=1)
Sep 03 04:08:16 dh.darknet kernel: wlp2s0: associated
Sep 03 04:08:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: authenticating -> associated
Sep 03 04:08:16 dh.darknet NetworkManager[17950]: <info>  (wlp2s0): supplicant interface state: associated -> completed
```

yuhhaurlin, 2015-09-03:
Yes, driver needs to ack every packets in AMSDU to mac80211.

notnyt, 2015-09-03:
@yuhhaurlin any eta on fixes?

rpallai, 2015-09-03:
Looking further I have 2 independent issues here:
1. "check ba result error" causes long traffic stalling lasts even for a half of a minute. It is likely prevent even authentication on the AP.
2. client getting disconnected every 5-10 minutes without error message at the AP side.

If both problems strikes at once then client gets disconnected permanently. Resolving one of them resolves the permanent client disconnection what is my first problem.

yuhhaurlin, 2015-09-03:
If 10.3.0.3 has this problem, I need to check what the problem is. 10.3.0.3 already acked every packets. Does anyone have this kind of problem with 10.3.0.3?

yuhhaurlin, 2015-09-03:
Please use "iw event -f" to get event log. From our test here, we did not find this kind of problem on 10.3.0.3.

rpallai, 2015-09-03:
@yuhhaurlin: The message was slightly different with 10.3.0.3 but appears at same rate here. Should I run "iw event -f" with RC3 and post the results?

```
phy1: result error
phy1: error code: -22
```

yuhhaurlin, 2015-09-03:
Sorry. From our test here, 10.3.0.3 will come out check ba error sometimes. I will try to fix it. But we did not find connect/disconnect problem on 10.3.0.3. "iw event -f" will show connecton related events. I wonder if connect/disconnect problem only happened for specific clients?

rpallai, 2015-09-03:
@yuhhaurlin: the connect/disconnect problem related to this ticket?

"check ba result error" can happen without disconnection and disconnection can happen without "check ba result error". They are not joined together, just happens at once sometimes.

Anyway, I definitely have the connection stability problem with 10.3.0.3 too, but I'm on the latest currently. I'll downgrade to RC3 and show the results of "iw event -f". I had only one client connected to the AP.
Should I connect more clients for the test?

yuhhaurlin, 2015-09-03:
You can offer information to me. I will focus on check ba error and issue #21 first. Thanks.

rpallai, 2015-09-03:
@yuhhaurlin Logs posted about connect/disconnect problem with 10.3.0.3 in #26. I'd leave this ticket for the traffic stalling issue followed by message "result error".

TireMeat, 2015-09-04:
With 10.3.0.8 "iw event -f" reports this right before wifi dies:

root@WRT1900:~# iw event -f
wlan1 (phy #1): mgmt TX status (cookie e): acked
wlan1 (phy #1): mgmt TX status (cookie f): acked

TireMeat, 2015-09-08:
I made the modification below to tx.c in my build environment, (note, lines 363-372 are commented out, and if statement added after).   Performance is on-par with my D825 and no panics/halts --- would you guys check on your end?

(BTW, looking at some other 802.11ac drivers, I don't think this is how ampdu streams are supposed to be set up)

	/* reset the packet count after each second elapses.  If the number of
	 * packets ever exceeds the ampdu_min_traffic threshold, we will allow
	 * an ampdu stream to be started.
	 *
	if (jiffies - tx_stats->start_time > HZ) {
		tx_stats->pkts = 0;
		tx_stats->start_time = 0;
	} else {
		tx_stats->pkts++;
	}*/

	if (jiffies - tx_stats->start_time < HZ)
		tx_stats->pkts++;

notnyt, 2015-09-09:
@yuhhaurlin  any comments on previous post?

yuhhaurlin, 2015-09-09:
No.

rpallai, 2015-09-12:
Seems like this issue depends on device used on the client side.

I've not seen this error after I connected with a RTL8812AE or Intel-7260 PCIE-E WIFI at 2.4 Ghz/HT40 driven by Fedora. My previous device triggered this issue was an AR9287 PCI-E WIFI.

notnyt, 2015-09-17:
@yuhhaurlin any chance of seeing 10.3.0.9 soon?

notnyt, 2015-10-17:
@yuhhaurlin this is still broken in 10.3.0.10.  I thought this was supposed to be fixed?

When I connect my macbook it gets awful speeds and the kernel generates these errors.
````
[ 1333.191803] ieee80211 phy1: ampdu start error code: -22
[ 1333.238473] ieee80211 phy1: check ba result error 1
[ 1333.243536] ieee80211 phy1: ampdu start error code: -22
[ 1333.277240] ieee80211 phy1: check ba result error 1
[ 1333.282259] ieee80211 phy1: ampdu start error code: -22
[ 1333.346929] ieee80211 phy1: check ba result error 1
[ 1333.351911] ieee80211 phy1: ampdu start error code: -22
[ 1333.386907] ieee80211 phy1: check ba result error 1
[ 1333.391905] ieee80211 phy1: ampdu start error code: -22
[ 1333.447593] ieee80211 phy1: check ba result error 1
[ 1333.452599] ieee80211 phy1: ampdu start error code: -22
[ 1333.496957] ieee80211 phy1: check ba result error 1
[ 1333.502156] ieee80211 phy1: ampdu start error code: -22
[ 1333.557225] ieee80211 phy1: check ba result error 1
[ 1333.562226] ieee80211 phy1: ampdu start error code: -22
[ 1333.694616] ieee80211 phy1: check ba result error 1
[ 1333.699549] ieee80211 phy1: ampdu start error code: -22
[ 1333.754431] ieee80211 phy1: check ba result error 1
[ 1333.759513] ieee80211 phy1: ampdu start error code: -22

````

notnyt, 2015-10-17:
@yuhhaurlin I tried @TireMeat 's fix and it seems to resolve the issue.  I do not know if it has any other negative effects, but you should consider looking into it further as previously requested.

notnyt, 2015-10-17:
Sorry, @TireMeat 's fix completely limits the speed.

yuhhaurlin, 2015-10-17:
Thanks, I will check this one.

Gazoo, 2015-10-17:
It seems that 10.3.0.10 has introduced some more trouble. Quote: Now one CPU is averaging 54% busy with just a few wireless clients connected. Watching /proc/interrupts, it looks like each mwlwifi is generating nearly 2K interrupts per second.

@yuhhaurlin for more info:
https://forum.openwrt.org/viewtopic.php?pid=296347#p296347

I appreciate all the time you are putting into getting this solved!

TireMeat, 2015-10-17:
@yuhhaurlin - great clean-up in 10.3.0.10.    I'm looking at the rx.c changes you made -- so cool, I think those were badly needed.  

I wanted to share some cleanup that's worked well for my custom build -- I'm working with branch, since I see the direction the DD guys are going is good but pretty steep to adjust my setup to that right now.  I figure to give the transmit side (tx, mac80211) a little more work and then see how that works in trunk before adapting to DD.  

Anyway, below are the latest I have on the tx side of things, (below the highlights listed is my patch):
1) fwcmd.c - removed the threshold condition to start ampdu.   This yielded the biggest performance increase I've seen yet.
2) fwdl.c - increased the time the firmware has to load.   On my v1 Linksys, 1 ms was too fast and it would lock up or reset - (am using a small prime number because it's a cheap way to prevent race conditions and such with whatever else going on that's time bound).
3) mac80211.c - simplified the ampdu action logic.   My thinking is that this is a full MAC - let mac80211 tell it what needs to be done instead of doing it (i.e. sequencing).   I don't have any Apple stacks besides my and my wife's iPhone 6s, but I did see that those are sensitive to sequencing - this clean-up helped to stabilize Apple access for me.
4) tx.c - removed the packet counting function altogether and also the calls in xmit.   I took out the fwcmd condition as mentioned in #1 above.   I also noticed in xmit that some variables were defined and used without (immediate) explicit initialization (like mwl_priv) -- this is sometimes problematic because the variables (pointers) could reference structures like buffers, streams, hw, etc. in-memory that shouldn't be there or start acting on these in an unknown state.   For the same reason, I also cleaned-out the ampdu action logic belonging in mac80211, and limited tx to just handle the QoS'ness of the packets.  From a stability standpoint, depending on the conditions underlying an IRQ, the kernel could enter mac80211.ko or mwlwifi.ko ... the two could be acting on ampdu streams and relying on stream->state is not enough to prevent lockups, resets, etc.  For me this improved the stability a great deal -- also, looking at data transfer patterns (i.e. the traffic graph in Luci for instance or a the traffic pattern in uTorrent or similar), the streams are much tighter and fewer gaps (i.e. less context switching and IRQ handling).

I am actively looking at the performance problem - good news, is under certain circumstances I can get within ~1MB/s of the stock Linksys firmware.  Bad news is that I have narrowed down the circumstances, but not yet nailed the root cause preventing faster speed.  I believe the issue lies somewhere in the receive side (hence the reason I dug into your rx.c so quickly when I saw you posted  :-)).    Any info you have that you can share on how rx behaves in this firmware, I'd love to hear about.  

On the IRQ affinity - I found it helps quite a bit to put both mwlwifi (phy0 & phy1) and the mvneta (eth0 & eth1) on the second CPU core.  My guess is that the whole network stack caches nicely in CC and makes prevents expensive memory access when handling IRQs, and the cores don't seem to share cache ?    I purposely left the USB and other (much) slower stuff on the first core.   

Last, there's some really good stuff in rx in 10.3.0.10 like I said, and I'll try to integrate those changes in my build and let you know how I do.

Here is my wireless = 

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
	option htmode 'HT20'
	option country 'US'
	option channel '4'
	option txpower '30'

config wifi-iface
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'O5'
	option encryption 'psk2+ccmp'
	option key 'sljfs;dfjas'
	option hidden '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:03.0/0000:03:00.0'
	option htmode 'VHT80'
	option country 'US'
	option channel '149'
	option txpower '30'

config wifi-iface
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'O5'
	option encryption 'psk2+ccmp'
	option key 'sljfs;dfjas'
	option hidden '1'

Patch =

diff original_CC_branch/fwcmd.c updated_CC_branch/fwcmd.c
2955,2956c2955
< 	return (sta_info->is_ampdu_allowed &&
< 		tx_stats->pkts > SYSADPT_AMPDU_PACKET_THRESHOLD);
---
> 	return (sta_info->is_ampdu_allowed);
diff original_CC_branch/fwdl.c updated_CC_branch/fwdl.c
20c20
< #define FW_CHECK_MSECS                  1
---
> #define FW_CHECK_MSECS                  3
diff original_CC_branch/mac80211.c updated_CC_branch/mac80211.c
598c598
< 	int i, rc = 0;
[... 249 lines cut ...]
< 				return;
< 			}
< 		} else {
< 			/* Defer calling mwl8k_start_stream so that the current
< 			 * skb can go out before the ADDBA request.  This
< 			 * prevents sequence number mismatch at the recipient
< 			 * as described above.
< 			 */
< 			if (mwl_fwcmd_ampdu_allowed(sta, tid)) {
< 				stream = mwl_fwcmd_add_stream(hw, sta, tid);
< 
< 				if (stream)
< 					start_ba_session = true;
< 			}
638d567
< 
660,667d588
< 
< 	/* Initiate the ampdu session here */
< 	if (start_ba_session) {
< 		spin_lock(&priv->stream_lock);
< 		if (mwl_fwcmd_start_stream(hw, stream))
< 			mwl_fwcmd_remove_stream(hw, stream);
< 		spin_unlock(&priv->stream_lock);
< 	}

yuhhaurlin, 2015-10-17:
Thanks a lot. I will check it.

yuhhaurlin, 2015-10-17:
BTW, can you share me what iperf number you got for UDP and TCP after the modificaion you done?

10.3.0.10 has enhancement of Tx. Can you add your modificaion based on 10.3.0.10, and let me know what throughput number you got before and after your modification? Thanks.

I will check your modification on AMPDU.

notnyt, 2015-10-18:
@TireMeat was getting about 20mbps with the original change you listed, vs 100+mbps with 10.3.0.3.  Please use 4 back ticks to surround code so it doesn't get processed by markdown.

TireMeat, 2015-10-19:
@notnyt  --  thanks for the formatting heads-up.

@yuhhaurlin  --  I did an overall patch from original 10.3.0.10 you posted Friday with the mods I'd done for 10.3.0.3.   Also ran iperf before & after (posted below after the patch) -- again, I'm using CC branch with default .config build.

From what I can tell, 10.3.0.10 is tighter than 10.3.0.3 (and 10.3.0.8 as I recall).  By "tighter" I'm mean that it seems to do fewer IRQs.   

I kept an eye out for CPU load like Gazoo mentioned, but haven't seen that at all.   At peak sustained transfer, CPU got to 3% maybe.

I found and got another v1 box yesterday, installed this driver with CC-branch on it, and replaced my D825 for long(er) term use/observation -- (our household is a heavy streaming and mobile-over-WiFi & VoIP environment, with lots of interference from ~12+ other APs within read-distance).

Looking forward to your feedback.

(Meanwhile, I'm going to take a look at the channel, power & cap part of the code ... am curious why initialization seems to go astray.)

```
diff -r original_CC_branch/fwcmd.c updated_CC_branch/fwcmd.c
2294,2297c2294
< 	tx_stats = &sta_info->tx_stats[tid];
< 
< 	return (sta_info->is_ampdu_allowed &&
< 		tx_stats->pkts > SYSADPT_AMPDU_PACKET_THRESHOLD);
---
> 	return (sta_info->is_ampdu_allowed);
diff -r original_CC_branch/fwdl.c updated_CC_branch/fwdl.c
28c28
< #define FW_CHECK_MSECS                  1
---
> #define FW_CHECK_MSECS                  3
diff -r original_CC_branch/mac80211.c updated_CC_branch/mac80211.c
580,583d579
< 	spin_lock_bh(&priv->stream_lock);
< 
< 	stream = mwl_fwcmd_lookup_stream(hw, addr, tid);
< 
589,638c585,595
< 		/* By the time we get here the hw queues may contain outgoing
< 		 * packets for this RA/TID that are not part of this BA
< 		 * session.  The hw will assign sequence numbers to these
< 		 * packets as they go out.  So if we query the hw for its next
< 		 * sequence number and use that for the SSN here, it may end up
< 		 * being wrong, which will lead to sequence number mismatch at
< 		 * the recipient.  To avoid this, we reset the sequence number
< 		 * to O for the first MPDU in this BA stream.
< 		 */
< 		*ssn = 0;
< 
< 		if (!stream) {
< 			/* This means that somebody outside this driver called
< 			 * ieee80211_start_tx_ba_session.  This is unexpected
< 			 * because we do our own rate control.  Just warn and
< 			 * move on.
< 			 */
< 			stream = mwl_fwcmd_add_stream(hw, sta, tid);
< 		}
< 
< 		if (!stream) {
< 			wiphy_warn(hw->wiphy, "no stream found\n");
< 			rc = -EBUSY;
< 			break;
< 		}
< 
< 		stream->state = AMPDU_STREAM_IN_PROGRESS;
< 
< 		/* Release the lock before we do the time consuming stuff */
< 		spin_unlock_bh(&priv->stream_lock);
< 
< 		/* Check if link is still valid */
< 		if (!sta_info->is_ampdu_allowed) {
< 			spin_lock_bh(&priv->stream_lock);
< 			mwl_fwcmd_remove_stream(hw, stream);
[... 307 lines cut ...]
[  5]   0.00-10.16  sec   125 MBytes   103 Mbits/sec                  receiver
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.2.227, port 3824
[  5] local 192.168.2.1 port 5201 connected to 192.168.2.227 port 61645
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  5]   0.00-1.00   sec   312 KBytes  2.55 Mbits/sec  3.385 ms  0/39 (0%)  
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec  3.385 ms  0/0 (nan%)  
[  5]   2.00-3.00   sec  48.0 KBytes   394 Kbits/sec  3.913 ms  0/6 (0%)  
[  5]   3.00-4.00   sec   128 KBytes  1.05 Mbits/sec  1.531 ms  0/16 (0%)  
[  5]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec  0.718 ms  0/16 (0%)  
[  5]   5.00-6.00   sec   128 KBytes  1.05 Mbits/sec  0.596 ms  0/16 (0%)  
[  5]   6.00-7.00   sec   128 KBytes  1.05 Mbits/sec  0.401 ms  0/16 (0%)  
[  5]   7.00-8.00   sec   128 KBytes  1.05 Mbits/sec  0.384 ms  0/16 (0%)  
[  5]   8.00-9.00   sec   120 KBytes   983 Kbits/sec  0.342 ms  0/15 (0%)  
[  5]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec  0.396 ms  0/16 (0%)  
[  5]  10.00-10.20  sec  24.0 KBytes   968 Kbits/sec  0.359 ms  0/3 (0%)  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  5]   0.00-10.20  sec  1.24 MBytes  1.02 Mbits/sec  0.359 ms  0/159 (0%)  
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
```

yuhhaurlin, 2015-10-19:
Can you let me know what client and configuration (setting on WRT1900AC or WRT1200AC) you use?

For WRT1200AC (2x2) and Macbook Air (2x2), 10.3.0.10 can get 4xx to 5xx Mbps for UDP/TCP (5 pairs) tests on 5g (channel 149 80 MHz).

BTW, can you cat "/sys/kernel/debug/ieee80211/phy0/mwlwifi/ampdu" and "/sys/kernel/debug/ieee80211/phy1/mwlwifi/ampdu" for me?

Thanks.

notnyt, 2015-10-19:
A 2008 unibody macbook generates the errors listed.  This information was provided in the original bug report.  I cannot even run a speed test properly the performance is so poor.  Other devices, my LG G4 for example, do not seem to have that issue.  It seems to be related to ampdu and the macbook.  As @TireMeat  suggested, the code may need some improvement in that area.

yuhhaurlin, 2015-10-19:
I think it could be IOT problem. I will check it later. Thanks.

johnnysl, 2015-10-19:
i have a V2 in a 90% apple household (Ipad2, mac mini, macbook air 2011, iphone 5, iphone 5s, HP folio 9470m). seeing the same errors as notnyt sees, but not the huge performance impact. After a certain amount of data transferred the errors start to appear and performance drops to about 10mbit per client (regardless of client type).

If i have to say, the HP (with a centrino advanced-n 6235) is causing more grief. especially when used at the edges of the reception range.

@Tiremeat: would it be possible to supply a compiled driver that is compatible the the 15.05 final release? i could then test the influence of the changes on my situation with the V2.

TireMeat, 2015-10-19:
A few weeks back I had my two iPhones (with Apple network stacks) disconnect and generally misbehave, and came across Adrian's post here (http://adrianchadd.blogspot.com/2012/10/power-save-cabq-multicast-frames-eapol.html) ... this is what made me think the packet sequencing logic in mwlwifi (specifically in the tx xmit and mac80211 ampdu action functions) may be (or come to be) at odds in certain scenarios (like power save and beam shape) with the logic within the (full-mac) firmware.  Remember, these mwlwifi drivers were based off of an early Marvell product, and it could well be that product may not have been as full-featured as this one (88W8864) -- meaning, there may be driver code that's redundant.  I've looked for tech docs, SDK (usually these contain code samples or at least calls), etc. for the 88W8864 but these aren't available without contacting product sales folks or something.

@johnnysl  --  will put one together.

TireMeat, 2015-10-19:
@johnnysl  ---  sorry mate, just saw you had a v2.  I don't have that version, just a couple of v1s.  The binary I have won't work for you.

TireMeat, 2015-10-19:
@yuhhaurlin  ---  you're seeing 4-5Mbps UDP?  (1Mbps I got does seem low indeed) 

Also, not sure if you were addressing Rob or me in wanting to see the configs -- just in case me, I don't have an ampdu in my ieee80211 path - my path looks like this: 

```
root@WRT1900:/sys/kernel/debug/ieee80211/phy1# pwd
/sys/kernel/debug/ieee80211/phy1
root@WRT1900:/sys/kernel/debug/ieee80211/phy1# ls
fragmentation_threshold  netdev:wlan1             statistics
ht40allow_map            power                    total_ps_buffered
hwflags                  queues                   user_power
keys                     rts_threshold            wep_iv
long_retry_limit         short_retry_limit
root@WRT1900:/sys/kernel/debug/ieee80211/phy1# 
```

Here's my config:
```
root@WRT1900:/etc/config# cat wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
	option htmode 'HT20'
	option country 'US'
	option channel '4'
	option txpower '30'
	option hwmode '11n'

config wifi-iface
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2+ccmp'
	option key '{edited out}'
	option ssid 'O2'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:03.0/0000:03:00.0'
	option htmode 'VHT80'
	option country 'US'
	option hwmode '11ac'
	option txpower '17'
	option channel '36'

config wifi-iface
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'O5'
	option encryption 'psk2+ccmp'
	option key '{edited out}'
```

johnnysl, 2015-10-19:
@TireMeat , shouldn't the V1 and V2 use the same wifi driver?

TireMeat, 2015-10-19:
@johnnysl  --  the code is the same, as is the wifi's firmware, but the resulting ko binaries are different -- v1 is mamba, the v2 is cobra.  At best the wifi won't work on initialize, in which case you'll be able to restore with a wired-in connection.  Likely, the kernel will panic so fast on boot that you won't be able to recover without restoring  the firmware using the TFTP method or something like that.  It's really not worth the risk.

I could cross-build it for v2, but have no way to test it -- you'd be right back to running unknown kernel code.

Maybe one of the guys has a v2 setup and can patch the 10.3.0.10 code, build & test.

yuhhaurlin, 2015-10-20:
@johnnysl : Did you try 10.3.0.10? We find 10.3.0.10 has IOT problem with few clients.

@TireMeat: 
1. Maybe you need to specify -b for UDP test.
2. The speed depends on band, antenna and bandwidth. If you use 3x3, the speed will be higher. But if your client is 1x1, the speed is lower (except that the client had IOT problem with 10.3.0.10).

BTW, does anyone still see lock up or crash problem on 10.3.0.10?

yuhhaurlin, 2015-10-20:
@TireMeat: debugfs is supported on 10.3.0.10.

notnyt, 2015-10-20:
@yuhhaurlin can't test for lockups since I can't get a number of my clients online.  Can you describe the IOT problem?

iperf in udp mode is default 1mbps, that said he was showing loss at this speed.  You can specify a rate with -b 100M for 100mbps.

[143 more replies]

closed 2015-11-24 by yuhhaurlin