kaloz/mwlwifi #35: hang in mwl_tx_xmit

notnyt, 2015-09-09:
logread, console messages, and iw event -f all displayed in log below.  The mwlwifi module randomly hung during transmit, then was interrupted by a timer and the stack dump was generated.

This is running 10.3.0.3

````
Wed Sep  9 00:34:25 2015 daemon.info hostapd: wlan0: STA f4:1b:a1:c1:21:10 WPA: group key handshake completed (RSN)
wlan0 (phy #0): mgmt TX status (cookie 33c5): acked
wlan0: del station 00:23:12:54:04:63
Wed Sep  9 00:34:33 2015 daemon.info hostapd: wlan0: STA 00:23:12:54:04:63 IEEE 802.11: deauthenticated due to local deauth request
Wed Sep  9 00:34:38 2015 daemon.info hostapd: wlan1: STA 00:23:6c:97:d7:cf WPA: group key handshake completed (RSN)
Wed Sep  9 00:34:39 2015 daemon.info hostapd: wlan1: STA f8:95:c7:f6:64:06 WPA: group key handshake completed (RSN)
Wed Sep  9 00:34:39 2015 daemon.info hostapd: wlan1: STA bc:f5:ac:e0:5b:f0 WPA: group key handshake completed (RSN)
Wed Sep  9 00:35:48 2015 daemon.info dnsmasq-dhcp[2408]: DHCPINFORM(br-lan) 192.168.1.229 f4:6d:04:4f:d0:90
Wed Sep  9 00:35:48 2015 daemon.info dnsmasq-dhcp[2408]: DHCPACK(br-lan) 192.168.1.229 f4:6d:04:4f:d0:90 CCDESK
Wed Sep  9 00:35:52 2015 daemon.info dnsmasq-dhcp[2408]: DHCPREQUEST(br-lan) 192.168.1.32 00:1c:10:c2:fb:4c
Wed Sep  9 00:35:52 2015 daemon.info dnsmasq-dhcp[2408]: DHCPACK(br-lan) 192.168.1.32 00:1c:10:c2:fb:4c PAP2TALARM
wlan0 (phy #0): probe client 00:14:bf:eb:94:ef (cookie 33c6): acked
Wed Sep  9 00:42:18 2015 daemon.info dnsmasq-dhcp[2408]: DHCPINFORM(br-lan) 192.168.1.229 f4:6d:04:4f:d0:90
Wed Sep  9 00:42:18 2015 daemon.info dnsmasq-dhcp[2408]: DHCPACK(br-lan) 192.168.1.229 f4:6d:04:4f:d0:90 CCDESK
Wed Sep  9 00:44:25 2015 daemon.info hostapd: wlan0: STA cc:95:d7:1a:39:f7 WPA: group key handshake completed (RSN)
Wed Sep  9 00:44:25 2015 daemon.info hostapd: wlan0: STA 00:14:bf:eb:94:ef WPA: group key handshake completed (RSN)
Wed Sep  9 00:44:25 2015 daemon.info hostapd: wlan0: STA 0c:2a:69:08:87:d8 WPA: group key handshake completed (RSN)
Wed Sep  9 00:44:25 2015 daemon.info hostapd: wlan0: STA f4:1b:a1:c1:21:10 WPA: group key handshake completed (RSN)
Wed Sep  9 00:44:38 2015 daemon.info hostapd: wlan1: STA 00:23:6c:97:d7:cf WPA: group key handshake completed (RSN)
Wed Sep  9 00:44:39 2015 daemon.info hostapd: wlan1: STA bc:f5:ac:e0:5b:f0 WPA: group key handshake completed (RSN)
Wed Sep  9 00:44:39 2015 daemon.info hostapd: wlan1: STA f8:95:c7:f6:64:06 WPA: group key handshake completed (RSN)
Wed Sep  9 00:48:28 2015 daemon.info dnsmasq-dhcp[2408]: DHCPINFORM(br-lan) 192.168.1.229 f4:6d:04:4f:d0:90
Wed Sep  9 00:48:28 2015 daemon.info dnsmasq-dhcp[2408]: DHCPACK(br-lan) 192.168.1.229 f4:6d:04:4f:d0:90 CCDESK
wlan0 (phy #0): probe client 00:14:bf:eb:94:ef (cookie 33c7): acked
Wed Sep  9 00:50:12 2015 daemon.notice openvpn(xx)[2264]: Inactivity timeout (--ping-restart), restarting
Wed Sep  9 00:50:12 2015 daemon.notice openvpn(xx)[2264]: SIGUSR1[soft,ping-restart] received, process restarting
Wed Sep  9 00:50:17 2015 daemon.notice openvpn(xx)[2264]: Attempting to establish TCP connection with [AF_INET]x.x.x.x:47100 [nonblock]
Wed Sep  9 00:50:27 2015 daemon.err openvpn(xx)[2264]: TCP: connect to [AF_INET]x.x.x.x:47100 failed, will try again in 5 seconds: Operation timed out
Wed Sep  9 00:50:42 2015 daemon.err openvpn(xx)[2264]: TCP: connect to [AF_INET]x.x.x.x:47100 failed, will try again in 5 seconds: Operation timed out
[1029962.640185] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=2213659 c=2213658 q=146)
[1029962.649863] Task dump for CPU 0:
[1029962.653289] kworker/u4:0    R running      0     6      2 0x00000002
[1029962.659955] Workqueue: phy0 ieee80211_iface_work [mac80211]
[1029962.665743] Backtrace:
[1029962.668407] [<c001add8>] (dump_backtrace) from [<c001b140>] (show_stack+0x18/0x1c)
[1029962.676188]  r6:c0428dc0 r5:c0428e80 r4:cf830400 r3:00000006
[1029962.682136] [<c001b128>] (show_stack) from [<c0044cf0>] (sched_show_task+0xc8/0xfc)
[1029962.690006] [<c0044c28>] (sched_show_task) from [<c0046f88>] (dump_cpu_task+0x40/0x44)
[1029962.698133]  r5:c0428e80 r4:00000000
[1029962.701958] [<c0046f48>] (dump_cpu_task) from [<c005f62c>] (rcu_dump_cpu_stacks+0x70/0xb4)
[1029962.710434]  r4:00000000 r3:00000001
[1029962.714249] [<c005f5bc>] (rcu_dump_cpu_stacks) from [<c00621c4>] (rcu_check_callbacks+0x24c/0x668)
[1029962.723421]  r9:c0428dc0 r8:c0428dc0 r7:0f9bd000 r6:00000092 r5:c041c214 r4:cfdd9214
[1029962.731473] [<c0061f78>] (rcu_check_callbacks) from [<c006426c>] (update_process_times+0x48/0x68)
[1029962.740562]  r10:c0456930 r9:c0456930 r8:c04568b0 r7:00000000 r6:00000000 r5:cf830400
[1029962.748673]  r4:cf84a000
[1029962.751435] [<c0064224>] (update_process_times) from [<c007239c>] (tick_sched_timer+0x224/0x268)
[1029962.760434]  r7:cf84b5e0 r6:cfdd9570 r5:0003a8c6 r4:c0194dbd
[1029962.766366] [<c0072178>] (tick_sched_timer) from [<c0065028>] (__run_hrtimer.isra.35+0xa8/0x144)
[1029962.775364]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
[1029962.783486]  r4:cfdd9570
[1029962.786234] [<c0064f80>] (__run_hrtimer.isra.35) from [<c00652d8>] (hrtimer_interrupt+0x138/0x2ac)
[1029962.795405]  r7:00000001 r6:cfdd9368 r5:0003a8c6 r4:c019418d
[1029962.801353] [<c00651a0>] (hrtimer_interrupt) from [<c02599f8>] (armada_370_xp_timer_interrupt+0x4c/0x54)
[1029962.811048]  r10:c046eae8 r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c042cfac
[1029962.819159]  r4:cfddcc40
[1029962.821918] [<c02599ac>] (armada_370_xp_timer_interrupt) from [<c005bcc0>] (handle_percpu_devid_irq+0x74/0x8c)
[1029962.832136]  r4:cf803c40 r3:c02599ac
[1029962.835951] [<c005bc4c>] (handle_percpu_devid_irq) from [<c00580a8>] (generic_handle_irq+0x34/0x44)
[1029962.845211]  r8:cf805000 r7:00000001 r6:cf84bd68 r5:00000000 r4:00000010 r3:c005bc4c
[1029962.853264] [<c0058074>] (generic_handle_irq) from [<c00583b0>] (__handle_domain_irq+0x90/0xa4)
[1029962.862176]  r4:c041c8f8 r3:0000005e
[1029962.865990] [<c0058320>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
[1029962.875423]  r8:cf84b5e0 r7:00000001 r6:c046eafc r5:20000153 r4:c0011da4 r3:cf84b5e0
[1029962.883471] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
[... 83 lines cut ...]
[1029963.488794] bd60:                   ce41ae8c cd92a5e9 00000000 0000230f 00000000 00000000
[1029963.497193] bd80: ce4194c0 ce418b60 ce41ae78 ce41ae8c 00000000 cf84bdf4 000000f7 cf84bdb0
[1029963.505592] bda0: bf3c3c48 bf3be660 80000153 ffffffff
[1029963.510850]  r10:00000000 r9:ce41ae8c r8:ce41ae78 r7:cf84bd9c r6:ffffffff r5:80000153
[1029963.518962]  r4:bf3be660
[1029963.521804] [<bf3be5d8>] (mwl_mac80211_ampdu_action [mwlwifi]) from [<bf358cfc>] (__ieee80211_start_rx_ba_session+0x23c/0x478 [mac80211])
[1029963.534373]  r10:00000000 r9:00000000 r8:ce418b60 r7:ccb41940 r6:00000040 r5:00000001
[1029963.542497]  r4:cd92a000
[1029963.545325] [<bf358ac0>] (__ieee80211_start_rx_ba_session [mac80211]) from [<bf358f8c>] (ieee80211_process_addba_request+0x54/0x5c [mac80211])
[1029963.558329]  r10:ce418f5c r9:cd9624c0 r8:cd9628d4 r7:ce418b60 r6:00000001 r5:00000000
[1029963.566452]  r4:00000001
[1029963.569279] [<bf358f38>] (ieee80211_process_addba_request [mac80211]) from [<bf35d0a0>] (ieee80211_iface_work+0x1d0/0x3ac [mac80211])
[1029963.581499]  r6:cd9628c4 r5:cded8088 r4:cd30ca00
[1029963.586419] [<bf35ced0>] (ieee80211_iface_work [mac80211]) from [<c0039728>] (process_one_work+0x1f4/0x33c)
[1029963.596378]  r10:00000007 r9:00000000 r8:cf84a020 r7:cf831700 r6:cf830800 r5:cd9628c4
[1029963.604500]  r4:cf82db80
[1029963.607250] [<c0039534>] (process_one_work) from [<c003a2a0>] (worker_thread+0x31c/0x50c)
[1029963.615640]  r10:cf84a000 r9:00000088 r8:cf84a020 r7:cf82db98 r6:cf830814 r5:cf830800
[1029963.623764]  r4:cf82db80
[1029963.626513] [<c0039f84>] (worker_thread) from [<c003dcc0>] (kthread+0xf4/0xf8)
[1029963.633945]  r10:00000000 r9:00000000 r8:00000000 r7:c0039f84 r6:cf82db80 r5:00000000
[1029963.642067]  r4:cf82c9c0
[1029963.644813] [<c003dbcc>] (kthread) from [<c0008f20>] (ret_from_fork+0x14/0x34)
[1029963.652243]  r7:00000000 r6:00000000 r5:c003dbcc r4:cf82c9c0
````

notnyt, 2015-09-13:
And another.

````
[350901.980171] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=830004 c=830003 q=84)
[350901.989499] Task dump for CPU 0:
[350901.992837] kworker/u4:2    R running      0   376      2 0x00000002
[350901.999414] Workqueue: phy0 ieee80211_iface_work [mac80211]
[350902.005116] Backtrace:
[350902.007695] [<c001add8>] (dump_backtrace) from [<c001b140>] (show_stack+0x18/0x1c)
[350902.015388]  r6:c0428dc0 r5:c0428e80 r4:cfac0c00 r3:00000006
[350902.021244] [<c001b128>] (show_stack) from [<c0044cf0>] (sched_show_task+0xc8/0xfc)
[350902.029026] [<c0044c28>] (sched_show_task) from [<c0046f88>] (dump_cpu_task+0x40/0x44)
[350902.037065]  r5:c0428e80 r4:00000000
[350902.040805] [<c0046f48>] (dump_cpu_task) from [<c005f62c>] (rcu_dump_cpu_stacks+0x70/0xb4)
[350902.049186]  r4:00000000 r3:00000001
[350902.052922] [<c005f5bc>] (rcu_dump_cpu_stacks) from [<c00621c4>] (rcu_check_callbacks+0x24c/0x668)
[350902.062008]  r9:c0428dc0 r8:c0428dc0 r7:0f9bd000 r6:00000054 r5:c041c214 r4:cfdd9214
[350902.069959] [<c0061f78>] (rcu_check_callbacks) from [<c006426c>] (update_process_times+0x48/0x68)
[350902.078957]  r10:c0456930 r9:c0456930 r8:c04568b0 r7:00000000 r6:00000000 r5:cfac0c00
[350902.086993]  r4:cf186000
[350902.089654] [<c0064224>] (update_process_times) from [<c007239c>] (tick_sched_timer+0x224/0x268)
[350902.098564]  r7:cf187758 r6:cfdd9570 r5:00013f25 r4:a322ae1a
[350902.104421] [<c0072178>] (tick_sched_timer) from [<c0065028>] (__run_hrtimer.isra.35+0xa8/0x144)
[350902.113332]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
[350902.121364]  r4:cfdd9570
[350902.124028] [<c0064f80>] (__run_hrtimer.isra.35) from [<c00652d8>] (hrtimer_interrupt+0x138/0x2ac)
[350902.133112]  r7:00000001 r6:cfdd9368 r5:00013f25 r4:a322a19a
[350902.138960] [<c00651a0>] (hrtimer_interrupt) from [<c02599f8>] (armada_370_xp_timer_interrupt+0x4c/0x54)
[350902.148568]  r10:c046eae8 r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c042cfac
[350902.156604]  r4:cfddcc40
[350902.159265] [<c02599ac>] (armada_370_xp_timer_interrupt) from [<c005bcc0>] (handle_percpu_devid_irq+0x74/0x8c)
[350902.169394]  r4:cf803c40 r3:c02599ac
[350902.173135] [<c005bc4c>] (handle_percpu_devid_irq) from [<c00580a8>] (generic_handle_irq+0x34/0x44)
[350902.182308]  r8:cf805000 r7:00000001 r6:cf187d30 r5:00000000 r4:00000010 r3:c005bc4c
[350902.190271] [<c0058074>] (generic_handle_irq) from [<c00583b0>] (__handle_domain_irq+0x90/0xa4)
[350902.199089]  r4:c041c8f8 r3:0000005e
[350902.202824] [<c0058320>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
[350902.212171]  r8:cf187758 r7:00000001 r6:c046eafc r5:20000153 r4:c0011da4 r3:cf187758
[350902.220120] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
[350902.228599] Exception stack(0xcf187758 to 0xcf1877a0)
[350902.233774] 7740:                                                       cfb7ae78 00000000
[350902.242090] 7760: 0000e41f 0000e41e 00000000 cf114f40 00000000 cdbc7dd8 00000004 00000001
[350902.250405] 7780: cda931c8 cf1877ac cf1877b0 cf1877a0 bf3c4760 c0011da4 20000153 ffffffff
[350902.258702]  r10:cda931c8 r9:00000001 r8:00000004 r7:cf18778c r6:ffffffff r5:20000153
[350902.266735]  r4:c0011da4
[350902.269432] [<c0011d70>] (_raw_spin_lock) from [<bf3c4760>] (mwl_tx_xmit+0x200/0x8f0 [mwlwifi])
[350902.278295] [<bf3c4560>] (mwl_tx_xmit [mwlwifi]) from [<bf3bf6d4>] (mwl_mac80211_tx+0x50/0x54 [mwlwifi])
[350902.287905]  r10:cf38fc78 r9:00000000 r8:00000002 r7:00000000 r6:cfb78b60 r5:cf114f40
[350902.295940]  r4:cf114f40
[350902.298661] [<bf3bf684>] (mwl_mac80211_tx [mwlwifi]) from [<bf36e4f0>] (ieee80211_nullfunc_get+0x11c0/0x1380 [mac80211])
[350902.309663]  r4:cf1878d0 r3:bf3bf684
[350902.313486] [<bf36e224>] (ieee80211_nullfunc_get [mac80211]) from [<bf36e640>] (ieee80211_nullfunc_get+0x1310/0x1380 [mac80211])
[350902.325184]  r10:00000000 r9:00000000 r8:cfb78b60 r7:000005fe r6:00000008 r5:cfb78b60
[350902.333222]  r4:cf1878d0
[350902.335967] [<bf36e53c>] (ieee80211_nullfunc_get [mac80211]) from [<bf36f2fc>] (ieee80211_get_buffered_bc+0x210/0x82c [mac80211])
[350902.347751]  r8:cfb78b60 r7:000005fe r6:cf38f4c2 r5:00000000 r4:cf114f40
[350902.354754] [<bf36f230>] (ieee80211_get_buffered_bc [mac80211]) from [<bf370664>] (ieee80211_xmit+0xf0/0xf8 [mac80211])
[350902.365668]  r9:00000000 r8:cea964ae r7:cdbc7800 r6:cfb78b60 r5:cf38f4c0 r4:cf114f40
[350902.373719] [<bf370574>] (ieee80211_xmit [mac80211]) from [<bf37151c>] (__ieee80211_subif_start_xmit+0x654/0x6a8 [mac80211])
[350902.385069]  r8:cf38f4c0 r7:00000000 r6:cf38f000 r5:00000000 r4:00804bbc
[350902.392069] [<bf370ec8>] (__ieee80211_subif_start_xmit [mac80211]) from [<bf371584>] (ieee80211_subif_start_xmit+0x14/0x1c [mac80211])
[350902.404290]  r10:c0422358 r9:cf114f40 r8:cd8a5c68 r7:cf9aa600 r6:cd8a5c00 r5:cf38f000
[350902.412326]  r4:cf114f40
[350902.415038] [<bf371570>] (ieee80211_subif_start_xmit [mac80211]) from [<c027e818>] (dev_hard_start_xmit+0x270/0x31c)
[350902.425709] [<c027e5a8>] (dev_hard_start_xmit) from [<c0297894>] (sch_direct_xmit+0xb4/0x218)
[350902.434362]  r10:cf38f000 r9:cf114f40 r8:cd8a5c68 r7:cf9aa600 r6:cd8a5c00 r5:cd8a5c00
[350902.442398]  r4:00000000
[350902.445059] [<c02977e0>] (sch_direct_xmit) from [<c027eb00>] (__dev_queue_xmit+0x23c/0x524)
[350902.453536]  r10:00000000 r9:cd8a5c68 r8:00000000 r7:cf9aa600 r6:cf38f000 r5:cd8a5c00
[350902.461572]  r4:cf114f40
[... 39 lines cut ...]
[... cut ...]
[350902.848427] [<c0039534>] (process_one_work) from [<c003a2a0>] (worker_thread+0x31c/0x50c)
[350902.856729]  r10:cf186000 r9:00000088 r8:cf186020 r7:cf907f98 r6:cf830814 r5:cf830800
[350902.864766]  r4:cf907f80
[350902.867426] [<c0039f84>] (worker_thread) from [<c003dcc0>] (kthread+0xf4/0xf8)
[350902.874769]  r10:00000000 r9:00000000 r8:00000000 r7:c0039f84 r6:cf907f80 r5:00000000
[350902.882806]  r4:cf1c42c0
[350902.885465] [<c003dbcc>] (kthread) from [<c0008f20>] (ret_from_fork+0x14/0x34)
[350902.892808]  r7:00000000 r6:00000000 r5:c003dbcc r4:cf1c42c0

````

notnyt, 2015-09-17:
And again.
````
[347607.585448] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=791271 c=791270 q=83)
[347607.594778] Task dump for CPU 0:
[347607.598116] kworker/u4:2    R running      0   376      2 0x00000002
[347607.604693] Workqueue: phy0 ieee80211_iface_work [mac80211]
[347607.610394] Backtrace:
[347607.612971] [<c001add8>] (dump_backtrace) from [<c001b140>] (show_stack+0x18/0x1c)
[347607.620663]  r6:c0428dc0 r5:c0428e80 r4:cfa58800 r3:00000006
[347607.626522] [<c001b128>] (show_stack) from [<c0044cf0>] (sched_show_task+0xc8/0xfc)
[347607.634305] [<c0044c28>] (sched_show_task) from [<c0046f88>] (dump_cpu_task+0x40/0x44)
[347607.642344]  r5:c0428e80 r4:00000000
[347607.646084] [<c0046f48>] (dump_cpu_task) from [<c005f62c>] (rcu_dump_cpu_stacks+0x70/0xb4)
[347607.654465]  r4:00000000 r3:00000001
[347607.658201] [<c005f5bc>] (rcu_dump_cpu_stacks) from [<c00621c4>] (rcu_check_callbacks+0x24c/0x668)
[347607.667287]  r9:c0428dc0 r8:c0428dc0 r7:0f9bd000 r6:00000053 r5:c041c214 r4:cfdd9214
[347607.675239] [<c0061f78>] (rcu_check_callbacks) from [<c006426c>] (update_process_times+0x48/0x68)
[347607.684238]  r10:c0456930 r9:c0456930 r8:c04568b0 r7:00000000 r6:00000000 r5:cfa58800
[347607.692274]  r4:cf28e000
[347607.694934] [<c0064224>] (update_process_times) from [<c007239c>] (tick_sched_timer+0x224/0x268)
[347607.703845]  r7:cf28f5d8 r6:cfdd9570 r5:00013c26 r4:4de552e7
[347607.709701] [<c0072178>] (tick_sched_timer) from [<c0065028>] (__run_hrtimer.isra.35+0xa8/0x144)
[347607.718611]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
[347607.726645]  r4:cfdd9570
[347607.729308] [<c0064f80>] (__run_hrtimer.isra.35) from [<c00652d8>] (hrtimer_interrupt+0x138/0x2ac)
[347607.738391]  r7:00000001 r6:cfdd9368 r5:00013c26 r4:4de546b7
[347607.744239] [<c00651a0>] (hrtimer_interrupt) from [<c02599f8>] (armada_370_xp_timer_interrupt+0x4c/0x54)
[347607.753846]  r10:c046eae8 r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c042cfac
[347607.761882]  r4:cfddcc40
[347607.764543] [<c02599ac>] (armada_370_xp_timer_interrupt) from [<c005bcc0>] (handle_percpu_devid_irq+0x74/0x8c)
[347607.774672]  r4:cf803c40 r3:c02599ac
[347607.778413] [<c005bc4c>] (handle_percpu_devid_irq) from [<c00580a8>] (generic_handle_irq+0x34/0x44)
[347607.787586]  r8:cf805000 r7:00000001 r6:cf28fd30 r5:00000000 r4:00000010 r3:c005bc4c
[347607.795547] [<c0058074>] (generic_handle_irq) from [<c00583b0>] (__handle_domain_irq+0x90/0xa4)
[347607.804365]  r4:c041c8f8 r3:0000005e
[347607.808100] [<c0058320>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
[347607.817446]  r8:cf28f5d8 r7:00000001 r6:c046eafc r5:20000153 r4:c0011da4 r3:cf28f5d8
[347607.825394] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
[347607.833872] Exception stack(0xcf28f5d8 to 0xcf28f620)
[347607.839048] f5c0:                                                       cea22e78 00000000
[347607.847364] f5e0: 0000c3d5 0000c3d4 00000060 cc869340 00000000 cd3efdd8 00000000 00000001
[347607.855681] f600: cf1655c8 cf28f62c cf28f630 cf28f620 bf3c4760 c0011da4 20000153 ffffffff
[347607.863976]  r10:cf1655c8 r9:00000001 r8:00000000 r7:cf28f60c r6:ffffffff r5:20000153
[347607.872010]  r4:c0011da4
[347607.874707] [<c0011d70>] (_raw_spin_lock) from [<bf3c4760>] (mwl_tx_xmit+0x200/0x8f0 [mwlwifi])
[347607.883570] [<bf3c4560>] (mwl_tx_xmit [mwlwifi]) from [<bf3bf6d4>] (mwl_mac80211_tx+0x50/0x54 [mwlwifi])
[347607.893181]  r10:ce80cc78 r9:00000000 r8:00000002 r7:00000000 r6:cea20b60 r5:cc869340
[347607.901215]  r4:cc869340
[347607.903936] [<bf3bf684>] (mwl_mac80211_tx [mwlwifi]) from [<bf36e4f0>] (ieee80211_nullfunc_get+0x11c0/0x1380 [mac80211])
[347607.914937]  r4:cf28f750 r3:bf3bf684
[347607.918757] [<bf36e224>] (ieee80211_nullfunc_get [mac80211]) from [<bf36e640>] (ieee80211_nullfunc_get+0x1310/0x1380 [mac80211])
[347607.930455]  r10:00000000 r9:00000000 r8:cea20b60 r7:000005fe r6:00000008 r5:cea20b60
[347607.938491]  r4:cf28f750
[347607.941236] [<bf36e53c>] (ieee80211_nullfunc_get [mac80211]) from [<bf36f2fc>] (ieee80211_get_buffered_bc+0x210/0x82c [mac80211])
[347607.953020]  r8:cea20b60 r7:000005fe r6:ce80c4c2 r5:00000000 r4:cc869340
[347607.960020] [<bf36f230>] (ieee80211_get_buffered_bc [mac80211]) from [<bf370664>] (ieee80211_xmit+0xf0/0xf8 [mac80211])
[347607.970935]  r9:00000000 r8:cc45b06e r7:cd3ef800 r6:cea20b60 r5:ce80c4c0 r4:cc869340
[347607.978984] [<bf370574>] (ieee80211_xmit [mac80211]) from [<bf37151c>] (__ieee80211_subif_start_xmit+0x654/0x6a8 [mac80211])
[347607.990334]  r8:ce80c4c0 r7:00000000 r6:ce80c000 r5:00000000 r4:0097d021
[347607.997332] [<bf370ec8>] (__ieee80211_subif_start_xmit [mac80211]) from [<bf371584>] (ieee80211_subif_start_xmit+0x14/0x1c [mac80211])
[347608.009553]  r10:c0422358 r9:cc869340 r8:cd8f3a68 r7:cf9e5600 r6:cd8f3a00 r5:ce80c000
[347608.017590]  r4:cc869340
[347608.020300] [<bf371570>] (ieee80211_subif_start_xmit [mac80211]) from [<c027e818>] (dev_hard_start_xmit+0x270/0x31c)
[347608.030972] [<c027e5a8>] (dev_hard_start_xmit) from [<c0297894>] (sch_direct_xmit+0xb4/0x218)
[347608.039624]  r10:ce80c000 r9:cc869340 r8:cd8f3a68 r7:cf9e5600 r6:cd8f3a00 r5:cd8f3a00
[347608.047660]  r4:00000000
[347608.050321] [<c02977e0>] (sch_direct_xmit) from [<c027eb00>] (__dev_queue_xmit+0x23c/0x524)
[347608.058796]  r10:00000000 r9:cd8f3a68 r8:00000000 r7:cf9e5600 r6:ce80c000 r5:cd8f3a00
[347608.066833]  r4:cc869340
[... cut ...]
[347608.558182] [<c0039534>] (process_one_work) from [<c003a2a0>] (worker_thread+0x31c/0x50c)
[347608.566484]  r10:cf28e000 r9:00000088 r8:cf28e020 r7:cf9dcf98 r6:cf830814 r5:cf830800
[347608.574509]  r4:cf9dcf80
[347608.577182] [<c0039f84>] (worker_thread) from [<c003dcc0>] (kthread+0xf4/0xf8)
[347608.584519]  r10:00000000 r9:00000000 r8:00000000 r7:c0039f84 r6:cf9dcf80 r5:00000000
[347608.592552]  r4:cf1e3440
[347608.595212] [<c003dbcc>] (kthread) from [<c0008f20>] (ret_from_fork+0x14/0x34)
[347608.602556]  r7:00000000 r6:00000000 r5:c003dbcc r4:cf1e3440

````

notnyt, 2015-09-21:
And another...

````
[169630.912446] INFO: rcu_bh detected stalls on CPUs/tasks: { 0} (detected by 1, t=6002 jiffies, g=2564, c=2563, q=1)
[169630.922900] Task dump for CPU 0:
[169630.926232] kworker/u4:3    R running      0   408      2 0x00000002
[169630.932820] Workqueue: phy0 ieee80211_iface_work [mac80211]
[169630.938512] Backtrace:
[169630.941161] [<cf127ff4>] (0xcf127ff4) from [<00000088>] (0x88)
[169630.947133] [<cf26ff8c>] (0xcf26ff8c) from [<00000000>] (  (null))
[169630.953423] Backtrace aborted due to bad frame pointer <bf35ced0>
Mon Sep 21 04:49:22 2015 kern.err kernel: [169630.912446] INFO: rcu_bh detected stalls on CPUs/tasks: { 0} (detected by 1, t=6002 jiffies, g=2564, c=2563, q=1)
Mon Sep 21 04:49:22 2015 kern.info kernel: [169630.922900] Task dump for CPU 0:
Mon Sep 21 04:49:22 2015 kern.info kernel: [169630.926232] kworker/u4:3    R running      0   408      2 0x00000002
Mon Sep 21 04:49:22 2015 kern.info kernel: [169630.932820] Workqueue: phy0 ieee80211_iface_work [mac80211]
Mon Sep 21 04:49:22 2015 kern.warn kernel: [169630.938512] Backtrace:
Mon Sep 21 04:49:22 2015 kern.warn kernel: [169630.941161] [<cf127ff4>] (0xcf127ff4) from [<00000088>] (0x88)
Mon Sep 21 04:49:22 2015 kern.warn kernel: [169630.947133] [<cf26ff8c>] (0xcf26ff8c) from [<00000000>] (  (null))
Mon Sep 21 04:49:22 2015 kern.warn kernel: [169630.953423] Backtrace aborted due to bad frame pointer <bf35ced0>
[169632.702413] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=396538 c=396537 q=130)
[169632.711828] Task dump for CPU 0:
[169632.715166] kworker/u4:3    R running      0   408      2 0x00000002
[169632.721745] Workqueue: phy0 ieee80211_iface_work [mac80211]
[169632.727448] Backtrace:
[169632.730026] [<c001add8>] (dump_backtrace) from [<c001b140>] (show_stack+0x18/0x1c)
[169632.737718]  r6:c0428dc0 r5:c0428e80 r4:cfaca000 r3:00000006
[169632.743575] [<c001b128>] (show_stack) from [<c0044cf0>] (sched_show_task+0xc8/0xfc)
[169632.751359] [<c0044c28>] (sched_show_task) from [<c0046f88>] (dump_cpu_task+0x40/0x44)
[169632.759397]  r5:c0428e80 r4:00000000
[169632.763137] [<c0046f48>] (dump_cpu_task) from [<c005f62c>] (rcu_dump_cpu_stacks+0x70/0xb4)
[169632.771518]  r4:00000000 r3:00000001
[169632.775254] [<c005f5bc>] (rcu_dump_cpu_stacks) from [<c00621c4>] (rcu_check_callbacks+0x24c/0x668)
[169632.784339]  r9:c0428dc0 r8:c0428dc0 r7:0f9bd000 r6:00000082 r5:c041c214 r4:cfdd9214
[169632.792290] [<c0061f78>] (rcu_check_callbacks) from [<c006426c>] (update_process_times+0x48/0x68)
[169632.801289]  r10:c0456930 r9:c0456930 r8:c04568b0 r7:00000000 r6:00000000 r5:cfaca000
[169632.809325]  r4:cf128000
[169632.811986] [<c0064224>] (update_process_times) from [<c007239c>] (tick_sched_timer+0x224/0x268)
[169632.820897]  r7:cf1295e8 r6:cfdd9570 r5:00009a47 r4:d06ab3ae
[169632.826753] [<c0072178>] (tick_sched_timer) from [<c0065028>] (__run_hrtimer.isra.35+0xa8/0x144)
[169632.835664]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
[169632.843698]  r4:cfdd9570
[169632.846362] [<c0064f80>] (__run_hrtimer.isra.35) from [<c00652d8>] (hrtimer_interrupt+0x138/0x2ac)
[169632.855447]  r7:00000001 r6:cfdd9368 r5:00009a47 r4:d06aa706
[169632.861294] [<c00651a0>] (hrtimer_interrupt) from [<c02599f8>] (armada_370_xp_timer_interrupt+0x4c/0x54)
[169632.870902]  r10:c046eae8 r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c042cfac
[169632.878938]  r4:cfddcc40
[169632.881598] [<c02599ac>] (armada_370_xp_timer_interrupt) from [<c005bcc0>] (handle_percpu_devid_irq+0x74/0x8c)
[169632.891728]  r4:cf803c40 r3:c02599ac
[169632.895469] [<c005bc4c>] (handle_percpu_devid_irq) from [<c00580a8>] (generic_handle_irq+0x34/0x44)
[169632.904641]  r8:cf805000 r7:00000001 r6:cf129d40 r5:00000000 r4:00000010 r3:c005bc4c
[169632.912602] [<c0058074>] (generic_handle_irq) from [<c00583b0>] (__handle_domain_irq+0x90/0xa4)
[169632.921421]  r4:c041c8f8 r3:0000005e
[169632.925156] [<c0058320>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
[169632.934502]  r8:cf1295e8 r7:00000001 r6:c046eafc r5:20000153 r4:c0011da4 r3:cf1295e8
[169632.942450] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
[169632.950930] Exception stack(0xcf1295e8 to 0xcf129630)
[169632.956109] 95e0:                   ce9f6e78 00000000 00003a0d 00003a0c 00000060 ce3fb640
[169632.964426] 9600: 00000000 cd95d5d8 00000000 00000001 cfb0f1c8 cf12963c cf129640 cf129630
[169632.972736] 9620: bf3c4760 c0011da4 20000153 ffffffff
[169632.977897]  r10:cfb0f1c8 r9:00000001 r8:00000000 r7:cf12961c r6:ffffffff r5:20000153
[169632.985932]  r4:c0011da4
[169632.988629] [<c0011d70>] (_raw_spin_lock) from [<bf3c4760>] (mwl_tx_xmit+0x200/0x8f0 [mwlwifi])
[169632.997493] [<bf3c4560>] (mwl_tx_xmit [mwlwifi]) from [<bf3bf6d4>] (mwl_mac80211_tx+0x50/0x54 [mwlwifi])
[169633.007103]  r10:cf3a8c78 r9:00000000 r8:00000002 r7:00000000 r6:ce9f4b60 r5:ce3fb640
[169633.015139]  r4:ce3fb640
[169633.017860] [<bf3bf684>] (mwl_mac80211_tx [mwlwifi]) from [<bf36e4f0>] (ieee80211_nullfunc_get+0x11c0/0x1380 [mac80211])
[169633.028861]  r4:cf129760 r3:bf3bf684
[169633.032684] [<bf36e224>] (ieee80211_nullfunc_get [mac80211]) from [<bf36e640>] (ieee80211_nullfunc_get+0x1310/0x1380 [mac80211])
[169633.044384]  r10:00000000 r9:00000000 r8:ce9f4b60 r7:000005fe r6:00000008 r5:ce9f4b60
[169633.052410]  r4:cf129760
[... 69 lines cut ...]
[169633.545010]  r10:00000000 r9:00000006 r8:00000006 r7:cf129d74 r6:ffffffff r5:20000153
[... cut ...]
[169633.659867] [<c0039534>] (process_one_work) from [<c003a2a0>] (worker_thread+0x31c/0x50c)
[169633.668169]  r10:cf128000 r9:00000088 r8:cf128020 r7:cf26ff98 r6:cf830814 r5:cf830800
[169633.676206]  r4:cf26ff80
[169633.678867] [<c0039f84>] (worker_thread) from [<c003dcc0>] (kthread+0xf4/0xf8)
[169633.686211]  r10:00000000 r9:00000000 r8:00000000 r7:c0039f84 r6:cf26ff80 r5:00000000
[169633.694247]  r4:cf291bc0
[169633.696907] [<c003dbcc>] (kthread) from [<c0008f20>] (ret_from_fork+0x14/0x34)
[169633.704250]  r7:00000000 r6:00000000 r5:c003dbcc r4:cf291bc0

````

notnyt, 2015-10-12:
and again
````
[1809766.981089] INFO: rcu_sched self-detected stall on CPU { 0}  (t=6000 jiffies g=3802045 c=3802044 q=108)
[1809766.990767] Task dump for CPU 0:
[1809766.994193] kworker/u4:1    R running      0    15      2 0x00000002
[1809767.000858] Workqueue: phy0 ieee80211_iface_work [mac80211]
[1809767.006645] Backtrace:
[1809767.009311] [<c001add8>] (dump_backtrace) from [<c001b140>] (show_stack+0x18/0x1c)
[1809767.017082]  r6:c0428dc0 r5:c0428e80 r4:cf88d000 r3:00000006
[1809767.023036] [<c001b128>] (show_stack) from [<c0044cf0>] (sched_show_task+0xc8/0xfc)
[1809767.030905] [<c0044c28>] (sched_show_task) from [<c0046f88>] (dump_cpu_task+0x40/0x44)
[1809767.039031]  r5:c0428e80 r4:00000000
[1809767.042860] [<c0046f48>] (dump_cpu_task) from [<c005f62c>] (rcu_dump_cpu_stacks+0x70/0xb4)
[1809767.051338]  r4:00000000 r3:00000001
[1809767.055153] [<c005f5bc>] (rcu_dump_cpu_stacks) from [<c00621c4>] (rcu_check_callbacks+0x24c/0x668)
[1809767.064325]  r9:c0428dc0 r8:c0428dc0 r7:0f9bd000 r6:0000006c r5:c041c214 r4:cfdd9214
[1809767.072377] [<c0061f78>] (rcu_check_callbacks) from [<c006426c>] (update_process_times+0x48/0x68)
[1809767.081464]  r10:c0456930 r9:c0456930 r8:c04568b0 r7:00000000 r6:00000000 r5:cf88d000
[1809767.089577]  r4:cf8a2000
[1809767.092337] [<c0064224>] (update_process_times) from [<c007239c>] (tick_sched_timer+0x224/0x268)
[1809767.101336]  r7:cf8a3610 r6:cfdd9570 r5:00066df8 r4:be2a0911
[1809767.107269] [<c0072178>] (tick_sched_timer) from [<c0065028>] (__run_hrtimer.isra.35+0xa8/0x144)
[1809767.116267]  r10:cfdd9368 r9:00000000 r8:cfdd9378 r7:00000001 r6:cfdd9368 r5:cfdd93a0
[1809767.124390]  r4:cfdd9570
[1809767.127138] [<c0064f80>] (__run_hrtimer.isra.35) from [<c00652d8>] (hrtimer_interrupt+0x138/0x2ac)
[1809767.136309]  r7:00000001 r6:cfdd9368 r5:00066df8 r4:be29fd31
[1809767.142256] [<c00651a0>] (hrtimer_interrupt) from [<c02599f8>] (armada_370_xp_timer_interrupt+0x4c/0x54)
[1809767.151952]  r10:c046eae8 r9:000003ff r8:cfddcc40 r7:00000010 r6:cf80aa00 r5:c042cfac
[1809767.160064]  r4:cfddcc40
[1809767.162822] [<c02599ac>] (armada_370_xp_timer_interrupt) from [<c005bcc0>] (handle_percpu_devid_irq+0x74/0x8c)
[1809767.173039]  r4:cf803c40 r3:c02599ac
[1809767.176856] [<c005bc4c>] (handle_percpu_devid_irq) from [<c00580a8>] (generic_handle_irq+0x34/0x44)
[1809767.186115]  r8:cf805000 r7:00000001 r6:cf8a3d68 r5:00000000 r4:00000010 r3:c005bc4c
[1809767.194179] [<c0058074>] (generic_handle_irq) from [<c00583b0>] (__handle_domain_irq+0x90/0xa4)
[1809767.203091]  r4:c041c8f8 r3:0000005e
[1809767.206905] [<c0058320>] (__handle_domain_irq) from [<c0008698>] (armada_370_xp_handle_irq+0x64/0xe0)
[1809767.216339]  r8:cf8a3610 r7:00000001 r6:c046eafc r5:20000153 r4:c0011da4 r3:cf8a3610
[1809767.224389] [<c0008634>] (armada_370_xp_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
[1809767.232955] Exception stack(0xcf8a3610 to 0xcf8a3658)
[1809767.238211] 3600:                                     ceaa2e78 00000000 00001e05 00001e04
[1809767.246610] 3620: 00000060 cd633e80 00000000 cd644dd8 00000000 00000001 cca5afc8 cf8a3664
[1809767.255011] 3640: cf8a3668 cf8a3658 bf3c4760 c0011da4 20000153 ffffffff
[1809767.261836]  r10:cca5afc8 r9:00000001 r8:00000000 r7:cf8a3644 r6:ffffffff r5:20000153
[1809767.269949]  r4:c0011da4
[1809767.272746] [<c0011d70>] (_raw_spin_lock) from [<bf3c4760>] (mwl_tx_xmit+0x200/0x8f0 [mwlwifi])
[1809767.281697] [<bf3c4560>] (mwl_tx_xmit [mwlwifi]) from [<bf3bf6d4>] (mwl_mac80211_tx+0x50/0x54 [mwlwifi])
[1809767.291394]  r10:cf230c78 r9:00000000 r8:00000002 r7:00000000 r6:ceaa0b60 r5:cd633e80
[1809767.299506]  r4:cd633e80
[1809767.302325] [<bf3bf684>] (mwl_mac80211_tx [mwlwifi]) from [<bf36e4f0>] (ieee80211_nullfunc_get+0x11c0/0x1380 [mac80211])
[1809767.313415]  r4:cf8a3788 r3:bf3bf684
[1809767.317312] [<bf36e224>] (ieee80211_nullfunc_get [mac80211]) from [<bf36e640>] (ieee80211_nullfunc_get+0x1310/0x1380 [mac80211])
[1809767.329098]  r10:00000000 r9:00000000 r8:ceaa0b60 r7:000005fe r6:00000008 r5:ceaa0b60
[1809767.337222]  r4:cf8a3788
[1809767.340054] [<bf36e53c>] (ieee80211_nullfunc_get [mac80211]) from [<bf36f2fc>] (ieee80211_get_buffered_bc+0x210/0x82c [mac80211])
[1809767.351925]  r8:ceaa0b60 r7:000005fe r6:cf2304c2 r5:00000000 r4:cd633e80
[1809767.359000] [<bf36f230>] (ieee80211_get_buffered_bc [mac80211]) from [<bf370664>] (ieee80211_xmit+0xf0/0xf8 [mac80211])
[1809767.370001]  r9:00000000 r8:cc89c06e r7:cd644800 r6:ceaa0b60 r5:cf2304c0 r4:cd633e80
[1809767.378135] [<bf370574>] (ieee80211_xmit [mac80211]) from [<bf37151c>] (__ieee80211_subif_start_xmit+0x654/0x6a8 [mac80211])
[1809767.389574]  r8:cf2304c0 r7:00000000 r6:cf230000 r5:00000000 r4:00f89054
[1809767.396662] [<bf370ec8>] (__ieee80211_subif_start_xmit [mac80211]) from [<bf371584>] (ieee80211_subif_start_xmit+0x14/0x1c [mac80211])
[1809767.408971]  r10:c0422358 r9:cd633e80 r8:cd9d1468 r7:ce9a2e00 r6:cd9d1400 r5:cf230000
[1809767.417095]  r4:cd633e80
[1809767.419893] [<bf371570>] (ieee80211_subif_start_xmit [mac80211]) from [<c027e818>] (dev_hard_start_xmit+0x270/0x31c)
[1809767.430652] [<c027e5a8>] (dev_hard_start_xmit) from [<c0297894>] (sch_direct_xmit+0xb4/0x218)
[1809767.439394]  r10:cf230000 r9:cd633e80 r8:cd9d1468 r7:ce9a2e00 r6:cd9d1400 r5:cd9d1400
[1809767.447517]  r4:00000000
[1809767.450266] [<c02977e0>] (sch_direct_xmit) from [<c027eb00>] (__dev_queue_xmit+0x23c/0x524)
[1809767.458829]  r10:00000000 r9:cd9d1468 r8:00000000 r7:ce9a2e00 r6:cf230000 r5:cd9d1400
[... cut ...]
[1809767.933948] [<c0039534>] (process_one_work) from [<c003a2a0>] (worker_thread+0x31c/0x50c)
[1809767.942336]  r10:cf8a2000 r9:00000088 r8:cf8a2020 r7:cf866718 r6:cf830814 r5:cf830800
[1809767.950449]  r4:cf866700
[1809767.953210] [<c0039f84>] (worker_thread) from [<c003dcc0>] (kthread+0xf4/0xf8)
[1809767.960633]  r10:00000000 r9:00000000 r8:00000000 r7:c0039f84 r6:cf866700 r5:00000000
[1809767.968754]  r4:cf890440
[1809767.971516] [<c003dbcc>] (kthread) from [<c0008f20>] (ret_from_fork+0x14/0x34)
[1809767.978940]  r7:00000000 r6:00000000 r5:c003dbcc r4:cf890440
````

yuhhaurlin, 2015-10-21:
How about 10.3.0.10?

notnyt, 2015-10-21:
can't test until issue #34 is resolved

gufus, 2015-10-22:
@yuhhaurlin

Using username "root".
DD-WRT v3.0-r27950M std (c) 2015 NewMedia-NET GmbH
Release: 10/16/15
Authenticating with public key "rsa-key-20120810"
==========================================================

     ___  ___     _      _____  ______       ____  ___
    / _ \/ _ \___| | /| / / _ \/_  __/ _  __|_  / / _ \
   / // / // /___/ |/ |/ / , _/ / /   | |/ //_ <_/ // /
  /____/____/    |__/|__/_/|_| /_/    |___/____(_)___/

                       DD-WRT v3.0
                   http://www.dd-wrt.com

==========================================================

BusyBox v1.23.2 (2015-10-16 18:46:46 CEST) built-in shell (ash)

root@AC-DD-WRT:~# uptime
 06:53:07 up 1 day, 23:21,  load average: 0.01, 0.02, 0.04
root@AC-DD-WRT:~#

This verson of dd-wrt useswifi driver 10.3.0.10

FYI
wifi driver 10.3.0.3

yuhhaurlin, 2015-10-23:
Thanks for your information.

gufus, 2015-10-23:
@yuhhaurlin

DD-WRT v3.0-r27950M std (c) 2015 NewMedia-NET GmbH
Release: 10/16/15

1900AC V1 

ASUS USB-AC53 802.11ac Wireless USB Adapter (192.168.1.194) 

This verson of dd-wrt uses wifi driver 10.3.0.10

[code]
Oct 23 11:54:39 AC-DD-WRT daemon.info dnsmasq-dhcp[26658]: DHCPREQUEST(br0) 192.168.1.194 74:d0:2b:d0:1c:94 
Oct 23 11:54:39 AC-DD-WRT daemon.info dnsmasq-dhcp[26658]: DHCPACK(br0) 192.168.1.194 74:d0:2b:d0:1c:94 KEVIN-NB
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.217075] ieee80211 phy1: check ba result error 1
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.222114] ieee80211 phy1: ampdu start error code: -22
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.258206] ieee80211 phy1: check ba result error 1
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.263261] ieee80211 phy1: ampdu start error code: -22
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.297219] ieee80211 phy1: check ba result error 1
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.302328] ieee80211 phy1: ampdu start error code: -22
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.347087] ieee80211 phy1: check ba result error 1
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.352125] ieee80211 phy1: ampdu start error code: -22
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.387076] ieee80211 phy1: check ba result error 1
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.392111] ieee80211 phy1: ampdu start error code: -22
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.427094] ieee80211 phy1: check ba result error 1
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.432125] ieee80211 phy1: ampdu start error code: -22
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.467095] ieee80211 phy1: check ba result error 1
Oct 23 17:54:40 AC-DD-WRT kern.err kernel: [274975.472125] ieee80211 phy1: ampdu start error code: -22
[/code]

notnyt, 2015-10-23:
@gufus @yuhhaurlin that's issue #34

yuhhaurlin, 2015-10-23:
We will check it. Thanks.

yuhhaurlin, 2015-11-12:
Does anyone still encounter this problem with 10.3.0.14?

closed 2015-11-24 by yuhhaurlin