kaloz/mwlwifi #37: Wifi (5Ghz) won't work with channels above 48

Razerwire, 2015-10-08:
Hi i've recently acquired a WRT1900AC-V2 and flashed OpenWRT 15.05 on it. 

Problem is the 5Ghz radio refuses tot do it's job when not using channel 36, 40, 44 or 48. No matter what country code (regulatory domain) i set or what channel width i use, only channels 36, 40 , 44, 48 can be used. 

The Signal is reported being at 0 dBm in the WebIf, no wireless signal, no SSID to be from clients, nothing at all.

I don't know if it is related to DFS but the channels that won't work all require DFS, at least in some regions. 

Both 15.05 RC3 and 15.05 final are stuck by this issue. 

I'm not the only one reporting this problem see:
​https://forum.openwrt.org/viewtopic.php?pid=294739#p294739

Here's part of the syslog, it might be of help fixing the issue:
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): wlan0: interface state HT_SCAN->DFS
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): wlan0: DFS-CAC-START freq=5660 chan=132 sec_chan=1, width=1, seg0=138, seg1=0, cac_time=60s
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): DFS start_dfs_cac() failed, -1
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): Interface initialization failed
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): wlan0: interface state DFS->DISABLED
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): wlan0: AP-DISABLED
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): wlan0: interface state DISABLED->DISABLED
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): wlan0: AP-DISABLED
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): hostapd_free_hapd_data: Interface wlan0 wasn't started
Fri Jul 3 19:01:10 2015 daemon.notice netifd: radio0 (1096): nl80211: deinit ifname=wlan0 disabled_11b_rates=0

If also reported the problem here:
https://dev.openwrt.org/ticket/20690

If further info is needed, let me know.

Razerwire, 2015-10-10:
Okay so this problem is definitely tied to DFS. 

I set 'China' as the routers regulatory domain and again channels 36, 40, 44 and 48 were working other channels weren't. But, according to output from the "iw list" command,  the upper 5 Ghz channels (149, 153, 157, 161) don't require the use of DFS in China. 

I then managed to get 5Ghz wifi working by using channel 149.  

So my guess is that DFS is causing the issue.

kristrev, 2015-10-17:
This issue is present also with the latest driver, which makes 5GHz kinda useless in Europe if you want to stay on the right side of the law :)

TireMeat, 2015-10-17:
@kristrev ... i've seen this too.  Channel 36 & 149 are what I'm using while I'm looking at the performance problems ... quick question for you, what happens in your case if you issue "wifi" from the command line a few times (waiting like 10-15 seconds between each command issue)?   Does is start to work with, say 149?

raven-au, 2015-10-18:
LOL, the WRT1900AC-v1 has never allowed these upper channels on the vendor firmware either and yet on every other router I have, including Linksys routers, they are allowed.
I wonder how many years it will take for Linksys support to realize they were wrong in dismissing the problem I logged with this as caused by regulatory constraints.

Same thing happening here perhaps?

kristrev, 2015-10-18:
@TireMeat I will do some more testing when I have the time and the router is "available". However, I did quite a lot of wifi restarts after I installed a new image with latest driver. Every time I changed something, so the interval must have been at least 10.15 seconds between each restart. I still saw the problem. I did not try changing HT or power, will do that next time.

One thing I did notice was that after a couple of days of running stable with latest driver, 5GHz started acting up and clients were not able to connect. The error message said something about keys not being added to hardware (just got a quick glance at message before router had to be rebooted).

johnnysl, 2015-10-24:
Seeing the same issue, only by selecting china as a region i can sometimes select higher frequenties.
Likely because they are not dfs restricted in China

yuhhaurlin, 2015-10-28:
DFS will be supported later.

johnnysl, 2015-10-30:
@yuhhaurlin: do you have any roadmap you can share on features planned to be added?
that would save us bugging you every day on things we want/need in the driver.

e.g. i could really use the extra channels on the 5ghz range, so i'd love to hear when you think this would be added to the driver.

notnyt, 2015-10-30:
just add noscan 1 to your config. 
````
config wifi-device  radio1
        option type     mac80211
        option channel  157
        option hwmode   11n
        option path     'soc/soc:pcie-controller/pci0000:00/0000:00:03.0/0000:03:00.0'
        option htmode   VHT80
        option txpower  30
        option country 'US'
        option noscan '1'

````

johnnysl, 2015-10-30:
@notnyt: thanks for the tip! You are teaching me new stuff every day :-)

yuhhaurlin, 2015-10-31:
Set country code to US, you can still use 149  (80Mhz).

johnnysl, 2015-10-31:
@yuhhaurlin 
I don't get it why i couldn't get 157 to work, wi5j a US code, but 149 indeed works. Thanks.

notnyt, 2015-10-31:
157 works for me

yuhhaurlin, 2015-10-31:
Can you post hostapd's configuration?

yuhhaurlin, 2015-10-31:
BTW, you can check AP setting via "iw dev".

johnnysl, 2015-10-31:
@yuhhaurlin 
Nah nevermind, no need to debug this. 149 works now, so i'm happy :+1: 
 I'll patiently await a proper solution to be able to use the 100+ range as well :-)

[offtopic] but if you guys have an idea for this openwrt puzzle, please feel free to comment on the forum. https://forum.openwrt.org/viewtopic.php?pid=298019#p298019 [/offtopic]

notnyt, 2015-10-31:
no idea, I don't experience that issue

Razerwire, 2015-10-31:
Great to read that DFS support will be added in the future. I can't wait for the day that I’ll be able to fully utilise the 5Ghz band.

srdjanrosic, 2015-11-28:
149 might be legal at no more than 25mW in Europe soon, depending on how fast regulation moves.
The question only becomes, whether it's quicker to pass laws to make 149 use officially legal or implement DFS in driver/firmware.

kristrev, 2015-12-08:
@yuhhaurlin Any update on DFS support?

davidc502, 2015-12-27:
DESIGNATED DRIVER (Bleeding Edge, r48005)  4.1.13

channel 52-144 do not work. However, they were working in RC3.

Appears channels 36-44 are working.

yuhhaurlin, 2016-01-06:
DFS is added. Please check https://github.com/kaloz/mwlwifi/commit/99d3879cc72f2a25d44fb4bee96fd84eca028b04.

yuhhaurlin, 2016-01-06:
This issue is closed.

closed 2016-01-06 by yuhhaurlin

kristrev, 2016-01-06:
@yuhhaurlin DFS works great, thanks alot. I have hit one issue though, it takes a very long time for the interface to come up when I configure a DFS channel the first time, restart wifi or reboot router. When looking at the logs, it seems that it takes roughly a minute between DFS-CAC-START and DFS-CAC-COMPLETED. After this, the interface works fine and changing channel is more or less instant (also between DFS and non-DFS).

I guess this issue can considered an optimization, but I thought I should report my observation.

johnnysl, 2016-01-06:
@kristrev, that is by design. the driver needs to listenen for radar activity for 1 minute before it can activate.

kristrev, 2016-01-06:
@johnnysl Ok, thanks for letting me know. I now see this work other drivers too, as well as the cac_time-value. Not my proudest comment ...

yuhhaurlin, 2016-01-06:
CAC timer is 60 seconds. Once if CAC is done and no radar detected, the channel can be used directly.

gufus, 2016-01-06:
----

Thanks a bunch!

Can we get WDS now?

Regards.