kaloz/mwlwifi #39: Failed to set key-error

kristrev, 2015-10-29:
Since issue were I mentioned this was marked as resolved, I thought I should create a new one.

Dmes sometimes show this when clients (with different chipsets) connect to AP:

[476322.762953] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[476322.769166] ieee80211 phy0: return code: 0x1122
[476322.773811] ieee80211 phy0: timeout: 0x1122
[476322.778472] ieee80211 phy0: failed execution
[476322.782891] wlan0: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)

Repeated over and over. 2.4 GHz works fine and the only way to get out of this state seems to be through a reboot. This problem is not limited to Apple devices, the client that caused the error this time was a Lenovo X1 Carbon 2. gen with an Intel Wireless-card (7260). I have also seen it with an iPhone 6s. If it matters, this error seems to happen right after the devices wake up. Of course that is when they try to connect to wifi again, but maybe there are some power-saving stuff in router or something.

johnnysl, 2015-10-29:
Does this also happen with driver version 10.3.0.12?

kristrev, 2015-10-29:
I have not been able to test (yet), but looking at the changelog diff between 10 and 12 there is no reason to believe it will not happen. I will compile an image and start testing tonight.

yuhhaurlin, 2015-11-12:
Can we close this one and use #41 to track problem?

kristrev, 2015-11-12:
Yes. I have not seen it in a long time anyway, so it could be that it has been fixed already.

closed 2015-11-12 by kristrev