kaloz/mwlwifi #41: ba/ampdu error with 10.3.0.12

kristrev, 2015-11-01:
Opened a new issue on @yuhhaurlin request.

Things have improved significanlty since 10.3.0.10, but the ba/ampdu error still occurs:

[  562.110063] ieee80211 phy0: check ba result error 1
[  562.114983] ieee80211 phy0: ampdu start error code: -22
[  562.170025] ieee80211 phy0: check ba result error 1
[  562.174939] ieee80211 phy0: ampdu start error code: -22
[  562.220015] ieee80211 phy0: check ba result error 1
[  562.224924] ieee80211 phy0: ampdu start error code: -22
[  562.270014] ieee80211 phy0: check ba result error 1
[  562.274923] ieee80211 phy0: ampdu start error code: -22
[  562.320028] ieee80211 phy0: check ba result error 1
[  562.324939] ieee80211 phy0: ampdu start error code: -22
[  562.380019] ieee80211 phy0: check ba result error 1
[  562.384922] ieee80211 phy0: ampdu start error code: -22
[  562.430018] ieee80211 phy0: check ba result error 1
[  562.434919] ieee80211 phy0: ampdu start error code: -22

Things work fine for a while, then some random event X happens and the errors start happening (while transferring data). The devices connected to 5GHz is an iPhone 6 Plus and a Lenovo X1 Carbon 2. gen. After some time it seems the 5Ghz just gives up and takes a long nap. By connecting to the 2.4 GHz and restarting wifi, 5GHz works again but at reduced speed.

Another, potentially related error, is that the iPhone has trouble connecting to the 5GHz network. It keeps requesting me for the key, and seemingly only accepts the key when another devices (the laptop) has been connected. Otherwise it just gives me an error.

Debug fs output (in case it helps debugging):

```
/ampdu:
stream: 0
idx: 0
state: 3
mac address: 7c:7a:91:a3:37:c5
tid: 0
stream: 1
idx: 1
state: 3
mac address: e0:b5:2d:15:cf:da
tid: 0
stream: 2
idx: 0
state: 0
stream: 3
idx: 0
state: 0

/info:
driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.12
firmware version: 0x07020806
mac address: 00:25:9c:13:ab:6f
2g: disable
5g: enable
antenna: 2 2
irq number: 101
iobase0: e0e00000
iobase1: e1000000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
mfg mode: false

/sta

mac address: 7c:7a:91:a3:37:c5
aid: 1
ampdu: true
amsdu: false
[... 127 lines cut ...]
[67524.758920] mwl_rx_recv: ieee80211 phy0: 80:86:f2:09:0f:b5 - addba.resp.capab=0x1002
[89605.118409] mwl_rx_recv: ieee80211 phy0: 0c:8b:fd:08:09:71 - addba.resp.capab=0x1002
[91525.521926] mwl_rx_recv: ieee80211 phy0: 5c:0a:5b:79:ae:68 - addba.resp.capab=0x202
[108434.204598] mwl_rx_recv: ieee80211 phy0: 58:c3:8b:ec:66:82 - addba.resp.capab=0x202
[108445.369578] mwl_rx_recv: ieee80211 phy0: 0c:8b:fd:08:09:71 - addba.resp.capab=0x1002
[108671.317061] mwl_rx_recv: ieee80211 phy1: 58:c3:8b:ec:66:82 - addba.resp.capab=0x202
[108742.047095] ieee80211 phy0: check ba result error 1
[108742.052155] ieee80211 phy0: ampdu start error code: -22
[108742.086924] ieee80211 phy0: check ba result error 1
[108742.092003] ieee80211 phy0: ampdu start error code: -22
[108742.127182] ieee80211 phy0: check ba result error 1
[108742.132250] ieee80211 phy0: ampdu start error code: -22
[108742.166947] ieee80211 phy0: check ba result error 1
[108742.172050] ieee80211 phy0: ampdu start error code: -22
[108742.207132] ieee80211 phy0: check ba result error 1
[108742.212301] ieee80211 phy0: ampdu start error code: -22
[108742.247162] ieee80211 phy0: check ba result error 1
[108742.252319] ieee80211 phy0: ampdu start error code: -22
[108742.287046] ieee80211 phy0: check ba result error 1
[108742.292106] ieee80211 phy0: ampdu start error code: -22
[108742.480842] mwl_rx_recv: ieee80211 phy0: 58:c3:8b:ec:66:82 - addba.resp.capab=0x202
[111659.302524] mwl_rx_recv: ieee80211 phy0: 0c:8b:fd:08:09:71 - addba.resp.capab=0x1006
[114157.963627] mwl_rx_recv: ieee80211 phy0: 5c:0a:5b:79:ae:68 - addba.resp.capab=0x202
[164362.167093] mwl_rx_recv: ieee80211 phy0: 0c:8b:fd:08:09:71 - addba.resp.capab=0x1002
```

johnnysl, 2015-11-01:
Only remark i can make is that with over 2.5days of uptime and >100GB handled by the wireless interface, i do not see these check ba errors anymore. I assume that is because only a certain subset of devices still see this issue.

gufus, 2015-11-01:
DD-WRT v3.0-r28054 std (c) 2015 NewMedia-NET GmbH
Release: 10/30/15

ASUS USB-AC53 802.11ac Wireless USB Adapter  (192.168.1.194)

MAC  74:d0:2b:d0:1c:94

Nov  1 18:03:57 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 IEEE 802.11: authenticated
Nov  1 18:03:57 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 IEEE 802.11: associated (aid 1)
Nov  1 18:03:57 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 RADIUS: starting accounting session 00000005-00000000
Nov  1 18:03:57 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 WPA: pairwise key handshake completed (RSN)
Nov  1 18:06:55 AC-DD-WRT daemon.info hostapd: ath0: STA 20:aa:4b:3e:5e:39 WPA: group key handshake completed (RSN)
Nov  1 18:06:55 AC-DD-WRT daemon.info hostapd: ath0: STA be:a3:86:4b:61:ee WPA: group key handshake completed (RSN)
Nov  1 18:06:55 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 WPA: group key handshake completed (RSN)
Nov  1 18:07:45 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 IEEE 802.11: authenticated
Nov  1 18:07:45 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 IEEE 802.11: associated (aid 1)
Nov  1 18:07:45 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 RADIUS: starting accounting session 00000005-00000000
Nov  1 18:07:45 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 WPA: pairwise key handshake completed (RSN)
Nov  1 18:07:46 AC-DD-WRT kern.err kernel: [79272.767529] ieee80211 phy1: check ba result error 1
Nov  1 18:07:46 AC-DD-WRT kern.err kernel: [79272.772490] ieee80211 phy1: ampdu start error code: -22
Nov  1 18:07:46 AC-DD-WRT kern.err kernel: [79272.807529] ieee80211 phy1: check ba result error 1
Nov  1 18:07:46 AC-DD-WRT kern.err kernel: [79272.812544] ieee80211 phy1: ampdu start error code: -22
Nov  1 18:07:46 AC-DD-WRT kern.err kernel: [79272.847462] ieee80211 phy1: check ba result error 1
Nov  1 18:07:46 AC-DD-WRT kern.err kernel: [79272.852417] ieee80211 phy1: ampdu start error code: -22
Nov  1 18:07:46 AC-DD-WRT kern.err kernel: [79272.877304] ieee80211 phy1: check ba result error 1
Nov  1 18:07:46 AC-DD-WRT kern.err kernel: [79272.882345] ieee80211 phy1: ampdu start error code: -22
Nov  1 18:08:02 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 IEEE 802.11: authenticated
Nov  1 18:08:02 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 IEEE 802.11: associated (aid 1)
Nov  1 18:08:02 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 RADIUS: starting accounting session 00000005-00000000

Stil in mwlwifi driver 10.3.0.12.

Razerwire, 2015-11-01:
I've seen the same issue once using my laptop with a Intel Wireless AC-7265, i was using speedtest.net to test my connection at that time. Latency went trough the roof and the download speed was only about 1mbps. 

 1 21:49:50 2015 daemon.info hostapd: wlan0: STA 60:57:18:28:72:c4 IEEE 802.11: authenticated
Sun Nov  1 21:49:50 2015 daemon.info hostapd: wlan0: STA 60:57:18:28:72:c4 IEEE 802.11: associated (aid 1)
Sun Nov  1 21:49:50 2015 daemon.info hostapd: wlan0: STA 60:57:18:28:72:c4 WPA: pairwise key handshake completed (RSN)
Sun Nov  1 21:49:50 2015 daemon.info dnsmasq-dhcp[1533]: DHCPREQUEST(br-lan) 192.168.1.102 60:57:18:28:72:c4 
Sun Nov  1 21:49:50 2015 daemon.info dnsmasq-dhcp[1533]: DHCPACK(br-lan) 192.168.1.102 60:57:18:28:72:c4 Dell-E5250
Sun Nov  1 21:49:51 2015 daemon.warn odhcpd[1060]: DHCPV6 CONFIRM IA_NA from 000456224406e6e6b57416bf76ea59272c29 on br-lan: not on-link fdf9:52ca:5c09::542/128 
Sun Nov  1 21:49:52 2015 daemon.warn odhcpd[1060]: DHCPV6 SOLICIT IA_NA from 000456224406e6e6b57416bf76ea59272c29 on br-lan: ok fdf9:52ca:5c09::542/128 
Sun Nov  1 21:49:53 2015 daemon.warn odhcpd[1060]: DHCPV6 REQUEST IA_NA from 000456224406e6e6b57416bf76ea59272c29 on br-lan: ok fdf9:52ca:5c09::542/128 
Sun Nov  1 21:50:12 2015 daemon.info hostapd: wlan0: STA 60:57:18:28:72:c4 WPA: group key handshake completed (RSN)
Sun Nov  1 21:50:12 2015 daemon.info hostapd: wlan0: STA 3c:a9:f4:02:e4:64 WPA: group key handshake completed (RSN)
Sun Nov  1 21:50:12 2015 daemon.info hostapd: wlan0: STA a4:e4:b8:11:dd:af WPA: group key handshake completed (RSN)
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.717775] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.722774] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.757773] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.762751] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.807772] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.812690] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.847812] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.852824] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.887807] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.892795] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.927753] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.932748] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.967846] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21498.972858] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.007752] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.012751] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.047837] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.052835] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.087861] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.092773] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.127757] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.132763] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.167762] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.172762] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.207747] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.212738] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.247827] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.252745] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.287821] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:44 2015 kern.err kernel: [21499.292780] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.327760] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.332740] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.367817] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.372804] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.407781] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.412775] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.447789] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.452788] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.487772] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.492764] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.527856] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.532838] ieee80211 phy0: ampdu start error code: -22
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.567846] ieee80211 phy0: check ba result error 1
Sun Nov  1 21:50:45 2015 kern.err kernel: [21499.572835] ieee80211 phy0: ampdu start error code: -22

TireMeat, 2015-11-01:
I've been having decent luck with the below (so far) -- just dropped the ssn reset closer to the stream start, and am not starting the stream unless the firmware says it's OK.

Driver is 10.3.0.12 and I'm still building against branch.

```
diff -ur original_CC_branch/mac80211.c updated_CC_branch/mac80211.c
--- original_CC_branch/mac80211.c	2015-10-29 00:20:51.000000000 -0500
+++ updated_CC_branch/mac80211.c	2015-11-01 15:18:15.509075000 -0600
@@ -593,59 +593,45 @@
 	case IEEE80211_AMPDU_RX_STOP:
 		break;
 	case IEEE80211_AMPDU_TX_START:
-		/* By the time we get here the hw queues may contain outgoing
-		 * packets for this RA/TID that are not part of this BA
-		 * session.  The hw will assign sequence numbers to these
-		 * packets as they go out.  So if we query the hw for its next
-		 * sequence number and use that for the SSN here, it may end up
-		 * being wrong, which will lead to sequence number mismatch at
-		 * the recipient.  To avoid this, we reset the sequence number
-		 * to O for the first MPDU in this BA stream.
-		 */
-		*ssn = 0;
 
-		if (!stream) {
-			/* This means that somebody outside this driver called
-			 * ieee80211_start_tx_ba_session.  This is unexpected
-			 * because we do our own rate control.  Just warn and
-			 * move on.
-			 */
-			stream = mwl_fwcmd_add_stream(hw, sta, tid);
-		}
-
-		if (!stream) {
-			wiphy_warn(hw->wiphy, "no stream found\n");
-			rc = -EBUSY;
-			break;
-		}
-
-		stream->state = AMPDU_STREAM_IN_PROGRESS;
-
-		/* Release the lock before we do the time consuming stuff */
+		/* If ampdu is not allowed, don't bother with stream setup */ 
 		spin_unlock_bh(&priv->stream_lock);
-
-		/* Check if link is still valid */
 		if (!sta_info->is_ampdu_allowed) {
+			wiphy_warn(hw->wiphy, "ampdu not allowed\n");
+			rc = -EBUSY;
 			spin_lock_bh(&priv->stream_lock);
-			mwl_fwcmd_remove_stream(hw, stream);
-			spin_unlock_bh(&priv->stream_lock);
-			wiphy_warn(hw->wiphy,
-				   "link is not valid now\n");
-			return -EBUSY;
+			break;
 		}
-		rc = mwl_fwcmd_check_ba(hw, stream, vif);
-
 		spin_lock_bh(&priv->stream_lock);
 
-		if (rc) {
-			mwl_fwcmd_remove_stream(hw, stream);
-			wiphy_err(hw->wiphy,
-				  "ampdu start error code: %d\n", rc);
-			rc = -EBUSY;
-			break;
+		/* If stream doesn't exist, try to create it */ 
+		if (!stream) {
+			stream = mwl_fwcmd_add_stream(hw, sta, tid);
+			if (!stream) {
+				wiphy_warn(hw->wiphy, "no stream found\n");
+				rc = -EBUSY;
+				break;
+			}
 		}
 
-		ieee80211_start_tx_ba_cb_irqsafe(vif, addr, tid);
+		/* If stream exists, reset the sequence number and then
+		   check that the ba is working. If there is a problem,
+		   remove the stream, but if OK, start it */ 
+		if (stream) {
+			*ssn = 0;
+			stream->state = AMPDU_STREAM_IN_PROGRESS;
+			spin_unlock_bh(&priv->stream_lock);
+			rc = mwl_fwcmd_check_ba(hw, stream, vif);
+			spin_lock_bh(&priv->stream_lock);
+			if (rc) {
+				mwl_fwcmd_remove_stream(hw, stream);
+				wiphy_err(hw->wiphy,
+					  "ampdu start error code: %d\n", rc);
+				rc = -EBUSY;
+				break;
+			}
+			ieee80211_start_tx_ba_cb_irqsafe(vif, addr, tid);
+		}
 		break;
 	case IEEE80211_AMPDU_TX_STOP_CONT:
 	case IEEE80211_AMPDU_TX_STOP_FLUSH:
```

gufus, 2015-11-01:
@BrainSlayer

My clients do stay up

Nov  1 20:06:55 AC-DD-WRT daemon.info hostapd: ath0: STA be:a3:86:4b:61:ee WPA: group key handshake completed (RSN)
Nov  1 20:06:55 AC-DD-WRT daemon.info hostapd: ath0: STA 20:aa:4b:3e:5e:39 WPA: group key handshake completed (RSN)
Nov  1 20:06:55 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 WPA: group key handshake completed (RSN)
Nov  1 21:00:00 AC-DD-WRT kern.info kernel: [89607.077378] dc.sh (28867): drop_caches: 3
Nov  1 21:01:22 AC-DD-WRT authpriv.info dropbear[24798]: Exit (root): Exited normally
Nov  1 21:01:27 AC-DD-WRT authpriv.info dropbear[24019]: Exit (root): Exited normally
Nov  1 21:06:55 AC-DD-WRT daemon.info hostapd: ath0: STA be:a3:86:4b:61:ee WPA: group key handshake completed (RSN)
Nov  1 21:06:55 AC-DD-WRT daemon.info hostapd: ath0: STA 20:aa:4b:3e:5e:39 WPA: group key handshake completed (RSN)
Nov  1 21:06:55 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 WPA: group key handshake completed (RSN)
Nov  1 21:41:43 AC-DD-WRT authpriv.info dropbear[30335]: Child connection from 192.168.1.194:53620
Nov  1 21:41:44 AC-DD-WRT authpriv.notice dropbear[30335]: Pubkey auth succeeded for 'root' with key md5 dc:e8:48:e8:1c:d1:7d:c4:53:6b:07:ef:1a:06:86:96 from 192.168.1.194:53620
Nov  1 21:43:04 AC-DD-WRT authpriv.info dropbear[30335]: Exit (root): Exited normally
Nov  1 22:00:00 AC-DD-WRT kern.info kernel: [93207.731440] dc.sh (30965): drop_caches: 3
Nov  1 22:06:55 AC-DD-WRT daemon.info hostapd: ath0: STA be:a3:86:4b:61:ee WPA: group key handshake completed (RSN)
Nov  1 22:06:55 AC-DD-WRT daemon.info hostapd: ath0: STA 20:aa:4b:3e:5e:39 WPA: group key handshake completed (RSN)
Nov  1 22:06:55 AC-DD-WRT daemon.info hostapd: ath1: STA 74:d0:2b:d0:1c:94 WPA: group key handshake completed (RSN)

yuhhaurlin, 2015-11-02:
Thanks for all your information. BTW, can you let me know if the client still works well after these messages came out?

johnnysl, 2015-11-02:
@yuhhaurlin: my experience; not without a reconnect. I believe Kristrev and Kazerwire wrote something similar as well.

yuhhaurlin, 2015-11-02:
Client will reconnect automatically, so that means client can continue to work after these messages came out? How soon to make this kind of problem happened? Only with specific clients?

johnnysl, 2015-11-02:
@yuhhaurlin: with me after 3 days of usage, and with a mid 2012 mac-book air.

yuhhaurlin, 2015-11-02:
Thanks. The client can still work after these messages came out?  I think I will modify FW to report correct failed reason to debug this run time problem.

bmork, 2015-11-02:
@yuhhaurlin: More detailed errors from the firmware would be great!

I realize that you have to keep the actual firmware implementation private, but is there any chance of a little more documentation on the firmware <-> driver API at the same time?  Most of it is obvious from fwcmd.c, but things like e.g "SetOptimizationLevel" or "LinkAdaptCsMode" is not (at least to me). That is, I understand that you can set a "level" and a "mode", but I have no idea what the firmware use these values for.  Or even which values are valid.  I'm not asking for any large manual.  Just thinking of a small comment in fwcmd.c listing the supported values and some high
level description of the assoociated behaviour.  Just enough to be able to understand the difference.

I ask this because I assume the firmware error codes will be meaningless to us without some attached description. I understand that they will still be useful to you, but it is easier if we can make some sense out of the errors too, maybe correlating them with client quirks we know about.

And to answer your question: Yes, the client seems to work fine (for me) even when I see the occasional ba error messages.  Looks like the driver handles this as it should, falling back until the command succeeds.

yuhhaurlin, 2015-11-02:
Thanks for your information. I have no right to come out this kind of document.

Yes, that is what I think. I will add more specific failed reason in FW for BA related commands to see if this error is reasonable.

johnnysl, 2015-11-02:
ok, i'm seeing the issue here again (uptime 1 day), but i have so many clients active, that i'm not sure which one is the problem. is there any way to pinpoint that?

gufus, 2015-11-02:
From: yuhhaurlin [mailto:notifications@github.com] 
Sent: Sunday, November 01, 2015 6:38 PM
To: kaloz/mwlwifi
Cc: gufus
Subject: Re: [mwlwifi] ba/ampdu error with 10.3.0.12 (#41)

Thanks for all your information. BTW, can you let me know if the client still works well after these messages came out?
----

Yes.

The client still works well after these messages came out.

I get the error (only once in a while) using this:
ASUS USB-AC53 802.11ac Wireless USB Adapter

Also I don't get any negative side effects with this error.

kern.err kernel: [ 5601.602534] ieee80211 phy0: ampdu start error code: -22 
kern.err kernel: [ 5601.637635] ieee80211 phy0: check ba result error 1 
kern.err kernel: [ 6222.253363] ieee80211 phy1: ampdu start error code: -22 
kern.err kernel: [ 6222.298444] ieee80211 phy1: check ba result error 1 

—

yuhhaurlin, 2015-11-02:
@johnnysl I think there is no need to know which client makes these messages came out. It looks like it is not a real bug. I will let FW to report clear error reason to reconfirm this. Thanks.

notnyt, 2015-11-02:
I'm also still seeing this on 10.3.0.12.  I've tried tiremeat's patch to see if it helps.

TireMeat, 2015-11-03:
@notnyt ... it doesn't unfortunately

notnyt, 2015-11-05:
@TireMeat using your patch I'm not seeing any more ba errors, and a friend with the router reported better performance and less crashing.  *shrug*  maybe refine and submit a pull request if its worth it?

TireMeat, 2015-11-05:
@notnyt  ...  that's great to hear.   I tightened up the code a bit more - here are the highlights:
* In the start ampdu action, I'm resetting the sequence number right before stream actually starts, (idea is to reduce packet loss in case starting the stream is delayed).
* In the stop ampdu action, I'm calling stop_ba only when existing streams are flushed.   (I can't reproduce the polling interrupt problem, but idea is to prevent calls to ieee80211_stop_tx_ba_cb_irqsafe unless known-stop is happening)
* In the ampdu operational action, (similar code refactoring as in start ampdu), activate only streams successfully created by the firmware.
* Added comments.

Below is the patch ...  (what do you mean by submitting a pull request?  how do i do that?)

```
--- original_10.3.0.12/mac80211.c	2015-10-29 00:20:51.000000000 -0500
+++ updated_10.3.0.12/mac80211.c	2015-11-05 00:07:39.398401000 -0600
@@ -593,63 +593,53 @@
 	case IEEE80211_AMPDU_RX_STOP:
 		break;
 	case IEEE80211_AMPDU_TX_START:
-		/* By the time we get here the hw queues may contain outgoing
-		 * packets for this RA/TID that are not part of this BA
-		 * session.  The hw will assign sequence numbers to these
-		 * packets as they go out.  So if we query the hw for its next
-		 * sequence number and use that for the SSN here, it may end up
-		 * being wrong, which will lead to sequence number mismatch at
-		 * the recipient.  To avoid this, we reset the sequence number
-		 * to O for the first MPDU in this BA stream.
-		 */
-		*ssn = 0;
 
-		if (!stream) {
-			/* This means that somebody outside this driver called
-			 * ieee80211_start_tx_ba_session.  This is unexpected
-			 * because we do our own rate control.  Just warn and
-			 * move on.
-			 */
-			stream = mwl_fwcmd_add_stream(hw, sta, tid);
-		}
-
-		if (!stream) {
-			wiphy_warn(hw->wiphy, "no stream found\n");
-			rc = -EBUSY;
-			break;
-		}
-
-		stream->state = AMPDU_STREAM_IN_PROGRESS;
-
-		/* Release the lock before we do the time consuming stuff */
+		/* If ampdu is not allowed, don't bother with stream setup */ 
 		spin_unlock_bh(&priv->stream_lock);
-
-		/* Check if link is still valid */
 		if (!sta_info->is_ampdu_allowed) {
+			wiphy_warn(hw->wiphy, "ampdu not allowed\n");
+			rc = -EBUSY;
 			spin_lock_bh(&priv->stream_lock);
-			mwl_fwcmd_remove_stream(hw, stream);
-			spin_unlock_bh(&priv->stream_lock);
-			wiphy_warn(hw->wiphy,
-				   "link is not valid now\n");
-			return -EBUSY;
+			break;
 		}
-		rc = mwl_fwcmd_check_ba(hw, stream, vif);
-
 		spin_lock_bh(&priv->stream_lock);
 
-		if (rc) {
-			mwl_fwcmd_remove_stream(hw, stream);
-			wiphy_err(hw->wiphy,
-				  "ampdu start error code: %d\n", rc);
-			rc = -EBUSY;
-			break;
+		/* If stream does not exist, try to add it */ 
+		if (!stream) {
+			stream = mwl_fwcmd_add_stream(hw, sta, tid);
+			if (!stream) {
+				wiphy_warn(hw->wiphy, "no stream found\n");
+				rc = -EBUSY;
+				break;
+			}
 		}
 
-		ieee80211_start_tx_ba_cb_irqsafe(vif, addr, tid);
+		/* If stream exists, check that the ba is working.
+		   If the ba is not working, remove the added stream and exit.
+		   If stream is OK, reset the sequence number just before
+		   starting it */ 
+		if (stream) {
+			stream->state = AMPDU_STREAM_IN_PROGRESS;
+			spin_unlock_bh(&priv->stream_lock);
+			rc = mwl_fwcmd_check_ba(hw, stream, vif);
+			spin_lock_bh(&priv->stream_lock);
+			if (rc) {
+				mwl_fwcmd_remove_stream(hw, stream);
+				wiphy_err(hw->wiphy,
+					  "ampdu start error code: %d\n", rc);
+				rc = -EBUSY;
+				break;
+			}
+			*ssn = 0;
+			ieee80211_start_tx_ba_cb_irqsafe(vif, addr, tid);
+		}
 		break;
 	case IEEE80211_AMPDU_TX_STOP_CONT:
 	case IEEE80211_AMPDU_TX_STOP_FLUSH:
 	case IEEE80211_AMPDU_TX_STOP_FLUSH_CONT:
+		/* If stream exists and active, flush it and destroy ba.
+		   If stream exists but is not active, just remove it.
+		   Finally issue stop_ba to complete. */ 
 		if (stream) {
 			if (stream->state == AMPDU_STREAM_ACTIVE) {
 				mwl_tx_del_ampdu_pkts(hw, sta, tid);
@@ -658,31 +648,38 @@
 				mwl_fwcmd_destroy_ba(hw, idx);
 				spin_lock_bh(&priv->stream_lock);
 			}
-
 			mwl_fwcmd_remove_stream(hw, stream);
+			ieee80211_stop_tx_ba_cb_irqsafe(vif, addr, tid);
 		}
-
-		ieee80211_stop_tx_ba_cb_irqsafe(vif, addr, tid);
 		break;
 	case IEEE80211_AMPDU_TX_OPERATIONAL:
-		if (WARN_ON(stream->state != AMPDU_STREAM_IN_PROGRESS)) {
+		/* If stream is not in progress, warn and exit */ 
+		if (stream->state != AMPDU_STREAM_IN_PROGRESS) {
+			wiphy_err(hw->wiphy,
+				  "ampdu operation is not in progress");
 			rc = -EPERM;
 			break;
 		}
-		spin_unlock_bh(&priv->stream_lock);
-		rc = mwl_fwcmd_create_ba(hw, stream, buf_size, vif);
-		spin_lock_bh(&priv->stream_lock);
 
-		if (!rc) {
-			stream->state = AMPDU_STREAM_ACTIVE;
-		} else {
-			idx = stream->idx;
+		/* If stream exists, create ba and set active.
+		   If ba create failed, remove the stream and warn.
+		   No need to stop_ba since the ba create failed. */ 
+		if (stream) {
 			spin_unlock_bh(&priv->stream_lock);
-			mwl_fwcmd_destroy_ba(hw, idx);
+			rc = mwl_fwcmd_create_ba(hw, stream, buf_size, vif);
 			spin_lock_bh(&priv->stream_lock);
-			mwl_fwcmd_remove_stream(hw, stream);
-			wiphy_err(hw->wiphy,
-				  "ampdu operation error code: %d\n", rc);
+
+			if (!rc) {
+				stream->state = AMPDU_STREAM_ACTIVE;
+			} else {
+				idx = stream->idx;
+				spin_unlock_bh(&priv->stream_lock);
+				mwl_fwcmd_destroy_ba(hw, idx);
+				spin_lock_bh(&priv->stream_lock);
+				mwl_fwcmd_remove_stream(hw, stream);
+				wiphy_err(hw->wiphy,
+					  "ampdu operation error code: %d\n", rc);
+			}
 		}
 		break;
 	default:
```

yuhhaurlin, 2015-11-05:
Thanks. I will check it.

kristrev, 2015-11-06:
I just tested 10.3.0.14 and the error is still present. The client is an iPhone 6s and error message is the same (so I guess additional FW debug has not been added). Device works fine after the error though.

yuhhaurlin, 2015-11-06:
Thanks for your information. Due to this problem is not so severe, I will add correct reason in FW later.

johnnysl, 2015-11-08:
@yuhhaurlin: 
1. I get these errors only occassionally with the iphone 5s from my wife. As soon as the errors occur, her internet is pretty much unusable on the phone. Switching between wlan0 and wlan1 fixes it for a while (day?). So, not convinced the error is totally harmless. 

2. During my tests with IPV6, my router crashed once when switching from wlan0 to wlan1 oj her phone. I first thought it was unrelated, but i'm now getting the impression that the wifi driver might be the cause of that? Can't seem to get the router to store decent crash logs though :-(

kristrev, 2015-11-08:
I saw the same as @johnnysl . With an iPhone 6s, when I get the error, I have to switch between the 5 and 2.4 Ghz (and back again) to make the device work again. Other times, the device works just fine after this error.

johnnysl, 2015-11-08:
on the plus side though, 10.3.0.14 gives me the most stable and fastest WiFi connection (in response times) I've ever seen to date. So many thanks for that. There are just some small wrinkles left that need ironing out.

mystica555, 2015-11-09:
This is with 10.3.0.12; I have not yet upped to 10.3.0.14 yet which I hope will be part of the nov07 openwrt trunk build on downloads.openwrt.org. 

I was in ssh with cat /proc/kmsg but didn't get a full stack trace before the router rebooted.
root@OpenWrt:~# dmesg
[ 5461.467333] ieee80211 phy1: check ba result error 1
[ 5461.472233] ieee80211 phy1: ampdu start error code: -22
[ 5461.517332] ieee80211 phy1: check ba result error 1
[ 5461.522231] ieee80211 phy1: ampdu start error code: -22
[ 5461.567332] ieee80211 phy1: check ba result error 1
[ 5461.572231] ieee80211 phy1: ampdu start error code: -22
[ 5461.617331] ieee80211 phy1: check ba result error 1
[ 5461.622229] ieee80211 phy1: ampdu start error code: -22
[ 5461.667330] ieee80211 phy1: check ba result error 1
[ 5461.672228] ieee80211 phy1: ampdu start error code: -22
[ 5461.717329] ieee80211 phy1: check ba result error 1
[ 5461.722228] ieee80211 phy1: ampdu start error code: -22
[ 5461.767328] ieee80211 phy1: check ba result error 1
[ 5461.772226] ieee80211 phy1: ampdu start error code: -22
[ 5461.817352] ieee80211 phy1: check ba result error 1
[ 5461.822261] ieee80211 phy1: ampdu start error code: -22
[ 5566.586880] ieee80211 phy1: check ba result error 1
[ 5566.591801] ieee80211 phy1: ampdu start error code: -22
[ 5566.636873] ieee80211 phy1: check ba result error 1
[ 5566.641788] ieee80211 phy1: ampdu start error code: -22
[ 5566.686874] ieee80211 phy1: check ba result error 1
[ 5566.691789] ieee80211 phy1: ampdu start error code: -22
[ 5566.736871] ieee80211 phy1: check ba result error 1
[ 5566.741787] ieee80211 phy1: ampdu start error code: -22
[ 5566.786888] ieee80211 phy1: check ba result error 1
[ 5566.791803] ieee80211 phy1: ampdu start error code: -22
[ 5566.836870] ieee80211 phy1: check ba result error 1
[ 5566.841784] ieee80211 phy1: ampdu start error code: -22
[ 5566.886873] ieee80211 phy1: check ba result error 1
[ 5566.891787] ieee80211 phy1: ampdu start error code: -22
[ 5566.936869] ieee80211 phy1: check ba result error 1
[ 5566.941782] ieee80211 phy1: ampdu start error code: -22
[ 5567.246885] ieee80211 phy1: check ba result error 1
[ 5567.251800] ieee80211 phy1: ampdu start error code: -22
[ 5567.296872] ieee80211 phy1: check ba result error 1
[ 5567.301787] ieee80211 phy1: ampdu start error code: -22
[ 5567.336867] ieee80211 phy1: check ba result error 1
[ 5567.341781] ieee80211 phy1: ampdu start error code: -22
[ 5567.386885] ieee80211 phy1: check ba result error 1
[ 5567.391931] ieee80211 phy1: ampdu start error code: -22
[ 5567.436867] ieee80211 phy1: check ba result error 1
[ 5567.441782] ieee80211 phy1: ampdu start error code: -22
[ 6076.463496] ieee80211 phy1: check ba result error 1
[ 6076.468418] ieee80211 phy1: ampdu start error code: -22
[ 6076.503494] ieee80211 phy1: check ba result error 1
[ 6076.508410] ieee80211 phy1: ampdu start error code: -22
[ 6076.543501] ieee80211 phy1: check ba result error 1
[ 6076.548415] ieee80211 phy1: ampdu start error code: -22
[ 6076.583486] ieee80211 phy1: check ba result error 1
[ 6076.588401] ieee80211 phy1: ampdu start error code: -22
[ 6076.623485] ieee80211 phy1: check ba result error 1
[ 6076.628398] ieee80211 phy1: ampdu start error code: -22
[ 6076.673484] ieee80211 phy1: check ba result error 1
[ 6076.678397] ieee80211 phy1: ampdu start error code: -22
[ 6076.723484] ieee80211 phy1: check ba result error 1
[ 6076.728398] ieee80211 phy1: ampdu start error code: -22
[ 6076.773483] ieee80211 phy1: check ba result error 1
[ 6076.778396] ieee80211 phy1: ampdu start error code: -22
[ 6076.833482] ieee80211 phy1: check ba result error 1
[ 6076.838396] ieee80211 phy1: ampdu start error code: -22
[ 6076.883481] ieee80211 phy1: check ba result error 1
[ 6076.888395] ieee80211 phy1: ampdu start error code: -22
[ 6076.933493] ieee80211 phy1: check ba result error 1
[ 6076.938407] ieee80211 phy1: ampdu start error code: -22
[ 6076.983479] ieee80211 phy1: check ba result error 1
[ 6076.988392] ieee80211 phy1: ampdu start error code: -22
[... 100 lines cut ...]
<4>[25630.805401] pc : [<bf306554>]    lr : [<bf306544>]    psr: 60000013
<4>[25630.805401] sp : df7b3d98  ip : df7b3d98  fp : df7b3dcc
<4>[25630.816927] r10: 00000000  r9 : de3863bc  r8 : dd4525e4
<4>[25630.822173] r7 : dd4525d8  r6 : de3863b0  r5 : de3854c0  r4 : 0000258c
<4>[25630.828725] r3 : ffffffff  r2 : 00000000  r1 : 00000000  r0 : de3863bc
<4>[25630.835279] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
<4>[25630.842618] Control: 10c5387d  Table: 1d94404a  DAC: 00000015
<0>[25630.848386] Process kworker/u4:2 (pid: 326, stack limit = 0xdf7b2238)
<0>[25630.854853] Stack: (0xdf7b3d98 to 0xdf7b4000)
<0>[25630.859229] 3d80:                                                       00000004 ffffffff
<0>[25630.867442] 3da0: df7b3dbc de386f30 de384b60 dd4525d8 de386f44 dd4525e4 00000004 00000004
<0>[25630.875655] 3dc0: df7b3e0c df7b3dd0 bf2fe634 bf306518 df7b3dd0 00000000 ded4cc78 00000003
<0>[25630.883868] 3de0: df7b3dfc dd452000 dddc4dc0 dddc4e40 00000000 dd452064 00000004 de384b60
<0>[25630.892081] 3e00: df7b3e54 df7b3e10 bf297188 bf2fe480 00000004 00000000 00000000 dfbd8400
<0>[25630.900294] 3e20: dfbd8438 00000003 df659048 dd452450 dd452000 00000002 00000004 ded4c8d4
<0>[25630.908507] 3e40: ded4c4c0 de384f5c df7b3e74 df7b3e58 bf297644 bf296f10 ddb1ec40 ddf5c088
<0>[25630.916720] 3e60: ded4c8c4 de384b60 df7b3e8c df7b3e78 bf2963b8 bf29761c ddf5c088 ded4c8c4
[... cut ...]
<0>[25630.941358] 3ec0: dee94d00 df7b2020 00000000 00000000 df7b3f1c df7b3ee0 c003a010 bf29c1cc
<0>[25630.949572] 3ee0: df7b3f04 df5f0300 df42a800 df42a800 df5f0318 df5f0300 df42a800 df42a814
<0>[25630.957785] 3f00: df5f0318 df7b2020 00000088 df7b2000 df7b3f5c df7b3f20 c003ab88 c0039e28
<0>[25630.965998] 3f20: c00518f8 c048e100 c0493653 df42a800 c003a86c df73f740 00000000 df5f0300
<0>[25630.974211] 3f40: c003a86c 00000000 00000000 00000000 df7b3fac df7b3f60 c003e5cc c003a878
<0>[25630.982423] packet_write_wait: Connection to 192.168.1.1: Broken pipe

johnnysl, 2015-11-10:
@yuhhaurlin: I see the errors only appear after the device has been in-active for a while, and then becomes active again. (e.g. if you grab your ipad after a day). Device is then connected but unusable, need to disconnect fully and reconnect (or switch from 5 to 2.4) to get WiFi working again.

notnyt, 2015-11-14:
I still get these with .14 with an intel wifi card.

````
[657432.691173] ieee80211 phy0: check ba result error 1
[657432.696341] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.691173] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.696341] ieee80211 phy0: ampdu start error code: -22
[657432.740801] ieee80211 phy0: check ba result error 1
[657432.745815] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.740801] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.745815] ieee80211 phy0: ampdu start error code: -22
[657432.790802] ieee80211 phy0: check ba result error 1
[657432.795812] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.790802] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.795812] ieee80211 phy0: ampdu start error code: -22
[657432.850803] ieee80211 phy0: check ba result error 1
[657432.855812] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.850803] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.855812] ieee80211 phy0: ampdu start error code: -22
[657432.926177] ieee80211 phy0: check ba result error 1
[657432.931229] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.926177] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.931229] ieee80211 phy0: ampdu start error code: -22
[657432.975933] ieee80211 phy0: check ba result error 1
[657432.980982] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.975933] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:25 2015 kern.err kernel: [657432.980982] ieee80211 phy0: ampdu start error code: -22
[657433.025912] ieee80211 phy0: check ba result error 1
[657433.030964] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:25 2015 kern.err kernel: [657433.025912] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:25 2015 kern.err kernel: [657433.030964] ieee80211 phy0: ampdu start error code: -22
[657433.075983] ieee80211 phy0: check ba result error 1
[657433.081041] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.075983] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.081041] ieee80211 phy0: ampdu start error code: -22
[657433.126061] ieee80211 phy0: check ba result error 1
[657433.131120] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.126061] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.131120] ieee80211 phy0: ampdu start error code: -22
[657433.166163] ieee80211 phy0: check ba result error 1
[657433.171239] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.166163] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.171239] ieee80211 phy0: ampdu start error code: -22
[657433.216111] ieee80211 phy0: check ba result error 1
[657433.221170] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.216111] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.221170] ieee80211 phy0: ampdu start error code: -22
[657433.256137] ieee80211 phy0: check ba result error 1
[657433.261225] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.256137] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.261225] ieee80211 phy0: ampdu start error code: -22
[657433.350787] ieee80211 phy0: check ba result error 1
[657433.355798] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.350787] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.355798] ieee80211 phy0: ampdu start error code: -22
[657433.401604] ieee80211 phy0: check ba result error 1
[657433.407121] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.401604] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.407121] ieee80211 phy0: ampdu start error code: -22
[657433.451926] ieee80211 phy0: check ba result error 1
[657433.456979] ieee80211 phy0: ampdu start error code: -22
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.451926] ieee80211 phy0: check ba result error 1
Sat Nov 14 00:10:26 2015 kern.err kernel: [657433.456979] ieee80211 phy0: ampdu start error code: -22
````

kristrev, 2015-11-17:
I got this error while using my laptop yesterday. The error log contained only the "normal" error and I had to reconnect the laptop (switch to 2.4 then back to 5) for the wifi to work again. L2 was up as the wifi driver on the laptop still believed it had connectivity and I could see the 5Ghz network when scanning. This laptop has an Intel-card.

johnnysl, 2015-11-18:
I've got a new error for a change:

400969.337083] ieee80211 phy1: cmd 0x9125=BAStream timed out
[400969.342593] ieee80211 phy1: return code: 0x1125
[400969.347236] ieee80211 phy1: timeout: 0x1125
[400969.352447] ieee80211 phy1: check ba failed execution
[400969.357631] ieee80211 phy1: ampdu start error code: -5

notnyt, 2015-11-26:
Still seeing these in the latest 10.3.0.14 builds...

[68146.882917] ieee80211 phy1: check ba result error 1
[68146.888361] ieee80211 phy1: ampdu start error code: -22
[68146.923076] ieee80211 phy1: check ba result error 1
[68146.928167] ieee80211 phy1: ampdu start error code: -22
[68146.973516] ieee80211 phy1: check ba result error 1
[68146.978709] ieee80211 phy1: ampdu start error code: -22
[68261.112329] ieee80211 phy1: check ba result error 1
[68261.117357] ieee80211 phy1: ampdu start error code: -22

johnnysl, 2015-11-26:
@notnyt: Intel n wireless card? I'm away from home for a couple of days, carrying my laptop with intel n with me. Since then i do not see the errors in my linksys anymore, so it seems to related to my intel n adapter. Fruity devices are not causing it.

mystica555, 2015-11-26:
Of note, I also have 1 intel wireless laptop in this house, and it seems to
me that only this card is causing the issues over time. The rest:
broadcom/murata/atheros work fine for a long time.

johnnysl, 2015-11-26:
Returned home today and even with the latest wireless build, i see the errors again. 
So it's 95% certain its the intel advanced n-300 wireless card that is creating the issues.

TireMeat, 2015-11-27:
I saw these errors on my test V1 happen in the middle of the night / off-hours ... do you guys have Smart Connect by any chance?   (http://www.intel.com/support/services/smartconnect/sb/CS-033108.htm)

johnnysl, 2015-11-27:
I don't. It's a HP folio 9470m, doesn't seem to have those features

notnyt, 2015-11-28:
Yes, using an intel.  I believe a 7260.  Found out that with the latest build it barely works at all on 5ghz, and lots of ba errors.

notnyt, 2015-11-28:
also this issue wasn't present on the initial revision of .14, only on the latest.

yuhhaurlin, 2015-11-28:
The latest driver is almost the same as initial 10.3.0.14.

notnyt, 2015-11-28:
@yuhhaurlin They both had the BA errors, but this particular client stopped working on 5ghz, where it worked fine earlier.

johnnysl, 2015-11-30:
Issue is still present, even in the latest version with the kernel panic fix:

[59075.207395] ieee80211 phy1: check ba result error 1
[59075.212294] ieee80211 phy1: ampdu start error code: -22

yuhhaurlin, 2015-12-01:
But it looks like it is just a warning message. The system should still work well, right?

mystica555, 2015-12-01:
Hello,

My personal experience is that if i leave the AP running for > 1 week, i
get packet loss over for example Skype calls made. Skype itself reports the
packet loss, and my roommate also reports unplayable video games via wifi
from her xboxone and her intel-wifi laptop.  It is related to AMPDU.
Something still isnt right.

Mike

yuhhaurlin, 2015-12-01:
@mystica555 Can you test latest driver? Thanks.

johnnysl, 2015-12-01:
@yuhhaurlin: before the kernel panic fix it wasn't, it sometimes needed me to take the client off the wireless and reconnect. Yesterday i did had some weird issue with my intel laptop, causing all network traffic to freeze, reconnect didn't fix that, but a reboot of the laptop did. So not sure if the two issues were related, Will report back, if I encounter the issue again.

yuhhaurlin, 2015-12-01:
@johnnysl 
1. I think this problem should be gone with latest fix. Even though it still prints out this kind of messages, it should be just temporary warning messages.
2. You mean your laptop is locked? If this is the case, it should not be the problem of mwlwifi driver.

johnnysl, 2015-12-01:
Laptop worked, but wifi "locked" each time a few seconds after starting a new tcp session.
E.g. Ssh session to the router would "hang" after typing a few characters of my login name.
Websites in the browser would hang halfway loading a page, etc. very weird. 
Intel laptop was the only device with the issue, mac and other devices were fine, so i currently have dismissed it as a laptop issue.

notnyt, 2015-12-01:
I had the same issue until I set 5ghz to VHT40.  You should try

yuhhaurlin, 2015-12-01:
It is very hard for me to check what you described.

yuhhaurlin, 2015-12-01:
@notnyt You still encounter the same problem with latest fix?

johnnysl, 2015-12-01:
@yuhhaurlin i totally understand. That is why i didn't bring it up at first. For the moment, just ignore it.

notnyt, 2015-12-01:
@yuhhaurlin  yes

````
[86556.563166] ieee80211 phy1: check ba result error 1
[86556.568228] ieee80211 phy1: ampdu start error code: -22
[86556.612695] ieee80211 phy1: check ba result error 1
[86556.617686] ieee80211 phy1: ampdu start error code: -22
[86556.652789] ieee80211 phy1: check ba result error 1
[86556.657798] ieee80211 phy1: ampdu start error code: -22
[86556.702617] ieee80211 phy1: check ba result error 1
[86556.707768] ieee80211 phy1: ampdu start error code: -22
[86556.743119] ieee80211 phy1: check ba result error 1
[86556.748356] ieee80211 phy1: ampdu start error code: -22
[86556.802871] ieee80211 phy1: check ba result error 1
[86556.807920] ieee80211 phy1: ampdu start error code: -22
[86556.852921] ieee80211 phy1: check ba result error 1
[86556.858050] ieee80211 phy1: ampdu start error code: -22
[86556.893040] ieee80211 phy1: check ba result error 1
[86556.898015] ieee80211 phy1: ampdu start error code: -22
[86556.932622] ieee80211 phy1: check ba result error 1
[86556.937647] ieee80211 phy1: ampdu start error code: -22
[86556.992768] ieee80211 phy1: check ba result error 1
[86556.997756] ieee80211 phy1: ampdu start error code: -22
[86557.032732] ieee80211 phy1: check ba result error 1
[86557.037751] ieee80211 phy1: ampdu start error code: -22
[86557.072902] ieee80211 phy1: check ba result error 1
[86557.077925] ieee80211 phy1: ampdu start error code: -22
[86557.132865] ieee80211 phy1: check ba result error 1
[86557.137849] ieee80211 phy1: ampdu start error code: -22
[86557.172946] ieee80211 phy1: check ba result error 1
[86557.177966] ieee80211 phy1: ampdu start error code: -22
[86557.213302] ieee80211 phy1: check ba result error 1
[86557.218302] ieee80211 phy1: ampdu start error code: -22
[86557.252922] ieee80211 phy1: check ba result error 1
[86557.257927] ieee80211 phy1: ampdu start error code: -22

````

yuhhaurlin, 2015-12-01:
@notnyt It looks like your case is that AP can't establish AMPDU with your client (only for VHT80?). The client can work with other AP via VHT80.

notnyt, 2015-12-01:
When I enable VHT80, no clients can connect properly.  Listing stations shows they only use 20 and 40mhz channels and speed is incredibly slow or the clients get disassociated.  Those errors were generated when the AP was operating in VHT40 mode.

yuhhaurlin, 2015-12-01:
@notnyt Any working firmware on your device?

notnyt, 2015-12-01:
With VHT80?  Not sure.

johnnysl, 2015-12-01:
Maybe try stock firmware again?

yuhhaurlin, 2015-12-01:
@notnyt It looks like the image can't work on your device. it is not specific driver problem.

johnnysl, 2015-12-03:
@yuhhaurlin: those errors are definitively not just cosmetic. When i grab a wireless device after it had long been unused, it sometimes is not able to get any data across the wifi. Need to disconnect and reconnect to get it working. At the same time that happens, the log fills with these errors.

Some findings:
1. It seems to get worse after a longer uptime (uptime of 3 days now)
2. Its not only confined to my intel nic based laptop, ipad air 2 has it as well

Could there be another variable not properly initialized somewhere?

tangsoft, 2015-12-06:
I am not sure if what I got on my v1 with latest drive when it is DLNA streaming video on iPad and iPhone. And sometimes random reboot as well.

[ 4790.518054] ieee80211 phy1: failed to reallocate TX buffer
[28229.515899] ieee80211 phy1: failed to reallocate TX buffer
[28229.521519] ieee80211 phy0: failed to reallocate TX buffer

[72 more replies]

closed 2016-03-30 by yuhhaurlin