kaloz/mwlwifi #42: Support for WLAN IEs

dpnwr, 2015-11-02:
Hi,

I have configured time zone for hostapd but it did not show up in the beacons. The debugging the hostapd binary, I observe that all the configured parameters and IEs are being passed to driver APIs and also seen in fwcmd.c: mwl_fwcmd_parse_beacon(...).

Is there any restriction within Marwell's firmware that is does support only selected IEs? Do you have any documentation/link for supported hostcommands and parameters (in addition to the one listed in dev.h and fwcmd.c)?

Best Regards,
Deepak

yuhhaurlin, 2015-11-03:
I  think you need to modify the driver to keep IEs that you want to show up in the beacon (mwl_fwcmd_parse_beacon()) and set these IEs to firmware via mwl_fwcmd_set_ies().

No. File "hostcmd.h" is what you can get for open source.

dpnwr, 2015-11-03:
Thanks @yuhhaurlin , I am upgrading the driver for required IEs.

closed 2015-11-06 by yuhhaurlin