kaloz/mwlwifi #44: 10.0.3.14 Poor Wifi performance on FireTV box

davidc502, 2015-11-14:
Installed wifi driver 10.0.3.14, but it has problems with my FireTV boxes.... Nothing in the logs, but has very poor performance, and can't connect to my media server on my LAN. WireShark shows nothing, but ping results do show latency. As soon as I revert back to 10.3.0.3 they (firetv) work normally again.

On wifi driver 10.0.3.14

these are some of the better ping times... many go over 100ms and or time out (bouncing)
Reply from 192.168.1.130: bytes=32 time=15ms TTL=64
Reply from 192.168.1.130: bytes=32 time=20ms TTL=64
Reply from 192.168.1.130: bytes=32 time=15ms TTL=64
Reply from 192.168.1.130: bytes=32 time=24ms TTL=64
Reply from 192.168.1.130: bytes=32 time=29ms TTL=64
Reply from 192.168.1.130: bytes=32 time=12ms TTL=64
Reply from 192.168.1.130: bytes=32 time=17ms TTL=64
Reply from 192.168.1.130: bytes=32 time=14ms TTL=64
Reply from 192.168.1.130: bytes=32 time=20ms TTL=64
Reply from 192.168.1.130: bytes=32 time=13ms TTL=64
Reply from 192.168.1.130: bytes=32 time=18ms TTL=64

On wifi driver 10.3.0.3

Reply from 192.168.1.130: bytes=32 time=1ms TTL=64
Reply from 192.168.1.130: bytes=32 time=1ms TTL=64
Reply from 192.168.1.130: bytes=32 time=1ms TTL=64
Reply from 192.168.1.130: bytes=32 time=3ms TTL=64
Reply from 192.168.1.130: bytes=32 time=1ms TTL=64
Reply from 192.168.1.130: bytes=32 time=1ms TTL=64
Reply from 192.168.1.130: bytes=32 time=1ms TTL=64
Reply from 192.168.1.130: bytes=32 time=1ms TTL=64
Reply from 192.168.1.130: bytes=32 time=2ms TTL=64

Since FireTV is Android I figured more people would be having problems, but so far that hasn't been the case.

davidc502, 2015-11-14:
Just a FYI --  I've tried 10.0.3.12 and 10.0.3.13, and they have the same results.

yuhhaurlin, 2015-11-25:
Can you use iperf to get throughput for each of them? Thanks.

davidc502, 2015-11-27:
Side loaded and installed iperf onto FireTV, and seems to work without any issues. Currently running RC3, and will download/install latest trunk this afternoon (Compare stats). Should be able to show the results then.

davidc502, 2015-11-27:
Installed Trunk again today and ran some tests...   Here are the results.  I'm seeing a slowdown on the new driver, but even with the slow down streaming video should work, but yet it doesn't. Unknown reason why.

I've tried running the client as the server and vice versa -- Same results.

192.168.1.108 = FireTV box (iperf running in server mode)
192.168.1.2 = Client TV

CHAOS CALMER (15.05-rc3, r46163)
C:\Program Files\iperf>iperf -c 192.168.1.108 -i 1 -p 5001
------------------------------------------------------------
Client connecting to 192.168.1.108, TCP port 5001
TCP window size: 63.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.1.2 port 49176 connected with 192.168.1.108 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 1.0 sec  8.62 MBytes  72.4 Mbits/sec
[  3]  1.0- 2.0 sec  8.38 MBytes  70.3 Mbits/sec
[  3]  2.0- 3.0 sec  9.12 MBytes  76.5 Mbits/sec
[  3]  3.0- 4.0 sec  8.62 MBytes  72.4 Mbits/sec
[  3]  4.0- 5.0 sec  8.38 MBytes  70.3 Mbits/sec
[  3]  5.0- 6.0 sec  9.50 MBytes  79.7 Mbits/sec
[  3]  6.0- 7.0 sec  9.12 MBytes  76.5 Mbits/sec
[  3]  7.0- 8.0 sec  9.12 MBytes  76.5 Mbits/sec
[  3]  8.0- 9.0 sec  6.75 MBytes  56.6 Mbits/sec
[  3]  9.0-10.0 sec  8.62 MBytes  72.4 Mbits/sec
[  3]  0.0-10.0 sec  86.4 MBytes  72.4 Mbits/sec

UDP
C:\Program Files\iperf>iperf.exe -c 192.168.1.108 -p 2000 -u -i 1 -b 1000m
------------------------------------------------------------
Client connecting to 192.168.1.108, UDP port 2000
Sending 1470 byte datagrams
UDP buffer size: 63.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.1.2 port 57641 connected with 192.168.1.108 port 2000
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 1.0 sec  63.1 MBytes   529 Mbits/sec
[  3]  1.0- 2.0 sec  63.7 MBytes   534 Mbits/sec
[  3]  2.0- 3.0 sec  64.3 MBytes   540 Mbits/sec
[  3]  3.0- 4.0 sec  64.0 MBytes   537 Mbits/sec
[  3]  4.0- 5.0 sec  60.0 MBytes   504 Mbits/sec
[  3]  5.0- 6.0 sec  64.3 MBytes   539 Mbits/sec
[  3]  6.0- 7.0 sec  64.2 MBytes   539 Mbits/sec
[  3]  7.0- 8.0 sec  63.6 MBytes   534 Mbits/sec
[  3]  8.0- 9.0 sec  60.0 MBytes   504 Mbits/sec
[  3]  9.0-10.0 sec  59.7 MBytes   501 Mbits/sec
[  3]  0.0-10.0 sec   627 MBytes   526 Mbits/sec
[  3] Sent 447273 datagrams
[  3] WARNING: did not receive ack of last datagram after 10 tries.

Bleeding Edge -- Trunk
DESIGNATED DRIVER (Bleeding Edge, r47642)

Driver ---
root@OpenWrt:~# strings /lib/modules/*.*/mwlwifi.ko | grep "^10.3"
10.3.0.12
10.3.0.12

C:\Program Files\iperf>iperf -c 192.168.1.108 -i 1 -p 5001
------------------------------------------------------------
Client connecting to 192.168.1.108, TCP port 5001
TCP window size: 63.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.1.2 port 49829 connected with 192.168.1.108 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 1.0 sec  6.50 MBytes  54.5 Mbits/sec
[  3]  1.0- 2.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3]  2.0- 3.0 sec  7.00 MBytes  58.7 Mbits/sec
[  3]  3.0- 4.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3]  4.0- 5.0 sec  6.75 MBytes  56.6 Mbits/sec
[  3]  5.0- 6.0 sec  7.00 MBytes  58.7 Mbits/sec
[  3]  6.0- 7.0 sec  6.88 MBytes  57.7 Mbits/sec
[  3]  7.0- 8.0 sec  6.38 MBytes  53.5 Mbits/sec
[  3]  8.0- 9.0 sec  6.75 MBytes  56.6 Mbits/sec
[  3]  9.0-10.0 sec  7.12 MBytes  59.8 Mbits/sec
[  3]  0.0-10.0 sec  67.8 MBytes  56.7 Mbits/sec

C:\Program Files\iperf>iperf.exe -c 192.168.1.108 -p 2000 -u -i 1 -b 1000m
------------------------------------------------------------
Client connecting to 192.168.1.108, UDP port 2000
Sending 1470 byte datagrams
UDP buffer size: 63.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.1.2 port 63689 connected with 192.168.1.108 port 2000
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 1.0 sec  60.0 MBytes   503 Mbits/sec
[  3]  1.0- 2.0 sec  61.9 MBytes   519 Mbits/sec
[  3]  2.0- 3.0 sec  62.0 MBytes   520 Mbits/sec
[  3]  3.0- 4.0 sec  62.2 MBytes   522 Mbits/sec
[  3]  4.0- 5.0 sec  64.0 MBytes   537 Mbits/sec
[  3]  5.0- 6.0 sec  63.7 MBytes   535 Mbits/sec
[  3]  6.0- 7.0 sec  63.7 MBytes   535 Mbits/sec
[  3]  7.0- 8.0 sec  62.1 MBytes   521 Mbits/sec
[  3]  8.0- 9.0 sec  61.7 MBytes   518 Mbits/sec
[  3]  9.0-10.0 sec  63.9 MBytes   536 Mbits/sec
[  3]  0.0-10.0 sec   625 MBytes   524 Mbits/sec
[  3] Sent 445902 datagrams
[  3] WARNING: did not receive ack of last datagram after 10 tries.

notnyt, 2015-11-27:
@davidc502  you're not running iperf correctly in UDP mode.  The server doesn't listen for UDP by default, you need to run it with -u as well.  You can see it's not getting responses from the server.

That said, you should just use iperf*3* in tcp mode, it's much more informative.

davidc502, 2015-11-28:
Can't use iperf3 because none of the .apk packages for Android are iperf3 compatible. It's my understanding that 3 is not backward compatible.

notnyt, 2015-11-28:
https://play.google.com/store/apps/details?id=net.he.networktools&hl=en

yuhhaurlin, 2015-11-28:
How about TCP throught you got for 10.3.0.3?

yuhhaurlin, 2015-11-28:
For UDP, you should use "iperf -s -u -i 1" for server side and check throughput on server side.

yuhhaurlin, 2015-11-28:
I think you can help to get UDP and TCP throughput for 10.3.0.3 and latest driver first.

davidc502, 2015-11-28:
notnyt
I downloaded and installed that network tool (It was my first choice), and couldn't get it to work on the FireTV. The GUI doesn't respond to the FireTV remote commands :*(

davidc502, 2015-11-28:
yuhhaurlin,

Below are the speeds using 10.3.0.3 which is which RC3. 
root@OpenWrt:~# strings /lib/modules/*.*/mwlwifi.ko | grep "^10.3"
10.3.0.3

C:\Program Files\iperf>iperf -c 192.168.1.108 -i 1 -p 5001

[ 3] local 192.168.1.2 port 49176 connected with 192.168.1.108 port 5001
[ ID] Interval Transfer Bandwidth
[ 3] 0.0- 1.0 sec 8.62 MBytes 72.4 Mbits/sec
[ 3] 1.0- 2.0 sec 8.38 MBytes 70.3 Mbits/sec
[ 3] 2.0- 3.0 sec 9.12 MBytes 76.5 Mbits/sec
[ 3] 3.0- 4.0 sec 8.62 MBytes 72.4 Mbits/sec
[ 3] 4.0- 5.0 sec 8.38 MBytes 70.3 Mbits/sec
[ 3] 5.0- 6.0 sec 9.50 MBytes 79.7 Mbits/sec
[ 3] 6.0- 7.0 sec 9.12 MBytes 76.5 Mbits/sec
[ 3] 7.0- 8.0 sec 9.12 MBytes 76.5 Mbits/sec
[ 3] 8.0- 9.0 sec 6.75 MBytes 56.6 Mbits/sec
[ 3] 9.0-10.0 sec 8.62 MBytes 72.4 Mbits/sec
[ 3] 0.0-10.0 sec 86.4 MBytes 72.4 Mbits/sec

davidc502, 2015-11-28:
yuhhaurlin,
For UDP, you should use "iperf -s -u -i 1" for server side and check throughput on server side.

I will try this tomorrow morning and post the results.

davidc502, 2015-11-28:
Results below ---

[image]

yuhhaurlin, 2015-11-30:
It looks like throughput is almost the same between 10.3.0.3 and latest driver. Can you let me know what kind of traffic for FireTV?

yuhhaurlin, 2015-11-30:
For issue #49, I figure out the driver had problem to deal BA with VI. I don't know if it is related to the problem you encountered. Can you help to try the fix for issue #49? Thanks.

davidc502, 2015-12-01:
Not sure if it is related (don't think so) because I don't remember seeing any kernel issue.

I've used Wireshark and watched the traffic and it's TCP. DLNA may be used to kick of the streaming service.  

I'll take a closer look at 49, the next time I do some testing.

yuhhaurlin, 2015-12-01:
Thanks. Hopefully the latest driver can work for you.

davidc502, 2015-12-01:
Just installed --  DESIGNATED DRIVER (Bleeding Edge, r47665)
root@OpenWrt:~# strings /lib/modules/*.*/mwlwifi.ko | grep "^10.3"
10.3.0.14
10.3.0.14

I am seeing this in the kernel log quite a bit. I don't know if this is a indication of a issue or not.

[   27.431823] ieee80211 phy0: change: 0x40
[   27.611809] ieee80211 phy0: change: 0x40
[   27.791835] ieee80211 phy0: change: 0x40
[   27.961891] ieee80211 phy0: change: 0x40
[   28.131846] ieee80211 phy0: change: 0x40
[   28.311997] ieee80211 phy0: change: 0x40
[   28.491834] ieee80211 phy0: change: 0x40
[   28.553910] ieee80211 phy0: change: 0x100
[   28.559128] ieee80211 phy0: change: 0x100
[   28.564781] ieee80211 phy0: change: 0x40
[   28.741819] ieee80211 phy0: change: 0x40
[   28.921810] ieee80211 phy0: change: 0x40
[   29.101786] ieee80211 phy0: change: 0x40
[   29.281789] ieee80211 phy0: change: 0x40
[   29.461785] ieee80211 phy0: change: 0x40
[   29.641788] ieee80211 phy0: change: 0x40
[   29.821785] ieee80211 phy0: change: 0x40
[   30.001792] ieee80211 phy0: change: 0x40
[   30.182310] ieee80211 phy0: change: 0x40
[   30.351982] ieee80211 phy0: change: 0x40

At any rate, still the same problem... Can't stream local videos, though I can stream Netflix no problem. If I plug in a ethernet cable, no problems streaming local. However, refuses to stream local through wifi.

yuhhaurlin, 2015-12-01:
10.3.0.3 is all right?

davidc502, 2015-12-01:
Yeah, no problems with 10.3.0.3

yuhhaurlin, 2015-12-01:
ieee80211 phy0: change: 0x40 => It means change channel.
ieee80211 phy0: change: 0x100 => It means change idle.

I am curious why AP will change channel frequently?

yuhhaurlin, 2015-12-01:
[   28.119687] ieee80211 phy0: 2G: disable
[   28.123571] ieee80211 phy0: 5G: enable
[   28.127339] ieee80211 phy0: TX: 2 antennas
[   28.131463] ieee80211 phy0: RX: 2 antennas
[   28.292886] ieee80211 phy0: change: 0xffffffff
[   28.449648] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   28.456760] ieee80211 phy0: change: 0x100
[   28.461782] ieee80211 phy0: change: 0x60
[   28.653616] cfg80211: Found new beacon on frequency: 5745 MHz (Ch 149) on phy0
[   28.711214] ieee80211 phy0: change: 0x40
[   28.762736] cfg80211: Found new beacon on frequency: 5765 MHz (Ch 153) on phy0
[   28.961216] ieee80211 phy0: change: 0x60
[   29.102828] ieee80211 phy0: change: 0x100
[   29.160740] ieee80211 phy0: change: 0x100
[   29.165763] ieee80211 phy0: change: 0x62
[   29.336687] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   29.386455] device wlan0 entered promiscuous mode
[   29.392095] br-lan: port 3(wlan0) entered forwarding state
[   29.398946] br-lan: port 3(wlan0) entered forwarding state
[   31.391209] br-lan: port 3(wlan0) entered forwarding state
[   88.951214] cfg80211: Verifying active interfaces after reg chan

When hostapd is running, it will set change. So you will see change channel 0x40 one time.

yuhhaurlin, 2015-12-01:
Which band you used? 2.4g or 5g?

davidc502, 2015-12-01:
That's especially interesting when I have both Wifi interfaces set to a specific channel (NO Auto).

yuhhaurlin, 2015-12-01:
Can you try to test with 2.4g and 5g? Both of them have problem? For 2.4g, it is better to test with 20 Mhz. Phy0 is 2.4g?

johnnysl, 2015-12-01:
Couldn't this issue be related to #36?

yuhhaurlin, 2015-12-01:
@johnnysl I don't think so. For issue #36, notnyt mentioned "I've seen this as well on 10.3.0.3". But davidc502 reported this problem is all right on 10.3.0.3. I wonder if anyone had encountered the same problem?

johnnysl, 2015-12-01:
I don't, but then again, i don't own any streaming devices that strream over wireless.
Does monitor mode already work again with this driver? Maybe we can dump the traffic, and see what is going wrong?

davidc502, 2015-12-01:
yuhhaurlin,

FireTV has been on 5Ghz... Just tested 2.4Ghz, and the video stream is working.

yuhhaurlin, 2015-12-02:
@davidc502 So this problem is gone?

davidc502, 2015-12-03:
Yes, on 2.4Ghz the problem is gone.

yuhhaurlin, 2015-12-03:
How about 5G? If 2.4G is all right, 5G should have no problem too.

davidc502, 2015-12-03:
That's the problem, I've had both FireTV boxes on 5Ghz, and when I moved them to 2.4 they began to work. If I move them back to 5Ghz, they won't work properly again.

davidc502, 2015-12-05:
Just reading through the other problems, I believe this may be related to #36 

I say that because DLNA streaming is based on initially broadcasts and then unicasts.

davidc502, 2015-12-05:
Monitoring the logs -- these are new

[295222.247350] ieee80211 phy0: check ba result error 1
[295222.252420] ieee80211 phy0: ampdu start error code: -22
[295222.287418] ieee80211 phy0: check ba result error 1
[295222.292498] ieee80211 phy0: ampdu start error code: -22
[295222.338479] ieee80211 phy0: check ba result error 1
[295222.344103] ieee80211 phy0: ampdu start error code: -22
[295222.397572] ieee80211 phy0: check ba result error 1
[295222.402742] ieee80211 phy0: ampdu start error code: -22
[295222.458392] ieee80211 phy0: check ba result error 1
[295222.464225] ieee80211 phy0: ampdu start error code: -22
[295222.546289] ieee80211 phy0: check ba result error 1
[295222.551359] ieee80211 phy0: ampdu start error code: -22
[295222.606321] ieee80211 phy0: check ba result error 1
[295222.611682] ieee80211 phy0: ampdu start error code: -22
[295222.666418] ieee80211 phy0: check ba result error 1
[295222.671613] ieee80211 phy0: ampdu start error code: -22
[295222.706319] ieee80211 phy0: check ba result error 1
[295222.711397] ieee80211 phy0: ampdu start error code: -22
[295222.766317] ieee80211 phy0: check ba result error 1
[295222.771379] ieee80211 phy0: ampdu start error code: -22
[295222.806840] ieee80211 phy0: check ba result error 1
[295222.812042] ieee80211 phy0: ampdu start error code: -22
[295222.856561] ieee80211 phy0: check ba result error 1
[295222.861649] ieee80211 phy0: ampdu start error code: -22
[295222.906708] ieee80211 phy0: check ba result error 1
[295222.911806] ieee80211 phy0: ampdu start error code: -22
[295222.946667] ieee80211 phy0: check ba result error 1
[295222.951756] ieee80211 phy0: ampdu start error code: -22
[295222.986767] ieee80211 phy0: check ba result error 1
[295222.991968] ieee80211 phy0: ampdu start error code: -22
[295223.026735] ieee80211 phy0: check ba result error 1
[295223.031825] ieee80211 phy0: ampdu start error code: -22
[295223.066948] ieee80211 phy0: check ba result error 1
[295223.072050] ieee80211 phy0: ampdu start error code: -22
[295223.106750] ieee80211 phy0: check ba result error 1
[295223.111898] ieee80211 phy0: ampdu start error code: -22

yuhhaurlin, 2015-12-08:
Can you confirm your problem only happened for "server on 5g and client on 2.4g"? BTW, also confirm 10.3.0.3 is all right, but 10.3.0.14 had problem.

yuhhaurlin, 2015-12-08:
Also if server and client are on the same band, the problem is gone.

davidc502, 2015-12-08:
1. Confirmed 10.3.0.3 works fine. 
2. Client =FireTV (wireless client)  DLNA streaming Server=workstation (Wired Ethernet)
 If FireTV is on 2.4Ghz it works okay, but if FireTV is on 5Ghz band it does not. Server remains on Ethernet at all times.

yuhhaurlin, 2015-12-08:
Can you use iperf to test ethernet to 2.4g and ethernet to 5g and get throughput number? After test is done, please cat "/sys/kernel/debug/ieee80211/phy0 or phy1/mwlwifi/sta" and check what is the setting for amsdu and ampdu for the client You had told me this is TCP application, so I think you only need to get iperf TCP throughput number.

johnnysl, 2015-12-08:
@davidc502, did you read my reply on OpenWrt that the build you mentioned there did not include the kernel panic patch yet? Yuhhaurlin just uploaded a new patch last night as well. Might want to use the absolute latest version before starting the testing.

davidc502, 2015-12-08:
Is the new patch included in today's trunk build found on the wiki?

davidc502, 2015-12-09:
Installed DESIGNATED DRIVER (Bleeding Edge, r47745)
root@OpenWrt:~# strings /lib/modules/*.*/mwlwifi.ko | grep "^10.3"
10.3.0.14
10.3.0.14

Tested both 2.4Ghz and 5Ghz with FireTV. There has been a change.

No issues streaming LAN DLNA video with FireTV on **5Ghz** band (problem solved and works great)
However! Streaming LAN DLNA video with FireTV on **2.4Ghz** band no long works. To be more specific, the video streams very poorly (choppy and stutters).

So it appears to be the opposite issue of the previous build that was installed.

yuhhaurlin, 2015-12-09:
Your test result is the same as #52 now, For 2.4g, I suggest that you do iperf throughput test to make sure if the channel is good. BTW, can you do stress test on 5g to see if you will encounter the same problem  as reported by #52 which will let memory is gone. If not, I think both this issue and #52 should be closed.

davidc502, 2015-12-09:
I should have time to do some testing tomorrow night.. -- Thanks

yuhhaurlin, 2015-12-10:
This issue is closed.

closed 2015-12-10 by yuhhaurlin