kaloz/mwlwifi #48: WDS Support Required (not MESH!!!)

BrainSlayer, 2015-11-28:
since the WDS related was closed with bad reason, i simply reopen it here

WDS is currently not supported and it would be nice if it would be supported (or is not working at all) and WDS has nothing todo with Mesh. its about WDS AP Bridging, which does allow to associate 4 addr frame supporting clients (WDS STA), but standard clients as well. this is a important feature for reliable dedicated network bridges. 

so do not close with reason like "mesh is supported" mesh is something else and a different protocol. 
or lets face it. nobody will use mesh if he wants to connect 2 devices to build a network bridge.

and if it helps in competition. all other chipset vendors do support WDS modes and i run into troubles to explain DD-WRT used (by the way, i'm the maintainer of this firmware) why the Linksys WRT1900ac is not capable of doing WDS, if all other Devices from Atheros / Qualcomm and Broadcom do support these operation modes. i hope you understand

yuhhaurlin, 2015-11-30:
No plan to support WDS now.

BrainSlayer, 2015-11-30:
so why is it supported with the closed source marvell sdk? i mean it is supported my marvell. its just not supported in the mac80211 driver. so the chipset isnt the problem here and i dont think that the firmware is a problem here

thagabe, 2015-11-30:
@yuhhaurlin I second this. It seems useful for a home network. Maybe after we nail down all the bugs on the current code we could work on integrating WDS.

yuhhaurlin, 2015-12-01:
Priority would be lower. Thanks.

suihkulokki, 2015-12-01:
I did have wrt1900ac as WDS AP, with DIR-825 as WDS STA. I had to reverse roles because of  bug #36 .I didn't manage to make wrt1900ac work as WDS STA, therefor it is now doing NAT as regular STA.  So I think the other half (WDS AP) already works, and only WDS STA is missing. For priority, #36 is still more important for me.

Gazoo, 2015-12-01:
@yuhhaurlin +1 from me as well. This is the only thing that keeps me from using the open firmware. Many thanks for all your hard work.

gufus, 2015-12-01:
@yuhhaurlin 

Yes.

WDS is **important**.

gufus, 2015-12-10:
@yuhhaurlin 

Any update on WDS?

BrainSlayer, 2015-12-10:
you wont get any reply here. he said already it has low priority (which means translated, it will likelly never happen). so he will stay silent now. otherwise he would just create useless hope that this driver can be something like a wireless driver for professional applications. it can also be that this feature requires a firmware change and it might be that he isnt able todo any modifications on it.

johnnysl, 2015-12-12:
@BrainSlayer: the .15 code has several references to WDS added. Did you take a look at that already?

gufus, 2015-12-31:
@yuhhaurlin

We still need WDS

Why?

It's a MUCH BETTER than mesh. IMHO

suihkulokki, 2016-01-04:
There is two different usecases for WDS

1. Wireless bridging, as described as 4addr mode in https://wiki.openwrt.org/doc/howto/clientmode
- For bridging, mesh is not useful. Or if it is, there no documentation/example howto do it.
2. As wireless repeater. 
- For repeating, mesh might be better than WDS

I think most people here want usecase 1 (at least I do), and thus are unhappy that mesh is suggested as alternative.

gufus, 2016-02-12:
@yuhhaurlin

We still need WDS

Mrengles, 2016-02-22:
Yes @gufus we deffinlitly still need full WDS support. Like @BrainSlayer mentioned MESH alone won't do the trick.

Why aren't  the mac80211 opensource driver and closed (proprietary) driver not released in tandem/sync? After all opening up the driver and allowing us to test and bug-fix is the whole point. The **open community** could really **help** speed up the entire development process.

Chadster766, 2016-02-22:
Why does mac80211 need to be customized?

BrainSlayer, 2016-03-27:
he is talking about the mac80211 driver not mac80211 itself. marvell has 2 drivers. one is mac80211 (mwlwifi) and the closed source driver used by most vendors. bad for us is that the closed source driver does support WDS, the opensource driver does not

Chadster766, 2016-03-27:
Ah thanks for the clarification

mrefky, 2016-04-11:
WDS is perfect for home users.......

GamemanAdvKid, 2016-04-15:
made this account just to respond to this specific ticket. 

WDS is a necessary function for us home users. We need it not want it. Like brainslayer said, it creates a **real bridge**. 

This is an important function for when people want to hook up multiple computers without buying over 200 dollars worth of wifi adapter to hook up all their computers or just want to have secondary routers handling NAS with other routers deal with routing features and functions. And a REAL bridge is necessary for things like this to work properly. 

Client bridge, while it may help provide the internet to the computer connected to it, cannot handle the demands we require from it.

yuhhaurlin, 2016-04-15:
Busy for other jobs now. I will check this function later. Basically it looks like only some configurations needed to let OpenWrt 4 address WDS work.

mrefky, 2016-04-15:
Thanks, that is good news

mrefky, 2016-04-22:
@yuhhaurlin any news related to WDS support?

yolanjoy, 2016-04-28:
Adding WDS support will be awesome and a great help for home users. I've E4220v2 routers and would like to bridge them with WRT1200ac. 
Please help.  Thanks.

chinawrj, 2016-04-29:
@yolanjoy @mrefky  I think 4addr mode supported by OpenWrt should help on WDS. I already did it

mrefky, 2016-04-29:
Thanks for the reply @yolanjoy, however this is not working as expected in WRT1200AC. It is working perfectly using TP-Link TL-WDR4300 but not using WRT1200AC

chinawrj, 2016-05-01:
Hi,
  I have tried openwrt trunk on WRT1900ac v2. AP WDS works. BTW, I use
wdr1043 as STA WDS

Thanks

Gazoo, 2016-05-01:
@chinawrj Still doesn't work for me :(

yolanjoy, 2016-05-27:
@yuhhaurlin - Any update about WDS support please?

yuhhaurlin, 2016-05-27:
I will add it later.

gufus, 2016-05-27:
Please do soon..

Regarrds

yuhhaurlin, 2016-05-31:
Do not know if 10.3.0.17-20160531-1 is working for WDS?

yolanjoy, 2016-06-24:
Has anyone tested WDS on WRT1200ac?

mrefky, 2016-07-13:
Yes I have tested it and WDS is not working properly.

yuhhaurlin, 2016-07-14:
I will tries to test it later.

gufus, 2016-07-16:
@yuhhaurlin 
So, is it working?

yolanjoy, 2016-09-02:
Has anyone tested WDS yet?

Thanks

thagabe, 2016-09-02:
@yuhhaurlin ping

gufus, 2016-09-02:
Doesent work with  https://github.com/kaloz/mwlwifi/commit/d48847cc83954930fdcd98818bbf33e75707b8e1

Haven't tested WDS with https://github.com/kaloz/mwlwifi/commit/af606563453c819fac156faf2b15b9caef844329

Tw0Sp00ns, 2016-10-18:
This following is a list of supported modes for the WRT1900ACS running LEDE
**Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor**

I don't see mesh or wds in there. Does the driver actually support mesh and/or wds?

gufus, 2016-10-23:
@yuhhaurlin 

Is WDS fixed and working?

reference to fix:
http://marc.info/?l=linux-wireless&m=147560260029581&w=2

yuhhaurlin, 2016-10-24:
I will check it but priority would be lower. BTW, you can use "routed client with relayd" (https://wiki.openwrt.org/doc/recipes/relayclient) to connect two devices.

BrainSlayer, 2016-10-24:
according to testers it works inbetween marvell chipsets.  but you cannot connect other chipsets with wds.

gufus, 2016-10-24:
@yuhhaurlin

I'll try "routed client with relayd"  soon as I can

Just to refresh your memory, this is working here.

It's not WDS :(

But it works

-----
My setup is 2 wrt1900ac

1 unit is in AP mode, both (2g and 5g up)

2g
100% Mode: Master | SSID: gypsynet
BSSID: 00:25:9C:13:34:6A | Encryption: WPA2 PSK (CCMP)
Channel: 6 (2.437 GHz) | Tx-Power: 30 dBm
Signal: -30 dBm | Noise: -67 dBm
Bitrate: 229.5 Mbit/s | Country: CA

5g
82% Mode: Master | SSID: gypsynet_5g
BSSID: 00:25:9C:13:34:6B | Encryption: WPA2 PSK (CCMP)
Channel: 161 (5.805 GHz) | Tx-Power: 30 dBm
Signal: -52 dBm | Noise: -88 dBm
Bitrate: 1053.0 Mbit/s | Country: CA

wifi driver on AP unit 3.18.20+10.3.0.3-2015..4-1

2 unit is in Client Bridge (routed) mode

wifi driver on Client Bridge unit 10.3.1.0-20160917

All wifi settings the same on both unit(s)

The Client Bridge works great

Rates on Client Bridge both 2g and 5g
http://www.gypsy-designs.com/4.jpg

Tw0Sp00ns, 2016-10-26:
@gufus I have a similar setup to yours. And I managed to get WDS working with the kaloz image. Do try that, if you haven't already.

BrainSlayer, 2016-10-26:
@gufus: the client bridge mode in dd-wrt uses relayd :-)
@Tw0Sp00ns : read what i said. it works only in between marvell devices. you cannot connect to a atheros / qca device with wds. so its incompatible to all other chipsets on the market

BrainSlayer, 2016-10-26:
@gufus & @Tw0Sp00ns : i revalidated everything today. wds sta / wds ap does indeed work on mwlwifi. this issue can be closed now

closed 2016-10-26 by BrainSlayer

cowwoc, 2017-11-30:
@BrainSlayer Thank you. Can you please clarify what setup instructions we should follow for Marvel? Is this applicable? https://www.dd-wrt.com/wiki/index.php/WDS#Qualcomm_Atheros_Based_Devices

Are the `Virtual Interfaces`, `Short Preamble` and `Short  GI` applicable for Marvel?

BrainSlayer, 2017-11-30:
yes all. the setup is identical to qca devices. consider that short preamble is only relevant to 802.11b/g networks. 
also consider that wds will not work for the wrt3200acm. just for the older devices. the wrt3200acm has a different chipset which does not support wds (yet). marvell is working on this