kaloz/mwlwifi #49: Frequent kernel panics with 10.3.0.14 and an Intel 7260 (AC) wifi card

not-the-nrc, 2015-11-28:
I have a WRT1900AC (version 1) that is currently running OpenWrt Designated Driver r47665.  This includes kmod-mwlwifi_3.18.23+10.3.0.14-20151126-1.

I have an Acer laptop running Windows 10 with an Intel Dual Band Wireless-AC 7260 (driver version 17.15.0.5 - which is the most recent driver from Windows Update).

With the laptop connected to either the 2.4 or 5 GHz network, the router will frequently encounter a kernel panic and reboot.  This occurs with just casual browsing using Firefox.

With the laptop connected to the 2.4 GHz network, I managed to capture the following kernel panic by using a desktop PC with a wired connection to the router and a "cat /dev/kmsg" (apologies in advance for the horrible formatting):

1,544,285702386,-;Unable to handle kernel paging request at virtual address 0000258c
1,545,285709651,-;pgd = c0004000
1,546,285712437,-;[0000258c] *pgd=00000000
0,547,285716072,-;Internal error: Oops: 17 [#1] SMP ARM
4,548,285720803,c;Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE armada_thermal xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY thermal_sys slhc nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack libcrc32c iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt fuse sch_teql sch_tbf sch_sfq sch_red sch_prio sch_pie sch_netem sch_htb sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic
4,549,285793258,+; act_police act_ipt act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress tmp421 pwm_fan hwmon i2c_mv64xxx i2c_dev mwlwifi mac80211 cfg80211 compat ledtrig_usbdev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables msdos ifb vfat fat ntfs hfsplus nls_utf8 regmap_spi regmap_mmio usb_storage leds_tlc59116 regmap_i2c i2c_core xhci_plat_hcd xhci_pci xhci_hcd ohci_platform ohci_hcd
4,550,285863999,+; sata_mv libata ehci_orion ehci_platform ehci_hcd sd_mod scsi_mod gpio_button_hotplug f2fs ext4 jbd2 mbcache btrfs xor raid6_pq usbcore nls_base usb_common crc32c_generic crypto_hash
4,551,285880492,-;CPU: 1 PID: 369 Comm: kworker/u4:2 Not tainted 3.18.23 #1
4,552,285887039,-;Workqueue: phy0 ieee80211_iface_work [mac80211]
4,553,285892668,-;task: cfa11c00 ti: cf32a000 task.ti: cf32a000
4,554,285898134,-;PC is at mwl_tx_del_ampdu_pkts+0x4c/0x130 [mwlwifi]
4,555,285904111,-;LR is at mwl_tx_del_ampdu_pkts+0x48/0x130 [mwlwifi]
4,556,285910072,-;pc : [<bf459518>]    lr : [<bf459514>]    psr: 60000153\x0asp : cf32bd68  ip : cf32bd68  fp : cf32bd9c
4,557,285921605,-;r10: 00000000  r9 : ce4914d0  r8 : ce5eedc0
4,558,285926867,-;r7 : ce5eedb4  r6 : ce4923e4  r5 : ce4914e0  r4 : 0000258c
4,559,285933433,-;r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : ce4923f0
4,560,285940000,-;Flags: nZCv  IRQs on  FIQs off  Mode SVC_32  ISA ARM  Segment kernel
4,561,285947440,-;Control: 10c5387d  Table: 0d25406a  DAC: 00000015
0,562,285953225,-;Process kworker/u4:2 (pid: 369, stack limit = 0xcf32a238)
0,563,285959706,-;Stack: (0xcf32bd68 to 0xcf32c000)
0,564,285964104,-;bd60:                   fffffff0 ce4923f0 cfa11c00 ce493020 ce490b60 ce49302c
0,565,285972335,-;bd80: ce5eedb4 ce5eedc0 00000004 00000004 cf32bddc cf32bda0 bf451588 bf4594d8
0,566,285980566,-;bda0: cf32bdbc 00000000 ce56cca8 00000003 cf32bdec bf4513e8 00000000 00000000
0,567,285988796,-;bdc0: ce5ee800 00000004 ce5ee864 ce490b60 cf32be0c cf32bde0 bf3df46c bf4513f4
0,568,285997026,-;bde0: 00000004 00000000 00000000 00000000 cf32be0c 00000000 cf93d880 00000000
0,569,286005257,-;be00: cf32be54 cf32be10 bf3e8974 bf3df3ec 00000004 00000000 00000000 00000000
0,570,286013488,-;be20: cf93d900 00000003 cfde3438 ce5eec38 ce5ee800 00000004 00000002 ce56c4c0
0,571,286021719,-;be40: ce490b60 ce490f54 cf32be74 cf32be58 bf3e8d90 bf3e87a0 ce047e80 cc7f8088
0,572,286029952,-;be60: ce56c8c4 ce56c8d4 cf32be8c cf32be78 bf3e7ccc bf3e8d64 ce56c8c4 ce56c8d4
0,573,286038189,-;be80: cf32bedc cf32be90 bf3eda74 bf3e7c98 cfb1c040 cf32a000 cf32bebc cf32bea8
0,574,286046419,-;bea0: c003fb0c c003f9f0 0000001e ce56cca8 cf32bf0c ce56c8c4 cfb27180 cf830800
0,575,286054650,-;bec0: ce42bf00 00000000 00000000 cf32a020 cf32bf1c cf32bee0 c003a64c bf3ed8ac
0,576,286062880,-;bee0: 00000001 cfb27180 cf830800 cf830800 00000088 cfb27180 cfb27198 cf830800
0,577,286071110,-;bf00: 00000088 cf830814 cf32a020 cf830800 cf32bf5c cf32bf20 c003aaf4 c003a418
0,578,286079340,-;bf20: c005258c 00000000 c0498100 c049d653 c003a7c0 00000000 cf988ac0 cfb27180
0,579,286087570,-;bf40: c003a7c0 00000000 00000000 00000000 cf32bfac cf32bf60 c003eec4 c003a7cc
0,580,286095800,-;bf60: e59031e0 00000000 f57ff05b cfb27180 00000000 00000000 cf32bf78 cf32bf78
0,581,286104030,-;bf80: 00000000 00000000 cf32bf88 cf32bf88 cf988ac0 c003edd8 00000000 00000000
0,582,286112261,-;bfa0: 00000000 cf32bfb0 c0008ee0 c003ede4 00000000 00000000 00000000 00000000
0,583,286120492,-;bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
0,584,286128724,-;bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 e51b308c e51be090
4,585,286136937,-;Backtrace: 
4,586,286139470,-;[<bf4594cc>] (mwl_tx_del_ampdu_pkts [mwlwifi]) from [<bf451588>] (mwl_mac80211_ampdu_action+0x1a0/0x2b8 [mwlwifi])
4,587,286150908,-; r10:00000004 r9:00000004 r8:ce5eedc0 r7:ce5eedb4 r6:ce49302c r5:ce490b60
4,588,286158912,-; r4:ce493020
4,589,286161547,-;[<bf4513e8>] (mwl_mac80211_ampdu_action [mwlwifi]) from [<bf3df46c>] (drv_ampdu_action+0x8c/0xac [mac80211])
4,590,286172465,-; r10:ce490b60 r9:ce5ee864 r8:00000004 r7:ce5ee800 r6:00000000 r5:00000000
4,591,286180470,-; r4:bf4513e8
4,592,286183133,-;[<bf3df3e0>] (drv_ampdu_action [mac80211]) from [<bf3e8974>] (___ieee80211_stop_tx_ba_session+0x1e0/0x208 [mac80211])
4,593,286194832,-; r6:00000000 r5:cf93d880 r4:00000000
4,594,286199638,-;[<bf3e8794>] (___ieee80211_stop_tx_ba_session [mac80211]) from [<bf3e8d90>] (__ieee80211_stop_tx_ba_session+0x38/0x4c [mac80211])
4,595,286212382,-; r10:ce490f54 r9:ce490b60 r8:ce56c4c0 r7:00000002 r6:00000004 r5:ce5ee800
4,596,286220386,-; r4:ce5eec38
4,597,286223045,-;[<bf3e8d58>] (__ieee80211_stop_tx_ba_session [mac80211]) from [<bf3e7ccc>] (ieee80211_process_delba+0x40/0x48 [mac80211])
4,598,286235092,-; r7:ce56c8d4 r6:ce56c8c4 r5:cc7f8088 r4:ce047e80
4,599,286240966,-;[<bf3e7c8c>] (ieee80211_process_delba [mac80211]) from [<bf3eda74>] (ieee80211_iface_work+0x1d4/0x384 [mac80211])
4,600,286252372,-;[<bf3ed8a0>] (ieee80211_iface_work [mac80211]) from [<c003a64c>] (process_one_work+0x240/0x3b4)
4,601,286262159,-; r10:cf32a020 r9:00000000 r8:00000000 r7:ce42bf00 r6:cf830800 r5:cfb27180
4,602,286270164,-; r4:ce56c8c4
4,603,286272751,-;[<c003a40c>] (process_one_work) from [<c003aaf4>] (worker_thread+0x334/0x538)
4,604,286280967,-; r10:cf830800 r9:cf32a020 r8:cf830814 r7:00000088 r6:cf830800 r5:cfb27198
4,605,286288973,-; r4:cfb27180
[... 30 lines cut ...]
0,561,151402537,-;Process kworker/u4:3 (pid: 1848, stack limit = 0xcdb7e238)
0,562,151409104,-;Stack: (0xcdb7fd68 to 0xcdb80000)
0,563,151413501,-;fd60:                   fffffff0 cdc963f0 cf986400 cdc97020 cdc94b60 cdc97034
0,564,151421720,-;fd80: cd2e65b4 cd2e65c0 00000004 00000004 cdb7fddc cdb7fda0 bf451588 bf4594d8
0,565,151429945,-;fda0: cdb7fdbc 00000000 ce53cca8 00000003 cdb7fdec bf4513e8 00000000 00000000
0,566,151438174,-;fdc0: cd2e6000 00000004 cd2e6064 cdc94b60 cdb7fe0c cdb7fde0 bf3df46c bf4513f4
0,567,151446403,-;fde0: 00000004 00000000 00000000 00000000 cdb7fe0c 00000000 cdd7d700 00000000
0,568,151454633,-;fe00: cdb7fe54 cdb7fe10 bf3e8974 bf3df3ec 00000004 00000000 00000000 00000000
0,569,151462862,-;fe20: cdd7d780 00000003 cfdda438 cd2e6438 cd2e6000 00000004 00000002 ce53c4c0
0,570,151471081,-;fe40: cdc94b60 cdc94f54 cdb7fe74 cdb7fe58 bf3e8d90 bf3e87a0 cdd73f40 cd804088
[... cut ...]
1.  In the advanced settings for the driver of the Intel 7260, I can set the HT mode to either disabled, HT mode, or VHT mode.  The kernel panics when the HT mode is set to HT or VHT mode, but does NOT panic when the HT mode is set to disabled.  However, when the HT mode is set to disabled, speed is obviously greatly reduced.

2.  When the same laptop boots Linux (System Rescue CD running kernel 4.1.8 was the most recent I tried), I am unable to trigger any kernel panics.

If you need any additional information, please let me know.

Thanks.

johnnysl, 2015-11-28:
Curious question: do you use IPV6?

yuhhaurlin, 2015-11-28:
Thanks for your information.

not-the-nrc, 2015-11-28:
My ISP does not issue IPV6 addresses, however the WAN6 interface is NOT disabled.  When I look at the interfaces page on LuCI, it shows WAN6 has an uptime of 0h 0m 0s while the WAN interface currently has an uptime of 10h 22m 14s (and counting).

Additionally, only the Acer laptop with the Intel 7260 running Windows 10 will trigger the kernel panic.  Other devices on the network include an HTC/Google Nexus 9 tablet, an LG/Google Nexus 5 phone, a Chromecast 2015, and an Xbox One.  All are connected to the 5 GHz network.

johnnysl, 2015-11-28:
the virtual address 0000258c in your error is exactly the same as in my kernel panic -> https://github.com/kaloz/mwlwifi/issues/45 that is why i was asking. This issue could well be related, but you have been able to collect a complete crashdump. 
i get mine only when i have IPV6 enabled on my wan interface. as soon as i disable it, the issue is gone.
If @yuhhaurlin  can fix this for you, i have good hopes it will actually fix it on my system at the same time as well.

not-the-nrc, 2015-11-28:
I saw your bug report, but I do not believe that mine is IPV6 related since my ISP does not issue IPV6 addresses.  This report can be closed if it's redundant.

johnnysl, 2015-11-28:
let's leave both open until there is a fix :) 
But thanks very much for the complete trace.

not-the-nrc, 2015-11-28:
No problem.  Let me know if there's any other information I can provide.

not-the-nrc, 2015-11-28:
Just out of curiosity, I tried disabling IPv6.  From LuCI -> Network -> Interfaces, I clicked on the Edit button next to WAN6. On the General Setup tab, I changed Request IPv6-address from try to disabled.  On the Advanced Settings tab, I unchecked Bring up on boot.  I then rebooted the router.  I was still able to trigger a kernel panic in only a few minutes.

johnnysl, 2015-11-28:
I disable it on wan, advanced, use build in  ipv6 set to disable.
I haven't got your wifi card, from intel i only own an n300 card, so can't verify your issue :-(

not-the-nrc, 2015-11-28:
OK, I disabled Use builtin IPv6-management on the WAN interface.  I still hit a kernel panic after just a few minutes.  Too bad.  I was hoping that this would be at least a temporary workaround.

yuhhaurlin, 2015-11-30:
I think if you can build the driver by yourself, you can mark off mwl_tx_del_ampdu_pkts() to see if you still encounter this problem. I will try to fix it. Thanks.

yuhhaurlin, 2015-11-30:
I just commit the fix for this problem, please help to check if it fixes the problem you encounter.

not-the-nrc, 2015-11-30:
I don't have a system set up for building OpenWrt anymore.  I've been using the trunk builds from 

http://downloads.openwrt.org/

I'll either have to wait for the trunk builds to incorporate your fix, or I might be able to set one up in the next few days.  Work is a bit overwhelming at the moment.

johnnysl, 2015-11-30:
@not-the-nrc: Maybe i can build a build for you? Let me know what packages you need included.

wongsyrone, 2015-11-30:
@johnnysl  I suggest only include LuCI for testing.

johnnysl, 2015-11-30:
ok, here is one plain vanilla bleeding edge DD build with just luci and the latest mwlwfi driver for a WRT1900AC-V1 (Mamba): https://www.dropbox.com/sh/7837v2wuxnzrkqx/AABn62I2HMD-0kifeSdJXiLAa?dl=0

yuhhaurlin, 2015-12-01:
Fixed?

not-the-nrc, 2015-12-01:
I will test it as soon as I am off work.

-----

not-the-nrc, 2015-12-01:
I do apologize for taking so long.  Work and sleep got in the way.

@johnnysl Thank you for compiling an image for me.  I really appreciate it.

I've been running the updated mwlwifi driver for the last 1.5 hours with no kernel panics.  Before, I could trigger a kernel panic within a few minutes (skipping around a YouTube video would reliably trigger one, I discovered).  So far, everything is running great.

@yuhhaurlin Thank you for taking a look at this issue and providing a fast fix.  I guess this issue can be closed if you want, or I can close it for you.

Thanks again!

yuhhaurlin, 2015-12-01:
Thanks. I close this issue.

closed 2015-12-01 by yuhhaurlin

johnnysl, 2015-12-01:
@not-the-nrc kaloz has added the fix to trunk yesterday, so you should be able to use an image from the buildbot again, if you wish.

not-the-nrc, 2015-12-01:
@johnnysl I saw that the fix had been applied, but since we only get new builds every 2-3 days, I knew it would take a little bit before a build was uploaded that had the fix included.  I use the Image Builder to make my own custom images without having to compile everything myself, so your build is missing some of my tweaks, but that's OK.  It let me test the driver and I really do appreciate your help.