kaloz/mwlwifi #50: 5GHz stopped working until power cycle

notnyt, 2015-11-29:
Using the latest driver, 5Ghz stopped working after an Intel 7260 was connected.  It did not work again until the router was power cycled.  Reboots did nothing.  Nothing was present in any kernel or syslog to indicate trouble.

notnyt, 2015-11-29:
And a few hours later 5GHz is no longer usable.  Nothing in dmesg.

root@ZOMGWTFBBQWIFI:/# iwinfo wlan0 info
wlan0     ESSID: "ZOMG2.4"
          Access Point: 00:25:9C:13:05:56
          Mode: Master  Channel: 3 (2.422 GHz)
          Tx-Power: 30 dBm  Link Quality: 61/70
          Signal: -49 dBm  Noise: -80 dBm
          Bit Rate: 109.8 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 11AB:2A55 11AB:0000 [Marvell 88W8864]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
root@ZOMGWTFBBQWIFI:/# iwinfo wlan1 info
wlan1     ESSID: "ZOMGWTFBBQWIFI"
          Access Point: 00:25:9C:13:05:57
          Mode: Master  Channel: 157 (5.785 GHz)
          Tx-Power: 30 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -58 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 11AB:2A55 11AB:0000 [Marvell 88W8864]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

notnyt, 2015-11-29:
It seems this works fine when set to VHT40, but once set to VHT80, no devices can connect.

yuhhaurlin, 2015-11-30:
Use Intel 7260?

notnyt, 2015-11-30:
@yuhhaurlin  Negative,  using macbook and a lg g4.  I'm wondering fi this is driver issue or hardware failure, since it seemed to work previously.

yuhhaurlin, 2015-11-30:
Since this one is not clear yet. I will check intel 7260 first. Thanks.

notnyt, 2015-11-30:
This has nothing to do with 7260, just VHT80 not working, only VHT40

yuhhaurlin, 2015-11-30:
But you said it is working previously?

lantis1008, 2015-11-30:
I don't experience this issue 
Latest driver
Ch 149
Vht80
Country AU
No scan is not set (0)

notnyt, 2015-11-30:
Might be hardware failure since I'm pretty sure it was working before.

yuhhaurlin, 2015-11-30:
Yes, we had done a lot of Inter operation tests with different clients. I did not receive this kind of test report.

notnyt, 2015-11-30:
Seems to be a hardware issue.  When temp2 rises above 50C it seems to fail.  And on that note, I no longer have a properly functioning device to debug with.  Good luck w/ the rest of the bugs, I'll be ordering a UAP-AC-LR.  Cheers.

yuhhaurlin, 2015-11-30:
I will close this one. Thanks.

closed 2015-11-30 by yuhhaurlin