kaloz/mwlwifi #53: Low speeds on 2.4 ghz and problems with Intel 7260

thagabe, 2015-12-09:
I have ran a series of test (ping, speed, file transfer) and I have noticed the 2.4ghz maxed out at around 81 mbps (of my internet speed) compared to 180 mbps (the most I get from my isp). I assume this is hardware rather than software. 

Now I have ran many test on the wireless from my laptop (Bonobo Extreme w/ Intel 7260 AC Wireless Chip) and have noticed a few things:

1) On 2.4 ghz with 20mhz i get around 5-10 mbps
2) " " " " forced 40mhz (link speed 300 mhz) I get around  ~30 mbps
3) " " " I keep running into ampsu errors (-22) seems like the code isn't fully ready to take into account the problem (Intel 7260). However with the Nexsus 6p I managed to achive 81 mbps with an ac chip running with N compatibility (connecting to 2.5 ghz, link speed of only 144 mhz) and doesn't get the errors like the intel card.

thagabe, 2015-12-09:
Running .14 but also tested on .15

chazikai24, 2015-12-10:
Yes，I hava the same problems!

yuhhaurlin, 2015-12-10:
So the problem is related to 2.4g band with Intel 7260?

thagabe, 2015-12-10:
Yes, It seems looks like @bmork was on to something when he tried to make the code check if the client is able to use ampsu or not and chose correctly and/or the implementation doesn't work with the Laptop version of the Intel's 7260 AC cards.

yuhhaurlin, 2015-12-10:
Current code had already added this kind of ability. You can "cat /sys/kernel/debug/ieee80211/phy0 or phy1 /mwlwifi/sta" to check ampdu and amsdu capability.

thagabe, 2015-12-10:
Then where might the problem lie? I tested a broadcom card from a fruity device and had no problem connecting to 2.4 but the intel card does have the problem.

yuhhaurlin, 2015-12-10:
Can you use "cat /sys/kernel/debug/ieee80211/phy0 or phy1 /mwlwifi/sta" to let me know ampdu and amsdu status of Intel card after you completed your test?

yuhhaurlin, 2015-12-10:
BTW, can you help to check if this problem is existed with stock firmware? Thanks.

davidc502, 2015-12-10:
2 iperf tests on 2.4Ghz and 23Mbps was about max.

[image]

yuhhaurlin, 2015-12-10:
@davidc502 The client is also Intel 7260?

davidc502, 2015-12-10:
Intel Wifi Link 5100AGN << Wireless card

yuhhaurlin, 2015-12-10:
Under same environment and setting, what throughput number can you get with other AP? BTW, can you check if Stock firmware has the same number?

davidc502, 2015-12-10:
Doesn't look like I am comparing "apples to apples"

root@OpenWrt:~# strings /lib/modules/*.*/mwlwifi.ko |grep "firmware version"
firmware version: 0x%x
firmware version: 0x%08x

1st test (above) was with Lenovo Laptop
2nd test with FireTV >>> Same results -- around 22Mbps

yuhhaurlin, 2015-12-10:
I need to know this number is not due to noise environment, so I hope you can do the test with other AP. I also hope you can test with stock firmware under same setting and environment.

davidc502, 2015-12-10:
Started up the Actiontec V1000h, and started up 2.4Ghz band. Ran the same iperf test using Lenovo Laptop on 2.4Ghz, and averaged around 22Mbps.

Good call asking to test with different AP -- Also, I tried the 3 main channels.

yuhhaurlin, 2015-12-10:
Yes. So what you reported is not a problem. I will wait test result with Intel 7260. Thanks.

johnnysl, 2015-12-12:
@yuhhaurlin: might this not be this issue? -> http://www.snbforums.com/threads/linksys-wrt-1900acs-reviewed.27516/page-4#post-217590
Afaik it is fixed by an update of the .bin firmware file in the closed source stock driver. I haven't seen that updated .bin file in the opensource driver (yet?). Any plans to update the .bin blob?

e88z4, 2015-12-12:
I have the same experience. My setup is the following. 

WMP600N with b43 driver as client. Connecting to WRT1900AC v1 with CC Final with .13 driver. I have a NAS connected through gigabit line through one of the ethernet port. 

2.4Ghz band N-only 40Mhz setting resulting only 700-900kbps (5Mbps-6Mbps)
2.4Ghz band legacy 20Mhz setting resulting better performance around 2000-3000 kbps around  (17Mbps-20Mbps).

Both file transfer were done using smb.

johnnysl, 2015-12-12:
@e88z4: Never test with copying files back and forth, always use iperf3. 
@thegabe: my performance differs per reboot, and seems to be best after a complete cold boot (thus power cycle). Can't really make heads or tales of it. Currently i get 90mbit via iperf3 between router and client on 20mhz 2.4ghz, which actually is pretty good. But before the power cycle, i was stuck on 20mbit. No idea why a powercycle does more then a warm reboot.

johnnysl, 2016-01-01:
One of the things i'm still seeing is that after a couple of days the interface to the router on the 5ghz wifi interface gets laggy, with a lot of packet loss. I just have no idea what causes it (wireless driver i guess), but i've found a simple method to test for it.
could you guys test this for me as well?
To test:
1. have an uptime of the router of at least 2-3 days.
2. from a wireless client on the 5ghz interface, ping the router with a larger block (16k) size a 100 times and check for the average values afterwards:
3. switch the client to the 2.4ghz Interface and repeat.
e.g on a windows machine: ping 192.168.1.1 -l 16384 -n 100 
Normal ping times for a large ping like this should be ~6-7msec. But often i see packet loss or >100 msec response times.
e.g:
5ghz:
Ping statistics for 192.168.1.1:
    Packets: Sent = 100, Received = 57, Lost = 43 (43% loss),
Approximate round trip times in milli-seconds:
    Minimum = 6ms, Maximum = 121ms, Average = 11ms
2.4ghz:
Ping statistics for 192.168.1.1:
    Packets: Sent = 100, Received = 100, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 5ms, Maximum = 112ms, Average = 11ms
a reboot will resolve the issue for a day or two and then it is back again.
NB. small sized packets (i.ow. a normal ping) would not show any packetloss.

johnnysl, 2016-01-01:
For clarity, this is the ping avg. just after a reboot:

Ping statistics for 192.168.110.1:
    Packets: Sent = 50, Received = 50, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 2ms, Maximum = 84ms, Average = 6ms

so no packet loss, and way faster response times.

yuhhaurlin, 2016-03-26:
Please test driver.17. Thanks.

thagabe, 2016-03-29:
Sorry for the long overdue response @yuhhaurlin. Intel has since release 3 driver updates that have quite honestly solved many of the problems in 5GHz (both on .16 and .17) however I have yet to test 2.4GHz speeds. Since the release of the new drivers I have compiled my own image (so i presume the new firmware has been compiled along with the rest of openwrt) the new problem I'm experiencing (5GHz) is a random drop of connectivity for a few seconds then it connects again. This, however, might be due to openwrt itself since im connected via another ap that is then connected to the router via ethernet [since this started happening when i compiled .17 with the bm offload patches it might be one or the other tho im incline to think it might be the bm patches but i have noticed this when connecting directly to the ap of the wrt1900ac v1 too]

yuhhaurlin, 2016-03-30:
Can you connect to AP directly and get debug log messages from hostapd?

johnnysl, 2016-04-01:
with the .17 driver, i'm still seeing deteriorating speeds with Intel centrino 6235 on 2.4ghz as well (speeds drop to 50mbit/sec on 2.4ghz / 20mhz / N). much better then the fall back to 20mbit on the .16 release though.

Not sure if it is the intel card though, as after a reboot the speed is 100mbit for a while again.
so ignore this, if it might be another issue. 

Speed on 5ghz / n / 40mhz drop about 20% after a day, but stay stable after that (from ~180Mbit/sec to ~155Mbit/sec).

yuhhaurlin, 2016-05-26:
How about the new driver?

thagabe, 2016-05-27:
All good on my end. Please close this issue

closed 2016-05-27 by yuhhaurlin