kaloz/mwlwifi #54: wlan issue by more SSIDs

trustno1foxm, 2015-12-21:
Hello!

I, and some other guys on openwrt forum, have troubles with wifi and using at least 2 SSIDs on one radio. With driver .3 the issue appears after some hours. With .13 it will cause after some days. The only way to get wifi to work is to restart the device.

The problem looks like this: Wifi SSIDs are still shown on the network, but nobody is able to connect. After rebooting the device all Clients work fine, for some hours/days.

yuhhaurlin, 2015-12-22:
1. Please test with latest driver.
2. Please describe your problem clearly if you still get problem with latest driver.

trustno1foxm, 2015-12-22:
ad1) ok, thats .15? This driver has the softirq bug, so my latest version without that bug is .13?
ad2) ok, sorry for my bad english, it's not as easy to describe, because the wlan connection will be dropped and no new connection is possible. If the bug will still exist, do you want some log files? I have to restart the device, I can't let it be offline for days :/

Thx!

so long

yuhhaurlin, 2015-12-22:
1. The latest driver should be no software irq issue.
2. When the problem happened, can you use "dmesg" to see if there is any messages related to WiFi? BTW, is there any messages printed in console.

yuhhaurlin, 2015-12-22:
BTW, .13 has two potential problems to make kernel oops.

trustno1foxm, 2015-12-22:
Ok, thx for your very fast respsone! I'll install .15 and give you a feedback!

trustno1foxm, 2015-12-22:
Ok, I installed .15 driver but the softirq error is back again.
Should I really test this driver for the wifi issue?
thx&so long

yuhhaurlin, 2015-12-22:
What do you mean softirq problem? It will generate more interrupts, but it will not trigger lot of tasklet.

trustno1foxm, 2015-12-22:
Oh sorry, I thought that high generation of interruprts autom. results in softirq problem? If not, sorry my fault! I have .15 installed and going to test wifi with 2 SSIDs.

yuhhaurlin, 2015-12-22:
The problem is fixed, the driver won't trigger more than one tasket within one tick.

trustno1foxm, 2015-12-22:
Hello,

wifi crashed at this moment, here is the output from dmesg:

root@OpenWrt:~# dmesg
[   11.726898] ieee80211 phy1: pcmd = de668000
[   11.736887] ieee80211 phy1: firmware version: 0x7020806
[   11.756198] ieee80211 phy1: phy1: FCC
[   11.759879] ieee80211 phy1: Channel: 1: 0x0 0x0 0xf
[   11.764788] ieee80211 phy1: 19 14 14 14 13 13 13 13 10 10 10 10 0 0 0 0
[   11.771522] ieee80211 phy1: Channel: 2: 0x0 0x0 0xf
[   11.776423] ieee80211 phy1: 1a 19 18 17 19 19 17 16 14 14 14 14 0 0 0 0
[   11.783156] ieee80211 phy1: Channel: 3: 0x0 0x0 0xf
[   11.788057] ieee80211 phy1: 1a 19 18 17 19 19 17 16 14 14 14 14 0 0 0 0
[   11.794790] ieee80211 phy1: Channel: 4: 0x0 0x0 0xf
[   11.799691] ieee80211 phy1: 1a 19 18 17 19 19 17 16 14 14 14 14 0 0 0 0
[   11.806424] ieee80211 phy1: Channel: 5: 0x0 0x0 0xf
[   11.811328] ieee80211 phy1: 1a 19 18 17 19 19 17 16 14 14 14 14 0 0 0 0
[   11.818057] ieee80211 phy1: Channel: 6: 0x0 0x0 0xf
[   11.822961] ieee80211 phy1: 1a 19 18 17 19 19 17 16 14 14 14 14 0 0 0 0
[   11.829691] ieee80211 phy1: Channel: 7: 0x0 0x0 0xf
[   11.834595] ieee80211 phy1: 1a 19 18 17 19 19 17 16 14 14 14 14 0 0 0 0
[   11.841327] ieee80211 phy1: Channel: 8: 0x0 0x0 0xf
[   11.846228] ieee80211 phy1: 1a 19 18 17 19 19 17 16 14 14 14 14 0 0 0 0
[   11.852963] ieee80211 phy1: Channel: 9: 0x0 0x0 0xf
[   11.857865] ieee80211 phy1: 1a 19 18 17 19 19 17 16 14 14 14 14 0 0 0 0
[   11.864598] ieee80211 phy1: Channel: 10: 0x0 0x0 0xf
[   11.869585] ieee80211 phy1: 1a 19 18 17 19 19 17 16 14 14 14 14 0 0 0 0
[   11.876318] ieee80211 phy1: Channel: 11: 0x0 0x0 0xf
[   11.881310] ieee80211 phy1: 19 15 15 15 14 14 14 14 13 13 13 13 0 0 0 0
[   11.888803] ieee80211 phy1: 2G: enable
[   11.892593] ieee80211 phy1: 5G: disable
[   11.896452] ieee80211 phy1: TX: 2 antennas
[   11.900571] ieee80211 phy1: RX: 2 antennas
[   11.905267] i2c /dev entries driver
[   11.915037] ip_tables: (C) 2000-2006 Netfilter Core Team
[   11.922570] nf_conntrack version 0.5.0 (8056 buckets, 32224 max)
[   11.991688] armada38x-rtc f10a3800.rtc: rtc core: registered f10a3800.rtc as                                                                                                                                                              rtc0
[   12.005196] xt_time: kernel timezone is -0000
[   12.024309] cdc_ether 1-1:1.0 eth2: register 'cdc_ether' at usb-f1058000.usb-                                                                                                                                                             1, CDC Ethernet Device, 0c:5b:8f:27:9a:64
[   12.035165] usbcore: registered new interface driver cdc_ether
[   12.043586] PPP generic driver version 2.4.2
[   12.048240] NET: Registered protocol family 24
[   14.126828] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   14.135886] device eth1.1 entered promiscuous mode
[   14.140701] device eth1 entered promiscuous mode
[   14.146317] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   14.155065] cdc_ether 1-1:1.0 eth2: kevent 12 may have been dropped
[   14.163142] IPv6: ADDRCONF(NETDEV_UP): eth1.10: link is not ready
[   14.179630] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   14.733623] cfg80211: Calling CRDA for country: AT
[   14.738589] ieee80211 phy1: phy1: ETSI
[   14.742407] ieee80211 phy1: Channel: 1: 0x0 0x0 0xf
[   14.747346] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0
[   14.753052] ieee80211 phy1: Channel: 2: 0x0 0x0 0xf
[   14.757956] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0
[   14.763703] ieee80211 phy1: Channel: 3: 0x0 0x0 0xf
[   14.768616] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0
[   14.774314] ieee80211 phy1: Channel: 4: 0x0 0x0 0xf
[   14.779225] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0
[   14.784921] ieee80211 phy1: Channel: 5: 0x0 0x0 0xf
[   14.789823] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0
[   14.795513] ieee80211 phy1: Channel: 6: 0x0 0x0 0xf
[   14.800414] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0
[   14.806101] ieee80211 phy1: Channel: 7: 0x0 0x0 0xf
[   14.811007] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0
[   14.816690] ieee80211 phy1: Channel: 8: 0x0 0x0 0xf
[   14.821598] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0
[   14.827283] ieee80211 phy1: Channel: 9: 0x0 0x0 0xf
[   14.832190] ieee80211 phy1: d d d d d d d d d d d d 0 0 0 0
[... 85 lines cut ...]
[  143.140709] ieee80211 phy1: change: 0x40
[  143.207270] ieee80211 phy1: change: 0x100
[  143.526406] ieee80211 phy1: change: 0xffffffff
[  143.597511] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  143.604478] device wlan1 entered promiscuous mode
[  143.609270] br-lan: port 3(wlan1) entered forwarding state
[  143.614805] br-lan: port 3(wlan1) entered forwarding state
[  143.676866] ieee80211 phy1: change: 0x100
[  143.685971] ieee80211 phy1: change: 0x42
[  143.771384] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  143.829411] IPv6: ADDRCONF(NETDEV_UP): wlan1-1: link is not ready
[  143.911274] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-1: link becomes ready
[  145.605682] br-lan: port 3(wlan1) entered forwarding state
[ 8596.067030] nf_conntrack: automatic helper assignment is deprecated and it wi                                                                                                                                                             ll be removed soon. Use the iptables CT target to attach helpers instead.
[17730.151878] ieee80211 phy1: check ba result error 1
[17730.156783] ieee80211 phy1: ampdu start error code: -22
[17730.241876] ieee80211 phy1: check ba result error 1
[17730.246776] ieee80211 phy1: ampdu start error code: -22
[17730.291867] ieee80211 phy1: check ba result error 1
[17730.296766] ieee80211 phy1: ampdu start error code: -22

5ghz is still working, but both 2.4ghz ssid's are shown as up, but connection is refused.
pls help :/

thx!

yuhhaurlin, 2015-12-22:
From the log, I can't find what problem you encountered.

trustno1foxm, 2015-12-22:
I just rebooted the device and everything works again. What else could be the problem? It's like the password is wrong. But be sure it's not and after the reboot all two Wi-Fi are working.

yuhhaurlin, 2015-12-22:
Can you describe your setting? BTW, when the problem happened, only 2.4g got this problem? Only one client can't connect or no client can connect?

trustno1foxm, 2015-12-22:
ok: I have setup 1 wi-fi network @5ghz and 2 wi-fi networks @2.4ghz. the problem is when there a set up 2 wifi networks. it also happens on the 5ghz, but the 5ghz is not as good as the good old 2.4ghz.
1 wifi is setup like a lan connection, with all access to the devices in the lan. the 2nd wifi is set up for guests and has it's on interface with connection to the internet. I had the same config on my tp-link 3420 router with openwrt and it worked a long time.
when wifi crashes, all clients can't connect, it trys to connect but it fails. the connection is shown on android devices as "saved".
any ideas? sorry for my bad english, and it's late too ;)

thx!

yuhhaurlin, 2015-12-23:
Can you help to test with open mode without security?

trustno1foxm, 2015-12-23:
Both wifi networks @2.4ghz?

yuhhaurlin, 2015-12-23:
Yes.

trustno1foxm, 2015-12-23:
That's not that good :/ I can't promise, no other way to find out the problem?

muronghan, 2015-12-23:
I can agree with trustno1foxm. cc final + 10.3.0.3 driver on my WRT1200AC just has the same problem as mentioned: setting up two SSIDs on 2.4GHz and after a while all 2.4GHz devices can not be connected Only a reboot will fix it. .13 and .15 are similar. I tested 5GZ on .15 and it has the same problem. I do not find this problem on .14 but the situation is more serious as cc final+ .14 will cause the router halted after a few days.

So right now I am still using cc final + 10.3.0.3 and have to add one another router dedicated for guest wifi. Have only one SSID for each band running on WRT1200AC is perfect. 

The wifi .ipk driver I mentioned above was from other people. I did not compile them by myself. Could wifi ipk compiled and installed seperately, or should be compiled together with the whole system?

There might be an exception. As I tried CC trund (r478xx)+.14. It run for 5 days until my home power was off. But I set a cronjob which let the wifi down in the middle night and up after one minute. For some reason I should have to stick to cc final. So I do not try more on this e.g. delete this cronjob.

yuhhaurlin, 2015-12-23:
It looks like since .14 with kernel panic fix, mwlwifi driver is pretty stable. I did not receive crash report about mwlwifi driver since then. Our internal stress test shows the same result. About Multi BSSID issue, I hope you can do one more test with open mode to check if the problem is exited or not.

trustno1foxm, 2015-12-23:
@muronghan thx for you assistance!!
@yuhhaurlin I just deactivated my shields  ;) I will give a feedback!

muronghan, 2015-12-23:
All right, I may try open mode for both SSIDs at 2.4GHz this weekend for cc final +.15. As I could monitor the invaders much easily during that time period. Thank for your help. Let's get this thing done. And taking this opportunity, guys, Merry Chirstmas and Happy New Year!

trustno1foxm, 2015-12-23:
So my wifi crashed again. No connection possible. It has been open (w/o security) for some hours and crashed. Pls help.
Merry Christmas! ;)

so long

dmesg output:
root@OpenWrt:~# dmesg
[   14.962216] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   14.969303] ieee80211 phy0: Channel: 52: 0x0 0x0 0xf
[   14.974305] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   14.981395] ieee80211 phy0: Channel: 56: 0x0 0x0 0xf
[   14.986384] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   14.993471] ieee80211 phy0: Channel: 60: 0x0 0x0 0xf
[   14.998462] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.005548] ieee80211 phy0: Channel: 64: 0x0 0x0 0xf
[   15.010539] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.017627] ieee80211 phy0: Channel: 100: 0x0 0x0 0xf
[   15.022709] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.029788] ieee80211 phy0: Channel: 104: 0x0 0x0 0xf
[   15.034871] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.041957] ieee80211 phy0: Channel: 108: 0x0 0x0 0xf
[   15.047033] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.054120] ieee80211 phy0: Channel: 112: 0x0 0x0 0xf
[   15.059203] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.066307] ieee80211 phy0: Channel: 116: 0x0 0x0 0xf
[   15.071391] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.078470] ieee80211 phy0: Channel: 120: 0x0 0x0 0xf
[   15.083552] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.090631] ieee80211 phy0: Channel: 124: 0x0 0x0 0xf
[   15.095717] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.102804] ieee80211 phy0: Channel: 128: 0x0 0x0 0xf
[   15.107879] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.114964] ieee80211 phy0: Channel: 132: 0x0 0x0 0xf
[   15.120044] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.127128] ieee80211 phy0: Channel: 136: 0x0 0x0 0xf
[   15.132208] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.139287] ieee80211 phy0: Channel: 140: 0x0 0x0 0xf
[   15.144370] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.151455] ieee80211 phy0: Channel: 149: 0x0 0x0 0xf
[   15.156531] ieee80211 phy0: 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11
[   15.163619] cfg80211: Regulatory domain changed to country: AT
[   15.169478] cfg80211:  DFS Master region: ETSI
[   15.173769] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gai                                                                                                                                                                                                                                             n, max_eirp), (dfs_cac_time)
[   15.183566] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 m                                                                                                                                                                                                                                             Bm), (N/A)
[   15.191609] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AU                                                                                                                                                                                                                                             TO), (N/A, 2000 mBm), (N/A)
[   15.201135] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AU                                                                                                                                                                                                                                             TO), (N/A, 2000 mBm), (0 s)
[   15.210654] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700                                                                                                                                                                                                                                              mBm), (0 s)
[   15.218784] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 40                                                                                                                                                                                                                                             00 mBm), (N/A)
[   16.140886] ieee80211 phy1: change: 0xffffffff
[   16.221669] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   16.227765] mvneta f1070000.ethernet eth1: link up
[   16.232712] mvneta f1034000.ethernet eth0: link up
[   16.297691] ieee80211 phy0: change: 0xffffffff
[   16.378140] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   16.384239] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   16.390605] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   16.397060] br-lan: port 1(eth1.1) entered forwarding state
[   16.402712] br-lan: port 1(eth1.1) entered forwarding state
[   16.408463] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.10: link becomes ready
[   16.415380] device wlan1 entered promiscuous mode
[   16.429839] device wlan0 entered promiscuous mode
[   16.435274] ieee80211 phy0: change: 0x100
[   16.444333] ieee80211 phy0: change: 0x40
[... cut ...]
[58416.752849] br-lan: port 2(wlan1) entered forwarding state
[58416.758379] br-lan: port 2(wlan1) entered forwarding state
[58416.764898] ieee80211 phy1: change: 0x100
[58416.773926] ieee80211 phy1: change: 0x42
[58416.858663] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[58416.906504] IPv6: ADDRCONF(NETDEV_UP): wlan1-1: link is not ready
[58416.933047] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-1: link becomes ready
[58416.990991] ieee80211 phy1: change: 0x20
[58418.753192] br-lan: port 2(wlan1) entered forwarding state

johnnysl, 2015-12-23:
Do you really need to reboot the router, or is disconnecting and reconnectting the client also a workaround?

northbound1, 2015-12-24:
@yuhhaurlin  Could  you elaborate? On this comment please.
BTW, .13 has two potential problems to make kernel oops.
Thanks.

yuhhaurlin, 2015-12-24:
@northbound1 https://github.com/kaloz/mwlwifi/commit/352efe4bdc34204cc528ac9a979bb0032a6996e0
https://github.com/kaloz/mwlwifi/commit/274ec5979b2080dbfe523d1618e69cebc845c572

yuhhaurlin, 2015-12-24:
@trustno1foxm Thanks. We will check it.

northbound1, 2015-12-24:
Thank you @yuhhaurlin

trustno1foxm, 2015-12-24:
@johnnysl no thats not working :/ tried to disable and enable wlan, too. but without success
@yuhhaurlin thank you very much!

Koldur81, 2016-01-01:
I too experience this problem (WiFi stops working after about two days with two SSIDs on one radio) with my WRT1200AC, with the .13 and .15 driver both. I also have to do a full reboot of the device in order to get it working again, disabling and reenabling doesn't do anything.

trustno1foxm, 2016-01-01:
@Koldur81 thx for your feedback!
@yuhhaurlin is there a roadmap for this bug? I don't want to be nuisance, but I really need a second SSID, because the 5ghz radio is very weak :/

thx & so long

muronghan, 2016-01-02:
When the wireless stopped service, I read from kernel log as something like this:

[22867.335131] ieee80211 phy1: check ba result error 1
[22867.340080] ieee80211 phy1: ampdu start error code: -22
[22867.385077] ieee80211 phy1: check ba result error 1
[22867.389985] ieee80211 phy1: ampdu start error code: -22
[22867.435086] ieee80211 phy1: check ba result error 1
[22867.439988] ieee80211 phy1: ampdu start error code: -22
[22867.585083] ieee80211 phy1: check ba result error 1
[22867.589988] ieee80211 phy1: ampdu start error code: -22

I also have the same problem on trunk r4800x (kernel 4.1.13) with 10.3.0.15 on my WRT1200AC. BTW, 1900ACv2 has a similar structure. This problem only takes place on 1200ac but not for 1900acv2?

trustno1foxm, 2016-01-02:
I have the same errors, mmh the error code is similar to issue#41?

muronghan, 2016-01-07:
Same problem for 10.3.0.16 on WRT1200AC (cc final)

yuhhaurlin, 2016-01-07:
This issue will be checked later.

Koldur81, 2016-01-07:
Great to hear that this is still on your list, as it is a pretty annoying problem which should be a relatively easy fix.

yuhhaurlin, 2016-01-07:
Yes, all issues listed will be checked one by one later.

muronghan, 2016-01-18:
cc final+.14 run smoothly for more than 3 days so far on my WRT1200AC. However, I have to put "00 02 * * * /sbin/wifi restart" in cronjob otherwise the router will halt after a few days. 

Such way does not work on .03, 13, 15 or 16 but only on .14. I bring this as a reference for future development.... I look forward to not waiting too long and this bug can be fixed in .17.

trustno1foxm, 2016-01-23:
are there any updates? @yuhhaurlin
Is there a roadmap? thx!

muronghan, 2016-01-25:
just to report that cc final+.14 on WRT1200AC with everyday's wifi restart still crashed after 3 and half days. Meanwhile, flashed .16 instead, the SSID of 5GHz could not be found on mobile devices after 3 days (only one SSID).

Chadster766, 2016-01-31:
@trustno1foxm what is the output of the below commands:

strings /lib/modules/./mwlwifi.ko | grep "^10.3"
ps -aux|grep hostapd

The last command will list the hostapd conf files in use. What is the contents of those files:

Ex: "cat /etc/hostapd/wlan0.conf"

trustno1foxm, 2016-01-31:
Thx for your comment, but I have some problems with your commands:

> strings /lib/modules/./mwlwifi.ko | grep "^10.3"

```
10.3.0.13
10.3.0.13
```

> ps -aux|grep hostapd

`ps: invalid option -- a
`

> Ex: "cat /etc/hostapd/wlan0.conf"

`there is no directory "hostapd"`

Chadster766, 2016-01-31:
You need to upgrade your wireless driver to .16 in order to proceed with troubleshooting.

Try command "ps" without the "-aux" on the end.

trustno1foxm, 2016-01-31:
Nice, .16 is unstable. I will install .16 for testing purposes. I hope you can help me!
By using .16 I have more problems as by using .13, see: https://github.com/kaloz/mwlwifi/issues/59

So far, with .16 the soft irq "issue" is back again.

> strings /lib/modules/./mwlwifi.ko | grep "^10.3"

```
10.3.0.16-20160105
10.3.0.16-20160105
```

> ps | grep hostapd

```
 1378 root      1632 S    /usr/sbin/hostapd -P /var/run/wifi-phy1.pid -B /var/
 1447 root      1632 S    /usr/sbin/hostapd -P /var/run/wifi-phy0.pid -B /var/
 2242 root      1304 S    grep hostapd
```

> Ex: "cat /etc/hostapd/wlan0.conf"

`there is no directory "hostapd"`

Chadster766, 2016-01-31:
Sorry I don't agree about .13 being better that .16.

Which OpenWRT version are you running?

Try the below commands:

/var/run/hostapd-phy0.conf
/var/run/hostapd-phy1.conf

Chadster766, 2016-01-31:
Also you should disable IPv6 unless required by your ISP. OpenWRT odhcpc has some issues.

trustno1foxm, 2016-02-01:
Sorry for my late answer...

OpenWRT Version = 15.05 (final)

output from both conf files:

```
root@OpenWrt:~# cat /var/run/hostapd-phy0.conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=xx
ieee80211d=1
ieee80211h=1
hw_mode=a
channel=44

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][SU-BEAMFORMER][SU-BEAMFORMEE][RX-ANTENNA-PATTERN          ][TX-ANTENNA-PATTERN][RX-STBC1][MAX-MPDU-7991][MAX-A-MPDU-LEN-EXP7]

interface=wlan0
ctrl_interface=/var/run/hostapd
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=xx
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=xx
bridge=br-lan
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=xx

root@OpenWrt:~# cat /var/run/hostapd-phy1.conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=xx
ieee80211d=1
hw_mode=g
channel=11

ieee80211n=1
ht_coex=0
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40]

interface=wlan1
ctrl_interface=/var/run/hostapd
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=xx
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=xx
bridge=br-lan
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=xx

bss=wlan1-1
ctrl_interface=/var/run/hostapd
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=xx
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=xx
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=xx
```

Thx for your help!

thagabe, 2016-02-01:
@Chadster766 Running Openwrt 15.05 (V1) with wireless driver .16 with your patch sent to the forum. Everything is running smoothly in a wifi heavy home. Any test you would like me to test?

Chadster766, 2016-02-01:
Nope we just need reports on how it performs over time and if any errors are logged :smiley:

Chadster766, 2016-02-01:
If the patch properly applied you won't see any ampdu errors in the log.

trustno1foxm, 2016-02-02:
You want me to activate the 2 ssid on the radio? I only can test this at the weekend. :/ But be sure it will fail, and which logs do you mean? dmesg?

trustno1foxm, 2016-02-02:
@thagabe hello, do you have a 2 ssid set up on your radio? I have a second with it's own dhcp and own brdige for guests. after some hours it quits, maybe because of some devices...

thagabe, 2016-02-02:
@Chadster766 the router worked fine, no lock ups, but one of the laptops kept dropping it's signal so in the end i had to move back to mcwrt haha I'll look into mcdebian but we'll see.

thagabe, 2016-02-02:
@Chadster766 the laptop was connecting via wireless n in 2.4GHz to 1 SSID on the router via forced aes

Chadster766, 2016-02-02:
@thagabe did the router log or laptop eventvwr show any clues why the laptop didn't connect?

@trustno1foxm for 5Ghz channel 44 should vht_oper_centr_freq_seg0_idx=50?
disassoc_low_ack=1 is no longer required.
The spaces in this options is a bug? "[RX-ANTENNA-PATTERN          ]"

There is a lot more unnecessary settings in your hostapd conf files. 

Below are examples of my multi SSID host conf files:

wlan0.conf:
```
root@MCDEBIAN:~# cat /etc/hostapd/wlan0.conf
interface=wlan0
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ignore_broadcast_ssid=0
hw_mode=g
channel=6
wmm_enabled=1
ieee80211n=1
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40]
ieee80211ac=0
country_code=US

#Corp Wireless
ssid=Xxxxx

auth_algs=1
wpa=2
wpa_passphrase=X1x1xx1x1
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

#Guest Wireless
bss=wlan0_4
bssid=e6:c3:f9:72:71:b1
bridge=br1
ssid=Xxxxx_Guest

auth_algs=1
wpa=2
wpa_passphrase=xxxxx123
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
```

wlan1.conf
```
root@MCDEBIAN:~# cat /etc/hostapd/wlan1.conf
interface=wlan1
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ignore_broadcast_ssid=0
hw_mode=a
channel=36
wmm_enabled=1
ieee80211n=1
ht_capab=[LDPC][HT40+][SHORT-GI-20][SHORT-GI-40]
vht_capab=[MAX-MPDU-7991][RXLDPC][SHORT-GI-80][RX-STBC-1][SU-BEAMFORMER][SU-BEAMFORMEE][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
country_code=US

#Corp Wireless
ssid=Xxxx_AC

auth_algs=1
wpa=2
wpa_passphrase=X1x1xx1x1
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

#Guest Wireless
bss=wlan1_4
bssid=5e:95:e1:46:52:6d
bridge=br1
ssid=Xxxx_AC_Guest

auth_algs=1
wpa=2
wpa_passphrase=xxxxx123
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
```

Chadster766, 2016-02-03:
@trustno1foxm I recommend you upgrade to OpenWRT Designated Driver r48564 or newer which properly calculates vht_oper_centr_freq_seg0_idx.

trustno1foxm, 2016-02-03:
@Chadster766 Thx you for your reply! r48564 is a trunk image, isn't it? Is it possible to calculate the vht_oper_centr_freq_seg0_idx by myself? I really want to use the 15.05 final and no alpha/beta software.
Thx!

Chadster766, 2016-02-03:
AFAIK it's always +6

trustno1foxm, 2016-02-03:
Thx, I changed the file. I also checked your conf files so far. I will give it a try. Your config is ok for you? 2 ssids on the radio and there are no connection issues?

Chadster766, 2016-02-03:
Yup

Last login: Mon Feb  1 15:05:21 2016 from 192.168.60.20
root@MCDEBIAN:~# uptime
 12:24:22 up 6 days,  1:06,  1 user,  load average: 0.13, 0.20, 0.22
root@MCDEBIAN:~# iw wlan0 station dump |grep -c Station
11
root@MCDEBIAN:~# iw wlan1 station dump |grep -c Station
4

[47 more replies]

closed 2016-03-30 by yuhhaurlin