kaloz/mwlwifi #55: Kernel error, update encryption times out

johnnysl, 2015-12-25:
@yuhhaurlin, currently getting these errors, which break all wifi connectivity. Am using the latest driver, and Openwrt build 48005. (X'ed out end of the mac adresses for privacy)

Sat Dec 26 00:01:34 2015 daemon.info hostapd: wlan0: STA 78:7e:61:xx:xx:xx IEEE 802.11: disassociated due to inactivity
Sat Dec 26 00:01:37 2015 kern.err kernel: [101791.011923] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Dec 26 00:01:37 2015 kern.err kernel: [101791.018202] ieee80211 phy0: return code: 0x1122
Sat Dec 26 00:01:37 2015 kern.err kernel: [101791.022893] ieee80211 phy0: timeout: 0x1122
Sat Dec 26 00:01:37 2015 kern.err kernel: [101791.027262] ieee80211 phy0: failed execution
Sat Dec 26 00:01:37 2015 kern.err kernel: [101791.031709] wlan0: failed to remove key (0, 78:7e:61:b8:71:5d) from hardware (-5)
Sat Dec 26 00:01:37 2015 daemon.info hostapd: wlan0: STA 70:56:81:xx:xx:xx IEEE 802.11: disassociated due to inactivity
Sat Dec 26 00:01:39 2015 kern.err kernel: [101793.051670] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Dec 26 00:01:39 2015 kern.err kernel: [101793.057889] ieee80211 phy0: return code: 0x1122
Sat Dec 26 00:01:39 2015 kern.err kernel: [101793.062525] ieee80211 phy0: timeout: 0x1122
Sat Dec 26 00:01:39 2015 kern.err kernel: [101793.066837] ieee80211 phy0: failed execution
Sat Dec 26 00:01:39 2015 kern.err kernel: [101793.071228] wlan0: failed to remove key (0, 70:56:81:xx:xx:xx) from hardware (-5)
Sat Dec 26 00:01:39 2015 daemon.info hostapd: wlan0: STA 78:7e:61:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Dec 26 00:01:41 2015 kern.err kernel: [101795.096057] ieee80211 phy0: cmd 0x9125=BAStream timed out
Sat Dec 26 00:01:41 2015 kern.err kernel: [101795.101581] ieee80211 phy0: return code: 0x1125
Sat Dec 26 00:01:41 2015 kern.err kernel: [101795.106251] ieee80211 phy0: timeout: 0x1125
Sat Dec 26 00:01:41 2015 kern.err kernel: [101795.110581] ieee80211 phy0: destroy ba failed execution
Sat Dec 26 00:01:43 2015 kern.err kernel: [101797.144347] ieee80211 phy0: cmd 0x9125=BAStream timed out
Sat Dec 26 00:01:43 2015 kern.err kernel: [101797.149870] ieee80211 phy0: return code: 0x1125
Sat Dec 26 00:01:43 2015 kern.err kernel: [101797.154518] ieee80211 phy0: timeout: 0x1125
Sat Dec 26 00:01:43 2015 kern.err kernel: [101797.158825] ieee80211 phy0: destroy ba failed execution
Sat Dec 26 00:01:45 2015 kern.err kernel: [101799.184537] ieee80211 phy0: cmd 0x9125=BAStream timed out
Sat Dec 26 00:01:45 2015 kern.err kernel: [101799.190119] ieee80211 phy0: return code: 0x1125
Sat Dec 26 00:01:45 2015 kern.err kernel: [101799.194807] ieee80211 phy0: timeout: 0x1125
Sat Dec 26 00:01:45 2015 kern.err kernel: [101799.199174] ieee80211 phy0: destroy ba failed execution
Sat Dec 26 00:01:47 2015 kern.err kernel: [101801.223587] ieee80211 phy0: cmd 0x9125=BAStream timed out
Sat Dec 26 00:01:47 2015 kern.err kernel: [101801.229153] ieee80211 phy0: return code: 0x1125
Sat Dec 26 00:01:47 2015 kern.err kernel: [101801.233845] ieee80211 phy0: timeout: 0x1125
Sat Dec 26 00:01:47 2015 kern.err kernel: [101801.238211] ieee80211 phy0: destroy ba failed execution
Sat Dec 26 00:01:49 2015 kern.err kernel: [101803.265832] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Sat Dec 26 00:01:49 2015 kern.err kernel: [101803.271843] ieee80211 phy0: return code: 0x1111
Sat Dec 26 00:01:49 2015 kern.err kernel: [101803.276542] ieee80211 phy0: timeout: 0x1111
Sat Dec 26 00:01:49 2015 kern.err kernel: [101803.280902] ieee80211 phy0: failed execution
Sat Dec 26 00:01:49 2015 daemon.info hostapd: wlan0: STA 70:56:81:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Dec 26 00:01:51 2015 kern.err kernel: [101805.298077] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Sat Dec 26 00:01:51 2015 kern.err kernel: [101805.304037] ieee80211 phy0: return code: 0x1111
Sat Dec 26 00:01:51 2015 kern.err kernel: [101805.308692] ieee80211 phy0: timeout: 0x1111
Sat Dec 26 00:01:51 2015 kern.err kernel: [101805.312991] ieee80211 phy0: failed execution
Sat Dec 26 00:01:53 2015 kern.err kernel: [101807.338583] ieee80211 phy0: cmd 0x8127=SetInformationElements timed out
Sat Dec 26 00:01:53 2015 kern.err kernel: [101807.345402] ieee80211 phy0: return code: 0x0127
Sat Dec 26 00:01:53 2015 kern.err kernel: [101807.350090] ieee80211 phy0: timeout: 0x0127
Sat Dec 26 00:01:53 2015 kern.err kernel: [101807.354452] ieee80211 phy0: failed execution
Sat Dec 26 00:01:55 2015 kern.err kernel: [101809.380083] ieee80211 phy0: cmd 0x8050=broadcast_ssid_enable timed out
Sat Dec 26 00:01:55 2015 kern.err kernel: [101809.386813] ieee80211 phy0: return code: 0x0050
Sat Dec 26 00:01:55 2015 kern.err kernel: [101809.391507] ieee80211 phy0: timeout: 0x0050
Sat Dec 26 00:01:55 2015 kern.err kernel: [101809.395882] ieee80211 phy0: failed execution
Sat Dec 26 00:01:57 2015 kern.err kernel: [101811.431214] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
Sat Dec 26 00:01:57 2015 kern.err kernel: [101811.437583] ieee80211 phy0: return code: 0x001c
Sat Dec 26 00:01:57 2015 kern.err kernel: [101811.442273] ieee80211 phy0: timeout: 0x001c
Sat Dec 26 00:01:57 2015 kern.err kernel: [101811.446642] ieee80211 phy0: failed execution
Sat Dec 26 00:01:59 2015 kern.err kernel: [101813.470283] ieee80211 phy0: cmd 0x8126=SetFixedRate timed out
Sat Dec 26 00:01:59 2015 kern.err kernel: [101813.476211] ieee80211 phy0: return code: 0x0126
Sat Dec 26 00:01:59 2015 kern.err kernel: [101813.480901] ieee80211 phy0: timeout: 0x0126
Sat Dec 26 00:01:59 2015 kern.err kernel: [101813.485268] ieee80211 phy0: failed execution
Sat Dec 26 00:05:22 2015 kern.err kernel: [102016.775093] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Dec 26 00:05:22 2015 kern.err kernel: [102016.781317] ieee80211 phy0: return code: 0x1122
Sat Dec 26 00:05:22 2015 kern.err kernel: [102016.785973] ieee80211 phy0: timeout: 0x1122
Sat Dec 26 00:05:22 2015 kern.err kernel: [102016.790271] ieee80211 phy0: failed execution
Sat Dec 26 00:05:22 2015 kern.err kernel: [102016.794664] wlan0: failed to remove key (1, ff:ff:ff:ff:ff:ff) from hardware (-5)
Sat Dec 26 00:05:24 2015 kern.err kernel: [102018.813726] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Dec 26 00:05:24 2015 kern.err kernel: [102018.819947] ieee80211 phy0: return code: 0x1122
Sat Dec 26 00:05:24 2015 kern.err kernel: [102018.824603] ieee80211 phy0: timeout: 0x1122
Sat Dec 26 00:05:24 2015 kern.err kernel: [102018.828902] ieee80211 phy0: failed execution
Sat Dec 26 00:05:24 2015 kern.err kernel: [102018.833294] wlan0: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-5)
Sat Dec 26 00:15:26 2015 kern.err kernel: [102620.850188] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Dec 26 00:15:26 2015 kern.err kernel: [102620.856469] ieee80211 phy0: return code: 0x1122
Sat Dec 26 00:15:26 2015 kern.err kernel: [102620.861167] ieee80211 phy0: timeout: 0x1122
Sat Dec 26 00:15:26 2015 kern.err kernel: [102620.865523] ieee80211 phy0: failed execution
Sat Dec 26 00:15:26 2015 kern.err kernel: [102620.869963] wlan0: failed to remove key (2, ff:ff:ff:ff:ff:ff) from hardware (-5)
Sat Dec 26 00:15:28 2015 kern.err kernel: [102622.896660] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Dec 26 00:15:28 2015 kern.err kernel: [102622.902936] ieee80211 phy0: return code: 0x1122
Sat Dec 26 00:15:28 2015 kern.err kernel: [102622.907628] ieee80211 phy0: timeout: 0x1122
Sat Dec 26 00:15:28 2015 kern.err kernel: [102622.911994] ieee80211 phy0: failed execution
Sat Dec 26 00:15:28 2015 kern.err kernel: [102622.916414] wlan0: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)

jianyunet, 2016-02-10:
Reproduced the same issue (15.05 release, mwlwifi 10.3.0.16-20160105):

```
[38869.651806] nf_conntrack: automatic helper assignment is deprecated and it will be removed soon. Use the iptables CT target to attach helpers instead.
[45020.859958] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[45020.866082] ieee80211 phy0: return code: 0x1122
[45020.870635] ieee80211 phy0: timeout: 0x1122
[45020.874969] ieee80211 phy0: failed execution
[45020.879266] wlan0: failed to remove key (0, ac:d1:b8:e5:1c:85) from hardware (-5)
[45022.884961] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[45022.891085] ieee80211 phy0: return code: 0x1122
[45022.895633] ieee80211 phy0: timeout: 0x1122
[45022.900223] ieee80211 phy0: failed execution
[45022.904515] wlan0: failed to remove key (0, 34:13:e8:38:f4:0f) from hardware (-5)
[45024.910222] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[45024.916345] ieee80211 phy0: return code: 0x1122
[45024.920898] ieee80211 phy0: timeout: 0x1122
[45024.925275] ieee80211 phy0: failed execution
[45024.929573] wlan0: failed to remove key (0, a4:5e:60:d7:71:f9) from hardware (-5)
[45026.935868] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[45026.941992] ieee80211 phy0: return code: 0x1122
[45026.946541] ieee80211 phy0: timeout: 0x1122
[45026.951324] ieee80211 phy0: failed execution
[45026.955622] wlan0: failed to remove key (0, b8:e8:56:95:c1:3e) from hardware (-5)
[45028.961266] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[45028.967387] ieee80211 phy0: return code: 0x1122
[45028.971940] ieee80211 phy0: timeout: 0x1122
[45028.976378] ieee80211 phy0: failed execution
[45028.980680] wlan0: failed to remove key (2, ff:ff:ff:ff:ff:ff) from hardware (-5)
[45030.986144] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[45030.992268] ieee80211 phy0: return code: 0x1122
[45030.996816] ieee80211 phy0: timeout: 0x1122
[45031.001318] ieee80211 phy0: failed execution
[45031.005611] wlan0: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
[45033.010829] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[45033.016951] ieee80211 phy0: return code: 0x1122
[45033.021505] ieee80211 phy0: timeout: 0x1122
[45033.025914] ieee80211 phy0: failed execution
[45033.030212] wlan0: failed to remove key (0, 6c:72:e7:5a:90:5e) from hardware (-5)
```

francispereira, 2016-02-22:
Ran into this today. Devices were not able to connect to 2.4GHz. 5GHz was working fine. Had to reboot.

Marvell 802.11ac Wireless Network Driver version 10.3.0.16-20160105

root@MCDEBIAN:~# dmesg | grep phy0
[   15.065649] ieee80211 phy0: regdomain: FCC
[   15.065654] ieee80211 phy0: Channel: 1: 0x0 0x0 0xf
[   15.065665] ieee80211 phy0: 17 11 11 11 10 10 10 10 f f f f 0 0 0 0
[   15.065670] ieee80211 phy0: Channel: 2: 0x0 0x0 0xf
[   15.065680] ieee80211 phy0: 17 16 16 16 16 16 16 14 10 10 10 10 0 0 0 0
[   15.065685] ieee80211 phy0: Channel: 3: 0x0 0x0 0xf
[   15.065696] ieee80211 phy0: 17 16 16 16 16 16 16 14 10 10 10 10 0 0 0 0
[   15.065701] ieee80211 phy0: Channel: 4: 0x0 0x0 0xf
[   15.065711] ieee80211 phy0: 17 16 16 16 16 16 16 14 10 10 10 10 0 0 0 0
[   15.065716] ieee80211 phy0: Channel: 5: 0x0 0x0 0xf
[   15.065727] ieee80211 phy0: 17 16 16 16 16 16 16 14 10 10 10 10 0 0 0 0
[   15.065731] ieee80211 phy0: Channel: 6: 0x0 0x0 0xf
[   15.065742] ieee80211 phy0: 17 16 16 16 16 16 16 14 10 10 10 10 0 0 0 0
[   15.065747] ieee80211 phy0: Channel: 7: 0x0 0x0 0xf
[   15.065757] ieee80211 phy0: 17 16 16 16 16 16 16 14 10 10 10 10 0 0 0 0
[   15.065762] ieee80211 phy0: Channel: 8: 0x0 0x0 0xf
[   15.065773] ieee80211 phy0: 17 16 16 16 16 16 16 14 10 10 10 10 0 0 0 0
[   15.065778] ieee80211 phy0: Channel: 9: 0x0 0x0 0xf
[   15.065788] ieee80211 phy0: 17 16 16 16 16 16 16 14 10 10 10 10 0 0 0 0
[   15.065793] ieee80211 phy0: Channel: 10: 0x0 0x0 0xf
[   15.065804] ieee80211 phy0: 17 16 16 16 16 16 16 14 10 10 10 10 0 0 0 0
[   15.065809] ieee80211 phy0: Channel: 11: 0x0 0x0 0xf
[   15.065819] ieee80211 phy0: 17 11 11 11 11 11 11 11 10 10 10 10 0 0 0 0
[   15.305040] ieee80211 phy0: change: 0x120
[   15.310108] ieee80211 phy0: change: 0x42
[428303.570246] ieee80211 phy0: cmd 0x9125=BAStream timed out
[428303.575756] ieee80211 phy0: return code: 0x1125
[428303.580404] ieee80211 phy0: timeout: 0x1125
[428303.586920] ieee80211 phy0: destroy ba failed execution
[428305.586508] ieee80211 phy0: cmd 0x8205=GetWatchdogBitMap timed out
[428305.592809] ieee80211 phy0: return code: 0x0205
[428305.597439] ieee80211 phy0: timeout: 0x0205
[428305.604726] ieee80211 phy0: failed execution
[428307.602973] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[428307.609184] ieee80211 phy0: return code: 0x1122
[428307.613814] ieee80211 phy0: timeout: 0x1122
[428307.619078] ieee80211 phy0: failed execution
[428309.623387] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[428309.629334] ieee80211 phy0: return code: 0x1111
[428309.633965] ieee80211 phy0: timeout: 0x1111
[428309.639188] ieee80211 phy0: failed execution
 
 
root@MCDEBIAN:~# uptime
 01:35:02 up 4 days, 23:20,  1 user,  load average: 0.00, 0.02, 0.05
 
 
 root@MCDEBIAN:~#  journalctl 
Feb 22 01:13:11 MCDEBIAN hostapd[2575]: wlan0: STA aa:bb:d0:4d:8e:xx IEEE 802.11: authenticated
Feb 22 01:13:11 MCDEBIAN hostapd[2575]: wlan0: STA aa:bb:d0:4d:8e:xx IEEE 802.11: associated (aid 10)
Feb 22 01:13:12 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9125=BAStream timed out
Feb 22 01:13:12 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1125
Feb 22 01:13:12 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1125
Feb 22 01:13:12 MCDEBIAN kernel: ieee80211 phy0: destroy ba failed execution
Feb 22 01:13:14 MCDEBIAN kernel: ieee80211 phy0: cmd 0x8205=GetWatchdogBitMap timed out
Feb 22 01:13:14 MCDEBIAN kernel: ieee80211 phy0: return code: 0x0205
Feb 22 01:13:14 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x0205
Feb 22 01:13:14 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 22 01:13:16 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 22 01:13:16 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 22 01:13:16 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 22 01:13:16 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 22 01:13:16 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:d0:4d:8e:xx) from hardware (-5)
Feb 22 01:13:18 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Feb 22 01:13:18 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1111
Feb 22 01:13:18 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1111
Feb 22 01:13:18 MCDEBIAN kernel: ieee80211 phy0: failed execution

francispereira, 2016-02-23:
Happened again today. Could connect using 5GHz and reboot the device. Logs along with a kernel stack trace attached to help with debugging.

```
Feb 23 01:25:11 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9125=BAStream timed out
Feb 23 01:25:11 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1125
Feb 23 01:25:11 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1125
Feb 23 01:25:11 MCDEBIAN kernel: ieee80211 phy0: destroy ba failed execution
Feb 23 01:25:13 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:25:13 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:25:13 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:25:13 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:13 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:25:15 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Feb 23 01:25:15 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1111
Feb 23 01:25:15 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1111
Feb 23 01:25:15 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:17 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:25:17 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:25:17 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:25:17 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:17 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:25:19 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Feb 23 01:25:19 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1111
Feb 23 01:25:19 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1111
Feb 23 01:25:19 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:48 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:25:48 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:25:48 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:25:48 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:48 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:25:50 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:25:50 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:25:50 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:25:50 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:50 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:25:52 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:25:52 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:25:52 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:25:52 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:52 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:25:54 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:25:54 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:25:54 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:25:54 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:54 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:25:56 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:25:56 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:25:56 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:25:56 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:56 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:25:58 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:25:58 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:25:58 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:25:58 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:25:58 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:26:00 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:26:00 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:26:00 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:26:00 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:26:00 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:26:02 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:26:02 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:26:02 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:26:02 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:26:02 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
Feb 23 01:26:04 MCDEBIAN kernel: ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Feb 23 01:26:04 MCDEBIAN kernel: ieee80211 phy0: return code: 0x1122
Feb 23 01:26:04 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x1122
Feb 23 01:26:04 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:26:04 MCDEBIAN kernel: wlan1: failed to remove key (0, aa:bb:cc:dd:ee:ff) from hardware (-5)
[... 263 lines cut ...]
Feb 23 01:27:54 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x010a
Feb 23 01:27:54 MCDEBIAN kernel: ieee80211 phy0: failed execution
[... cut ...]
Feb 23 01:28:02 MCDEBIAN kernel: ieee80211 phy0: timeout: 0x0203
Feb 23 01:28:02 MCDEBIAN kernel: ieee80211 phy0: failed execution
Feb 23 01:28:02 MCDEBIAN kernel: br0: port 1(eth1) entered disabled state
Feb 23 01:28:02 MCDEBIAN kernel: device eth1 left promiscuous mode
Feb 23 01:28:02 MCDEBIAN kernel: br0: port 1(eth1) entered disabled state
Feb 23 01:28:02 MCDEBIAN kernel: IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
Feb 23 01:28:03 MCDEBIAN kernel: watchdog watchdog0: watchdog did not stop!
```

francispereira, 2016-03-14:
Few observations on the issue.

1. It always appears to happen after the dhcp server (ISC dhcpd) sends a DHCPACK to a client. 
2. I have seen this lockup being triggered for a wide variety of clients. Sometimes by a MacBook Pro, sometimes by a Lenovo all-in-one PC. It's certainly not limited to Apple devices
3. It only affects 2.4Ghz. 5GHz continues to work
4. Disabling 802.11ac on 5GHz does not help
5. Disabling the 5GHz radio does not help.

yuhhaurlin, 2016-03-25:
Can you try the new driver .17 and firmware image? Thanks.

jglooije, 2016-03-27:
@yuhhaurlin tested the .17 driver, the issue is still there

[111090.306937] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[111090.313155] ieee80211 phy1: return code: 0x1122
[111090.317808] ieee80211 phy1: timeout: 0x1122
[111090.322098] ieee80211 phy1: failed execution
[111090.326487] wlan1: failed to set key (2, ff:if:if:if:if:if) to hardware (-5)

yuhhaurlin, 2016-03-27:
With the new firmware? Thanks.

jglooije, 2016-03-28:
Yes it is

Date: Sun, 27 Mar 2016 14:11:44 -0700

yuhhaurlin, 2016-03-28:
Is there any way to reproduce this problem easily?

jglooije, 2016-03-28:
I don't know how to reproduce but  i think i have the starting event in this log. Maybe you can figure it out for a triggerevent. If there is anything else i can do to help letme please know.

Mon Mar 28 21:08:17 2016 daemon.info hostapd: wlan1: STA 64:fb:81:06:31:3a WPA: group key handshake completed (RSN)
Mon Mar 28 21:08:52 2016 daemon.info hostapd: wlan1: STA 78:0c:b8:5d:e1:6b IEEE 802.11: authenticated
Mon Mar 28 21:08:52 2016 daemon.info hostapd: wlan1: STA 78:0c:b8:5d:e1:6b IEEE 802.11: associated (aid 6)
Mon Mar 28 21:08:52 2016 daemon.info hostapd: wlan0: STA 60:21:c0:3b:56:ea IEEE 802.11: disassociated due to inactivity
Mon Mar 28 21:08:54 2016 kern.err kernel: [130850.359196] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Mon Mar 28 21:08:54 2016 kern.err kernel: [130850.365416] ieee80211 phy0: return code: 0x1122
Mon Mar 28 21:08:54 2016 kern.err kernel: [130850.370056] ieee80211 phy0: timeout: 0x1122
Mon Mar 28 21:08:54 2016 kern.err kernel: [130850.374355] ieee80211 phy0: failed execution
Mon Mar 28 21:08:54 2016 kern.err kernel: [130850.378735] wlan0: failed to remove key (0, 60:21:c0:3b:56:ea) from hardware (-5)
Mon Mar 28 21:08:54 2016 daemon.info hostapd: wlan0: STA 60:21:c0:3b:56:ea IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Mar 28 21:08:56 2016 kern.err kernel: [130852.386981] ieee80211 phy0: cmd 0x9125=BAStream timed out
Mon Mar 28 21:08:56 2016 kern.err kernel: [130852.392504] ieee80211 phy0: return code: 0x1125
Mon Mar 28 21:08:56 2016 kern.err kernel: [130852.397156] ieee80211 phy0: timeout: 0x1125
Mon Mar 28 21:08:56 2016 kern.err kernel: [130852.401462] ieee80211 phy0: destroy ba failed execution
Mon Mar 28 21:08:59 2016 kern.err kernel: [130854.407049] ieee80211 phy0: cmd 0x9125=BAStream timed out
Mon Mar 28 21:08:59 2016 kern.err kernel: [130854.412570] ieee80211 phy0: return code: 0x1125
Mon Mar 28 21:08:59 2016 kern.err kernel: [130854.417210] ieee80211 phy0: timeout: 0x1125
Mon Mar 28 21:08:59 2016 kern.err kernel: [130854.421515] ieee80211 phy0: destroy ba failed execution
Mon Mar 28 21:09:01 2016 kern.err kernel: [130856.427156] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Mon Mar 28 21:09:01 2016 kern.err kernel: [130856.433114] ieee80211 phy0: return code: 0x1111
Mon Mar 28 21:09:01 2016 kern.err kernel: [130856.437776] ieee80211 phy0: timeout: 0x1111
Mon Mar 28 21:09:01 2016 kern.err kernel: [130856.442073] ieee80211 phy0: failed execution
Mon Mar 28 21:09:01 2016 daemon.info hostapd: wlan1: STA 78:0c:b8:5d:e1:6b WPA: pairwise key handshake completed (RSN)
Mon Mar 28 21:09:02 2016 daemon.warn odhcpd[1163]: DHCPV6 SOLICIT IA_NA from 000100011dc49727d8cb8a84a80e on br-lan: ok fdfb:a642:1993::7f4/128 
Mon Mar 28 21:09:02 2016 daemon.warn odhcpd[1163]: DHCPV6 REQUEST IA_NA from 000100011dc49727d8cb8a84a80e on br-lan: ok fdfb:a642:1993::7f4/128 
Mon Mar 28 21:09:02 2016 daemon.info dnsmasq[8691]: read /etc/hosts - 4 addresses
Mon Mar 28 21:09:02 2016 daemon.info dnsmasq[8691]: read /tmp/hosts/dhcp - 7 addresses
Mon Mar 28 21:09:02 2016 daemon.info dnsmasq[8691]: read /tmp/hosts/odhcpd - 3 addresses
Mon Mar 28 21:09:02 2016 daemon.info dnsmasq-dhcp[8691]: read /etc/ethers - 0 addresses
Mon Mar 28 21:09:02 2016 daemon.info dnsmasq[8691]: read /etc/hosts - 4 addresses
Mon Mar 28 21:09:02 2016 daemon.info dnsmasq[8691]: read /tmp/hosts/dhcp - 7 addresses
Mon Mar 28 21:09:02 2016 daemon.info dnsmasq[8691]: read /tmp/hosts/odhcpd - 3 addresses
Mon Mar 28 21:09:02 2016 daemon.info dnsmasq-dhcp[8691]: read /etc/ethers - 0 addresses
Mon Mar 28 21:09:04 2016 daemon.info hostapd: wlan0: STA 78:0c:b8:5d:e1:6b IEEE 802.11: disassociated due to inactivity
Mon Mar 28 21:09:06 2016 kern.err kernel: [130861.460195] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Mon Mar 28 21:09:06 2016 kern.err kernel: [130861.466417] ieee80211 phy0: return code: 0x1122
Mon Mar 28 21:09:06 2016 kern.err kernel: [130861.471059] ieee80211 phy0: timeout: 0x1122
Mon Mar 28 21:09:06 2016 kern.err kernel: [130861.475365] ieee80211 phy0: failed execution
Mon Mar 28 21:09:06 2016 kern.err kernel: [130861.479768] wlan0: failed to remove key (0, 78:0c:b8:5d:e1:6b) from hardware (-5)
Mon Mar 28 21:09:06 2016 daemon.warn odhcpd[1163]: DHCPV6 CONFIRM IA_NA from 000100011dc49727d8cb8a84a80e on br-lan: not on-link fdfb:a642:1993::7f4/128 
Mon Mar 28 21:09:06 2016 daemon.info hostapd: wlan0: STA 78:0c:b8:5d:e1:6b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Mar 28 21:09:08 2016 kern.err kernel: [130863.487906] ieee80211 phy0: cmd 0x9125=BAStream timed out
Mon Mar 28 21:09:08 2016 kern.err kernel: [130863.493429] ieee80211 phy0: return code: 0x1125
Mon Mar 28 21:09:08 2016 kern.err kernel: [130863.498081] ieee80211 phy0: timeout: 0x1125
Mon Mar 28 21:09:08 2016 kern.err kernel: [130863.502411] ieee80211 phy0: destroy ba failed execution
Mon Mar 28 21:09:10 2016 kern.err kernel: [130865.508159] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Mon Mar 28 21:09:10 2016 kern.err kernel: [130865.514115] ieee80211 phy0: return code: 0x1111
Mon Mar 28 21:09:10 2016 kern.err kernel: [130865.518751] ieee80211 phy0: timeout: 0x1111
Mon Mar 28 21:09:10 2016 kern.err kernel: [130865.523044] ieee80211 phy0: failed execution
Mon Mar 28 21:09:10 2016 daemon.info dnsmasq-dhcp[8691]: DHCPDISCOVER(br-lan) 78:0c:b8:5d:e1:6b 
Mon Mar 28 21:09:10 2016 daemon.info dnsmasq-dhcp[8691]: DHCPOFFER(br-lan) 192.168.144.25 78:0c:b8:5d:e1:6b 
[... cut ...]
Mon Mar 28 21:09:34 2016 kern.err kernel: [130889.776598] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Mon Mar 28 21:09:34 2016 kern.err kernel: [130889.782813] ieee80211 phy0: return code: 0x1122
Mon Mar 28 21:09:34 2016 kern.err kernel: [130889.787449] ieee80211 phy0: timeout: 0x1122
Mon Mar 28 21:09:34 2016 kern.err kernel: [130889.791736] ieee80211 phy0: failed execution
Mon Mar 28 21:09:34 2016 kern.err kernel: [130889.796117] wlan0-1: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-5)

yuhhaurlin, 2016-03-29:
BTW, can you reconfirm the version of driver and firmware you used via "cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info"? Thanks.

francispereira, 2016-03-29:
@yuhhaurlin I think this is got to do with number of connected stations. My device at home does not run into this. 30 days uptime until I powered it down myself. Similar device (WRT1900ac v1) running in office runs into this issue every few days. Only happens in the mornings when the numbers of connected devices increase. At peak I have ~40 connected devices. I have never seen this issue occur overnight when there is no one in the office. How many devices do you have in your test setup ?

@jglooije What is the max number of stations you have connected ?

yuhhaurlin, 2016-03-29:
@ francispereira When the problem happened, only one device can't work? Can you help to enlarge following constant in fwcmd.c to see if this problem is gone?

#define MAX_WAIT_FW_COMPLETE_ITERATIONS         2000  => 10000

francispereira, 2016-03-29:
@Chadster766 set it to 10000 for me on .16 few weeks ago. 10000 because it was in the original driver released by Marvell. With 10000, the radio on which it triggers goes offline while the other one continues to operate.

yuhhaurlin, 2016-03-29:
So if this number is set to 10000, the problem is gone? Due to mwlwifi uses mutex now, we can even enlarge the number.

francispereira, 2016-03-29:
Not completely. The issue gets isolated to the radio where its triggered. The other radio continues to operate. We still need to gracefully recover from problem without bringing down the affected radio.

yuhhaurlin, 2016-03-29:
There are two 88W8864 modules. From your description, even though this number is set to 10000, this  
 kind of messages will still come out and the module can't work any more unless you reboot the device.

@jglooije How many client devices you used on your test environment?

francispereira, 2016-03-29:
@yuhhaurlin thats correct.

jglooije, 2016-03-29:
@yuhhaurlin here is the asked information.

I have around 10 active wifi clients and bellow is the driver information. 

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.17-20160324
firmware version: 0x07020902
mac address: 00:25:9c:xx:xx:xx
2g: disable
5g: enable
antenna: 4 4
irq number: 103
iobase0: e1080000
iobase1: e1200000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
qe trigger number: 1188644
mfg mode: false

Thanks for your active support, much appreciated.

jglooije, 2016-03-29:
@yuhhaurlin this was in my log a moment ago. I was online with my laptop doing a speed test and a radar detected came and my wifi did not return. See log below, for your information. 

[24856.984236] ieee80211 phy0: radar detected by firmware
[24857.516952] ieee80211 phy0: channel switch is done
[24857.516972] ieee80211 phy0: change: 0x40
[25171.740099] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[25171.746234] ieee80211 phy0: return code: 0x1122
[25171.750786] ieee80211 phy0: timeout: 0x1122
[25171.754999] ieee80211 phy0: failed execution
[25171.759311] wlan0: failed to remove key (0, 60:21:c0:3b:56:ea) from hardware (-5)
[25173.767360] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[25173.773221] ieee80211 phy0: return code: 0x1111
[25173.777782] ieee80211 phy0: timeout: 0x1111
[25173.781981] ieee80211 phy0: failed execution
[25183.124718] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[25183.130845] ieee80211 phy0: return code: 0x1122
[25183.135402] ieee80211 phy0: timeout: 0x1122
[25183.139602] ieee80211 phy0: failed execution
[25183.143890] wlan0: failed to remove key (0, 78:0c:b8:5d:e1:6b) from hardware (-5)
[25185.152200] ieee80211 phy0: cmd 0x9125=BAStream timed out
[25185.157631] ieee80211 phy0: return code: 0x1125
[25185.162179] ieee80211 phy0: timeout: 0x1125
[25185.166386] ieee80211 phy0: destroy ba failed execution
[25187.171826] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[25187.177691] ieee80211 phy0: return code: 0x1111
[25187.182239] ieee80211 phy0: timeout: 0x1111
[25187.186447] ieee80211 phy0: failed execution
[25189.191059] ieee80211 phy0: cmd 0x8050=broadcast_ssid_enable timed out
[25189.197621] ieee80211 phy0: return code: 0x0050
[25189.202170] ieee80211 phy0: timeout: 0x0050
[25189.206377] ieee80211 phy0: failed execution
[25191.210785] ieee80211 phy0: cmd 0x8127=SetInformationElements timed out
[25191.217434] ieee80211 phy0: return code: 0x0127
[25191.221983] ieee80211 phy0: timeout: 0x0127
[25191.226190] ieee80211 phy0: failed execution
[25193.230694] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
[25193.236908] ieee80211 phy0: return code: 0x001c
[25193.241456] ieee80211 phy0: timeout: 0x001c
[25193.245664] ieee80211 phy0: failed execution
[25195.250066] ieee80211 phy0: cmd 0x8126=SetFixedRate timed out
[25195.255841] ieee80211 phy0: return code: 0x0126
[25195.260390] ieee80211 phy0: timeout: 0x0126
[25195.264597] ieee80211 phy0: failed execution
[25197.269106] ieee80211 phy0: cmd 0x8050=broadcast_ssid_enable timed out
[25197.275667] ieee80211 phy0: return code: 0x0050
[25197.280214] ieee80211 phy0: timeout: 0x0050
[25197.284442] ieee80211 phy0: failed execution
[25199.288858] ieee80211 phy0: cmd 0x8127=SetInformationElements timed out

yuhhaurlin, 2016-03-29:
Thanks for your information.

werunthenite, 2016-04-05:
I have the same trouble, 2.4Ghz and MacBook Pro. 

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.16-20160105
firmware version: 0x07020902
mac address: 00:25:9c:13:30:2c
2g: enable
5g: disable
antenna: 4 4
irq number: 87
iobase0: d0e00000
iobase1: d1000000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 827
mfg mode: false

I don't know why the driver version is still in 10.3.0.16, I downloaded last 88W8864.bin and placed it into /lib/firmware/mwlwifi/

Any ideas?

northbound1, 2016-04-05:
You need to update the mwlwifi.ko also

werunthenite, 2016-04-05:
@northbound1 How can I do it?

werunthenite, 2016-04-05:
I updated it. Here you have:

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.17-20160324
firmware version: 0x07020902
mac address: 00:25:9c:13:30:2c
2g: enable
5g: disable
antenna: 4 4
irq number: 87
iobase0: d0e00000
iobase1: d1000000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 19419
mfg mode: false

Also here's the system log:

Tue Apr  5 12:49:41 2016 authpriv.notice dropbear[1582]: Password auth succeeded for 'root' from 192.168.1.180:54542
Tue Apr  5 12:50:03 2016 authpriv.info dropbear[1582]: Exit (root): Disconnect received
Tue Apr  5 12:50:09 2016 cron.err crond[1180]: time disparity of 16124 minutes detected
Tue Apr  5 12:51:45 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 IEEE 802.11: authenticated
Tue Apr  5 12:51:45 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 IEEE 802.11: associated (aid 2)
Tue Apr  5 12:51:45 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 WPA: pairwise key handshake completed (RSN)
Tue Apr  5 12:51:47 2016 daemon.info dnsmasq-dhcp[1523]: DHCPREQUEST(br-lan) 192.168.1.180 78:31:c1:d5:4f:b8 
Tue Apr  5 12:51:47 2016 daemon.info dnsmasq-dhcp[1523]: DHCPACK(br-lan) 192.168.1.180 78:31:c1:d5:4f:b8 Alejandros-MBP
Tue Apr  5 12:52:42 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 IEEE 802.11: authenticated
Tue Apr  5 12:52:42 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 IEEE 802.11: associated (aid 2)
Tue Apr  5 12:52:42 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 WPA: pairwise key handshake completed (RSN)
Tue Apr  5 12:52:44 2016 daemon.info dnsmasq-dhcp[1523]: DHCPREQUEST(br-lan) 192.168.1.180 78:31:c1:d5:4f:b8 
Tue Apr  5 12:52:44 2016 daemon.info dnsmasq-dhcp[1523]: DHCPACK(br-lan) 192.168.1.180 78:31:c1:d5:4f:b8 Alejandros-MBP
Tue Apr  5 12:53:05 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 IEEE 802.11: authenticated
Tue Apr  5 12:53:05 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 IEEE 802.11: associated (aid 2)
Tue Apr  5 12:53:05 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 WPA: pairwise key handshake completed (RSN)
Tue Apr  5 12:53:07 2016 daemon.info dnsmasq-dhcp[1523]: DHCPREQUEST(br-lan) 192.168.1.180 78:31:c1:d5:4f:b8 
Tue Apr  5 12:53:07 2016 daemon.info dnsmasq-dhcp[1523]: DHCPACK(br-lan) 192.168.1.180 78:31:c1:d5:4f:b8 Alejandros-MBP
Tue Apr  5 12:53:33 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 IEEE 802.11: authenticated
Tue Apr  5 12:53:33 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 IEEE 802.11: associated (aid 2)
Tue Apr  5 12:53:34 2016 daemon.info hostapd: wlan0: STA 78:31:c1:d5:4f:b8 WPA: pairwise key handshake completed (RSN)
Tue Apr  5 12:53:35 2016 kern.err kernel: [  300.316038] ieee80211 phy0: check ba result error 1
Tue Apr  5 12:53:35 2016 kern.err kernel: [  300.321219] ieee80211 phy0: ampdu start error code: -22
Tue Apr  5 12:53:35 2016 daemon.info dnsmasq-dhcp[1523]: DHCPREQUEST(br-lan) 192.168.1.180 78:31:c1:d5:4f:b8 
Tue Apr  5 12:53:35 2016 daemon.info dnsmasq-dhcp[1523]: DHCPACK(br-lan) 192.168.1.180 78:31:c1:d5:4f:b8 Alejandros-MBP

robd3, 2016-04-06:
I might have the same issue. See dmesg and mwlwifi debug output below. I am using davidc502's build on  wrt1200ac.
There are only few devices connected.

# dmesg
[ 1681.646866] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1681.653003] ieee80211 phy1: return code: 0x1122
[ 1681.657556] ieee80211 phy1: timeout: 0x1122
[ 1681.661774] ieee80211 phy1: failed execution
[ 1681.666070] wlan1: failed to remove key (0, 20:d3:90:a1:b8:34) from hardware                       (-5)
[ 1683.674366] ieee80211 phy1: cmd 0x9125=BAStream timed out
[ 1683.679791] ieee80211 phy1: return code: 0x1125
[ 1683.684350] ieee80211 phy1: timeout: 0x1125
[ 1683.688551] ieee80211 phy1: destroy ba failed execution
[ 1685.694006] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[ 1685.699865] ieee80211 phy1: return code: 0x1111
[ 1685.704422] ieee80211 phy1: timeout: 0x1111
[ 1685.708623] ieee80211 phy1: failed execution
[ 1690.716691] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1690.722825] ieee80211 phy1: return code: 0x1122
[ 1690.727379] ieee80211 phy1: timeout: 0x1122
[ 1690.731594] ieee80211 phy1: failed execution
[ 1690.735887] wlan1: failed to remove key (0, e8:99:c4:8b:8b:91) from hardware                       (-5)
[ 1692.743976] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1692.750101] ieee80211 phy1: return code: 0x1122
[ 1692.754664] ieee80211 phy1: timeout: 0x1122
[ 1692.758874] ieee80211 phy1: failed execution
[ 1692.763179] wlan1: failed to remove key (0, 5c:51:4f:62:ba:06) from hardware                       (-5)
[ 1694.771304] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[ 1694.777164] ieee80211 phy1: return code: 0x1111
[ 1694.781722] ieee80211 phy1: timeout: 0x1111
[ 1694.785922] ieee80211 phy1: failed execution
[ 1696.790992] ieee80211 phy1: cmd 0x9125=BAStream timed out
[ 1696.796428] ieee80211 phy1: return code: 0x1125
[ 1696.800981] ieee80211 phy1: timeout: 0x1125
[ 1696.805199] ieee80211 phy1: destroy ba failed execution
[ 1698.810668] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[ 1698.816536] ieee80211 phy1: return code: 0x1111
[ 1698.821099] ieee80211 phy1: timeout: 0x1111
[ 1698.825306] ieee80211 phy1: failed execution
[ 1700.829874] ieee80211 phy1: cmd 0x8050=broadcast_ssid_enable timed out
[ 1700.836437] ieee80211 phy1: return code: 0x0050
[ 1700.840987] ieee80211 phy1: timeout: 0x0050
[ 1700.845193] ieee80211 phy1: failed execution
[ 1702.849648] ieee80211 phy1: cmd 0x8127=SetInformationElements timed out
[ 1702.856300] ieee80211 phy1: return code: 0x0127
[ 1702.860850] ieee80211 phy1: timeout: 0x0127
[ 1702.865057] ieee80211 phy1: failed execution
[ 1704.869586] ieee80211 phy1: cmd 0x801c=80211RadioControl timed out
[ 1704.875807] ieee80211 phy1: return code: 0x001c
[ 1704.880360] ieee80211 phy1: timeout: 0x001c
[ 1704.884578] ieee80211 phy1: failed execution
[ 1706.889069] ieee80211 phy1: cmd 0x8126=SetFixedRate timed out
[ 1706.894851] ieee80211 phy1: return code: 0x0126
[ 1706.899404] ieee80211 phy1: timeout: 0x0126
[ 1706.903619] ieee80211 phy1: failed execution
[ 1828.421129] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1828.427258] ieee80211 phy1: return code: 0x1122
[ 1828.431826] ieee80211 phy1: timeout: 0x1122
[ 1828.436033] ieee80211 phy1: failed execution
[ 1828.440328] wlan1: failed to remove key (2, ff:ff:ff:ff:ff:ff) from hardware                       (-5)
[ 1830.448137] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1830.454270] ieee80211 phy1: return code: 0x1122
[ 1830.458824] ieee80211 phy1: timeout: 0x1122
[ 1830.463038] ieee80211 phy1: failed execution
[ 1830.467333] wlan1: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
[ 2432.473577] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 2432.479703] ieee80211 phy1: return code: 0x1122
[ 2432.484261] ieee80211 phy1: timeout: 0x1122
[ 2432.488462] ieee80211 phy1: failed execution
[ 2432.492750] wlan1: failed to remove key (1, ff:ff:ff:ff:ff:ff) from hardware                       (-5)
[ 2434.500498] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 2434.506626] ieee80211 phy1: return code: 0x1122
[ 2434.511175] ieee80211 phy1: timeout: 0x1122
[ 2434.515382] ieee80211 phy1: failed execution
[ 2434.519669] wlan1: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-5)

# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.17-20160324
firmware version: 0x07020902
mac address: [...mac address..]
2g: disable
5g: enable
antenna: 2 2
irq number: 103
iobase0: e1080000
iobase1: e1200000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 115062
mfg mode: false

# uname -a
Linux 4.4.6 #1 SMP Fri Mar 25 18:05:58 UTC 2016 armv7l GNU/Linux

jglooije, 2016-04-06:
@everybody
I update my build to openwrt mrfrezee instead of davidc502 build. And did not see the update encryption error anymore for a week now. Before it was at least once a day or more. I am still a bit cautious, but it looks good. I also had a lot of:
[103558.651054] ieee80211 phy1: check ba result error 1
[103558.656055] ieee80211 phy1: ampdu start error code: -22
but it changed the regulatory domain to different once on both interface and for the 5Ghz i choose a country with only DFS and no auto-bw and outdoor. I had the above errors once after a channel switch, so also this looks good. I will try not to play with the router for a week and see how the results are and let you know. An other thing i noticed that the regulatory domain 00-world can not be used in combination with a guest wifi on 5Ghz. The interface would not come up.

robd3, 2016-04-06:
@jglooije 
Thank you for your post!
I changed to mrfrezee (2016-04-03) instead of davidc502 build and the errors are gone so far.

yuhhaurlin, 2016-04-08:
@jglooije what is the mrfrezee (2016-04-03) build?

jglooije, 2016-04-08:
mrfrezee is building the latest trunk version(DD - Designated Driver) of openwrt with some changes. He is using kernel 4.4 instead of 4.1 and some patches that I believe or not yet in trunk. He is posting in this forum https://forum.openwrt.org/viewtopic.php?id=50173 and has is own website where you can upload his build https://openwrt.nc-network.io/.But yesterday I ran in trouble with his build also. I use a nfs kernel server and every time i do a backup the free memory goes down until it reached zero.(file cache design of the kernel) Not really a problem because the free memory is used as a file cache. But the wifi trouble came back at that moment. So I ran a drop_cache on the router and the memory was free again. No memory free could lead to timing problems because of  when a process needs memory it needs to be made free before it can be used. So I give this a try and I automated the drop_cache and tuned some kernel parameters so this should not be an issue anymore. I will give it a try.I don't know if the wifi driver is sensitive to memory allocation or timing. Date: Fri, 8 Apr 2016 00:06:20 -0700

yuhhaurlin, 2016-04-11:
@all Please help to check if this problem only happened when hostapd tried to remove key. Thanks.

lantis1008, 2016-04-14:
@yuhhaurlin
I have a user experiencing this issue as well. And yes this happens when hostapd tries to remove the key.

lantis1008, 2016-04-14:
```
Thu Apr 14 16:00:39 2016 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0)
Thu Apr 14 16:00:39 2016 user.notice firewall: Reloading firewall due to ifup of wan (eth0)
Thu Apr 14 19:30:21 2016 cron.err crond[3413]: time disparity of 209 minutes detected
Thu Apr 14 19:31:42 2016 daemon.info hostapd: wlan1: STA 00:6d:52:19:8c:2a IEEE 802.11: authenticated
Thu Apr 14 19:31:42 2016 daemon.info hostapd: wlan1: STA 00:6d:52:19:8c:2a IEEE 802.11: associated (aid 1)
Thu Apr 14 19:31:42 2016 daemon.info hostapd: wlan1: STA 00:6d:52:19:8c:2a RADIUS: starting accounting session 570F31E6-00000000
Thu Apr 14 19:31:42 2016 daemon.info hostapd: wlan1: STA 00:6d:52:19:8c:2a WPA: pairwise key handshake completed (RSN)
Thu Apr 14 19:31:42 2016 daemon.info dnsmasq-dhcp[2810]: DHCPREQUEST(br-lan) 192.168.1.130 00:6d:52:19:8c:2a 
Thu Apr 14 19:31:42 2016 daemon.info dnsmasq-dhcp[2810]: DHCPACK(br-lan) 192.168.1.130 00:6d:52:19:8c:2a Matthews-iPhone
Thu Apr 14 19:37:15 2016 daemon.info hostapd: wlan1: STA 00:6d:52:19:8c:2a IEEE 802.11: disassociated due to inactivity
Thu Apr 14 19:37:17 2016 kern.err kernel: [  487.225980] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
Thu Apr 14 19:37:17 2016 kern.err kernel: [  487.232104] ieee80211 phy1: return code: 0x1122
Thu Apr 14 19:37:17 2016 kern.err kernel: [  487.236660] ieee80211 phy1: timeout: 0x1122
Thu Apr 14 19:37:17 2016 kern.err kernel: [  487.240859] ieee80211 phy1: failed execution
Thu Apr 14 19:37:17 2016 kern.err kernel: [  487.245152] wlan1: failed to remove key (0, 00:6d:52:19:8c:2a) from hardware (-5)
Thu Apr 14 19:37:17 2016 daemon.info hostapd: wlan1: STA 00:6d:52:19:8c:2a IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Apr 14 19:37:19 2016 kern.err kernel: [  489.246620] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
Thu Apr 14 19:37:19 2016 kern.err kernel: [  489.252479] ieee80211 phy1: return code: 0x1111
Thu Apr 14 19:37:19 2016 kern.err kernel: [  489.257034] ieee80211 phy1: timeout: 0x1111
Thu Apr 14 19:37:19 2016 kern.err kernel: [  489.261233] ieee80211 phy1: failed execution
Thu Apr 14 19:38:26 2016 authpriv.info dropbear[5202]: Child connection from 192.168.1.197:35868
Thu Apr 14 19:38:30 2016 authpriv.notice dropbear[5202]: Password auth succeeded for 'root' from 192.168.1.197:35868
Thu Apr 14 19:39:29 2016 kern.err kernel: [  619.615928] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
Thu Apr 14 19:39:29 2016 kern.err kernel: [  619.622056] ieee80211 phy1: return code: 0x1122
Thu Apr 14 19:39:29 2016 kern.err kernel: [  619.626611] ieee80211 phy1: timeout: 0x1122
Thu Apr 14 19:39:29 2016 kern.err kernel: [  619.630810] ieee80211 phy1: failed execution
Thu Apr 14 19:39:29 2016 kern.err kernel: [  619.635104] wlan1: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
Thu Apr 14 19:44:12 2016 kern.info kernel: [  902.616834] device wlan1 left promiscuous mode
Thu Apr 14 19:44:12 2016 kern.info kernel: [  902.621321] br-lan: port 2(wlan1) entered disabled state
Thu Apr 14 19:44:14 2016 kern.err kernel: [  904.620361] ieee80211 phy1: cmd 0x9100=BssStart timed out
Thu Apr 14 19:44:14 2016 kern.err kernel: [  904.625784] ieee80211 phy1: return code: 0x1100
Thu Apr 14 19:44:14 2016 kern.err kernel: [  904.630339] ieee80211 phy1: timeout: 0x1100
Thu Apr 14 19:44:14 2016 kern.err kernel: [  904.634537] ieee80211 phy1: failed execution
Thu Apr 14 19:44:14 2016 kern.debug kernel: [  904.638833] ieee80211 phy1: change: 0x40
Thu Apr 14 19:44:16 2016 kern.err kernel: [  906.632324] ieee80211 phy1: cmd 0x9114=SetApMode timed out
Thu Apr 14 19:44:16 2016 kern.err kernel: [  906.637835] ieee80211 phy1: return code: 0x1114
Thu Apr 14 19:44:16 2016 kern.err kernel: [  906.642382] ieee80211 phy1: timeout: 0x1114
Thu Apr 14 19:44:16 2016 kern.err kernel: [  906.646581] ieee80211 phy1: failed execution
Thu Apr 14 19:44:18 2016 kern.err kernel: [  908.644405] ieee80211 phy1: cmd 0x8129=LinkAdaptCsMode timed out
Thu Apr 14 19:44:18 2016 kern.err kernel: [  908.650440] ieee80211 phy1: return code: 0x0129
Thu Apr 14 19:44:18 2016 kern.err kernel: [  908.654988] ieee80211 phy1: timeout: 0x0129
Thu Apr 14 19:44:18 2016 kern.err kernel: [  908.659192] ieee80211 phy1: failed execution
Thu Apr 14 19:44:20 2016 kern.err kernel: [  910.656968] ieee80211 phy1: cmd 0x810a=SetRfChannel timed out
Thu Apr 14 19:44:20 2016 kern.err kernel: [  910.662738] ieee80211 phy1: return code: 0x010a
Thu Apr 14 19:44:20 2016 kern.err kernel: [  910.667290] ieee80211 phy1: timeout: 0x010a
Thu Apr 14 19:44:21 2016 kern.err kernel: [  910.671489] ieee80211 phy1: failed execution
Thu Apr 14 19:44:21 2016 kern.debug kernel: [  910.675777] ieee80211 phy1: change: 0x100
Thu Apr 14 19:44:23 2016 kern.err kernel: [  912.669203] ieee80211 phy1: cmd 0x801c=80211RadioControl timed out
Thu Apr 14 19:44:23 2016 kern.err kernel: [  912.675408] ieee80211 phy1: return code: 0x001c
Thu Apr 14 19:44:23 2016 kern.err kernel: [  912.679960] ieee80211 phy1: timeout: 0x001c
Thu Apr 14 19:44:23 2016 kern.err kernel: [  912.684159] ieee80211 phy1: failed execution
Thu Apr 14 19:44:23 2016 daemon.notice netifd: Network device 'wlan1' link is down
Thu Apr 14 19:44:25 2016 kern.err kernel: [  914.682158] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
Thu Apr 14 19:44:25 2016 kern.err kernel: [  914.688281] ieee80211 phy1: return code: 0x1122
Thu Apr 14 19:44:25 2016 kern.err kernel: [  914.692829] ieee80211 phy1: timeout: 0x1122
[... cut ...]
Thu Apr 14 19:44:43 2016 daemon.notice netifd: radio1 (5425): nl80211 driver initialization failed.
Thu Apr 14 19:44:43 2016 daemon.notice netifd: radio1 (5425): wlan1: interface state UNINITIALIZED->DISABLED
Thu Apr 14 19:44:43 2016 daemon.notice netifd: radio1 (5425): wlan1: AP-DISABLED 
Thu Apr 14 19:44:43 2016 daemon.notice netifd: radio1 (5425): hostapd_free_hapd_data: Interface wlan1 wasn't started
Thu Apr 14 19:44:43 2016 daemon.notice netifd: radio1 (5425): Device setup failed: HOSTAPD_START_FAILED
```

I hope this is helpful @yuhhaurlin

yuhhaurlin, 2016-04-14:
Yes, thanks.

gsustek, 2016-05-08:
Hi i have  1900acs using arokh r41966 build from which i removed_mwan3,sqm,nzgbet,strongswan, and some other packages. i have also connected on usb3 one USB 2.0 hub who connects one camera L C270 and USB HDD, on usb2.0 i have another usb 2.0 which connects two L c270 camera, mjpg_streamer using those cameras(3MB throughput whole 3 cameras), this USB hub has external power.  There are two clients iphone 4 conected to 2,4GHz, and ipad Air 2 connected to 5GHZ and Dell laptop Intel Corporation Wireless 7265 (rev 59) as well.

here is the dmesg output after 16 hours.

cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.17-20160324
firmware version: 0x07020902
mac address: 00:25:9c:13:b4:b0
2g: disable
5g: enable
antenna: 4 4
irq number: 104
iobase0: e1200000
iobase1: e1380000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
qe trigger number: 2343855
mfg mode: false

root@1900acs ~# dmesg 
[93180.052755] wlan0-1: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
[93777.953494] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[93777.959755] ieee80211 phy0: return code: 0x1122
[93777.964463] ieee80211 phy0: timeout: 0x1122
[93777.968781] ieee80211 phy0: failed execution
[93777.973165] wlan0: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-5)
[93782.052371] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[93782.058584] ieee80211 phy0: return code: 0x1122
[93782.063266] ieee80211 phy0: timeout: 0x1122
[93782.067536] ieee80211 phy0: failed execution
[93782.071929] wlan0-1: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-5)
[94379.972778] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[94379.979007] ieee80211 phy0: return code: 0x1122
[94379.983639] ieee80211 phy0: timeout: 0x1122
[94379.987950] ieee80211 phy0: failed execution
[94379.992337] wlan0: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
[94384.071519] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[94384.077801] ieee80211 phy0: return code: 0x1122
[94384.082537] ieee80211 phy0: timeout: 0x1122

johnnysl, 2016-07-09:
@yuhhaurlin, as you've closed my second topic (sorry forgot about this one), i can add to the remarks that the same driver works fine with an older build of LEDE, but is absolutly horrible with the new build. 

it almost seems like a race condition triggered only with certain builds?

yuhhaurlin, 2016-07-09:
I will check it. Thanks.

mystica555, 2016-07-10:
My previous installation was openwrt-trunk as of april. It did not have this particular issue with the 4.1 kernel it had.. it may be some combination of brand new kernel...

closed 2016-08-13 by yuhhaurlin

InkblotAdmirer, 2016-09-09:
Just an FYI that I'm not sure this issue is completely closed.
Recent LEDE commits related to hostapd and some kernel changes seem to be causing issues -- but only on the ACS model (not observed on ACV1).
Commit:  no 5G lockups -- https://git.lede-project.org/?p=source.git;a=commit;h=42f559ed70897a7b74dd3e6293b42e6d2e511eaa
Commit:  5G lockups, typically within a few hours -- https://git.lede-project.org/?p=source.git;a=commit;h=083ef3fefecd024b2b9cdc496e1f7811c225041e

jklap, 2016-09-09:
Agreed. See https://github.com/kaloz/mwlwifi/issues/102

mystica555, 2016-10-02:
2016-09-08 Felix Fietkau mac80211: remove the fq-disable hack, now that
reorderi
2016-09-08 Felix Fietkau mac80211: fix packet loss on fq reordering
2016-09-08 Felix Fietkau hostapd: update to version 2016-09-05
2016-09-08 Felix Fietkau kernel: merge a softirq performance improvement
patch
are all the changes past the commit you mention:

2016-09-06 Rafał Miłecki kernel: backport upstream overlayfs fixes commit |
commitdiff | tree | snapshot

So the question is bisecting to find the proper commit affecting mwlwifi's
interaction with any of : softirq and the kernel, hostapd, or mac80211
layers...

the other commits in the mean-time were adding support for entirely
unrelated items and I can't imagine would themselves be the cause.

mystica555, 2016-10-02:
Sorry, linking might be useful.

On Sun, Oct 2, 2016 at 1:21 PM, Mike Hodson <mystica@gmail.com> wrote:

>
> 2016-09-08 Felix Fietkau mac80211: remove the fq-disable hack, now that
> reorderi
>
https://git.lede-project.org/?p=source.git;a=commitdiff;h=c4bfb119d85bcd5faf569f9cc83628ba19f58a1f

> 2016-09-08 Felix Fietkau mac80211: fix packet loss on fq reordering
>
https://git.lede-project.org/?p=source.git;a=commitdiff;h=a194ffd4a89588bc75aeb9a27f59c36afd3d24bd

> 2016-09-08 Felix Fietkau hostapd: update to version 2016-09-05
>
https://git.lede-project.org/?p=source.git;a=commitdiff;h=859d940c79e8cb1c66a3d0cccc18e9eac499415b

> 2016-09-08 Felix Fietkau kernel: merge a softirq performance improvement
> patch
>
https://git.lede-project.org/?p=source.git;a=commitdiff;h=b6cd42a54e901f21d47f52a6e7a61bab83a50f3b

arebokert, 2017-08-24:
I just experienced this issue as well.

`
Thu Aug 24 19:14:39 2017 kern.err kernel: [241258.717570] ieee80211 phy0: cmd 0x9125=BAStream timed out
Thu Aug 24 19:14:39 2017 kern.err kernel: [241258.723147] ieee80211 phy0: return code: 0x1125
Thu Aug 24 19:14:39 2017 kern.err kernel: [241258.727955] ieee80211 phy0: timeout: 0x1125
Thu Aug 24 19:14:41 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.151 to 192.168.1.169
Thu Aug 24 19:14:42 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.169 to 192.168.1.248
Thu Aug 24 19:14:54 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.248 to 192.168.1.151
Thu Aug 24 19:14:58 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 2c:54:cf:e7:26:be
Thu Aug 24 19:14:59 2017 daemon.info dnsmasq-dhcp[26680]: DHCPDISCOVER(br-lan) 2c:54:cf:e7:26:be 
Thu Aug 24 19:14:59 2017 daemon.info dnsmasq-dhcp[26680]: DHCPOFFER(br-lan) 192.168.1.147 2c:54:cf:e7:26:be 
Thu Aug 24 19:14:59 2017 daemon.info dnsmasq-dhcp[26680]: DHCPREQUEST(br-lan) 192.168.1.147 2c:54:cf:e7:26:be 
Thu Aug 24 19:14:59 2017 daemon.info dnsmasq-dhcp[26680]: DHCPACK(br-lan) 192.168.1.147 2c:54:cf:e7:26:be android-35f86d62f09fb010
Thu Aug 24 19:15:00 2017 daemon.info odhcpd[1648]: Using a RA lifetime of 0 seconds on br-lan
Thu Aug 24 19:15:08 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.151 to 192.168.1.248
Thu Aug 24 19:15:11 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.248 to 192.168.1.147
Thu Aug 24 19:15:13 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.147 to 192.168.1.169
Thu Aug 24 19:15:25 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.169 to 192.168.1.147
Thu Aug 24 19:15:25 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.147 to 192.168.1.151
Thu Aug 24 19:15:36 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.151 to 192.168.1.147
Thu Aug 24 19:15:45 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.147 to 192.168.1.151
Thu Aug 24 19:16:04 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 2c:54:cf:e7:26:be
Thu Aug 24 19:17:08 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.151 to 192.168.1.248
Thu Aug 24 19:17:13 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.248 to 192.168.1.169
Thu Aug 24 19:17:25 2017 user.warn igmpproxy[26185]: The origin for route 239.255.255.250 changed from 192.168.1.169 to 192.168.1.151
Thu Aug 24 19:17:58 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 2c:54:cf:e7:26:be
Thu Aug 24 19:18:01 2017 daemon.info dnsmasq-dhcp[26680]: DHCPREQUEST(br-lan) 192.168.1.147 2c:54:cf:e7:26:be 
Thu Aug 24 19:18:01 2017 daemon.info dnsmasq-dhcp[26680]: DHCPACK(br-lan) 192.168.1.147 2c:54:cf:e7:26:be android-35f86d62f09fb010
Thu Aug 24 19:18:01 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 2c:54:cf:e7:26:be
Thu Aug 24 19:18:11 2017 user.warn igmpproxy[26185]: MRT_DEL_MFC; Errno(2): No such file or directory
Thu Aug 24 19:18:12 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 2c:54:cf:e7:26:be
Thu Aug 24 19:18:13 2017 daemon.info odhcpd[1648]: Using a RA lifetime of 0 seconds on br-lan
Thu Aug 24 19:18:19 2017 daemon.info odhcpd[1648]: Using a RA lifetime of 0 seconds on br-lan
Thu Aug 24 19:18:23 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:54:cf:e7:26:be
Thu Aug 24 19:18:27 2017 kern.err kernel: [241486.937131] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Thu Aug 24 19:18:27 2017 kern.err kernel: [241486.943545] ieee80211 phy0: return code: 0x1122
Thu Aug 24 19:18:27 2017 kern.err kernel: [241486.948208] ieee80211 phy0: timeout: 0x1122
Thu Aug 24 19:18:27 2017 kern.err kernel: [241486.952497] wlan0: failed to remove key (0, 2c:54:cf:e7:26:be) from hardware (-5)
Thu Aug 24 19:18:27 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 2c:54:cf:e7:26:be
Thu Aug 24 19:18:27 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED c0:ee:fb:4b:4c:8c
Thu Aug 24 19:18:31 2017 kern.err kernel: [241490.979168] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Thu Aug 24 19:18:31 2017 kern.err kernel: [241490.985448] ieee80211 phy0: return code: 0x1122
Thu Aug 24 19:18:31 2017 kern.err kernel: [241490.990143] ieee80211 phy0: timeout: 0x1122
Thu Aug 24 19:18:31 2017 kern.err kernel: [241490.994468] wlan0: failed to remove key (0, c0:ee:fb:4b:4c:8c) from hardware (-5)
Thu Aug 24 19:18:31 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED f4:f5:d8:66:bd:b6
Thu Aug 24 19:18:35 2017 kern.err kernel: [241495.003133] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Thu Aug 24 19:18:35 2017 kern.err kernel: [241495.009411] ieee80211 phy0: return code: 0x1122
Thu Aug 24 19:18:35 2017 kern.err kernel: [241495.014082] ieee80211 phy0: timeout: 0x1122
Thu Aug 24 19:18:35 2017 kern.err kernel: [241495.018400] wlan0: failed to remove key (0, f4:f5:d8:66:bd:b6) from hardware (-5)
Thu Aug 24 19:18:40 2017 kern.err kernel: [241499.055073] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Thu Aug 24 19:18:40 2017 kern.err kernel: [241499.061360] ieee80211 phy0: return code: 0x1122
[... cut ...]
Thu Aug 24 19:19:16 2017 kern.err kernel: [241535.260707] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
Thu Aug 24 19:19:16 2017 kern.err kernel: [241535.267101] ieee80211 phy0: return code: 0x001c
Thu Aug 24 19:19:16 2017 kern.err kernel: [241535.271830] ieee80211 phy0: timeout: 0x001c
Thu Aug 24 19:19:20 2017 kern.err kernel: [241539.276654] ieee80211 phy0: cmd 0x8126=SetFixedRate timed out
Thu Aug 24 19:19:20 2017 kern.err kernel: [241539.282530] ieee80211 phy0: return code: 0x0126
Thu Aug 24 19:19:20 2017 kern.err kernel: [241539.287235] ieee80211 phy0: timeout: 0x0126
`