kaloz/mwlwifi #57: .16 driver, unusable 5ghz wifi

johnnysl, 2016-01-10:
With my V2, i'm currently seeing 2 issues on this new driver:

1. DFS always reverts my channel back to 36. if it is not immediately, it will do it a couple of minutes later. Yet i'm 100KM from the nearest airport, so no radar. 
2. performance is unbelievably bad with speeds on avg <1mbit a second.

2.4ghz seems to be fine and quite stable with up and download speeds of around 100mbit/sec (20mhz channel)

APLandry, 2016-01-10:
Having the same issue with V1. 5GHz Wifi rarely actually turns on, is frequently very poor performance, can't get it to run on any combo of mode/channel/width.

Chadster766, 2016-01-10:
I compiled the .16 driver but will some exceptions:

Pulled from kaloz/mwlwifi up to commit 99d3879cc72f2a25d44fb4bee96fd84eca028b04

The below changes I declined from kaloz/mwlwifi commit c8165c47910169d6d73983b11743fa75941b5b76:

1. Use mutex to protect firmware command instead of spinlock.
2. Change mdelay(1000) to msleep(1000) in firmware command
   mwl_fwcmd_get_hw_specs().
3. Add might_sleep() in mwl_fwcmd_exec_cmd() to detect if firmware
   command is called within atomic context.
4. Change compile variable from "CONFIG_SUPPORT_MFG" to "SUPPORT_MFG".

https://github.com/Chadster766/McDebian/tree/master/linux-4.2.4/drivers/mwlwifi

Wireless throughput was exceptional:

Testing command used:
iperf3 -c 192.168.1.5 -P6 -t 600

Results:
[SUM]   0.00-600.00 sec  17.5 GBytes   250 Mbits/sec                  sender
[SUM]   0.00-600.00 sec  17.5 GBytes   250 Mbits/sec                  receiver

Below are some wireless configs and stats that might help you troubleshoot the performance issue:

```
root@MCDEBIAN:/etc/hostapd# cat wlan1.conf
interface=wlan1
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ssid=McDebian50
ignore_broadcast_ssid=0
hw_mode=a
channel=36
auth_algs=1
wmm_enabled=1
ieee80211n=1
ht_capab=[LDPC][HT40+][SHORT-GI-20][SHORT-GI-40]
ieee80211ac=1
vht_capab=[MAX-MPDU-7991][RXLDPC][SHORT-GI-80][RX-STBC-1][SU-BEAMFORMER][SU-BEAMFORMEE][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211h=1
ieee80211d=1

wpa=2
wpa_passphrase=mcdebian123
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
country_code=US
```

```
root@MCDEBIAN:/etc/hostapd# iw wlan1 station dump
Station c8:d7:19:be:5f:91 (on wlan1)
        inactive time:  10 ms
        rx bytes:       743279387
        rx packets:     14547200
        tx bytes:       677693311
        tx packets:     5959585
        tx retries:     0
        tx failed:      0
        signal:         -47 dBm
        signal avg:     -47 dBm
        tx bitrate:     866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
        rx bitrate:     866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
        authorized:     yes
        authenticated:  yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
```

Chadster766, 2016-01-10:
I can't get the 5Ghz in 80Mhz to change to any other channel than 36 (including DFS channels).

If AC mode is disabled (ieee80211ac=0) the 5Ghz channel can successful be changed to 36, 44 and 149 but it fallsback to 20Mhz only. Oddly enough if the channel is set to 157 it will enable 40Mhz.

All DFS channel report  "DFS-RADAR-DETECTED" or "not found from the channel list". Testing was done in a rural area hours away from any airport with RADAR capability and in a concrete basement. No other DFS wireless signals detected by Acrylic Wifi.

Radar detected but then fallback to non DFS channel:

```
root@MCDEBIAN:/etc/hostapd# hostapd /etc/hostapd/wlan1.conf
Configuration file: /etc/hostapd/wlan1.conf
rfkill: Cannot open RFKILL control device
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan1: interface state COUNTRY_UPDATE->HT_SCAN
wlan1: interface state HT_SCAN->DFS
wlan1: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=42, seg1=0, cac_time=60s
wlan1: DFS-CAC-COMPLETED success=0 freq=5260 ht_enabled=1 chan_offset=1 chan_width=2 cf1=5270 cf2=0
wlan1: DFS-RADAR-DETECTED freq=5260 ht_enabled=1 chan_offset=1 chan_width=2 cf1=5270 cf2=0
wlan1: DFS-NEW-CHANNEL freq=5220 chan=44 sec_chan=1
Using interface wlan1 with hwaddr 00:25:9c:13:07:31 and ssid "McDebian50"
wlan1: interface state DFS->ENABLED
wlan1: AP-ENABLED
```
To me it looks like the Radar detection times out with a radar detected state.

Chadster766, 2016-01-10:
My testing was done on a WRT1900AC V1.

Chadster766, 2016-01-12:
I updated the country code reg database using files for the kernel dev source:
https://wireless.wiki.kernel.org/en/developers/regulatory/wireless-regdb

This resolved some of the issue but I also needed to change my hostapd 5Ghz wlan config file.

I changed ht_capab to a single 80Mhz option, removed vht_capab entirely and configured vht_oper_centr_freq_seg0_idx to the base channel + 6.

```
interface=wlan1
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ssid=McDebian50
ignore_broadcast_ssid=0
hw_mode=a
channel=100
auth_algs=1
wmm_enabled=1
ieee80211n=1
ht_capab=[HT40+]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=106
ieee80211h=1
ieee80211d=1

wpa=2
wpa_passphrase=mcdebian123
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
country_code=US
```
Thanks for your help :smile: 

I now get 80Mhz on the entire 5Ghz channel range including DFS!

yuhhaurlin, 2016-01-12:
@Chadster766  Yes. I updated hostapd-dfs.conf to GitHub when I released the code. You need to add ieee80211d, country_code and ieee80211h in hostapd configuration file.

yuhhaurlin, 2016-01-12:
@Chadster766 But I think you need to keep HT and VHT capability. Please reference file hostapd-dfs.conf.

Chadster766, 2016-01-12:
Thanks I will look at the hostapd-dfs.conf file.

I did throughput testing and removing the vht_capab didn't cause any performance issues.

insignia96, 2016-01-16:
Also having the same issue as the OP on a WRT1200AC. 2.4 is stable, but 5Ghz is bugged out in a strange way. Speed tests show good throughput (250 Mbps, which is our speed from the ISP) but downloading any actual file fails after a few KB have been downloaded. Even images of a few KB stall out and never finish. I'm not 100% sure how both things could happen at the same time, but this is what my experience has been.

Chadster766, 2016-01-16:
To confirm the issue isn't related to the spinlock to mutex change try the below driver for comparison:
[McDebian mwlwifi](https://github.com/Chadster766/McDebian/blob/master/rootfs/lib/modules/4.2.4/kernel/drivers/mwlwifi/mwlwifi.ko)

yuhhaurlin, 2016-03-25:
So, DFS is all right? Can I close this one? Thanks.

closed 2016-03-29 by yuhhaurlin