kaloz/mwlwifi #58: Can't get DFS started

lantis1008, 2016-01-10:
Doesn't matter which DFS channel i choose, it won't start up.
```
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): Configuration file: /var/run/hostapd-phy1.conf`
`Mon Jan 11 10:05:42 2016 kern.debug kernel: [224705.777105] ieee80211 phy1: change: 0xffffffff`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): wlan1: interface state COUNTRY_UPDATE->HT_SCAN`
`Mon Jan 11 10:05:42 2016 kern.info kernel: [224705.863499] device wlan1 entered promiscuous mode`
`Mon Jan 11 10:05:42 2016 kern.debug kernel: [224705.870654] ieee80211 phy1: change: 0x100`
`Mon Jan 11 10:05:42 2016 kern.debug kernel: [224705.876458] ieee80211 phy1: change: 0x60`
`Mon Jan 11 10:05:42 2016 kern.debug kernel: [224706.058116] ieee80211 phy1: change: 0x40`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): wlan1: interface state HT_SCAN->DFS`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): wlan1: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): DFS start_dfs_cac() failed, -1`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): Interface initialization failed`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): wlan1: interface state DFS->DISABLED`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): wlan1: AP-DISABLED`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): wlan1: interface state DISABLED->DISABLED`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): wlan1: AP-DISABLED`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): hostapd_free_hapd_data: Interface wlan1 wasn't started`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): nl80211: deinit ifname=wlan1 disabled_11b_rates=0`
`Mon Jan 11 10:05:42 2016 kern.debug kernel: [224706.238321] ieee80211 phy1: change: 0x60`
`Mon Jan 11 10:05:42 2016 kern.debug kernel: [224706.311432] ieee80211 phy1: change: 0x100`
`Mon Jan 11 10:05:42 2016 kern.info kernel: [224706.329516] device wlan1 left promiscuous mode`
`Mon Jan 11 10:05:42 2016 kern.info kernel: [224706.334179] br-lan: port 2(wlan1) entered disabled state`
`Mon Jan 11 10:05:42 2016 daemon.notice netifd: radio1 (30219): ELOOP: remaining socket: sock=19 eloop_data=0x106cf40 user_data=(nil) handler=0x38e38`
`Mon Jan 11 10:05:42 2016 kern.debug kernel: [224706.520898] ieee80211 phy1: change: 0xffffffff`
```
```/etc/config/wireless
config wifi-device 'radio1'
    option type 'mac80211'
    option channel '52
    option hwmode '11a'
    option path 'soc/soc:pcie-controller/pci0000:00/0000:00:03.0/0000:03:00.0'
    option country 'AU'
    option htmode 'VHT80'
config wifi-iface 'ap_a'
    option device 'radio1'
    option mode 'ap'
    option network 'lan'
    option disassoc_low_ack '0'
    option ssid 'xxxxxxxxxx'
    option encryption 'psk2'
    option key 'xxxxxxxxx'
```
```
strings /lib/modules/*.*/mwlwifi.ko | grep "^10.3"
10.3.0.16-20160105
10.3.0.16-20160105
```

It works fine on non-DFS channels.

Chaos Calmer build with the latest driver included.

Chadster766, 2016-01-10:
Which WRT model do you have?

lantis1008, 2016-01-10:
1900ac v1

lantis1008, 2016-01-10:
Thinking it is possibly a regulation issue? I'm in australia, which seems to have everything allowed (in our regulatory body) but the openwrt regulatory file might be out of date?. BUT, when i set my location to US, i can get DFS to start on channel 52.
US
```
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 17), (N/A)
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)
```
```
 Frequencies:
                        * 5180 MHz [36] (17.0 dBm)
                        * 5200 MHz [40] (17.0 dBm)
                        * 5220 MHz [44] (17.0 dBm)
                        * 5240 MHz [48] (17.0 dBm)
                        * 5260 MHz [52] (23.0 dBm) (radar detection)
                          DFS state: available (for 162 sec)
                          DFS CAC time: 60000 ms
                        * 5280 MHz [56] (23.0 dBm) (radar detection)
                          DFS state: available (for 162 sec)
                          DFS CAC time: 60000 ms
                        * 5300 MHz [60] (23.0 dBm) (radar detection)
                          DFS state: available (for 162 sec)
                          DFS CAC time: 60000 ms
                        * 5320 MHz [64] (23.0 dBm) (radar detection)
                          DFS state: available (for 162 sec)
                          DFS CAC time: 60000 ms
                        * 5500 MHz [100] (disabled)
                        * 5520 MHz [104] (disabled)
                        * 5540 MHz [108] (disabled)
                        * 5560 MHz [112] (disabled)
                        * 5580 MHz [116] (disabled)
                        * 5600 MHz [120] (disabled)
                        * 5620 MHz [124] (disabled)
                        * 5640 MHz [128] (disabled)
                        * 5660 MHz [132] (disabled)
                        * 5680 MHz [136] (disabled)
                        * 5700 MHz [140] (disabled)
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
```

AU
```
country AU: DFS-UNSET
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 17), (N/A)
        (5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS
        (5490 - 5710 @ 160), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
```
```
Frequencies:
                        * 5180 MHz [36] (17.0 dBm)
                        * 5200 MHz [40] (17.0 dBm)
                        * 5220 MHz [44] (17.0 dBm)
                        * 5240 MHz [48] (17.0 dBm)
                        * 5260 MHz [52] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5280 MHz [56] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5300 MHz [60] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5320 MHz [64] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5500 MHz [100] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5520 MHz [104] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5540 MHz [108] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5560 MHz [112] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5580 MHz [116] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5600 MHz [120] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5620 MHz [124] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5640 MHz [128] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5660 MHz [132] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5680 MHz [136] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5700 MHz [140] (24.0 dBm) (radar detection)
                          DFS state: usable (for 50 sec)
                          DFS CAC time: 60000 ms
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
```

Chadster766, 2016-01-11:
No i did testing with country code RU which had the most DFS channels and results were the same. Testing in below posts. 

https://github.com/kaloz/mwlwifi/issues/57

lantis1008, 2016-01-11:
Rebuilt with an updated reg domain database. DFS now works fine for me. Closing this issue.

lantis1008, 2016-01-11:
DFS-CAC-COMPLETED success=1 :+1: 
```
Mon Jan 11 22:23:13 2016 kern.info kernel: [  369.255210] device wlan1 left promiscuous mode
Mon Jan 11 22:23:13 2016 kern.info kernel: [  369.259768] br-lan: port 3(wlan1) entered disabled state
Mon Jan 11 22:23:13 2016 daemon.notice netifd: Network device 'wlan1' link is down
Mon Jan 11 22:23:13 2016 kern.debug kernel: [  369.288021] ieee80211 phy1: change: 0x40
Mon Jan 11 22:23:13 2016 kern.debug kernel: [  369.362236] ieee80211 phy1: change: 0x100
Mon Jan 11 22:23:13 2016 kern.info kernel: [  369.368063] device wlan0 left promiscuous mode
Mon Jan 11 22:23:13 2016 kern.info kernel: [  369.372644] br-lan: port 2(wlan0) entered disabled state
Mon Jan 11 22:23:13 2016 daemon.notice netifd: Network device 'wlan0' link is down
Mon Jan 11 22:23:13 2016 kern.debug kernel: [  369.422946] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:13 2016 kern.debug kernel: [  369.484958] ieee80211 phy0: change: 0x100
Mon Jan 11 22:23:14 2016 daemon.notice netifd: radio1 (11585): Configuration file: /var/run/hostapd-phy1.conf
Mon Jan 11 22:23:14 2016 kern.debug kernel: [  370.344519] ieee80211 phy1: change: 0xffffffff
Mon Jan 11 22:23:14 2016 daemon.notice netifd: radio0 (11618): Configuration file: /var/run/hostapd-phy0.conf
Mon Jan 11 22:23:14 2016 daemon.notice netifd: radio1 (11585): wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Jan 11 22:23:14 2016 kern.info kernel: [  370.476964] device wlan1 entered promiscuous mode
Mon Jan 11 22:23:14 2016 daemon.notice netifd: radio1 (11585): wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Mon Jan 11 22:23:14 2016 kern.debug kernel: [  370.484168] ieee80211 phy1: change: 0x100
Mon Jan 11 22:23:14 2016 kern.debug kernel: [  370.494623] ieee80211 phy1: change: 0x60
Mon Jan 11 22:23:14 2016 kern.debug kernel: [  370.713584] ieee80211 phy1: change: 0x40
Mon Jan 11 22:23:14 2016 kern.debug kernel: [  370.714136] ieee80211 phy0: change: 0xffffffff
Mon Jan 11 22:23:14 2016 kern.info kernel: [  370.784318] device wlan0 entered promiscuous mode
Mon Jan 11 22:23:14 2016 kern.info kernel: [  370.789112] br-lan: port 3(wlan0) entered forwarding state
Mon Jan 11 22:23:14 2016 kern.info kernel: [  370.794701] br-lan: port 3(wlan0) entered forwarding state
Mon Jan 11 22:23:14 2016 kern.debug kernel: [  370.801621] ieee80211 phy0: change: 0x100
Mon Jan 11 22:23:14 2016 kern.debug kernel: [  370.806891] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:14 2016 daemon.notice netifd: radio0 (11618): wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Jan 11 22:23:14 2016 daemon.notice netifd: radio0 (11618): wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  370.934008] ieee80211 phy1: change: 0x60
Mon Jan 11 22:23:15 2016 daemon.notice netifd: radio1 (11585): wlan1: interface state HT_SCAN->DFS
Mon Jan 11 22:23:15 2016 daemon.notice netifd: radio1 (11585): wlan1: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  370.973572] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  371.007164] ieee80211 phy1: change: 0x100
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  371.012508] ieee80211 phy1: change: 0x100
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  371.017635] ieee80211 phy1: change: 0x62
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  371.090162] ieee80211 phy1: change: 0x40
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  371.183984] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  371.363613] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:15 2016 kern.info kernel: [  371.464168] br-lan: port 3(wlan0) entered disabled state
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  371.534078] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  371.713578] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:15 2016 kern.debug kernel: [  371.893576] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:16 2016 kern.debug kernel: [  372.063615] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:16 2016 kern.debug kernel: [  372.243869] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:16 2016 kern.debug kernel: [  372.413631] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:16 2016 kern.debug kernel: [  372.584168] ieee80211 phy0: change: 0x40
Mon Jan 11 22:23:16 2016 kern.debug kernel: [  372.646161] ieee80211 phy0: change: 0x100
Mon Jan 11 22:23:16 2016 daemon.notice netifd: radio0 (11618): Using interface wlan0 with hwaddr xxxxxxxxxxx and ssid "xxxxxxxxxx"
Mon Jan 11 22:23:16 2016 kern.debug kernel: [  372.893935] ieee80211 phy0: change: 0x100
Mon Jan 11 22:23:16 2016 kern.debug kernel: [  372.898958] ieee80211 phy0: change: 0x42
Mon Jan 11 22:23:17 2016 kern.info kernel: [  372.990199] br-lan: port 3(wlan0) entered forwarding state
Mon Jan 11 22:23:17 2016 kern.info kernel: [  372.995824] br-lan: port 3(wlan0) entered forwarding state
Mon Jan 11 22:23:17 2016 daemon.notice netifd: radio0 (11618): wlan0: interface state HT_SCAN->ENABLED
Mon Jan 11 22:23:17 2016 daemon.notice netifd: radio0 (11618): wlan0: AP-ENABLED
Mon Jan 11 22:23:17 2016 daemon.notice netifd: Network device 'wlan0' link is up
Mon Jan 11 22:23:19 2016 kern.info kernel: [  374.993612] br-lan: port 3(wlan0) entered forwarding state
Mon Jan 11 22:24:15 2016 kern.debug kernel: [  431.195553] ieee80211 phy1: change: 0x40
Mon Jan 11 22:24:15 2016 kern.debug kernel: [  431.269638] ieee80211 phy1: change: 0x100
[... cut ...]
Mon Jan 11 22:24:15 2016 kern.info kernel: [  431.700468] br-lan: port 2(wlan1) entered forwarding state
Mon Jan 11 22:24:15 2016 kern.info kernel: [  431.706085] br-lan: port 2(wlan1) entered forwarding state
Mon Jan 11 22:24:15 2016 daemon.notice netifd: radio1 (11585): wlan1: interface state DFS->ENABLED
Mon Jan 11 22:24:15 2016 daemon.notice netifd: radio1 (11585): wlan1: AP-ENABLED
Mon Jan 11 22:24:15 2016 daemon.notice netifd: Network device 'wlan1' link is up
Mon Jan 11 22:24:17 2016 kern.info kernel: [  433.714812] br-lan: port 2(wlan1) entered forwarding state
```

closed 2016-01-11 by lantis1008

Chadster766, 2016-01-11:
How did you updated reg domain database?

lantis1008, 2016-01-11:
I recompiled and added a patch for it. 
I have the patch available here.
https://github.com/lantis1008/gargoyle/blob/9df043d4884a630f366d47c7daa793680b262b87/patches-generic/120-latest-reg-domain-db.patch

The chaos calmer reg domain is like 12 months out of date. A lot of changes since then.

lantis1008, 2016-01-11:
I can also confirm with a spectrum analyser that I indeed am achieving 80MHz width @Chadster766 
I follow the forum thread on openwrt and see that everyone is discussing that currently

Chadster766, 2016-01-11:
Is the analyzer showing 80Mhz for upper non-DFS channel +149? Please test any one of DFS channels your country code allows.

I'm guess this ability is tied to the OpenWRT hostapd custom code for the "option noscan".

lantis1008, 2016-01-11:
@Chadster766 yes it does. My settings are the same as my first comment except changed channel 52 to 149. I **_do not_** use no-scan option.

Screenshot attached.
[image]

Chadster766, 2016-01-11:
Awesome, I must need your patch! I will post back with the results when completed.

Chadster766, 2016-01-12:
I updated the country code reg database using files for the kernel dev source:
https://wireless.wiki.kernel.org/en/developers/regulatory/wireless-regdb

This resolved some of the issue but I also needed to change my hostapd 5Ghz wlan config file.

I changed ht_capab to a single 80Mhz option, removed vht_capab entirely and configured vht_oper_centr_freq_seg0_idx to the base channel + 6.

```
interface=wlan1
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ssid=McDebian50
ignore_broadcast_ssid=0
hw_mode=a
channel=100
auth_algs=1
wmm_enabled=1
ieee80211n=1
ht_capab=[HT40+]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=106
ieee80211h=1
ieee80211d=1

wpa=2
wpa_passphrase=mcdebian123
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
country_code=US
```
Thanks for your help :smile: 

I now get 80Mhz on the entire 5Ghz channel range including DFS!

lantis1008, 2016-01-12:
Yep that is the git source i wrote my patch from. Glad it could help you!