kaloz/mwlwifi #59: .16 driver buggy

trustno1foxm, 2016-01-16:
hello,

by using .16 I have major problems with my wlan. a nexus 5 device and a surface pro 4 have very high ping times, over 300ms.
by using .13 everything is fine.
maybe because of a changing from spinlock to mutex?!

pls fix!

thx!

yuhhaurlin, 2016-01-17:
How about the throughput you can get?

trustno1foxm, 2016-01-17:
which device should i test? on the nexus 9 there weren't any issues, but with the devices above the times have been very high.

yuhhaurlin, 2016-01-17:
I suspect that client entered power save mode. You can disable power save mode of client and ping again.

trustno1foxm, 2016-01-17:
I'll try again later, but it's useless for me, why should I deactivate power save mode? There are no problems with .13? Battery run-time is more important for me.

yuhhaurlin, 2016-01-17:
Just check if it is due to pwer save mode. If.13 is all right, .16 should be no problem too.

trustno1foxm, 2016-01-22:
I have tested it, but with .16 the ping times are ok now. The router was rebootet because of the installation of the driver. Could it be that the problem appears after some runtime?

closed 2016-03-26 by yuhhaurlin