kaloz/mwlwifi #60: kernel crash

rspanbauer, 2016-01-21:
When I create monitor interface, kernel crashes in mwl_tx_done.  Using the trunk version 
of openwrt.

Platform is Linksys WRT1900ACS (Shelby).

To produce bug:

iw dev wlan0 interface add mon0 type monitor
ifconfig mon0 up
tcpdump -i mon0

Kernel crashes when a bit of traffic is run through the wlan0 inteface.

16:32:37.525162 5180 MHz 11a -70dB signal antenna 0 243.0 Mb/s MCS 14 40 MHz lon GI CF +QoS Data IV:3aa0
16:32:37.528896 [bit 15] CF +QoS Data IV:706 Pad 20 KeyID 0
16:32:37.531475 [bit 15] CF +QoS Data IV:707 Pad 20 Ke[  526.849754] Unable to handle kernel NULL point4
-------------------------------------------------------------
[  526.858814] pgd = c0004000
[  526.861530] [00000004] *pgd=00000000
[  526.865130] Internal error: Oops: 817 [#1] SMP ARM
[  526.869938] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntracn
[  526.958035] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 3.18.23 #3
[  526.964065] task: c0419398 ti: c0412000 task.ti: c0412000
[  526.969488] PC is at skb_dequeue+0x50/0x70
[  526.973599] LR is at skb_dequeue+0x1c/0x70
[  526.977711] pc : [<c0267908>]    lr : [<c02678d4>]    psr: a0000193
[  526.977711] sp : c0413d78  ip : 00000000  fp : c0413d94
[  526.989236] r10: defecb40  r9 : 00000000  r8 : de815b6c
[  526.994480] r7 : 0fc01511  r6 : defee8a0  r5 : defee894  r4 : deeab100
[  527.001034] r3 : defee894  r2 : 00000000  r1 : 20000113  r0 : 20000113
[  527.007587] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[  527.015012] Control: 10c5387d  Table: 1da5404a  DAC: 00000015
[  527.020780] Process swapper/0 (pid: 0, stack limit = 0xc0412238)
[  527.026809] Stack: (0xc0413d78 to 0xc0414000)
[  527.031183] 3d60:                                                       00000000 c0415044
[  527.039395] 3d80: de815ac0 00000101 c0413e34 c0413d98 bf23b008 c02678c4 00000000 00000128
[  527.047606] 3da0: dfbd89c0 c0414100 c0413dc4 c0413db8 c001d128 c01a523c c0413df4 c0413dc8
[  527.055817] 3dc0: c0076698 c001d0f8 00000000 00000000 defee340 defed874 00000000 c041a2d0
[  527.064028] 3de0: c0415084 c0415044 defee880 defee894 00000004 defed840 c0413e1c c0413e08
[  527.072239] 3e00: c0029cb4 de9f1668 00000001 defee8f4 defee8f8 00000000 c0436780 c040f13c
[  527.080451] 3e20: c0412000 00000006 c0413e5c c0413e38 c0029674 bf23aeb8 00000000 c0414098
[  527.088663] 3e40: c0414080 40000006 c0412000 00000100 c0413ebc c0413e60 c0029840 c00295e4
[  527.096874] 3e60: c0413e84 df406e80 c0421020 00200000 c0377854 c0412020 0000589f c0414100
[  527.105085] 3e80: c0436780 0000000a c040f1b0 c0414080 df406e80 c040f8f8 00000000 00000000
[  527.113296] 3ea0: 00000001 df407000 00000000 c04150cc c0413ed4 c0413ec0 c0029bf0 c0029748
[  527.121507] 3ec0: 00000100 c040f8f8 c0413efc c0413ed8 c00582cc c0029b5c c0413f20 e080210c
[  527.129718] 3ee0: c0415ca8 c0413f20 e0802100 c0412000 c0413f1c c0413f00 c0008758 c0058244
[  527.137929] 3f00: c00187bc 60000013 ffffffff c0413f54 c0413f74 c0413f20 c00097a0 c000871c
[  527.146141] 3f20: ffffffed 00000000 008dc2ba c000a280 c0412038 00000000 00000000 c0377844
[  527.154352] 3f40: c0412000 00000000 c04150cc c0413f74 c0413f78 c0413f68 c00187b8 c00187bc
[  527.162564] 3f60: 60000013 ffffffff c0413fa4 c0413f78 c005171c c001878c c0413f94 c0410380
[  527.170775] 3f80: c0418dbf c0410380 c040fa40 ffffffed c000edc8 c0415000 c0413fb4 c0413fa8
[  527.178987] 3fa0: c000d264 c00515e0 c0413ff4 c0413fb8 c03dfdb4 c000d1f4 ffffffff ffffffff
[  527.187198] 3fc0: c03df6dc 00000000 00000000 c0407230 c0435fa4 c041507c c040722c c04181d4
[  527.195409] 3fe0: 0000406a 414fc091 00000000 c0413ff8 00008074 c03df988 00000000 00000000
[  527.203617] Backtrace:
[  527.206093] [<c02678b8>] (skb_dequeue) from [<bf23b008>] (mwl_tx_done+0x15c/0x18b0 [mwlwifi])
[  527.214651]  r6:00000101 r5:de815ac0 r4:c0415044 r3:00000000
[  527.220383] [<bf23aeac>] (mwl_tx_done [mwlwifi]) from [<c0029674>] (tasklet_action+0x9c/0x108)
[  527.229027]  r10:00000006 r9:c0412000 r8:c040f13c r7:c0436780 r6:00000000 r5:defee8f8
[  527.236939]  r4:defee8f4
[  527.239494] [<c00295d8>] (tasklet_action) from [<c0029840>] (__do_softirq+0x104/0x234)
[  527.247440]  r9:00000100 r8:c0412000 r7:40000006 r6:c0414080 r5:c0414098 r4:00000000
[  527.255271] [<c002973c>] (__do_softirq) from [<c0029bf0>] (irq_exit+0xa0/0xb0)
[  527.262520]  r10:c04150cc r9:00000000 r8:df407000 r7:00000001 r6:00000000 r5:00000000
[  527.270432]  r4:c040f8f8
[  527.272989] [<c0029b50>] (irq_exit) from [<c00582cc>] (__handle_domain_irq+0x94/0xa4)
[  527.280848]  r4:c040f8f8 r3:00000100
[  527.284459] [<c0058238>] (__handle_domain_irq) from [<c0008758>] (gic_handle_irq+0x48/0x64)
[  527.292841]  r8:c0412000 r7:e0802100 r6:c0413f20 r5:c0415ca8 r4:e080210c r3:c0413f20
[  527.300672] [<c0008710>] (gic_handle_irq) from [<c00097a0>] (__irq_svc+0x40/0x54)
[  527.308184] Exception stack(0xc0413f20 to 0xc0413f68)
[  527.313256] 3f20: ffffffed 00000000 008dc2ba c000a280 c0412038 00000000 00000000 c0377844
[  527.321467] 3f40: c0412000 00000000 c04150cc c0413f74 c0413f78 c0413f68 c00187b8 c00187bc
[  527.329678] 3f60: 60000013 ffffffff
[  527.333177]  r7:c0413f54 r6:ffffffff r5:60000013 r4:c00187bc
[  527.338901] [<c0018780>] (arch_cpu_idle) from [<c005171c>] (cpu_startup_entry+0x148/0x2a8)
[  527.347204] [<c00515d4>] (cpu_startup_entry) from [<c000d264>] (rest_init+0x7c/0x80)
[  527.354977]  r7:c0415000
[  527.357535] [<c000d1e8>] (rest_init) from [<c03dfdb4>] (start_kernel+0x438/0x444)
[  527.365053] [<c03df97c>] (start_kernel) from [<00008074>] (0x8074)
[  527.371258] Code: e5853008 e894000c e584c000 e584c004 (e5823004)
[  527.377375] ---[ end trace af23f3409568ff6c ]---
[  527.382009] Kernel panic - not syncing: Fatal exception in interrupt

yuhhaurlin, 2016-04-11:
Monitor interface will be checked with lower priority.

yuhhaurlin, 2017-01-11:
Root cause is found. I will fix this problem later.

steffanson, 2017-08-24:
> Root cause is found. I will fix this problem later.

@yuhhaurlin Has the problem been fixed yet?

yuhhaurlin, 2017-08-24:
If you use WRT3200ACM, you will not encounter this problem. For 88W8864, this problem will be fixed later.

thagabe, 2017-11-08:
@yuhhaurlin Is it later yet?

yuhhaurlin, 2017-11-08:
OK. I will fix 60 and 107 before end of this year.

thagabe, 2017-11-08:
Perfect!

yuhhaurlin, 2018-03-05:
Fixed by https://github.com/kaloz/mwlwifi/commit/4591708bfd92b99102976b7ae2571101cbf8518e.

closed 2018-03-05 by yuhhaurlin