kaloz/mwlwifi #61: .16 driver Incompatability with with Dell 1504 wireless card and HP officejet printer 8600

johnnysl, 2016-01-22:
Both devices work horrible bad with the current driver. No errors in the logs, but very frequent disconnects on the 2.4ghz interface. 
I've fired up a Linksys EA4500 V2, with an identical Openwrt build (and identical wireless config), and both devices work great with the EA4500. This forces me to believe the issue is related to the current wireless driver.

Chadster766, 2016-01-22:
What errors are you seeing in the WRT log? You should be able to grep the log for the problem devices MAC Addresses.

johnnysl, 2016-01-22:
@chadster, what log do you mean? system log and kernel log are not showing any warnings or errors

Chadster766, 2016-01-22:
Those are the logs. Odd no errors.

jsebean, 2016-02-12:
and also have this issue with force disconnects on 5GHz for a Nexus Player and an Asus X550Z Broadcom wireless chip. Constantly have to force reconnect to gain connection again but it's driven me up the wall to the point I don't even use the wifi on the WRT1900ACv2 any longer.

jsebean, 2016-02-13:
Just had it happen to my laptop (Jonah-laptop)... Wifi died completely and had to reset the adapter to reconnect. This is all the log shows from DD-WRT:
http://i.imgur.com/G5hGXEm.png

yuhhaurlin, 2016-03-29:
How about .17 and new firmware?

johnnysl, 2016-03-29:
Still frequent disconnects after ~10 minutes on the Dell laptop.

yuhhaurlin, 2016-03-29:
Can you get log from hostapd? BTW, this is the only problem for.17 and new firmware now?

jsebean, 2016-03-30:
I can't say whether or not older versions have the issue or not, but... unless this is placebo effect I find setting the transmit power lower than 20dbm seems to help, but doesn't completely mitigate the issue. Eventually it gets bad again where rebooting the router then helps for a while.

yuhhaurlin, 2016-03-30:
.16 has throughput becomes lower problem. It is fixed by.17.

johnnysl, 2016-04-01:
stability for the dell laptop has improved greatly, after the firmware change / change back trick.
seems to be solved, but have not tested the printer yet, will report back after this weekend.

johnnysl, 2016-04-02:
@yuhhaurlin, this issue can be closed. Both devices are now running perfectly stable on .17.

I'm a happy camper now :-) :+1:

closed 2016-04-02 by yuhhaurlin