kaloz/mwlwifi #62: Spinlock contention causing Slow Path at drivers/mwlwifi/tx.c:901

Chadster766, 2016-01-29:
I see the below periodically in the journal at one site:

```
Jan 29 09:08:15 MCDEBIAN kernel: ------------[ cut here ]------------
Jan 29 09:08:15 MCDEBIAN kernel: WARNING: CPU: 0 PID: 2830 at drivers/mwlwifi/tx.c:901 mwl_tx_xmit+0x1628/0x167c [mwlwifi]()
Jan 29 09:08:15 MCDEBIAN kernel: Modules linked in: ip6table_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack iptable_mangle iptable_filte
Jan 29 09:08:15 MCDEBIAN kernel: CPU: 0 PID: 2830 Comm: hostapd Tainted: G        W       4.2.4 #64
Jan 29 09:08:15 MCDEBIAN kernel: Hardware name: Marvell Armada 370/XP (Device Tree)
Jan 29 09:08:15 MCDEBIAN kernel: [<c00167d4>] (unwind_backtrace) from [<c0012874>] (show_stack+0x10/0x14)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0012874>] (show_stack) from [<c0550e7c>] (dump_stack+0x84/0x94)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0550e7c>] (dump_stack) from [<c0025220>] (warn_slowpath_common+0x80/0xb0)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0025220>] (warn_slowpath_common) from [<c00252ec>] (warn_slowpath_null+0x1c/0x24)
Jan 29 09:08:15 MCDEBIAN kernel: [<c00252ec>] (warn_slowpath_null) from [<bf091ad0>] (mwl_tx_xmit+0x1628/0x167c [mwlwifi])
Jan 29 09:08:15 MCDEBIAN kernel: [<bf091ad0>] (mwl_tx_xmit [mwlwifi]) from [<c052ebf8>] (ieee80211_tx_frags+0xd8/0x2e8)
Jan 29 09:08:15 MCDEBIAN kernel: [<c052ebf8>] (ieee80211_tx_frags) from [<c0530634>] (__ieee80211_tx.isra.33+0x68/0x130)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0530634>] (__ieee80211_tx.isra.33) from [<c05307ac>] (ieee80211_tx+0xb0/0xdc)
Jan 29 09:08:15 MCDEBIAN kernel: [<c05307ac>] (ieee80211_tx) from [<c0527450>] (ieee80211_probe_client+0x178/0x1d0)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0527450>] (ieee80211_probe_client) from [<c04f57d4>] (nl80211_probe_client+0xbc/0x150)
Jan 29 09:08:15 MCDEBIAN kernel: [<c04f57d4>] (nl80211_probe_client) from [<c0425768>] (genl_rcv_msg+0x230/0x3a4)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0425768>] (genl_rcv_msg) from [<c0424b94>] (netlink_rcv_skb+0xac/0xc0)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0424b94>] (netlink_rcv_skb) from [<c0425528>] (genl_rcv+0x24/0x34)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0425528>] (genl_rcv) from [<c0424324>] (netlink_unicast+0xf4/0x1e0)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0424324>] (netlink_unicast) from [<c0424920>] (netlink_sendmsg+0x46c/0x534)
Jan 29 09:08:15 MCDEBIAN kernel: [<c0424920>] (netlink_sendmsg) from [<c03e3f14>] (sock_sendmsg+0x14/0x24)
Jan 29 09:08:15 MCDEBIAN kernel: [<c03e3f14>] (sock_sendmsg) from [<c03e47f8>] (___sys_sendmsg+0x1cc/0x1d4)
Jan 29 09:08:15 MCDEBIAN kernel: [<c03e47f8>] (___sys_sendmsg) from [<c03e5410>] (__sys_sendmsg+0x3c/0x6c)
Jan 29 09:08:15 MCDEBIAN kernel: [<c03e5410>] (__sys_sendmsg) from [<c000f440>] (ret_fast_syscall+0x0/0x3c)
Jan 29 09:08:15 MCDEBIAN kernel: ---[ end trace 5aa6433fc264c416 ]---
```

This seems to indicate that the lock is held too long in sub `mwl_tx_xmit` for the ADDBA:

```
if (sta && sta->ht_cap.ht_supported && !eapol_frame &&
            ieee80211_is_data_qos(wh->frame_control)) {
                tid = qos & 0xf;
                mwl_tx_count_packet(sta, tid);

                spin_lock_bh(&priv->stream_lock);
                stream = mwl_fwcmd_lookup_stream(hw, sta->addr, tid);

                if (stream) {
                        if (stream->state == AMPDU_STREAM_ACTIVE) {
                                if (WARN_ON(!(qos &
**Line 901**                    MWL_QOS_ACK_POLICY_BLOCKACK))) {
                                        spin_unlock_bh(&priv->stream_lock);
                                        dev_kfree_skb_any(skb);
                                        return;
                                }
```

I wonder if the lock release and return drops the stream.

Chadster766, 2016-01-29:
I should mention this warning doesn't happen a lot and seems to have very little overall effect on operation. Once I have more run time I will report the number of occurrences.

Chadster766, 2016-02-04:
The number of occurrences are very low and it don't seem to have any impact on operations:

```
root@MCDEBIAN:~# uptime
 11:47:44 up 7 days, 30 min,  1 user,  load average: 0.32, 0.30, 0.31
root@MCDEBIAN:~# journalctl | grep -c "cut here"
6
```

yuhhaurlin, 2016-03-25:
Please help to check if you still encounter this kind of problem of 10.3.0.17-20160324. Basically we did not encounter this kind of problem.

Chadster766, 2016-03-25:
Awesome, will do :-)

yuhhaurlin, 2016-03-30:
I this problem won't happen, I will close it.

closed 2016-03-30 by yuhhaurlin