kaloz/mwlwifi #63: Wifi performance drops over time

MagicSimi, 2016-02-02:
I experience a on-going drop of my wifi performance to the point where it barely reacts (for example, just logging in with ssh and doing a reboot takes several minutes). A reboot resets the wifi performance, until it degrades again. It appears to me that if the wifi is used alot, the performance loss happens faster than with little wifi usage.

It seems like the irq problemes (descripted in issue [#34](https://github.com/kaloz/mwlwifi/issues/34) and [#43](https://github.com/kaloz/mwlwifi/issues/43) ) aren't really solved.

Router: WRT1900ACS
Openwrt version: r48564 (trunk)
Kernel: 4.1.16
mwlwifi.ko version: 10.3.0.16-20160105

If any more info is needed, I will do my best to provide it.

Chadster766, 2016-02-02:
Rebuild OpenWRT with this mwlwifi patch and test please:
http://pastebin.com/Vp19wxR4

MagicSimi, 2016-02-02:
Thanks for you quick answer. I will test the patch, but it might take me a few days to find time to rebuild an image and test it. I will report back then.

balintkozma, 2016-02-03:
I experience exactly the same. 

Router: WRT1200AC
Openwrt version: r48461(trunk)
Kernel: 4.1.15
mwlwifi.ko version: 10.3.0.16-20160105

I put a fan on my device, to rule out overheating. The idle temp is around 80C, goes above 85C under load. With fan the temp is 60C-68C. I will report  in two days.

balintkozma, 2016-02-03:
Degraded with fan too.

sam-sam-1, 2016-02-06:
I experience same issue. Significant ssh connection lags, lost pings to router.
Command : "echo 3 > /proc/sys/vm/drop_caches" helps to keep router stable (no router forced resets needed).
Router: Linksys WRT1900ACv2
openwrt version: Chaos Calmer 15.05 (r47515)
Kernel: 3.18

MagicSimi, 2016-02-11:
I did some testing with Chadster's patch now. It did not resolve the problem completely. It seems like the decrease takes longer but it still happens. Then also sam-sam's suggestion doesn't make a difference, only a reboot helps.

I started testing the wifi with iperf. 
After a reboot I get like 140 Mbits/sec client to Openwrt and 175 Mbits/sec Openwrt to client. It stays like that for a day or two.
Today with an uptime of two days I got: 135 Mbits/sec client to Openwrt and 27.3 **Kbits**/sec Openwrt to client.

yuhhaurlin, 2016-03-03:
When the problem happened, can you help to do iperf UDP test?

yuhhaurlin, 2016-03-03:
Reproduce and try to fix it.

yuhhaurlin, 2016-03-03:
In order to make sure you encounter the same problem as I try to fix, can you remove following line from mwl_rx_enable_sta_amsdu() of rx.c to see if the problem is gone?

sta_info->is_amsdu_allowed = true;

stmtpr, 2016-03-15:
@yuhhaurlin, not sure if this helps. 

```
root@mamba4:~# uptime
 07:13:19 up 20 days, 22:25,  load average: 0.10, 0.06, 0.05

root@mamba4:~# uname -a
Linux mamba4 4.4.0 #1 SMP Tue Feb 23 13:10:45 UTC 2016 armv7l GNU/Linux

root@mamba4:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='Bleeding Edge'
DISTRIB_REVISION='r48753'
DISTRIB_CODENAME='designated_driver'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_DESCRIPTION='OpenWrt Designated Driver r48753'
DISTRIB_TAINTS='no-all'
root@mamba4:~# grep "^10" /lib/modules/4.4.0/mwlwifi.ko 
10.3.0.16-20160105
10.3.0.16-20160105

kirkb@mbpw ~> iperf3 -c 192.168.1.1 -t 10 -u
Connecting to host 192.168.1.1, port 5201
[  4] local 192.168.1.4 port 60002 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bandwidth       Total Datagrams
[  4]   0.00-1.01   sec   120 KBytes   978 Kbits/sec  15  
[  4]   1.01-2.01   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   2.01-3.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   3.00-4.00   sec   128 KBytes  1.04 Mbits/sec  16  
[  4]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   5.00-6.01   sec   128 KBytes  1.04 Mbits/sec  16  
[  4]   6.01-7.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   7.00-8.01   sec   128 KBytes  1.04 Mbits/sec  16  
[  4]   8.01-9.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   9.00-10.01  sec   128 KBytes  1.04 Mbits/sec  16  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  4]   0.00-10.01  sec  1.24 MBytes  1.04 Mbits/sec  0.322 ms  0/159 (0%)  
[  4] Sent 159 datagrams

iperf Done.
kirkb@mbpw ~> iperf3 -c 192.168.1.1 -t 10
Connecting to host 192.168.1.1, port 5201
[  4] local 192.168.1.4 port 52727 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  1.27 MBytes  10.6 Mbits/sec                  
[  4]   1.00-2.00   sec  4.31 MBytes  36.1 Mbits/sec                  
[  4]   2.00-3.00   sec  8.55 MBytes  71.7 Mbits/sec                  
[  4]   3.00-4.00   sec  12.0 MBytes   101 Mbits/sec                  
[  4]   4.00-5.00   sec  8.97 MBytes  75.3 Mbits/sec                  
[  4]   5.00-6.00   sec  11.1 MBytes  93.1 Mbits/sec                  
[  4]   6.00-7.00   sec  9.20 MBytes  77.1 Mbits/sec                  
[  4]   7.00-8.00   sec  11.5 MBytes  96.2 Mbits/sec                  
[  4]   8.00-9.00   sec  10.0 MBytes  84.2 Mbits/sec                  
[  4]   9.00-10.00  sec  10.1 MBytes  84.2 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  86.9 MBytes  72.9 Mbits/sec                  sender
[  4]   0.00-10.00  sec  86.7 MBytes  72.7 Mbits/sec                  receiver
```

After performing /etc/init.d/network restart but NO router reboot.
```

kirkb@mbpw ~> iperf3 -c 192.168.1.1 -t 10 -u
Connecting to host 192.168.1.1, port 5201
[  4] local 192.168.1.4 port 60199 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bandwidth       Total Datagrams
[  4]   0.00-1.00   sec   120 KBytes   980 Kbits/sec  15  
[  4]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   3.00-4.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   5.00-6.01   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   6.01-7.01   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   7.01-8.01   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   8.01-9.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   9.00-10.01  sec   128 KBytes  1.04 Mbits/sec  16  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  4]   0.00-10.01  sec  1.24 MBytes  1.04 Mbits/sec  0.328 ms  0/159 (0%)  
[  4] Sent 159 datagrams

iperf Done.
kirkb@mbpw ~> iperf3 -c 192.168.1.1 -t 10
Connecting to host 192.168.1.1, port 5201
[  4] local 192.168.1.4 port 52742 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  34.3 MBytes   286 Mbits/sec                  
[  4]   1.00-2.00   sec  1.10 MBytes  9.20 Mbits/sec                  
[  4]   2.00-3.00   sec  13.8 MBytes   116 Mbits/sec                  
[  4]   3.00-4.00   sec  41.3 MBytes   347 Mbits/sec                  
[  4]   4.00-5.00   sec  51.5 MBytes   432 Mbits/sec                  
[  4]   5.00-6.00   sec  62.7 MBytes   526 Mbits/sec                  
[  4]   6.00-7.00   sec  68.4 MBytes   574 Mbits/sec                  
[  4]   7.00-8.00   sec  31.5 MBytes   265 Mbits/sec                  
[  4]   8.00-9.01   sec  42.2 MBytes   352 Mbits/sec                  
[  4]   9.01-10.00  sec  28.0 MBytes   236 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   375 MBytes   314 Mbits/sec                  sender
[  4]   0.00-10.00  sec   374 MBytes   314 Mbits/sec                  receiver
```

yuhhaurlin, 2016-03-25:
Please help to try 10.3.0.17-20160324. Thanks.

stmtpr, 2016-03-25:
@yuhhaurlin, I give it spin and let you know.  Is there anything specific you would like to see?

Chadster766, 2016-03-25:
I'm testing driver version 10.3.0.17-20160324 as well.

yuhhaurlin, 2016-03-25:
From our internal test, this problem is gone. Please let me know if this promblem is also gone on your site.

balintkozma, 2016-03-26:
The problem is gone.

yuhhaurlin, 2016-03-26:
Thanks. I will close this one.

closed 2016-03-26 by yuhhaurlin

MagicSimi, 2016-05-04:
I just wanted to say thanks for all you effort. The new module works great, router uptime of 21 days now. It seems to run really stable now. Cheers!

marconett, 2017-01-10:
The problem is still there for me.

Router: WRT1200AC
Openwrt version: r48532 (15.05.1)
Kernel: 3.18.23
mwlwifi.ko version: 10.3.0.17-20160324

Any insight on this?

I'm going to try 10.3.0.17-20160531 for now (the lastest build I found here: https://github.com/NemoAlex/mwlwifi-bin).

yuhhaurlin, 2017-01-10:
Please use latest driver. Thanks.