kaloz/mwlwifi #64: Abysmal 802.11N 5Ghz performance and performance variation

Razerwire, 2016-02-06:
I'm  experiencing pretty disappointing and fluctuating wireless performance with my 802.11N clients on the 5Ghz band with my WRT1900ACv2.
The performance issues are there under Windows 10 as well as under Linux, although under Linux the performance is worse than when using the same clients running Windows 10. 

My ISP provides me with a internet connection capable of  120Mbit down and 12mbit upload speed.
I've tested my WRT1900ACv2 with several laptops, one uses a Intel WiFi Link 5100AGN WiFi adapter the other laptop is equipped with a Intel Ultimate-N 6300AGN wireless card. 

For the tests I've used the 13.0.15 driver, WIFI channel is set to 116 (upper) Regulatory Domain is set to Netherlands, channel width is set to 80Mhz (although I’m aware that 802.11N only utilizes 40mhz wide channels). 

The WRT1900ACv2 is positioned about 8 meters away and although there are some walls I can sill see the router from where the clients are located. 
Also, i want to make clear that there aren't any other WIFI networks that overlap with mine.

The test was carried out at regular intervals over the course of a day one client at a time. Naturally the same speedtest.net server was used every time. 

`

       Windows 10       /      Intel Ultimate-N 6300:

              Ping       Downstream     Upstream
              16ms       85.48 Mbps	    12.35 Mbps
              19ms 		 88.88 Mbps		12.68 Mbps
              19ms		 42.09 Mbps		12.46 Mbps
              16ms		 76.12 Mbps		12.67 Mbps      
              14ms		 91.40 Mbps		12.49 Mbps
              15ms		 64.33 Mbps		12.49 Mbps
              17ms		 33.12 Mbps		12.57 Mbps
              15ms		 85.01 Mbps	    12.70 Mbps

       Linux Kernel 4.5.1rc1  / 	Intel Ultimate-N 6300:

            Ping	    Downstream      Upstream
            21ms		57.08 Mbps		12.49 Mbps
            23ms		56.74 Mbps		12.53 Mbps
            19ms		53.53 Mbps		12.35 Mbps
            18ms		48.39 Mbps		12.33 Mbps
            21ms		97.11 Mbps		12.40 Mbps	
            20ms		39.20 Mbps		12.30 Mbps
            29ms		44.74 Mbps		12.40 Mbps
            18ms		61.54 Mbps		12.41 Mbps

       Windows 10            /   	Intel  WiFi Link 5100:

            Ping       Downstream       Upstream 
            17ms	   115.30 Mbps		12.77 Mbps
            20ms		50.46 Mbps		12.58 Mbps
            18ms		77.30 Mbps		12.53 Mbps
            19ms		85.73 Mbps		12.65 Mbps
            16ms	   114.15 Mbps		12.80 Mbps
            18ms	   117.24 Mbps		12.72 Mpbs
            17ms	   114.92 Mbps		12.80 Mbps
            17ms	   112.67 Mbps		12.59 Mbps

        Linux Kernel 3.19.0-32	/  Intel  WiFi Link 5100:

           ping     Downstream      Upstream 
           23ms		46.01 Mbps		12.64 Mbps
           19ms		56.30 Mbps		12.44 Mbps
           17ms		84.60 Mbps		12.60 Mbps
           19ms		62.92 Mbps		12.51 Mbps
           19ms		33.79 Mbps		12.52 Mbps
           25ms		15.43 Mbps		01.43 Mbps
           19ms		42.84 Mbps		00,77 Mbps
           22ms		32.98 Mbps		04.68 Mbps

`

Now, please note the extreme performance variation with the 5Ghz WiFi. Also, performance is much worse on Linux then it is running Windows. 

Furthermore the max performance with the Ultimate-N 6300 on Windows is lower than that of the WiFi Link 5100 even-though the Ultimate-N 6300 has three antennas versus the two of the 5100AGN.

I Would like to point out that even when I place my laptops at about two meters from the WRT1900ACv2 performance is similar, that is to say: really bad. 

I own a Netgear R7000 and with that router I've never experienced performance like this. With the R7000 I always manage to get the full 120mbit down and 12mbit up, with the same clients.

I have several clients that can connect to the WRT1900ACv2 using the 802.11AC protocol and those are capable of maxing out my internet connection, using the same method, time and time again. 

So in summary:
Although with the WRT1900ACv2 (and the 10.3.0.15 driver) 802.11AC performance is fine, 802.11N performance definitely needs fixing. 

-----------------------------------------------------------------------------------------------------------------------------
P.S.
Shouldn't the wireless firmware (the 88W8864.bin) for the 88W8864 SoC be updated it hasn't been updated for 6 months and the Linksys firmware does get updated wireless firmware:
Firmware version:	1.0.0.169041
Release date:		November 10, 2015
- Integrated an updated wireless driver (v7.2.9.2)

yuhhaurlin, 2016-02-08:
Yes, 7.2.9.6 is under verification and this version of firmware will be updated here later. Once if 7.2.9.6 is released, we can debug issues still existed based on this version of firmware.

Razerwire, 2016-02-08:
All right I'll just wait for 7.2.9.6 then. 

Thanks for the info.

balintkozma, 2016-02-20:
Any update on 7.2.9.6 is welcome.

yuhhaurlin, 2016-03-03:
It is under verification.

haryadiaja, 2016-03-17:
hi there, any update on the update?

Mrengles, 2016-03-21:
How long does it take to get a open source release verified?

yuhhaurlin, 2016-03-25:
Please help to try 10.3.0.17-20160324. I also upgrade firmware to 7.2.9.2. Thanks.

Razerwire, 2016-03-29:
> Please help to try 10.3.0.17-20160324. I also upgrade firmware to 7.2.9.2. Thanks.

Well the new firmware and binary isn't pushed to OpenWRT trunk yet, i'll run the tests with DD-WRT (as i did before) over the weekend. DD-WRT devs already integrated .17 and the 7.2.9.2 binary.

yuhhaurlin, 2016-03-29:
Thanks for your information.

johnnysl, 2016-04-02:
@Razerwire, probably an "open door" did you run acrylic wifi to check if there are not any neighbours on the same channel?

Razerwire, 2016-04-05:
> @Razerwire, probably an "open door" did you run acrylic wifi to check if there are not any neighbours on the same channel?

Well of course, I’m blessed with a reasonably quit 5Ghz band, i always chose an empty channel or channel range when setting up a router.

Furthermore, as I’ve stated the problem described in this issue is limited to 802.11N clients 802.11AC clients do not exhibit the behaviour described in this issue.

Razerwire, 2016-04-06:
@yuhhaurlin 
I don't understand why issue #68 was closed, all test performed by me, with regards to #68 were carried out with 802.11AC cliënts, no 802.11N clients involved.

Besides that there are more people running DD-WRT that have been affected by this issue:
http://www.dd-wrt.com/phpBB2/viewtopic.php?p=1024842&sid=a246b1c4df495374b48c09b9448bc611#1024842

Please respond.

yuhhaurlin, 2016-04-06:
@Razerwire How about the throughput you got if you run stock firmware on your deivce? Obviously .17 does not have problem with V2 hardware because someone can run OpenWrt with .17 on it and throughput will not be blocked at 25 mbps. About 11n client on 5g band, we will check it later to see if the driver had this kind of problem.

johnnysl, 2016-04-06:
@Razerwire or if you are interested, you are welcome to test my personal build.

Razerwire, 2016-04-07:
> @Razerwire How about the throughput you got if you run stock firmware on your deivce? Obviously .17 does not have problem with V2 hardware because someone can run OpenWrt with .17 on it and [...]

I've just tried stock firmware, no problems there......

Razerwire, 2016-04-07:
>@Razerwire or if you are interested, you are welcome to test my personal build.

I would like to test you're OpenWRT build provided it uses the .17 driver and 7.2.9.2. firmware.

johnnysl, 2016-04-07:
Here you go, only for V2. https://www.dropbox.com/sh/kdo4gdr5d0pi7r2/AABKYtaFi2Cur2_ezlSaDGsOa?dl=0

Razerwire, 2016-04-07:
> Here you go, only for V2. https://www.dropbox.com/sh/kdo4gdr5d0pi7r2/AABKYtaFi2Cur2_ezlSaDGsOa?dl=0

Thanks a lot, i'll try it, maybe the problem is (limited to) DD-WRT

johnnysl, 2016-04-07:
Let me know how it works out for you, am really curious. If i can help setup or tweak, let me k ow as wel

johnnysl, 2016-04-07:
Bleh, just did another round of iperf testing after being very happy with the speed for over 8 days, and while 2.4ghz speeds is still quite ok, the 5ghz speeds are back to 25-35mbit range again. (Testing this with an ipad air 2 AC-device, to lazy to find and grab my laptop currently)

@yuhhaurlin: haven't rebooted yet, but pretty sure that will fix it again.
Any thing i can gather/test? Station dump shows proper 866mhz connected speeds, vht-mcs 9, no errors on dmesg.

chrismrutherford, 2016-04-08:
I'm seeing something strange on the Caimen/WRT1200AC/.17/openWRT/DD.  I just wanted to clarify and ask some questions.
If I connect 2 Android phones to the 5GHz wifi and run iperf on them both, one as client and one as server I get about 250mbps, which is really good.  However if I run iperf server on the router its self and use an android device as a client I only get about 25mbps.  If I run iperf client and server on the router and go via local host I get several gbps which is what I would expect.
Also if I run samba server on the router and transfer files from ram disk to an android client I also get about 25mbps.
Does the wifi chipset perform some layer 2 switching in hardware, allowing data transfer between the android phones without sending data to/from the kernel/main memory?  i.e. hardware accelerated?
Am I seeing 25mbps when transferring from the router because the kernel / driver / main memory is involved, requiring some degree of software switching?
Does this imply that when routing from an Ethernet port to wifi then this will similarly be limited to 25mbps?  Has anyone tested this?

johnnysl, 2016-04-09:
... And my speeds on 5ghz are since yesterday evening back to normal, without even touching the router. Seems that i have some gremlins playing with my setup or something :-/ 

I'm just lost why it behaves like that.

johnnysl, 2016-04-10:
interesting issue. So i'm experiencing speeds that go down on 5ghz, and fix themselves again after a few hours. if i reset the interface (wifi down, wifi up) it is instantly fixed again. Maybe i should open a separate thread on this, as it has nothing to do with Wireless-N.

Razerwire, 2016-04-10:
Maybe @yuhhaurlin should just reopen issue #68 again since it does affect multiple people.

yuhhaurlin, 2016-04-11:
@johnnysl  How many clients used by you?

johnnysl, 2016-04-11:
@yuhhaurlin: Usually around 8, max 12.

yuhhaurlin, 2016-04-11:
@johnnysl Thanks. When this problem happened, can you disconnect all other clients and redo the iperf throughput test again to see if throughput is all right?

johnnysl, 2016-04-11:
@yuhhaurlin: will try, but might take some time, as i'm travelling this week.

yuhhaurlin, 2016-04-11:
@johnnysl No problem, Thanks.

johnnysl, 2016-04-17:
ok, have been doing endless iperf3 testing with any combination and number of clients, and it doesn't seem to be related to the number of clients i'm using. have tested any number of clients on 5ghz from 1 to 10 and i cannot trigger the issue based on amount of clients.

johnnysl, 2016-04-17:
@yuhhaurlin: I do not like results i cannot reproduce, but i've been testing primarily on my HP Elitebook Folio 9470M, with intel Centrino Advanced-N 6235 adapter, connected to the 5ghz radio.

I cannot trigger the issue by adding clients, but when the issue occurred, removing a mid 2012 Macbook air from the router fixed the issue immediately. Annoyingly re-adding the macbook to 5ghz radio keeps the speed up. At this moment only 5 devices are connected wirelessly. Will try to do more testig when i see the issue occur.

This is current ly my client list:
Iphone 5 
Iphone 5S*
Ipad Air 2
Mid 2012 Macbook air
HP Folio 9470M

*connected to the 2.4ghz radio

Razerwire, 2016-04-21:
So i've just installed OpenWRT trunk build of 20th april and my Wifi speed is capped @20mpbs just like with DD-WRT.... 

@yuhhaurlin you need any logs or config?

jglooije, 2016-04-28:
I'm having the same kind of problems on 5GHz. My speed is 95% of the time around 80Mpbs, some times around 50Mpbs or 250Mbps. I'm running my own build of openwrt DD-driver whit the .17 driver. The build is stable and the only thing i changed in the driver is that i removed the ampdu errors. These where filling up my kernel log.

yuhhaurlin, 2016-05-23:
How about the version 10.3.0.17-20160520-1? Does this problem still exist?

jglooije, 2016-05-23:
The driver at this moment looks stable, but i need some more time to evaluate this. The mpdu message i already took out a time ago and was no problem. The speed of the 10.3.0.17-20160520-1 was a bit slower then the previous version around the 60Mb/s. But i didn't test this very long because i added the '200-fix_excessive_delays.patch' from the lede project. My speed now looks stable at just above 80Mb/s to the internet, the ping time has improved dramaticlly. (from 20-30mS to 10mS)
If you like to know how the driver is without the lede patch let me know, then i will test it for you.
Date: Sun, 22 May 2016 19:03:48 -0700

yuhhaurlin, 2016-05-23:
Where can I get the patch? Thanks.

jglooije, 2016-05-23:
Here is the link : https://github.com/Xotic750/mirror-lede/tree/master/package/kernel/mwlwifi/patches

Date: Mon, 23 May 2016 13:54:37 -0700

johnnysl, 2016-05-24:
Same here. I've been testing the last few days with the LEDE solution and the driver currently seems stable. there seems to be some performance degradation over time (about 10%), but no longer the "capped" 20mbit performance issues i saw with previous builds.
Will keep an eye on it, over the next days.

yuhhaurlin, 2016-05-26:
So this problem is fixed with updated driver?

johnnysl, 2016-05-26:
Only thing i could nag about is that performance for an N-client is not symmetrical. I get about 160mbit up (normal iperf3 test)  and 110 down (iperf3 -R, both values seen from client), but otherwise it is stable for me.

Can others chip in as well on their findings?

Razerwire, 2016-05-26:
Well, I can't comment yet on the <25mbps 'cap' that i saw with DD-WRT since there isn't a DD-WRT build incorporating the "10.3.0.17-20160523" driver yet. 

Regarding the issue i initially complained about the fluctuating performance:
While i no longer own the cliënt with the Ultimate-N 6300 wifi-device performance seems improved with the WiFi Link 5100: it's under linux it's now between 65- and 90mbps. 

I'll do some benchmarking under Windows over the weekend and report back.

jglooije, 2016-05-26:
I tested on a day to day basis and the speed was around the 80 Mb/s. Only today i tested it for around 15 minuts, so many tests after each other, and then i see the speed gradualy decline to 55 MB/s. After about an hour is is still slow. The ping times are still good.

Razerwire, 2016-05-29:
So I've done some testing under Windows with the system utilizing a WiFi Link 5100AGN, and performance issues and performance variation are still there. 

Performance variates all the way from 10mbps to 110mpbs both under Linux and under Windows, tested using a non-congested WiFi channel

@yuhhaurlin If you need more thorough reporting like in the first post let me know. 

P.S. 
802.11AC performance with the latest driver is really really good i'm seeing peak speeds of 510mbps with a 2x2 802.11AC client.

verbit, 2016-05-29:
I reported my 5GHz 802.11N performance on another issue: https://github.com/kaloz/mwlwifi/issues/72#issuecomment-222375840.
My results seemed pretty good to me.

@Razerwire Strange, that our results differ so much. Are you using the official DD-WRT beta builds? I remember that I had performance issues with the official beta releases which were gone when I switched to Kong builds.

Razerwire, 2016-05-29:
@verbit 
I'm using LEDE at the moment quite similar to OpenWRT.

Difference in result's don't seem strange to me we're both using different systems, probably with different WNIC's and different drivers. 

Thing is:  those issues don't show when using a Netgear R7000,  which shows me that the WRT1900AC's drivers and/or firmware probably need improving.

verbit, 2016-05-29:
True, I will try to get access to some other WiFi clients. Maybe I can reproduce the issues.

yuhhaurlin, 2016-05-30:
@Razerwire 
"one uses a Intel WiFi Link 5100AGN WiFi adapter the other laptop is equipped with a Intel Ultimate-N 6300AGN wireless card" --> So you still encounter the same problem with latest driver if running these n clients on 5 GHz, right? How about running these clients on 2.4g?

yuhhaurlin, 2016-05-30:
@Razerwire BTW, can you help to check if stock firmware has the same problem? Thanks a lot.

Razerwire, 2016-05-30:
> So you still encounter the same problem with latest driver if running these n clients on 5 GHz, right? 
How about running these clients on 2.4g?

@yuhhaurlin 
Yes, i do although as i've stated before i no longer own the client with the "Ultimate-N 6300AGN" WNIC. 
I'll check if using 2.4Ghz yields the same results / problem

> BTW, can you help to check if stock firmware has the same problem? Thanks a lot.

@yuhhaurlin 
Yes, I’ll check if stock firmware has the same problem.
Do you want me tot test with stock firmware for the WRT1900ACS?  It has the same wireless driver (7.2.9.2) as is used for open source. I could test stock firmware for WRT1900ACv2 but, IIRC, it uses a older wireless driver.

yuhhaurlin, 2016-05-30:
Yes, please give me more information. Thanks a lot.

Razerwire, 2016-05-30:
Al-right  I'll use the stock WRT1900ACS firmware from Linksys to test then.

Razerwire, 2016-06-05:
Here's the performance of 2.4Ghz band using the '10.3.0.17-20160520-1' open source driver.

` 

    Linux Kernel 3.19.0-32              /               Intel  WiFi Link 5100 (2.4Ghz )

    Ping		Downstream		Upstream
    18ms		52.61Mbps		15.51Mbps
    20ms		67.77Mbps		14.57Mbps
    20ms		83.58mbps		14.47mbps
    18ms		88.45mbps		15.31mbps
    20ms		76.99mbps		15.41mbps
    20ms		39.80mbps		14.85mbps
    18ms		88.52mbps		15.50mbps
    20ms		74.02mbps		15.49mbps
    20ms		82.04mbps		15.19mbps
    18ms		43.58mbps		15.08mbps

    Windows 10			/		Intel  WiFi Link 5100  (2.4Ghz)

    Ping		Downstream		Upstream
    16ms		59.07mbps		15.59mbps
    18ms		34.30mbps		15.24mbps
    21ms		77.89mbps		15.72mbps
    16ms		67.77mbps		15.80mbps		
    17ms		91.15mbps		15.82mbps
    19ms		97.76mbps		15.76mbps
    17ms		95.10mbps		15.70mbps
    17ms		95.69mbps		15.62mbps
    17ms		87.78mbps		15.80mbps
    18ms		79.82mbps		15.78mbps

`
So performance is still isn't stable (especially on Linux) but it isn't nearly as bad as on the 5Ghz band. 

Next I'll test 5Ghz performance running Linksys (stock) WRT1900ACS firmware

bbogey, 2016-06-05:
Here is iperf test with N client on 5GHz with macbook pro.
Running LEDE r484 with mwlwifi version 10.3.0.17-20160603

```
root@lede:~# iwinfo wlan0 assoclist

4:26:96:CD:B9:19  -61 dBm / -90 dBm (SNR 29)  0 ms ago
	RX: 450.0 MBit/s, MCS 23, 40MHz                15760 Pkts.
	TX: 450.0 MBit/s, MCS 23, 40MHz                13882 Pkts.
```

Here is for Download:

```
Macbook-Jukka:Documents jukka$ ./iperf301  -c 192.168.1.33 -i 10 -t 60 -p 5201    -R
Connecting to host 192.168.1.33, port 5201
Reverse mode, remote host 192.168.1.33 is sending
[  4] local 192.168.1.14 port 60581 connected to 192.168.1.33 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   388 MBytes   325 Mbits/sec              
[  4]  10.00-20.00  sec   393 MBytes   330 Mbits/sec              
[  4]  20.00-30.00  sec   393 MBytes   330 Mbits/sec              
[  4]  30.00-40.01  sec   394 MBytes   330 Mbits/sec              
[  4]  40.01-50.01  sec   394 MBytes   330 Mbits/sec              
[  4]  50.01-60.00  sec   394 MBytes   330 Mbits/sec              
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-60.00  sec  2.30 GBytes   330 Mbits/sec    0         sender
[  4]   0.00-60.00  sec  2.30 GBytes   329 Mbits/sec              receiver

iperf Done.
```

And for upload
```
Macbook-Jukka:Documents jukka$ ./iperf301  -c 192.168.1.33 -i 10 -t 60 -p 5201    
Connecting to host 192.168.1.33, port 5201
[  4] local 192.168.1.14 port 60583 connected to 192.168.1.33 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.01  sec   291 MBytes   244 Mbits/sec              
[  4]  10.01-20.02  sec   321 MBytes   269 Mbits/sec              
[  4]  20.02-30.00  sec   320 MBytes   269 Mbits/sec              
[  4]  30.00-40.02  sec   321 MBytes   269 Mbits/sec              
[  4]  40.02-50.00  sec   320 MBytes   269 Mbits/sec              
[  4]  50.00-60.00  sec   317 MBytes   266 Mbits/sec              
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-60.00  sec  1.85 GBytes   264 Mbits/sec              sender
[  4]   0.00-60.00  sec  1.85 GBytes   264 Mbits/sec              receiver

iperf Done.
```

So for me upstream is bit slower.

johnnysl, 2016-06-05:
On 5ghz I get 360mbit down, 540mbit up on an ipad air2 to a mac on the wired lan (with iperf3)
With an N-client i get 110 down and 165 up. So i see similar speed differences.

2.4ghz i simply ignore due to overcrowded wireless channels around me.

yuhhaurlin, 2016-06-05:
I think this problem is that Razerwire can't get good download speed on two N clients on 5 GHz. I will focus on that.

johnnysl, 2016-06-05:
Sure, any reason why up and download differ so much in speed? Or is that normal?

jglooije, 2016-06-05:
Here is iperf test with with MSI 2QC apache, AC client on 5GHz. Intel dualband AC3160 
Running wrt1900acs, LEDE Reboot r492 with mwlwifi version 10.3.0.17-20160603 Latest commit 8635191

root@OpenWrt:/usr/bin# ./iperf3  -c 192.168.144.29 -i 10 -t 60 -p 5201 -R
Connecting to host 192.168.144.29, port 5201
Reverse mode, remote host 192.168.144.29 is sending
[  4] local 192.168.144.1 port 38176 connected to 192.168.144.29 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  77.5 MBytes  65.0 Mbits/sec
[  4]  10.00-20.00  sec  86.4 MBytes  72.5 Mbits/sec
[  4]  20.00-30.00  sec  89.6 MBytes  75.2 Mbits/sec
[  4]  30.00-40.00  sec  86.6 MBytes  72.6 Mbits/sec
[  4]  40.00-50.00  sec  86.0 MBytes  72.1 Mbits/sec
[  4]  50.00-60.00  sec  79.7 MBytes  66.9 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-60.00  sec   506 MBytes  70.7 Mbits/sec                  sender
[  4]   0.00-60.00  sec   506 MBytes  70.7 Mbits/sec                  receiver

iperf Done.
root@OpenWrt:/usr/bin# ./iperf3  -c 192.168.144.29 -i 10 -t 60 -p 5201
Connecting to host 192.168.144.29, port 5201
[  4] local 192.168.144.1 port 38180 connected to 192.168.144.29 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-10.00  sec  49.6 MBytes  41.6 Mbits/sec    0    211 KBytes
[  4]  10.00-20.00  sec  77.9 MBytes  65.4 Mbits/sec    0    211 KBytes
[  4]  20.00-30.00  sec  72.7 MBytes  61.0 Mbits/sec    0    211 KBytes
[  4]  30.00-40.00  sec  73.9 MBytes  62.0 Mbits/sec    0    211 KBytes
[  4]  40.00-50.00  sec  81.1 MBytes  68.0 Mbits/sec    0    211 KBytes
[  4]  50.00-60.00  sec  75.1 MBytes  63.0 Mbits/sec    2    211 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-60.00  sec   430 MBytes  60.2 Mbits/sec    2             sender
[  4]   0.00-60.00  sec   430 MBytes  60.1 Mbits/sec                  receiver

iperf Done.

jbsky, 2016-06-05:
@jglooije wow! what WRT1X00AC? v1 v2 1200 or ACS?

yuhhaurlin, 2016-06-05:
@jglooije Can you create another issue?

Razerwire, 2016-06-12:
So here's the result for stock firmware (Linksys firmware Ver. 1.0.0.172978 ) on the 5ghz band:

```
Linux Kernel 3.19.0-32		/               Intel  WiFi Link 5100 (5.2Ghz )

Ping		        Downstream      Upstream
19ms		        18.54mbps	    15.58mbps 
21ms		        13.94mbps	    14.79mbps
17ms		        55.17mbps	    15.47mbps
17ms		        72.11mbps	    15.73mbps
21ms		        14.88mbps	    13.14mbps
22ms		        17.29mbps	    13.43mbps
20ms		        36.18mbps	    15.06mbps
38ms		        35.39mbps	    15.62mbps
20ms		        15.26mbps	    15.20mbps
25ms		        15.05mbps	    14.77mbps

Windows 10          		/		Intel  WiFi Link 5100  (5.2Ghz)

Ping		       Downstream      Upstream
15ms		       41.89mbps	   15.55mbps
16ms		       70.17mbps	   15.62mbps
16ms		       57.40mbps	   15.68mbps
17ms		       64.81mbps	   15.65mbps
15ms		       42.55mbps	   15.68mbps
16ms		       26.12mbps	   15.33mbps
17ms		       61.05mbps	   15.54mbps
14ms		       52.38mbps	   15.65mbps
15ms		       33.90mbps	   15.61mbps
16ms		       29.39mbps	   15.69mbps
```

[2 more replies]

closed 2016-10-17 by yuhhaurlin