kaloz/mwlwifi #66: multicast udp packet reordering

xia-harry, 2016-02-26:
All,

Does anyone see this issue that multicast udp packet reordering from AP which is mwlwifi driver?

Thanks
Harry

chinawrj, 2016-02-26:
Hi, I am experiencing this issue too. I am digging on it, but still no clue.

xia-harry, 2016-02-29:
Chinawrj,

I will also look at this. We can update each other if we have any findings.

yuhhaurlin, 2016-03-03:
Check it now.

yuhhaurlin, 2016-03-03:
Reproduce and try to fix it.

yuhhaurlin, 2016-03-03:
In order to make sure you encounter the same problem as I try to fix, can you remove following line from mwl_rx_enable_sta_amsdu() of rx.c to see if the problem is gone?

sta_info->is_amsdu_allowed = true;

chinawrj, 2016-03-03:
Hi, Thanks for the update. I will have a try tomorrow.
I have a idea for this.

sta_info->is_amsdu_allowed = true;
is trying to disable a-msdu for station. A-MSDU is not used for multicast
packet? So the line here is trying to disable A-MSDU for the station when
station send multicast packet to AP？

yuhhaurlin, 2016-03-03:
I already removed the comment. The problem still happened. I will try to fix this problem. Thanks.

xia-harry, 2016-03-07:
yuhhaurlin,

Do you have any more information about the root cause of this issue? Thanks

yuhhaurlin, 2016-03-08:
This problem is under investigation. If this problem is fixed, the fix will be updated here with F/W version 7.2.9.6.

yuhhaurlin, 2016-03-25:
Please help to try 10.3.0.17-20160324. Thanks.

northbound1, 2016-03-25:
Sorry to say this, but this is a downgrade from standard trunk 4.4.6
But keep trying please. I am willing to try again when you update wlwifi. I can use this version and will see what happens after a few days of uptime.
Ftp. Seems ok but iperf3 is much slower
C:\iperf3>iperf3 -c 192.168.1.1 -t 10
Connecting to host 192.168.1.1, port 5201
[  4] local 192.168.1.131 port 50571 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.01   sec  13.8 MBytes   115 Mbits/sec
[  4]   1.01-2.00   sec  20.5 MBytes   173 Mbits/sec
[  4]   2.00-3.00   sec  15.9 MBytes   133 Mbits/sec
[  4]   3.00-4.01   sec  8.12 MBytes  67.3 Mbits/sec
[  4]   4.01-5.00   sec  14.9 MBytes   126 Mbits/sec
[  4]   5.00-6.02   sec  24.8 MBytes   205 Mbits/sec
[  4]   6.02-7.01   sec  15.5 MBytes   130 Mbits/sec
[  4]   7.01-8.00   sec  12.9 MBytes   110 Mbits/sec
[  4]   8.00-9.00   sec  11.9 MBytes  99.4 Mbits/sec
[  4]   9.00-10.00  sec  13.9 MBytes   117 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   152 MBytes   128 Mbits/sec                  sender
[  4]   0.00-10.00  sec   152 MBytes   127 Mbits/sec                  receiver

iperf Done.

C:\iperf3>iperf3 -c 192.168.1.1 -t 10 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[  4] local 192.168.1.131 port 50574 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  26.5 MBytes   222 Mbits/sec
[  4]   1.00-2.00   sec  28.5 MBytes   239 Mbits/sec
[  4]   2.00-3.00   sec  28.4 MBytes   238 Mbits/sec
[  4]   3.00-4.00   sec  29.2 MBytes   245 Mbits/sec
[  4]   4.00-5.00   sec  29.5 MBytes   248 Mbits/sec
[  4]   5.00-6.00   sec  27.4 MBytes   230 Mbits/sec
[  4]   6.00-7.00   sec  27.0 MBytes   227 Mbits/sec
[  4]   7.00-8.00   sec  28.6 MBytes   239 Mbits/sec
[  4]   8.00-9.00   sec  29.6 MBytes   249 Mbits/sec
[  4]   9.00-10.00  sec  29.8 MBytes   250 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec   285 MBytes   239 Mbits/sec    2             sender
[  4]   0.00-10.00  sec   285 MBytes   239 Mbits/sec                  receiver

iperf Done.

yuhhaurlin, 2016-03-25:
From our internal test, this problem is gone and we did not find impact on performance.

Chadster766, 2016-03-25:
2.4Ghz Results:

C:\Users\chad\Downloads\iperf-3.0.11-win32>iperf3 -c 192.168.200.3 -B 192.168.200.11
Connecting to host 192.168.200.3, port 5201
[  4] local 192.168.200.11 port 43663 connected to 192.168.200.3 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  14.5 MBytes   122 Mbits/sec
[  4]   1.00-2.00   sec  16.9 MBytes   142 Mbits/sec
[  4]   2.00-3.00   sec  16.6 MBytes   139 Mbits/sec
[  4]   3.00-4.00   sec  15.9 MBytes   134 Mbits/sec
[  4]   4.00-5.00   sec  16.6 MBytes   139 Mbits/sec
[  4]   5.00-6.00   sec  16.1 MBytes   135 Mbits/sec
[  4]   6.00-7.00   sec  16.6 MBytes   139 Mbits/sec
[  4]   7.00-8.00   sec  16.4 MBytes   137 Mbits/sec
[  4]   8.00-9.00   sec  16.2 MBytes   136 Mbits/sec
[  4]   9.00-10.00  sec  16.6 MBytes   139 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   162 MBytes   136 Mbits/sec                  sender
[  4]   0.00-10.00  sec   162 MBytes   136 Mbits/sec                  receiver

iperf Done.

xia-harry, 2016-03-26:
yuhhaurlin,

thanks your fix.

closed 2016-03-26 by xia-harry