kaloz/mwlwifi #69: WPS support

hdiniz, 2016-03-31:
Any change we get WPS support on a future release?

yuhhaurlin, 2016-04-01:
Yes.

hdiniz, 2016-04-01:
Thank you, I am actually studying this and would be very interested in contributing to it somehow.

I am still a newbie and have dome doubts, so please correct me if I am wrong.. 
From the driver side the only modification for WPS is including the appropriate IEs in management frames (Beacon, Probe Response, Association Response), correct?
These IEs come from hostapd, if built to support it, and we would only need to handle them in modem side?
Additional steps from WPS are in the EAP layer and passed to /handled by hostapd, meaning no interference from driver?

yuhhaurlin, 2016-05-27:
The way done by you is not correct. I will update code later and you can help to test it. Thanks.

hdiniz, 2016-05-30:
Thanks, glad to help.

yuhhaurlin, 2016-05-31:
Please test 10.3.0.17-20160531.

yuhhaurlin, 2016-06-03:
@hdiniz How about the test? Thanks.

hdiniz, 2016-06-03:
Sorry about the delay, I will test asap.

verbit, 2016-06-03:
Does the WPS support include the *Push Button Configuration* method?

hdiniz, 2016-06-03:
@yuhhaurlin I tested today and PBC is not working. I followed readme instruction to compile, so this is the version tested: PKG_VERSION:=10.3.0.17-20160601 & PKG_SOURCE_VERSION:=4bb95ba1aeccce506a95499b49b9b844ecfae8a1

I did a quick check using airpcap and I could not see WPS IE being sent. I can support next week by collecting logs, let me know what you need.

hdiniz, 2016-06-03:
@yuhhaurlin actually, its very strange. Only Beacon frames are carrying WPS IE, Probe Responses are not. I will upload pcap soon.

yuhhaurlin, 2016-06-03:
I will modify the code. Thanks.

yuhhaurlin, 2016-06-03:
@verbit Yes. You can configure hostage to do that.

yuhhaurlin, 2016-06-03:
@hdiniz  A quick update. Please help to verify it. Thanks.

hdiniz, 2016-06-03:
I am without access to router, on Monday I can verify for you.
Thank you for the prompt support.

hdiniz, 2016-06-07:
@yuhhaurlin I just tested and it is working, sorry for the delay.

PKG_VERSION:=10.3.0.17-20160607
PKG_SOURCE_VERSION:=2a4f8bab4fe4a801affede8178e8fbc578d14f45
Latest release.

Thank you!

closed 2016-06-07 by hdiniz

verbit, 2016-06-07:
@hdiniz: "I tested today and PBC is not working"

Is WPS PBC also working?

hdiniz, 2016-06-07:
My hostapd setup only supports WPS PBC for the moment, so yes it is working.
I have not tested other methods such as PIN, NFC, UPNP, etc.

verbit, 2016-06-07:
Ok, thanks!

hdiniz, 2016-06-07:
@verbit btw, as long as the driver is properly including the IE it receives from upper layer (which looks to be the case now, I haven't checked air logs), every WPS method should be supported if hostapd is set up for that.

yuhhaurlin, 2016-06-08:
Yes.